nxp-archive/openil #3: Openil build fails

DoogRC, 2018-05-31:
Hi,
It's my first time using openil and I/m trying to do a build for the NXP LS1021A TSN box using openil -201712 and the nxp board config "nxp_ls1021atsn_defconfig".

The build fails with the following..
Makefile:418: recipe for target 'create_inode.o' failed
make [3]: *** [create_inode.o] Error 1
..
Makefile:405: recipe for target 'all-progs-recursive' failed
make [2]: *** [all-progs-recursive] Error 1
..
Makefile:340: recipe for target 'all' failed
make [1]: *** [all] Error 2

Any help would be much appreciated :-)

vladimiroltean, 2018-05-31:
Hello there!
The `create_inode.o` file is part of e2fsprogs.
I don't see any particular reason why that would fail to build.
Please try to rebuild only that package and post the full output again, if it still fails:
```
$ rm -rf output/build/e2fsprogs*
$ rm -rf dl/e2fsprogs*
$ make e2fsprogs
```

DoogRC, 2018-05-31:
Thanks - I'll give it a go.
The error is to do with conflicting types for 'file_copy_range'.
I'll post the full output if it still fails.