nxp-archive/openil #6: Master sync timeout

echizenr, 2018-07-16:
Hello,

I have a problem with my two LS1021ATSN boards. When I try to establish a basic master slave sync between my two Boards with the following command : 
**$ ptp4l -i eth0 -p /dev/ptp0 -f /etc/ptp4l_default.cfg –m** 
The sync fail with the following message (continuously send) : 

**ptp4l[3998.755]: port 1: announce timeout
ptp4l[3998.755]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[3998.755]: selected best master clock 00049f.fffe.ef0707
ptp4l[3998.755]: assuming the grand master role
ptp4l[3998.760]: port 1: master tx announce timeout
ptp4l[3998.760]: port 1: setting asCapable
ptp4l[3998.885]: port 1: master sync timeout
ptp4l[3999.015]: port 1: master sync timeout
ptp4l[3999.145]: port 1: master sync timeout
ptp4l[3999.275]: port 1: master sync timeout
ptp4l[3999.405]: port 1: master sync timeout
ptp4l[3999.535]: port 1: master sync timeout
ptp4l[3999.665]: port 1: master sync timeout
ptp4l[3999.795]: port 1: master sync timeout
ptp4l[3999.925]: port 1: master sync timeout
ptp4l[4000.055]: port 1: master sync timeout
ptp4l[4000.185]: port 1: master sync timeout
ptp4l[4000.315]: port 1: master sync timeout
ptp4l[4000.445]: port 1: master sync timeout
ptp4l[4000.575]: port 1: master sync timeout
ptp4l[4000.705]: port 1: master sync timeout
ptp4l[4000.765]: port 1: master tx announce timeout**
 
Also when I try to run a sync between two Ordinary clocks with the following command : 

**ptp4l -i eth0 -p /dev/ptp0 -2 -m**

I have the same result on the master but the slave displays the following message : 

**ptp4l[1562.770]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1562.770]: selected best master clock 00049f.fffe.ef0707
ptp4l[1562.770]: assuming the grand master role
ptp4l[1565.112]: port 1: new foreign master 00049f.fffe.ef0101-1
ptp4l[1569.122]: selected best master clock 00049f.fffe.ef0101
ptp4l[1569.122]: port 1: MASTER to UNCALIBRATED on RS_SLAVE
ptp4l[1570.132]: master offset     795419 s0 freq   -5400 path delay      2147
ptp4l[1571.137]: master offset     802150 s1 freq   +1298 path delay      1421
ptp4l[1572.142]: master offset      -1800 s2 freq    -502 path delay      1421
ptp4l[1572.143]: port 1: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[1573.147]: master offset      -2173 s2 freq   -1415 path delay      2674
ptp4l[1574.152]: master offset       -664 s2 freq    -558 path delay      3175
ptp4l[1575.157]: master offset        223 s2 freq    +130 path delay      3188
ptp4l[1576.162]: master offset        503 s2 freq    +477 path delay      3188
ptp4l[1577.167]: master offset        384 s2 freq    +509 path delay      3262
ptp4l[1578.172]: master offset        229 s2 freq    +469 path delay      3322
ptp4l[1579.177]: master offset        163 s2 freq    +472 path delay      3368
ptp4l[1580.182]: master offset         -7 s2 freq    +350 path delay      3418
ptp4l[1581.187]: master offset        163 s2 freq    +518 path delay      3418
ptp4l[1582.192]: master offset         35 s2 freq    +439 path delay      3426
ptp4l[1583.197]: master offset        -95 s2 freq    +320 path delay      3426
ptp4l[1584.202]: master offset        150 s2 freq    +536 path delay      3426
ptp4l[1585.207]: master offset        -85 s2 freq    +346 path delay      3426
ptp4l[1586.212]: master offset         60 s2 freq    +466 path delay      3426
ptp4l[1587.217]: master offset        -50 s2 freq    +374 path delay      3426
ptp4l[1588.222]: master offset        169 s2 freq    +578 path delay      3397
ptp4l[1589.227]: master offset       -103 s2 freq    +356 path delay      3409
ptp4l[1590.232]: master offset        -15 s2 freq    +414 path delay      3411
ptp4l[1591.237]: master offset         79 s2 freq    +503 path delay      3402
ptp4l[1592.242]: master offset        -41 s2 freq    +407 path delay      3402
ptp4l[1593.247]: master offset        -52 s2 freq    +383 path delay      3423
ptp4l[1594.252]: master offset        -31 s2 freq    +389 path delay      3407
ptp4l[1595.257]: master offset         24 s2 freq    +435 path delay      3407
ptp4l[1596.262]: master offset         91 s2 freq    +509 path delay      3405
ptp4l[1597.267]: master offset         24 s2 freq    +469 path delay      3417**

Could you please help me with this problem ?

I also have a question regarding the NXP openIl User guide. Do you know how they got the graphics named : **Offset from master in start up state** and **Offset from master in stable state** section 4.4.6 p 33

Thanks in advance !

vladimiroltean, 2018-07-16:
This is typical output of the ptp4l program.
On the master, the `master sync timeout` message means that the waiting timeout for sending a new SYNC message (part of the 1588 protocol) has expired, and a new SYNC packet can therefore be sent.
On the other board, I can see it correctly transitioning from the MASTER -> UNCALIBRATED -> SLAVE states and correctly adjust its PHC via its servo algorithm.
It is normal behavior that only the slave clock to be adjusted, since the master is the reference.

vladimiroltean, 2018-07-16:
One possible way to plot the master-to-slave offsets over time is to run the following on the slave board:
```
$ ptp4l -i eth0 -4 -m -l 7 | awk '/master offset/ {time=gensub(/^ptp4l\[(.*)\]:/, "\\1", "g", $1); print time, $4; }' > ptp4l.log
```
And on a Linux PC, after retrieving the `ptp4l.log` file:
```
$ gnuplot --persist -e 'set xlabel "Time (seconds)"; set ylabel "Offset from master (nanoseconds)"; set term svg; set output "ptp4l_offsets.svg"; plot "ptp4l.log" with lines;'
```

echizenr, 2018-07-16:
Thank you for your quick answer. It will help me a lot.

I'm a new user so could you please explain me what is PHC ?

Moreover is there a way or an equipment that can be use to precisely measure the time that a message will take to get through the network.

Thanks in advance !

vladimiroltean, 2018-07-16:
PHC is PTP Hardware Clock. On the LS1021, this is a software construct (the `/dev/ptp0` char device in Linux) that manages access to the hardware TMR_CNT register of the eTSEC.

The 1588 standard mentions 2 methods to approximate the network delay in "software":
 * end-to-end: Master and slave exchange DELAY_REQUEST and DELAY_RESPONSE messages. May be inaccurate if master and slave are not directly connected, and if network delay is asymmetrical
 * peer-to-peer: All directly connected nodes in the network exchange PDELAY_REQUEST and PDELAY_RESPONSE messages. Asymmetrical link delays are also solved by keeping bidirectional peer delay information in every node (as opposed to end-to-end where only the slave keeps that info).
Works better in the ideal case, but the whole network needs to be aware of peer-to-peer delay measurements.

The above software-based network delay measurement methods are used by stacks like ptp4l. They have the disadvantage that the clocks need to already be somewhat in sync for the measurements to be relevant.

The Meinberg blog has a nice blogpost that gives some more details: https://blog.meinbergglobal.com/2013/09/19/end-end-versus-peer-peer/

Alternatively, it may be possible that network testing equipment manufacturers (Ixia, Spirent etc) have dedicated hardware-based solutions for measuring this delay. I am not up-to-date with their offering unfortunately.

echizenr, 2018-07-17:
Thank you for your quick answer and your help.

I will try your graphic solution and gather more information on network testing equipment manufacturers like Ixia or Spirent.

I close this Issue

closed 2018-07-17 by echizenr

vladimiroltean, 2018-07-17:
Aditionally, NXP has this presentation which I personally find very helpful in understanding the hardware support for 1588: http://cache.freescale.com/files/training/doc/ftf/2014/FTF-NET-F0071.pdf

echizenr, 2018-07-18:
Thanks I will take a look at it and hope it will help me understand the hardware support for 1588.