nxp-archive/openil #8: TSN demo : Test and LBT

echizenr, 2018-08-24:
Hello,

I have a question regarding the TSN demo.

I did the tests in the TSN demo (except the synchronized Qbv) and I would need some clarification about the Iperf flows used. Indeed, when I run the demo: standard, scheduling, prioritizing, policing from the LBT and disconnects the serial link between the PC and the LS1021ATSN cards, the Iperf streams are stopped. The same thing happens when I disconnect the Ethernet link between card 1 and the standard management switch (not TSN). In addition, when I disconnect the Ethernet links connecting the eth0 interfaces of cards 2 and 3 to the interfaces of the standard management switch, the LBT continues to work and returns me results from these two cards while the connection is interrupted. When I disconnect only the cable connecting the interface eth0 of the card 2 to the interface of the standard management switch and run a test from the LBT application, it returns an error code (which is normal).

I would therefore like to know who generates the Iperf flows and by which link they are transmitted (is it the Boards that generate them or the PC with the LBT interface?).
In addition, I would like to know the links type that are connected the SJA1105 and the LS1021A ARM SoC? Is the information of each RGMII port connected to the eth2 to 5 interfaces transmitted to the LS1021A ARM SoC on a single link or is it done differently?

Finally, I would like to know if you have a LS1021ATSN card datasheet and a user manual for the LS1021ATSN socket explaining the hardware part of the card and the transmission paths for the demonstration streams?

I thank you in advance.

vladimiroltean, 2018-08-24:
The source code for the entire Latency and Bandwidth Tester web app is available [here](https://github.com/openil/openil/tree/master/package/nodejs-lbt/src). It is written in Javascript and it is only a convenience wrapper over shell commands (to make the demo easier to run). There is a server.js that runs on the board and a client.js that runs in your browser. 
The job of the LBT server is to initiate iperf (or ping) sessions from the source IPs you specify, to the destinations. For this, it needs to run commands remotely on the other machines, and it does this over SSH. The interfaces through which traffic passes are decided by the standard Linux IP routing rules (nothing special done there).

vladimiroltean, 2018-08-24:
The port numbering of the SJA1105 is somewhat of a mess. But it is actually described in [this table](https://github.com/openil/sja1105-tool#port-numbering-on-the-nxp-ls1021atsn-board) in the sja1105-tool README. The internal switch port (4) is connected to eth2 and the others to the BCM5464R, which is a quad PHY. All switch port connections are RGMII (1x MAC-to-MAC and 4x MAC-to-PHY).
I'm not up to date with what is the externally shareable info for the LS1021A-TSN board. I'll figure it out and come back to you.

echizenr, 2018-08-27:
I thank you for your answer.

So if I understand, Boards are the ones which send Iperf flow through an SSH connection made by the LBT , right ? But in that case, Why did I have those results ? If boards send Iperf flow, why when I disconnect the ethernet link between Board 2 and Board 3, Iperf flow is not interrupt on LBT output ?

Ok, I will wait your answer regarding LS1021ATSN datasheet. Also, could you ask for SJA1105 datasheet ?

I thank you in advance.

vladimiroltean, 2018-08-27:
> If boards send Iperf flow, why when I disconnect the ethernet link between Board 2 and Board 3, Iperf flow is not interrupt on LBT output ?

This behavior suggests to me that the iperf packets are being routed through the management network (the `eth0` ports) instead of the TSN network (the `eth2` ports). You should make sure your IP routing settings are correct and that the IP addresses that you put in the LBT are those of the `eth2` ports. You can also run `tcpdump -i eth2` to make sure packets arrive on the correct interfaces.

> Also, could you ask for SJA1105 datasheet ?

This one is public: [UM10944.pdf](https://www.nxp.com/docs/en/user-guide/UM10944.pdf)

echizenr, 2018-08-27:
> This behavior suggests to me that the iperf packets are being routed through the management network (the eth0 ports) instead of the TSN network (the eth2 ports). You should make sure your IP routing [...]

Yes I checked it and IP adresses that I put in the LBT are those of eth2 ports. So I dont understand what happened. Moreover, In the user guide, Instructions tells me to disconnect serial connection between boards and the host PC but when I do that, LBT traffic output is interrupt. According to what you said earlier, it's seams due to SSH connection failure (but in that case, maybe a modification of this part of user guide : 7.6 Hardware setup will be usefull to prevent misunderstanding).

vladimiroltean, 2018-08-27:
Please find attached the documentation for the LS1021A-TSN board
* [Reference Manual](https://github.com/openil/openil/files/2324611/VVDN_TSN_User.Reference.Manual.pdf)
* [Getting Started Guide](https://github.com/openil/openil/files/2324622/LS1021ATSN-PA_Getting_Started_Guide_A0-04.pdf)

vladimiroltean, 2018-08-27:
> Moreover, In the user guide, Instructions tells me to disconnect serial connection between boards and the host PC but when I do that, LBT traffic output is interrupt. According to what you said [...]

If you're talking about this portion:
> Now, disconnect the USB-serial cables and assemble the 3 boards, PC, and L2 switch in their final position: Place the 3 LS1021ATSN boards in a stack, with Board 1 on top and Board 3 at the bottom.

I don't see what's wrong with this. It doesn't say to disconnect the USB-serial cable while running the test. It says that you no longer need the serial cables (and therefore can remove them), and that you should access the board Linux shells using SSH from thereon.

echizenr, 2018-08-28:
> Please find attached the documentation for the LS1021A-TSN board
> 
>     Reference Manual
>     Getting Started Guide
> 
Thank you for the Ref Manual and Getting started Guide

> I don't see what's wrong with this. It doesn't say to disconnect the USB-serial cable while running the test. It says that you no longer need the serial cables (and therefore can remove them), and that you should access the board Linux shells using SSH from thereon.

Yes I'm talking about this portion. From what I understand and according to the TSN demo schematics, I thought that I must disconnect USB serial cable before running tests (once configuration scripts implemented). So that's what I thought, USB serial cable allow boards to initiate an SSH connection with Host PC in order to gives information selected in LBT.

Moreover, is it normal that the LBT does not immediately detect the disconnection of an ethernet link between the Boards and the regular L2 switch or between Boards themselves?

vladimiroltean, 2018-08-28:
Nothing is transmitted through the USB-serial cable, other than a serial console where a login shell is spawned. If you run `/usr/lib/node_modules/lbt/server.js` from a shell running over the serial console, and then remove the cable, it is completely expected that the process dies when you remove the cable.

> Moreover, is it normal that the LBT does not immediately detect the disconnection of an ethernet link between the Boards and the regular L2 switch or between Boards themselves?

It depends on what you mean by "immediately".

> I would like to know how Board 2 and 3 send results to the LBT server database in order to allow it to display those result in graphics of LBT application.

The LBT web server has the following application flow:
* Spawns a SSH login shell to the iperf server (the receiver of traffic)
* Opens an iperf server through SSH, and waits for the server to start listening
* Spawns a SSH login shell to the iperf client (the sender of traffic)
* Opens an iperf client through SSH. This triggers the traffic start.
* Collects the stdout of the remote iperf server process (over SSH). This is configured to print bandwidth statistics every half a second.
* Passes the collected bandwidth statistics to the feedgnuplot program, which is a Perl wrapper over gnuplot to make it real-time.
* Feedgnuplot generates an SVG image that represents the plotted bandwidth chart.
* Sends the generated SVG to the web client (your browser)
* Your browser renders it.

Therefore, SSH and iperf communication between the boards happens over the TSN network. Web (HTTP) communication with the PC happens over the management network.

echizenr, 2018-08-28:
> It depends on what you mean by "immediately".

If I disconnect the link between 2 boards, the LBT application detect the link missing approximately 1minute after (worst case : it doesn't detect it except if I stop the test and restart it).

Thank you for your answer, I have a question regarding the type of flow collected by the LBT (bandwidth statistics). What kind of flow is it and do you prioritize it (with a hight priority to be sure not losing trafic)? It is just to know if it's possible to lose information ; for example if I overload the network (Is there a particular mechanism prevent boards lose "bandwidth statistics" ?)

vladimiroltean, 2018-08-28:
> I have a question regarding the type of flow collected by the LBT (bandwidth statistics). What kind of flow is it and do you prioritize it (with a hight priority to be sure not losing trafic)?

It is SSH traffic with plaintext payload. We don't prioritize it, so it gets mixed up with the iperf.

> It is just to know if it's possible to lose information ; for example if I overload the network (Is there a particular mechanism prevent boards lose "bandwidth statistics" ?)

It is possible to prioritize it, yes, but the demo setup will change a little bit.
The SJA1105 switch assigns VLAN tags internally to packets based on the port it receives them from. So it can't distinguish between iperf and SSH traffic, because both these flows end up being tagged the same way, since they come from the same port.
What you have to do is tag one of these flows (iperf or SSH) with VLAN when you send them from the LS1021. The SJA1105 only puts the default per-port VLAN on untagged packets.
The 3-board synchronized Qbv demo makes use of this functionality. The [vlan-setup](https://github.com/openil/openil/blob/master/board/nxp/ls1021atsn/rootfs_overlay/usr/bin/vlan-setup) script, called from [/etc/network/interfaces](https://github.com/openil/openil/blob/master/board/nxp/ls1021atsn/rootfs_overlay/etc/network/interfaces), creates an additional VLAN tagged network over `eth2`. So if you send a `ping 172.15.0.1`, the ping packet will go through `eth2` and be untagged, but if you run `ping 172.15.100.1` the packet will go through `eth2.100` and get the VLAN tagging configured on that port.
I should probably explain what this line does:

```
ip link add link eth2 name eth2.100 type vlan id 100 egress 0:0 1:1 2:2 3:3 4:4 5:5 6:6 7:7
```

* Create a VLAN-tagged interface called `eth2.100` that maps over the `eth2` physical interface
* Tag packets that go through the `eth2.100` with VLAN ID 100
* Establish the following priority mapping on outgoing packets:

|  SKB (socket buffer) priority  |  VLAN priority  |
|--------------------------------|-----------------|
|                0               |        0        |
|                1               |        1        |
|                2               |        2        |
|                3               |        3        |
|                4               |        4        |
|                5               |        5        |
|                6               |        6        |
|                7               |        7        |

You may now ask: what is SKB priority and how do I set that?
Packets are called "socket buffers" in the Linux network stack and have this priority property that can be acted upon, for QoS, by the stack or the network card driver. By default, the SKB priority of all packets is 0, but we are applying this netfilter rule to mangle ICMP packets on the POSTROUTING chain, and modify their SKB priority (or traffic class - same thing) from 0 to 7 (highest priority). It looks like this:
```
iptables -t mangle -A POSTROUTING -p icmp -j CLASSIFY --set-class 0:7
```

So what happens is:
* You send a `ping 172.15.100.1`.
* The packet gets routed to the `eth2.100` interface, based on the match on the netmask (`ip addr show eth2.100` shows `172.15.0.2`)
* The packet hits the POSTROUTING chain, and the CLASSIFY target from the mangle table is hit. Its SKB priority is remapped from 0 to 7.
* The packet is tagged with VLAN ID 100 and PCP 7, based on the `ip link` settings, and gets forwarded to the driver of the `eth2` interface.
* The switch recognizes the VLAN header from the ping packet, so:
  - it doesn't apply the default tagging
  - it places the in the highest priority queue / traffic class (7) where it is forwarded before any other packet from lower priority queues
  - it can apply custom QoS rules on it (policing, shaping), as well as Qbv time-based scheduling.

vladimiroltean, 2018-08-29:
Not available on the SJA1105 hardware.
Of the features you mentioned, the upcoming LS1028 chip will support:
* Frame preemption on the Ethernet controller ports
* Frame preemption and seamless redundancy on the integrated L2 switch