nxp-archive/openil #11: LBT Synchronized Qbv

echizenr, 2018-09-04:
Hello

I'm currently trying to run the Synchronized Qbv demo and I faced a problem.

When I run the following command on the 3 boards : 

> ptp4l -i eth0 -p /dev/ptp0 -m -l 7 -t 1000

Boards are displaying the following output : 

> ptp4l[1100.545]: sja1105: sync timer timeout
> ptp4l[1100.545]: sja1105 reset requested
> src/lib/ptp/ptp.c@221: sja1105_ptp_reset: Resetting PTP clock
> src/lib/ptp/ptp.c@313: sja1105_ptp_clk_get: sja1105_ptp_clk_get: returned zero
> ptp4l[1100.546]: rc1 0 rc2 -11 rc3 0
> ptp4l[1100.546]: sja1105: calculating got time error
> ptp4l[1100.675]: sja1105: sync timer timeout
> ptp4l[1100.675]: sja1105 reset requested
> src/lib/ptp/ptp.c@221: sja1105_ptp_reset: Resetting PTP clock
> src/lib/ptp/ptp.c@313: sja1105_ptp_clk_get: sja1105_ptp_clk_get: returned zero
> ptp4l[1100.676]: rc1 0 rc2 -11 rc3 0
> ptp4l[1100.676]: sja1105: calculating got time error

Moreover, when I enable the 1-hop flow (ping), the following message appear on LBT Web Application : 
[image]

Boards are able to ping each other from their eth2 interfaces but not from eth2.100

Could you help me please ?

I thank you in advance.

vladimiroltean, 2018-09-04:
> src/lib/ptp/ptp.c@221: sja1105_ptp_reset: Resetting PTP clock
> src/lib/ptp/ptp.c@313: sja1105_ptp_clk_get: sja1105_ptp_clk_get: returned zero
> ptp4l[1100.546]: rc1 0 rc2 -11 rc3 0
> ptp4l[1100.546]: sja1105: calculating got time error

I have seen these errors in the past on some boards. It may be due to an SPI timing violation. What is your kernel version? Do you have [this patch](https://github.com/openil/linux/commit/d2672c6a33cdbc0ea7a6fb79a81b920c216a8e98) applied? Also see this [NXP community response](https://community.nxp.com/message/1052443?commentID=1052443&et=watches.email.thread#comment-1049339).

vladimiroltean, 2018-09-04:
> Boards are able to ping each other from their eth2 interfaces but not from eth2.100

To pass VLAN 100 traffic, all 3 switches need to have VLAN 100 configured in their XML. The standard XML's don't have it, only the ones generated by the scripts for this demo do. Which XML configurations do you have loaded into the switches?

echizenr, 2018-09-04:
It's not really a packet loss. It seams that the SSH connection can't be establish.

> sja1105-tool status port

Display : 

[image]

> Compile the latest kernel from the master branch and it will be included.

When you said "Compile", where am I supposed to compile it ? (it seams that I didn't used kernel from any github branch since I started my tests)

echizenr, 2018-09-05:
Really grateful for your reply. I understand now exactly what happened.

Ok, so I need to rebuild the whole sdcard.img in order to make it work, am I right ?

vladimiroltean, 2018-09-05:
Rebuild the whole sdcard.img, replace just uImage and ls1021a-tsn.dtb files on the first partition (FAT32).

echizenr, 2018-09-05:
Ok I will do that.

Thank you for your answer !

echizenr, 2018-09-05:
I got a problem during sdcard.img building. Indeed, I cloned OpenIL201808 and run : 

> make nxp_ls1021atsn_defconfig 

I obtain the following : make: Nothing to be done for 'nxp_ls1021atsn_defconfig'.

And when I just run make 2>&1, the terminal reply my the following thing : 

> mkdir -p /home/alain/openil-OpenIL-201808/output/build/buildroot-config/lxdialog
> PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
>     obj=/home/alain/openil-OpenIL-201808/output/build/buildroot-config -C support/kconfig -f Makefile.br mconf
> make[1]: Entering directory '/home/alain/openil-OpenIL-201808/support/kconfig'
> /usr/bin/gcc -DCURSES_LOC="<curses.h>" -DLOCALE  -I/home/alain/openil-OpenIL-201808/output/build/buildroot-config -DCONFIG_=\"\"  -MM *.c > /home/alain/openil-OpenIL-201808/output/build/buildroot-config/.depend 2>/dev/null || :
> /usr/bin/gcc -DCURSES_LOC="<curses.h>" -DLOCALE  -I/home/alain/openil-OpenIL-201808/output/build/buildroot-config -DCONFIG_=\"\"   -c conf.c -o /home/alain/openil-OpenIL-201808/output/build/buildroot-config/conf.o
>  *** Unable to find the ncurses libraries or the
>  *** required header files.
>  *** 'make menuconfig' requires the ncurses libraries.
>  *** 
>  *** Install ncurses (ncurses-devel) and try again.
>  *** 
> Makefile:199: recipe for target '/home/alain/openil-OpenIL-201808/output/build/buildroot-config/dochecklxdialog' failed
> make[1]: *** [/home/alain/openil-OpenIL-201808/output/build/buildroot-config/dochecklxdialog] Error 1
> make[1]: Leaving directory '/home/alain/openil-OpenIL-201808/support/kconfig'
> Makefile:813: recipe for target '/home/alain/openil-OpenIL-201808/output/build/buildroot-config/mconf' failed
> make: *** [/home/alain/openil-OpenIL-201808/output/build/buildroot-config/mconf] Error 2

But it seams that ncurses can't be found when I try to install it

echizenr, 2018-09-05:
problem solved with 

> sudo apt-get install libncurses5-dev

echizenr, 2018-09-05:
When I have to insert the SDcard into the board (LS1021ATSN getting started : booting from SD card), to set the environment variables, should I need to complete those steps : 

> Setenv bootargs console=ttyS0, 115200 root=/dev/mmcblk0p1 rootwait
> Setenv load_kernel ext2load mmc :1 82000000 boot/uImage-ls1021atsn.bin
> Setenv load_dbt ext2load mmc :1 8f000000 boot/uImage-ls1021a-tsn.dtb
> Setenv bootcmd 'mmc rescan; setenv bootcmd 'mmc rescan; run load_kernel load_dtb; bootm 82000000 -8f000000';
> saveenv
> run bootcmd

echizenr, 2018-09-06:
Thank you, the sync work now but I have the same mistake than before (VLAN 100).

SSH connection can't be establish.

However, I load qbv-ptp-board{i} (i = 1, 2,3) on each board with those command : 

> sja1105-tool config default ls
> sja1105-tool config save qbv-ptp-board{i}
> sja1105-tool config load qbv-ptp-board{i}

Do you have a solution to fix it ?

echizenr, 2018-09-06:
I also have another question regarding Synchronized Qbv test. Synchronization between boards is done throught eth0 interfaces right ? Are PTP messages transmitted over management network or TSN ?

The purpose of this question is to know if it's a software synchronisation or if TAS of SJA1105 are synchronized over TSN network when I run the following command : 

> ptp4l -i eth0 /dev/ptp0 -m -l 7 -t 1000

echizenr, 2018-09-07:
VLAN 100 and SSH connection failed problem are solved. I change for master branch of sja1105-tool and it's working now. 

Just a little thing, when I run 1 and 3 hop (ICMP), sometimes the PIT value rise to 60 or 90 ms. Is it normal ? Do you know what this it about ?

vladimiroltean, 2018-09-10:
> Just a little thing, when I run 1 and 3 hop (ICMP), sometimes the PIT value rise to 60 or 90 ms. Is it normal ? Do you know what this it about ?

If the packet interarrival times reported by ping are not exactly equal to the length of the Qbv schedule (30 ms), it can mean 2 things:

- the ping packet missed the synchronized timeslot on at least 1 switch, leading to the 2x delay (60 ms) or on both switches, leading to 3x delay (90 ms). Could you please inspect the synchronization values reported by ptp4l?
- the ping packets were not **sent** in time to catch consecutive timeslots configured on the switches. Keep in mind that at the moment, there is no real-time guarantee for ping packet delivery within the LS1021 network stack for this demo. Generally this happens if the network stack is overloaded with other traffic to process. If you are under this scenario, then the PIT value should go back to 30 when you don't send any other background traffic, and increase to 30 or 60 when you start iperf.

echizenr, 2018-09-11:
> the ping packet missed the synchronized timeslot on at least 1 switch, leading to the 2x delay (60 ms) or on both switches, leading to 3x delay (90 ms). Could you please inspect the synchronization values reported by ptp4l?

I see, so it could be due to a sync problem between Boards.

> Generally this happens if the network stack is overloaded with other traffic to process. If you are under this scenario, then the PIT value should go back to 30 when you don't send any other background traffic, and increase to 30 or 60 when you start iperf.

Yes I saw when I only run ping test that PIT value is 30. When I also run Iperf flow, PIT value could rise to 60 or 90.

> Why did you chose to perform synchronization over separate cabling ? It is possible to perform this synchronization through TSN network without degrading performance ?

Regarding my previous question above, do you have an explanation to provide me ? It is more interesting for us to perform synchronization through TSN network if it could be done without degrading performance. 

Also, I've got a problem with policing.xml file, I can't run the test, as if configuration wasn't take into account (flow 1 is limited to 600 mbps (normal behavior). Except that, both flow acting like in standard configuration. I've tried the master branch, the 1.0 release and 201712 without solving the problem. Did you modified sja1105-tool during last month ? Because I was able to run this tests before. Also I've tried to modified the policing.sh script in order to add an other configured flow but it doesn't work (maybe a problem during xml generation).

Also, in 802.1Qci, is it possible to configure Time slice per VLAN number (not only per VLAN priority) ?

vladimiroltean, 2018-09-11:
> Yes I saw when I only run ping test that PIT value is 30. When I also run Iperf flow, PIT value could rise to 60 or 90.

This means that the problem are the endpoints (the LS1021 chips). Iperf packet processing is causing many interrupts which prevent the ping packets from being sent/received in time. Increasing the guard band time and decreasing the iperf band time [in here](https://github.com/openil/sja1105-tool/blob/master/src/helpers/configs/synchronized-qbv/qbv-ptp.sh#L78) should tell for sure whether this is the problem. We are facing this demo limitation because LS1021 does not currently have a real-time network stack.

> Why did you chose to perform synchronization over separate cabling ? It is possible to perform this synchronization through TSN network without degrading performance ?

I have not done the test, but I think it should be possible to switch the PTP synchronization network to eth2. Congestion management with the iperf traffic would need to be handled somehow, though.

> Also, in 802.1Qci, is it possible to configure Time slice per VLAN number (not only per VLAN priority) ?

In the implementation on the SJA1105, no. VLAN ID alone is not a selection criterion for flows.

echizenr, 2018-09-12:
Ok I will try to increase Guard Band time.

> I have not done the test, but I think it should be possible to switch the PTP synchronization network to eth2. Congestion management with the iperf traffic would need to be handled somehow, though.

Thank you, the purpose of this question was to know if it was possible to have only one network through which the PTP synchronization and the flows would pass.

> In the implementation on the SJA1105, no. VLAN ID alone is not a selection criterion for flows.

The use case here : if two Endpoints (A & B) send flows, have the same VLAN priority and for example, Endpoint B start to acting weirdly (send to much flow and prevent Endpoint A to send flow). Does the switch be able to identify which Endpoint acting weirdly and discard it's flows ?

vladimiroltean, 2018-09-12:
> the purpose of this question was to know if it was possible to have only one network through which the PTP synchronization and the flows would pass.

In the general case this is something that should indeed work, however the demo does not cover this usage due to concerns that there were already too many things that could go wrong.

> if two Endpoints (A & B) send flows, have the same VLAN priority and for example, Endpoint B start to acting weirdly (send to much flow and prevent Endpoint A to send flow). Does the switch be able to identify which Endpoint acting weirdly and discard it's flows ?

I think you're looking at Virtual Links here. I remember quite a while ago playing around with this sort-of-ARINC664, relatively obscure feature of the SJA1105. I actually found a writeup I did for an old version of the 3-board TSN demo, before it turned into synchronized Qbv. I don't remember much but I can leave you a link: [Open Industrial Linux User Guide Release v0.2.pdf](https://github.com/openil/openil/files/2375989/Open.Industrial.Linux.User.Guide.Release.v0.2.pdf). See chapter "**7.2 TTEthernet (SAE AS6802) features**" for an explanation of what the hw can do, and let me know if this is what you're looking for.

echizenr, 2018-09-13:
> In the general case this is something that should indeed work, however the demo does not cover this usage due to concerns that there were already too many things that could go wrong.

Sure, in any case, if it's possible it's ok for me because it's easier and better for us to deploy all TSN things over one network.

> I think you're looking at Virtual Links here. I remember quite a while ago playing around with this sort-of-ARINC664, relatively obscure feature of the SJA1105. I actually found a writeup I did for [...]

I was not particularly interested in setting up VLs because of the heaviness of their implementation. I am trying to prove the flows segregation on the TSN network. If there is another mechanism as robust as the VL to do so, I am very interested. Otherwise, although VL requires a very heavy implementation, it is perfectly suitable since this type of communication is already implemented in civil avionics for the flows segregation (ARINC 664 P7).

However, are VLs compatible with other features of TSN? (synchronization, TAS, Qbv Scheduling and Qci)
Is the "7.2 TTEthernet (SAE AS6802) features" part of user guide v0.2 still available on the latest releases of the OpenIL and sja1105-tool master branch?

echizenr, 2018-09-13:
> Not compliant per se with IEEE stuff, but TTEthernet looks similar to Qbv (scheduled traffic). In the Schedule Table I see that it should be possible to schedule time-triggered virtual links onto timeslots, but personally I never tested it and can't give you an exact demo of how it works.

Ok, that could be a problem for civilian avionics because the network should be segregate. I will continue my investigation and see if it's possible to prove the segregation over TSN network.

> Not tested recently, but configuring virtual links with the latest sja1105-tool should hopefully still work.

I hope so, I will try to run it and see how it work.

vladimiroltean, 2018-09-26:
> How does the time aware scheduler of the switch get the network clock (master or slave)?

In sja1105-tool, the `schedule-entry-points-params` field `clksrc` denotes the source for the Qbv clock. In the synchronized Qbv demo we set it to "_11 selects the internal PTP clock which is under the control of the host as clock source._"
In practice, the Qbv engine of the switch uses the 64-bit `PTPCLKVAL` register as a time base.
We don't have a standard API at the moment for controlling `PTPCLKVAL`, but instead there are hooks in our patched `ptp4l` program which keep the switch clock in sync with the Ethernet clock. [This file](https://github.com/openil/linuxptp/blob/master/sja1105.c) does that.

> is it through the serial interface between the subscriber managed by the LS1021atsn and the sja1105

Through SPI, yes.

> are there any absorption mechanisms under switch register level ?

Could you please rephrase this? I don't understand what you mean.

echizenr, 2018-09-26:
thank you for your answer. It's precise and clear for me now.

> Could you please rephrase this? I don't understand what you mean.

Is it necessary to have an Endpoint and a switch in the same board in order to have a clock synchronization or does the switch have an absorption mecanisms to capture and use the clock value in switch register table in order to synchronize it ? (I do not know if it's more explicit now, I had myself some trouble to understand my boss's question)

echizenr, 2018-09-26:
Thank you very much for your answer, it is complete and clear.

> I don't know of any switches that have the feedback loop embedded inside them and perform this operation autonomously. If you do, please let me know.

Today, I do not know any either. However, in the context of a civil avionics network, this could be interesting. Indeed, if there is a way to "pass" the `correctionField` to the switch without a dedicated processor or synchronize the switch like if was acting as an ordinary clock, this could be very attractive for avionic.

echizenr, 2018-11-05:
Hello,

I'm currently trying to rebuild the TSN demo and I've a problem with a server error return from LBT when I try to run the 1 Hop and 3 Hops ping test. It seems that the problem come from VLAN 100 with the following error : 

[image]

Last time, I had solved the problem by switching for the sja1105 master branch but now I'm currently use it so if you have an idea to how I can fix this problem it will help me a lot. Perhaps I need to separate the source and destination IP from source and destination Login.

vladimiroltean, 2018-11-05:
Is VLAN 100 present in the switch configuration XML? If it is, could you post the XML?

echizenr, 2018-11-05:
I'm not sure but I think it is. 

Here you can find XML files for each boards : 

[qbv-ptp-board1.txt](https://github.com/openil/openil/files/2548930/qbv-ptp-board1.txt)
[qbv-ptp-board2.txt](https://github.com/openil/openil/files/2548932/qbv-ptp-board2.txt)
[qbv-ptp-board3.txt](https://github.com/openil/openil/files/2548933/qbv-ptp-board3.txt)

echizenr, 2018-11-06:
In **/etc /init.d** There is no S45vlan script. So I created it but I'm not sure it was taken into account during the testing process.

vladimiroltean, 2018-11-06:
There is no longer any S45vlan script.
VLAN networks are created like this now (/etc/network/interfaces):
```
auto eth2
iface eth2 inet static
        address 172.15.0.1
        netmask 255.255.255.0
        up   /usr/bin/vlan-setup start 172.15.100.1/24
        down /usr/bin/vlan-setup stop
```
The vlan-lookup-table in the switches looks ok. You have VLANID 0 and VLANID 100 configured.
Do you have your IP addresses configured ok in /etc/network/interfaces? The `vlan-setup` should assign 172.15.100.{1,2,3} on eth2.100 on the 3 boards. Is that the case for you?
Can you `ping -I eth2.100 172.15.100.3` from board 1?

vladimiroltean, 2018-11-06:
What happens if you run:
```
ifdown eth2
ifup eth2
```
?

Correct behavior should be:
```
[root@OpenIL:~]# ifdown eth2
Removing VLAN-tagged interface: OK
Removing iptables rules: OK
[root@OpenIL:~]# ifup eth2
[698876.234727] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
Creating VLAN-tagged interface: [698876.257348] IPv6: ADDRCONF(NETDEV_UP): eth2.100: link is not ready
OK
Adding iptables rules: OK
[root@OpenIL:~]# [698877.287952] fsl-gianfar soc:ethernet@2d90000 eth2: Link is Up - 1Gbps/Full - flow control off
[698877.296744] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[698877.303640] IPv6: ADDRCONF(NETDEV_CHANGE): eth2.100: link becomes ready
```

echizenr, 2018-11-06:
I don't have it.

Here you could find mine : 

[image]

echizenr, 2018-11-06:
Yes I have the same thing.
[image]

echizenr, 2018-11-06:
I'm running this test on the same board but I will try with the full test bench when OpenIL sdcard.img will be done.

echizenr, 2018-11-06:
It seems that I have an error when I try to rebuild the sdcard.img : 

cc1: all warnings being treated as errors
CMakeFiles/open62541-object.dir/build.make:203: recipe for target 'CMakeFiles/open62541-object.dir/open62541.c.o' failed
make[3]: *** [CMakeFiles/open62541-object.dir/open62541.c.o] Error 1
make[3]: Leaving directory '/home/Documents/openil-master/output/build/libopen62541-v0.2'
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/open62541-object.dir/all' failed
make[2]: *** [CMakeFiles/open62541-object.dir/all] Error 2
make[2]: Leaving directory '/home/Documents/openil-master/output/build/libopen62541-v0.2'
Makefile:149: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/Documents/openil-master/output/build/libopen62541-v0.2'
package/pkg-generic.mk:216: recipe for target '/home/Documents/openil-master/output/build/libopen62541-v0.2/.stamp_built' failed
make: *** [/home/Documents/openil-master/output/build/libopen62541-v0.2/.stamp_built] Error 2

Do you know to what it's due ?

vladimiroltean, 2018-11-06:
Not unless you print the full output of:
```
rm -rf dl/libopen62541-v0.2.tar.gz
rm -rf output/build/libopen62541-v0.2/
make libopen62541
```
I just tried that and I'm able to build it successfully.

echizenr, 2018-11-06:
Sorry ! 

It seems to work by making **sudo make** instead of **make**

Just to be sure I will try your solution.

closed 2018-11-07 by echizenr