nxp-archive/openil #12: node_modules in not available in the  /usr/lib/ LS1021ATSN-PA board

rubidebnath, 2018-09-18:
Hi,

I am trying to run the TSN and all other demos on the LS1021ATSN-PA board according to the OPENIL_UG.pdf. 
To check the Latency and bandwidth tester I want to run server.js and hence followed the process which is as follows.

[root@openil] $ /usr/lib/node_modules/lbt/server.js

But there is not node_modules inside the lib folder.

Would someone be kind enough to tell me how to fix this?

Thanks in advance.

Best Regards,
Rubi

vladimiroltean, 2018-09-18:
What software do you have on the board?

rubidebnath, 2018-09-18:
The board has the Linux operating system which was pre-loaded in the SD card. It was provided by the vendors from whom we bought the device. I can boot the device using USB and also via Ethernet port. I generated the standard.xml files and also all the other demo XML files. The only thing that I am missing is the node_modules inside my lib folder. 

I am very new to this so might be I am missing something.

Thanks

vladimiroltean, 2018-09-18:
Please try to compile OpenIL for the LS1021A-TSN board from the sources located here on Github (preferably from the OpenIL-201808 tag). Unfortunately the matter of fact is that we don't have a very precise way of controlling what software gets shipped with the boards. We'll take that into consideration as a separate topic, but currently the most productive thing to do is to go ahead and recompile the operating system from a well known and supported version.

closed 2018-11-22 by vladimiroltean