nxp-archive/openil #22: Layer 2 sync problem

Meng0527, 2019-04-11:
Hello，
When I run the following command on two boards；
ptp4l -i eth0 -m -2
One of the board displaying continuously the following output:
ptp4l[274756.039]: selected /dev/ptp1 as PTP clock
ptp4l[274756.076]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[274756.076]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[274762.975]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[274762.976]: selected local clock 6805ca.fffe.21d777 as best master
ptp4l[274762.977]: assuming the grand master role
ptp4l[274810.082]: port 1: bad message
ptp4l[274811.082]: port 1: bad message
ptp4l[274812.082]: port 1: bad message
ptp4l[274813.082]: port 1: bad message
ptp4l[274814.083]: port 1: bad message
ptp4l[274815.083]: port 1: bad message
The same problem occurs when I use two servers connected by sja1105 or one server and one board for Layer 2 time synchronization.
L3 sync works well.
Could you please help me with this problem ?
I also have a question with  PTPSCHTIME parameter，in UM10944,Table 60. PTP control registers 2 to 6,it says this parameter specifies the time on the synchronized PTP clock at whichthe switch integrates into the schedule.But this parameter is not readable or writable. How is it generated and works?
Thanks！

Meng0527, 2019-04-11:
Thid is the frame I captured when I use  one server and one board for Layer 2 time synchronization.

[error frame.zip](https://github.com/openil/openil/files/3069204/error.frame.zip)

vladimiroltean, 2019-04-11:
Hello Meng,
Thank you for your report. I think we have seen this before in a different environment.
The OpenIL kernel uses this patch backported from 4.17.0-rc4: [packet: in packet_snd start writing at link layer allocation](https://github.com/openil/linux/commit/01a658c1b9d4b5393c38d5a92d9112ab1425382a). It looks like this patch is introducing a regression: the network stack is increasing skb->len past the actual frame size.
The upstream fixup came in 4.17.0-rc6: [packet: fix reserve calculation](https://github.com/torvalds/linux/commit/9aad13b087ab).
I don't see the fixup commit backported in the OpenIL kernel. Could you please cherry-pick it and see if it fixes your issue?

vladimiroltean, 2019-04-11:
> I also have a question with PTPSCHTIME parameter，in UM10944,Table 60. PTP control registers 2 to 6,it says this parameter specifies the time on the synchronized PTP clock at whichthe switch integrates into the schedule.But this parameter is not readable or writable. How is it generated and works?

What do you mean PTPSCHTIME is not readable nor writable? It is not accessible in the sja1105-tool (since it's useless to set it statically), but accessible at runtime [through the libsja1105](https://github.com/openil/linuxptp/blob/master/sja1105.c#L305).
Please also check out [this response](https://github.com/openil/sja1105-tool/issues/45#issuecomment-470951954) for more details about how the time-aware scheduler works.