nxp-archive/openil #26: Problem with time sync

Norman03, 2019-09-16:
Hello, 

I have synchronized the global NTP time by running 'ntp' service in the switch and by connecting the internet to **eth1**  in the non-tsn switch ports. Now I used command 'date' to verify the current global time and I'm able to see current time successfully in the switch. 

Next, I have ran the ptp master command in the switch and ptp slave command in other PC's connected to the sja1105 switched ports. 
Ptp master: `ptp4l -i eth2 -2 -mq`
Ptp slaves: `ptp4l -i enp3s0 -2 -mq -s`

Now, my slaves are getting epoch time because the master is sharing the epoch timestamp to other slaves. My questions/doubts are:

1. How to configure the /dev/ptp0 to use the NTP time in the switch; so that the switch shares the ptp time(NTP global time rather epoch) with other ptp slaves? as of now /dev/ptp0 has the epoch timestamp.

**Switch Hardware:**
LS1021ATSN switch 
** Switch Software:**
Kernel: 4.14.47-ipipe
openIL version: OpenIL-201904

vladimiroltean, 2019-09-17:
So if you run `phc_ctl CLOCK_REALTIME`, you get a date in 2019, but if you run `phc_ctl /dev/ptp0` you get a date in 1970?
How about you run a command such as `phc2sys -a -r -r -m`? That will detect whether there is a ptp4l master instance on the board, and serve the CLOCK_REALTIME over through /dev/ptp0 and into the PTP network. In your case, you mentioned that CLOCK_REALTIME is already being disciplined by NTP.

Norman03, 2019-09-17:
> So if you run `phc_ctl CLOCK_REALTIME`, you get a date in 2019, but if you run `phc_ctl /dev/ptp0` you get a date in 1970?
Yes I get this in my slave PC's.
> How about you run a command such as `phc2sys -a -r -r -m`? That will detect whether there is a ptp4l master instance on the board, and serve the CLOCK_REALTIME over through /dev/ptp0 and into the PTP network. In your case, you mentioned that CLOCK_REALTIME is already being disciplined by NTP.

I don't see PHC2SYS package available in the switch.

vladimiroltean, 2019-09-17:
> I don't see PHC2SYS package available in the switch.

I stopped keeping track of which linuxptp applications does each version of OpenIL ship. The 'linuxptp' package in upstream Buildroot ships them all, including phc2sys. If you use the OpenIL fork I mentioned in the other thread it will be there.

closed 2020-08-06 by vladimiroltean