nxp-archive/openil #27: frequent offset exceeded in LS1021ATSN synchroniyed-qbv demo

laravich, 2019-09-17:
Hello,

i am trying the synchronized qbv demo with 3 LS1021ATSN boards as given in the below openil guide (https://www.openil.org/news/files/OpenIL_User_Guide_Rev1.2.pdf)
The output of the ptp synchronization is not stable after some time. i observe a huge offset values frequently only on the board 2 specifically. due to which the qbv monitor of the board2 goes to "state disabled" state and it takes 3 sec to come back to 2state running" state. 
I also observe that in some scenarios the board 2 qbv monitor goes to "surprisingly disabled" state even without the offset exceeding. i have shown the ptp4l log below:

ptp4l[5451.581]: sja1105 ratio: 1.008937
ptp4l[5451.581]: sja1105_qbv_monitor: state running
ptp4l[5451.582]: time since started: [238.366216464]
ptp4l[5451.627]: port 1: delay timeout
ptp4l[5451.627]: Received Delay_Resp: correction 0 ns
ptp4l[5451.627]: delay   filtered       3023   raw       2939
ptp4l[5451.707]: sja1105: sync timer timeout
ptp4l[5451.708]: sja1105: offset    213507 ns, delay    160137 ns
ptp4l[5451.709]: sja1105 ratio: 0.998029
ptp4l[5451.709]: sja1105_qbv_monitor: state running
ptp4l[5451.709]: sja1105_qbv_monitor: surprisingly stopped
ptp4l[5451.834]: sja1105: sync timer timeout
ptp4l[5451.835]: sja1105: offset    -40635 ns, delay    159702 ns
ptp4l[5451.835]: sja1105 ratio: 1.000497
ptp4l[5451.836]: sja1105_qbv_monitor: state disabled
ptp4l[5451.962]: sja1105: sync timer timeout
ptp4l[5451.963]: sja1105: offset     13132 ns, delay    159837 ns
ptp4l[5451.963]: sja1105 ratio: 0.999972
ptp4l[5451.963]: sja1105_qbv_monitor: state enabled not running
ptp4l[5451.964]: time to start: [2.901344872]
ptp4l[5452.089]: sja1105: sync timer timeout
ptp4l[5452.090]: sja1105: offset      1425 ns, delay    161710 ns
ptp4l[5452.090]: sja1105 ratio: 1.000083
ptp4l[5452.090]: sja1105_qbv_monitor: state enabled not running
ptp4l[5452.091]: time to start: [2.774262128]
ptp4l[5452.136]: Received Sync: correction 0 ns
ptp4l[5452.136]: Received Follow_Up: correction 0 ns
ptp4l[5452.136]: master offset        -70 s2 freq  -47731 path delay      3023
ptp4l[5452.216]: sja1105: sync timer timeout
ptp4l[5452.217]: sja1105: offset      2562 ns, delay    160092 ns
ptp4l[5452.217]: sja1105 ratio: 1.000071
ptp4l[5452.217]: sja1105_qbv_monitor: state enabled not running
ptp4l[5452.218]: time to start: [2.647250088]
ptp4l[5452.343]: sja1105: sync timer timeout
ptp4l[5452.344]: sja1105: offset      2933 ns, delay    159295 ns
ptp4l[5452.344]: sja1105 ratio: 1.000065
ptp4l[5452.344]: sja1105_qbv_monitor: state enabled not running
ptp4l[5452.345]: time to start: [2.520312384]
ptp4l[5452.470]: sja1105: sync timer timeout
ptp4l[5452.471]: sja1105: offset      4925 ns, delay    154187 ns
ptp4l[5452.471]: sja1105 ratio: 1.000042
ptp4l[5452.471]: sja1105_qbv_monitor: state enabled not running
ptp4l[5452.472]: time to start: [2.393336704]
ptp4l[5452.597]: sja1105: sync timer timeout
ptp4l[5452.598]: sja1105: offset     -1959 ns, delay    160762 nsptp4l[5452.598]: sja1105_qbv_monitor: state enabled not running
ptp4l[5452.599]: time to start: [2.266347760]
ptp4l[5452.724]: sja1105: sync timer timeout
ptp4l[5452.725]: sja1105: offset      3573 ns, delay    160402 ns
ptp4l[5452.725]: sja1105 ratio: 1.000054
ptp4l[5452.725]: sja1105_qbv_monitor: state enabled not running
ptp4l[5452.726]: time to start: [2.139361728]
ptp4l[5452.851]: sja1105: sync timer timeout
ptp4l[5452.852]: sja1105: offset      3317 ns, delay    159882 ns
ptp4l[5452.852]: sja1105 ratio: 1.000054
ptp4l[5452.852]: sja1105_qbv_monitor: state enabled not running
ptp4l[5452.853]: time to start: [2.012330912]
ptp4l[5452.978]: sja1105: sync timer timeout
ptp4l[5452.979]: sja1105: offset       390 ns, delay    159302 ns
ptp4l[5452.979]: sja1105 ratio: 1.000082
ptp4l[5452.979]: sja1105_qbv_monitor: state enabled not running
ptp4l[5452.980]: time to start: [1.885382416]
ptp4l[5453.105]: sja1105: sync timer timeout
ptp4l[5453.106]: sja1105: offset      2615 ns, delay    160877 ns
ptp4l[5453.106]: sja1105 ratio: 1.000059
ptp4l[5453.106]: sja1105_qbv_monitor: state enabled not running
ptp4l[5453.107]: time to start: [1.758238352]
ptp4l[5453.136]: Received Sync: correction 0 ns
ptp4l[5453.136]: Received Follow_Up: correction 0 ns
ptp4l[5453.136]: master offset         13 s2 freq  -47669 path delay      3023
ptp4l[5453.144]: port 1: delay timeout
ptp4l[5453.145]: Received Delay_Resp: correction 0 ns
ptp4l[5453.145]: delay   filtered       3023   raw       3087
ptp4l[5453.232]: sja1105: sync timer timeout
ptp4l[5453.233]: sja1105: offset     -1697 ns, delay    162180 ns
ptp4l[5453.233]: sja1105 ratio: 1.000101
ptp4l[5453.233]: sja1105_qbv_monitor: state enabled not running
ptp4l[5453.234]: time to start: [1.631009368]
ptp4l[5453.359]: sja1105: sync timer timeout
ptp4l[5453.360]: sja1105: offset      3309 ns, delay    161325 ns
ptp4l[5453.361]: sja1105 ratio: 1.000051
ptp4l[5453.361]: sja1105_qbv_monitor: state enabled not running
ptp4l[5453.361]: time to start: [1.503733768]
ptp4l[5453.486]: sja1105: sync timer timeout
ptp4l[5453.488]: sja1105: offset      3000 ns, delay    159495 ns
ptp4l[5453.488]: sja1105 ratio: 1.000052
ptp4l[5453.488]: sja1105_qbv_monitor: state enabled not running
ptp4l[5453.488]: time to start: [1.376507784]
ptp4l[5453.614]: sja1105: sync timer timeout
ptp4l[5453.615]: sja1105: offset       585 ns, delay    159827 ns
ptp4l[5453.615]: sja1105 ratio: 1.000074
ptp4l[5453.615]: sja1105_qbv_monitor: state enabled not running
ptp4l[5453.616]: time to start: [1.249320992]
ptp4l[5453.741]: sja1105: sync timer timeout
ptp4l[5453.742]: sja1105: offset       907 ns, delay    159532 ns
ptp4l[5453.742]: sja1105 ratio: 1.000070
ptp4l[5453.742]: sja1105_qbv_monitor: state enabled not running
ptp4l[5453.743]: time to start: [1.122148864]
ptp4l[5453.868]: sja1105: sync timer timeout
ptp4l[5453.869]: sja1105: offset      1802 ns, delay    158980 ns
ptp4l[5453.869]: sja1105 ratio: 1.000061
ptp4l[5453.869]: sja1105_qbv_monitor: state enabled not running
ptp4l[5453.870]: time to start: [0.995043800]
ptp4l[5453.995]: sja1105: sync timer timeout
ptp4l[5453.996]: sja1105: offset        62 ns, delay    160497 ns
ptp4l[5453.996]: sja1105 ratio: 1.000077
ptp4l[5453.997]: sja1105_qbv_monitor: state enabled not running
ptp4l[5453.997]: time to start: [0.867838984]
ptp4l[5454.122]: sja1105: sync timer timeout
ptp4l[5454.123]: sja1105: offset      1083 ns, delay    159932 ns
ptp4l[5454.124]: sja1105 ratio: 1.000067
ptp4l[5454.124]: sja1105_qbv_monitor: state enabled not running
ptp4l[5454.124]: time to start: [0.740684232]
ptp4l[5454.136]: Received Sync: correction 0 ns
ptp4l[5454.136]: Received Follow_Up: correction 0 ns
ptp4l[5454.136]: master offset         45 s2 freq  -47633 path delay      3023
ptp4l[5454.249]: sja1105: sync timer timeout
ptp4l[5454.251]: sja1105: offset     -1272 ns, delay    157897 ns
ptp4l[5454.251]: sja1105 ratio: 1.000090
ptp4l[5454.251]: sja1105_qbv_monitor: state enabled not running
ptp4l[5454.251]: time to start: [0.613508072]
...
ptp4l[5454.760]: time to start: [0.104869288]
ptp4l[5454.885]: sja1105: sync timer timeout
ptp4l[5454.886]: sja1105: offset       937 ns, delay    159067 ns
ptp4l[5454.887]: sja1105 ratio: 1.000065
ptp4l[5454.887]: sja1105_qbv_monitor: state enabled not running
ptp4l[5454.887]: sja1105_qbv_monitor: transitioned to running state
ptp4l[5455.012]: sja1105: sync timer timeout
ptp4l[5455.014]: sja1105: offset     -1022 ns, delay    160822 ns
ptp4l[5455.014]: sja1105 ratio: 1.000084
ptp4l[5455.014]: sja1105_qbv_monitor: state running

How to solve this issue?

vladimiroltean, 2019-09-17:
Hi there,

You raise a good question. The sync imprecision in the synchronized scheduled traffic demo is a known and tricky issue.
Basically there are 2 ways you can synchronize the PTP clock of the switch:
- Through hardware timestamping of Ethernet frames (ptp4l). This is not supported in OpenIL and I would say it would be impossible to add it with the libsja1105 programming model.
- Through software timestamping (synchronizing to another local clock - phc2sys). This is basically what OpenIL uses for the demo.

The issue is that software synchronization of the PTP clock requires SPI I/O to read/write the timer counter. This takes a nondeterministic amount of time because of the way the SPI controller driver (spi-fsl-dspi) is written: it transfers one byte at a time, and is interrupt-triggered. So transferring a 12-byte frame means it must service 12 interrupts. And the time it takes to do all of that adds up as an error/latency between clock A and clock B, which are software-synchronized (write A to B).
In fact this problem is not specific to SPI-controlled PTP clocks. It also affects MDIO devices in much the same way: see Marvell MDIO-controlled DSA switches. For those, ptp4l works perfectly, phc2sys not so much.

So you'd say that switching the spi-fsl-dspi driver to poll mode solves the problem?
Well, not so fast. Indeed it is much quicker and hence more stable, but it is still non-deterministic. Basically the core can at any time service an interrupt, or the spi kernel thread can get preempted, which means that the latency between "read clock A" and "write clock B" is still accounting for unrelated stuff that happens in-between.

Enter the PTP_SYS_OFFSET_EXTENDED ioctl added to the Linux kernel and to linuxptp by Miroslav Lichvar.

I recently participated in a long discussion on the netdev mailing list related to both [MDIO](https://lkml.org/lkml/2019/8/19/830) and [SPI](https://lwn.net/Articles/798467/), where we tried to understand what it would take to completely solve this problem.

The basic idea is that if you correlate the system time (a low latency clock) with the exact instant that the PTP time (a high latency clock) was snapshotted, then you are basically able to compensate for whatever latency there was between "read clock A" and "write clock B".
Now the ugly part is that this requires the introduction of a software timestamping mechanism deep into the guts of SPI and MDIO drivers. I would say this is the main reason why we didn't come to a conclusion yet: we need to understand that this is a sensible enough change and that there is enough justification to make it.

As for the LS1021A-TSN, well I already mentioned that OpenIL only supports so much. I keep mentioning in other threads that [there's a community kernel I maintain](https://github.com/vladimiroltean/ls1021atsn-linux) which includes the results of my findings. And because other people are asking to test this out easier, I also had to create [an OpenIL community fork](https://github.com/vladimiroltean/openil) with everything pre-packaged out of the box.

If you want to look at the community kernel, you'll find that instead of sja1105-tool there is now a kernel driver which uses the DSA abstraction to represent the 4 front-panel ports as net devices. Those support proper PTP using the L2 transport (the device is especially useful as a P2P_TC switch). Additionally you will find that the Qbv configuration is now managed using a tc-taprio offload.

Unfortunately I don't have any documentation put together for it, since it's all very new. You'll find a lot of info if you sift through the commit log. Basically what I'm trying to say is that the software synchronization on the LS1021A-TSN is now a solved problem in that particular kernel. The offset does never exceed +/- 200 ns under normal conditions (which includes cores under load), even after days of use. It is almost as good as hardware synchronization in that regard. If you are serious in getting this issue solved I would suggest to give the community kernel a try. Just ask if you have any trouble and I will take the opportunity to lay down some documentation as I'm writing further answers.

Hope this helps,
-Vladimir

laravich, 2019-09-18:
Ok, i am trying to understand your above answer i will go through the community kernel. what about the qbv monitor of board2 going under "surprisingly stopped" state? i observe this quite frequently even though the offset haven't exceeded. More over this produce a slight variation in packet inter arrival time (a sharp peak more then 60 to 90ms is observed) in LBT demo. Moreover i also observe all this only on the board2 for the given demo setup. even if i change the hardware i observe the same only on board2. Why is that?
one more thing regarding the first question. if i observe a offset exceeding situation, i refresh the config upload on each boards, which gives me a stable clock for sometime. However i consider this non ideal. so is there anything that i can do with the config upload to synchronize the clocks better?

vladimiroltean, 2019-09-18:
> what about the qbv monitor of board2 going under "surprisingly stopped" state? i observe this quite frequently even though the offset haven't exceeded.

There are some conditions that must be respected in order for the TAS engine to be able to properly integrate the PTP time into its own tick counter (the internal syntonization as described in the sja1105_tas.c file). Namely, clock stepping must not be done, and drift correction must not exceed a certain frequency.
I haven't looked at it since long ago, but the user space (patched ptp4l) implementation only observes the clock stepping requirement as far as I remember. It is possible that the frequency that user space tries to correct may exceed the hardware threshold. When that happens, I think the TAS will shut down automatically, leading to the 'surprisingly stopped' message.
In the kernel implementation there is more control over the ppb value used for correction, and the limit is configured as such that frequency steering is always a safe operation. At least I never saw the 'surprisingly stopped' condition using the kernel impl.

> More over this produce a slight variation in packet inter arrival time (a sharp peak more then 60 to 90ms is observed) in LBT demo

This means that a frame loses the schedule deadline, either on the sender, on the 2 switches that transport it, or the response misses the deadline. The kernel re-implementation will fix the 'deadline missed on switches' scenario.

> Moreover i also observe all this only on the board2 for the given demo setup. even if i change the hardware i observe the same only on board2. Why is that?

Coincidence?
Please remember that the PTP network in the OpenIL demo is not through the switch ports, but through a separate network of eth0 devices and a dumb switch. Maybe board 2 is the master, and it's all the other boards that went out of sync?

> if i observe a offset exceeding situation, i refresh the config upload on each boards, which gives me a stable clock for sometime. However i consider this non ideal. so is there anything that i can do with the config upload to synchronize the clocks better?

No.
The SPI access is simply non-deterministic without PTP_SYS_OFFSET_EXTENDED and even if it works for a while, it can fail any time.

Also with the community kernel there's the non-negligible benefit of not needing software synchronization at all, if you only need hardware timestamping. And since the sync offset is lower than +/- 200 ns, you can bring the cycle time to quite lower values than the 30 ms above (which I had forgot about) without fear of going out of sync. Of course, that is assuming that you can produce data that fast, for which you would need an RT kernel, I think.

laravich, 2019-11-06:
Hello,

As suggested by you i am trying out the openIL community kernel. when i insert the TSN board with the new sdcard.img it's asking for openIL username and password. where previously is used to login automatically. Am i doing something wrong? if not what is the username and password?

If you find my question silly, i am so sorry i am so new to this!

Thank you.

vladimiroltean, 2019-11-06:
User name is "root". When you input that, it should not ask for any password (because there isn't any).
As for SSH, logging in as root is disabled by default. However there is a command you can run at your own risk called "passwordless-ssh-login" which enables SSH access to the board as root, with no password.

laravich, 2019-11-06:
Thank you, i could login now.

But there seems to be no Sja1105 tools loaded under drivers/net/dsa
it says:

[    1.964339] sja1105 spi0.1: Probed switch chip: SJA1105T                     
[    2.926902] sja1105 spi0.1: Expected device ID 0x9e00030e but read 0x5e00030e
[    2.934247] sja1105 spi0.1: Device ID check failed: -19

How can i change that?

vladimiroltean, 2019-11-06:
That is very odd. It indicates to me that there is some variability in the SPI electrical interface of this board which I did not predict and do not understand.
So the driver checks the device id against these values: https://git.kernel.org/pub/scm/linux/kernel/git/davem/net-next.git/tree/drivers/net/dsa/sja1105/sja1105_static_config.h#n91
As you can see there is no 0x5e00030e. That is just a bit flip (0x9 became 0x5) which is probably a SPI bus error.
As a debugging step, can you please try to decrease the SPI frequency from 12 MHz to, say, 6 MHz? The device tree is located at this path: https://git.kernel.org/pub/scm/linux/kernel/git/davem/net-next.git/tree/arch/arm/boot/dts/ls1021a-tsn.dts#n45

As to your question:

> there seems to be no Sja1105 tools loaded under drivers/net/dsa

Yes, that is by design. With a kernel driver the sja1105-tool is of no use.

closed 2020-08-06 by vladimiroltean