nxp-archive/openil #29: Does openil support the LS1028ARDB board?

liu-jeff, 2019-11-14:
Does openil support the LS1028ARDB board?

vladimiroltean, 2019-11-14:
Yes.

liu-jeff, 2019-11-14:
> Yes.

miriac SBC-LS1028A is also supported？

vladimiroltean, 2019-11-14:
No.

liu-jeff, 2019-11-14:
> No.
Is LS1028ARDB better than LS1021ATSN?

vladimiroltean, 2019-11-14:
Better for what?

liu-jeff, 2019-11-14:
> No.

> Better for what?

liu-jeff, 2019-11-14:
> > No.
> 
> > Better for what?

About the performance of their TSN

liu-jeff, 2019-11-14:
> > > No.
> > 
> > 
> > > Better for what?
> 
> About the performance of their T

liu-jeff, 2019-11-14:
> Better for what?

For example, the speed of processing packets, delays?

vladimiroltean, 2019-11-14:
The LS1021A-TSN has a non-TSN Ethernet controller (eTSEC, using the Gianfar driver) for the regular ports and a standalone Ethernet switch with TSN.
The LS1028A-RDB has a TSN-enabled Ethernet controller (ENETC) as well as a TSN-enabled embedded switch (Felix).
The designs have some similarities (the switch is connected the same way to the system, aka via an Ethernet port), but the LS1028A has a faster connection to the switch (2.5Gbps), cut-through switching and some extra TSN features: frame preemption and a pre-standard version of 802.1CB seamless redundancy.
I don't have access to fully-optimized software environments to judge the performance. The expectation is that under similar conditions (i.e. Ethernet link speeds, no cut-through, no frame preemption) they perform similarly for packet switching. Whether the extra features make a difference depends on the use case, frame size, required latency, guard bands, etc etc. There is also the core performance (2 Cortex A72 at 1.5GHz vs 2 Cortex A7 at 1.0 or 1.2GHz) that might make a difference if you need significant processing on the cores.
In terms of software enablement in OpenIL, the LS1021A-TSN lags a bit behind due to a limited user-space driver for the switch, but there is mainline Linux support to compensate. Whereas on the LS1028A-RDB, mainline support is ongoing and OpenIL or LSDK is the better alternative.

liu-jeff, 2019-11-14:
> The LS1021A-TSN has a non-TSN Ethernet controller (eTSEC, using the Gianfar driver) for the regular ports and a standalone Ethernet switch with TSN.
> The LS1028A-RDB has a TSN-enabled Ethernet [...]

thank you very much,I also have some questions when  compiling the kernel，I want to run make menuconfig, but it appears init/Kconfig:1950: can't open file "arch/x86/xenomai/Kconfig"，I don't how to solve

vladimiroltean, 2019-11-14:
And how is this related to the topic?
You need a cross-compiler (for example [Linaro](https://releases.linaro.org/components/toolchain/binaries/latest-7/aarch64-linux-gnu/)), to add the binaries somewhere visible in `$PATH`, and to set `ARCH=arm64` and `CROSS_COMPILE=aarch64-linux-gnu-`. There are many tutorials online on how to  cross-compile a Linux kernel.