nxp-archive/openil #35: Linuxptp supports 802.1AS, can LS021 synchronize with other tsn devices that support 802.1AS?

liu-jeff, 2019-11-19:
Linuxptp supports 802.1AS, can LS021 synchronize with other tsn devices that support 802.1AS? For example, Cisco IE4000 type switch? Could you please advise？

vladimiroltean, 2019-11-22:
The answer is yes and no.
The mainline version of linuxptp supports 802.1AS for endpoints. The OpenIL version of linuxptp adds support for 802.1AS bridges.
The OpenIL version of linuxptp will not work for the sja1105 ports in OpenIL. You can use linuxptp only on eth0 and eth1 at the moment.
The OpenIL version of linuxptp should work with the sja1105 driver from the mainline kernel. However I have found that there is an issue with frame reordering that I am currently investigating. So synchronization stops after a while. I will update this thread (and others asking for this feature) once I figure out what the reordering issue is, and once time-aware bridging works properly.

liu-jeff, 2019-11-22:
> The answer is yes and no.
> The mainline version of linuxptp supports 802.1AS for endpoints. The OpenIL version of linuxptp adds support for 802.1AS bridges.
> The OpenIL version of linuxptp will not [...]
After my debugging, I found that the LS1021 and the Cisco switch can be synchronized.

vladimiroltean, 2019-11-22:
> After my debugging, I found that the LS1021 and the Cisco switch can be synchronized.

Yes, but the timestamps are taken at eth2, not at the switch MACs. Generally you want it the other way around.

liu-jeff, 2019-12-02:
> The answer is yes and no.
> The mainline version of linuxptp supports 802.1AS for endpoints. The OpenIL version of linuxptp adds support for 802.1AS bridges.
> The OpenIL version of linuxptp will not [...]

It is found that when performing 802.1AS synchronization with other tsn devices, there is indeed instability and sometimes it stops

vladimiroltean, 2019-12-13:
Update: I managed to figure out the PTP freezes on the LS1021A-TSN board, you need to update the kernel after this commit:
https://github.com/vladimiroltean/openil-community/commit/c3a7ef8b0a0d6e1d5cf6e8b64014f9eafc75abee

liu-jeff, 2019-12-13:
Thank you very much, that say I must use the openil-community?  but I do not found the kernel code 
In the ls1021 configuration file, I found it to download the linux kernel from this link
[image]

vladimiroltean, 2019-12-13:
Yes that is correct. The base kernel is mainline (davem == David Miller, networking maintainer) plus some extra downstream patches in `board/nxp/ls1021atsn/patches/linux` which the build system applies automatically on top of it.

liu-jeff, 2019-12-13:
Thanks again, but for the ls1021,I do not find the sja-tool in openil-community configuration file

vladimiroltean, 2019-12-13:
I have created a basic guide [here](https://github.com/openil/openil/issues/48#issuecomment-565465139) to explain how the sja1105-tool functionality has been replaced with the kernel driver in openil-community.

closed 2020-08-06 by vladimiroltean

liing0228, 2022-06-11:
> The answer is yes and no.
> The mainline version of linuxptp supports 802.1AS for endpoints. The OpenIL version of linuxptp adds support for 802.1AS bridges.
> The OpenIL version of linuxptp will not [...]

@liu-jeff 
Hi Liu can you sync ie4000 with general linux device?
Because I fount that my ptp4l say ie4000 doesn’t use pro timescale .
And what version iOS do you use?