nxp-archive/openil #37: Does the LS1021ATSN have a wifi module?

liu-jeff, 2019-11-24:
Does the LS1021ATSN have a wifi module? If so, how do you open it?

vladimiroltean, 2019-11-24:
No. It has a mini PCIe slot though and holes for antennas in the chassis though, if you want to install a Wi-Fi module.

liu-jeff, 2019-11-24:
> No. It has a mini PCIe slot though and holes for antennas in the chassis though, if you want to install a Wi-Fi module. I have not tested it.

[image]
But I saw it in the manual, as long as insert the wifi driver module

vladimiroltean, 2019-11-24:
I would read that paragraph "how would you test a WiFi module if you had one".

liu-jeff, 2019-11-25:
> I would read that paragraph "how would you test a WiFi module if you had one".

Thank you very much,Do you have a recommended wifi module?

liu-jeff, 2019-11-25:
> I would read that paragraph "how would you test a WiFi module if you had one".

Does the LS1021ATSN have special requirements for its wifi module?

vladimiroltean, 2019-11-25:
I don't know. I will try to test a Compex WLE900VX in a few weeks.

liu-jeff, 2019-11-25:
> I don't know. I will try to test a Compex WLE900VX in a few weeks.

Ok ,thank you very much,I also want to know that  canthe wifi module communicate with the sja1105?Or data sent via wifi can also be scheduled through QBV？

vladimiroltean, 2019-11-25:
There is no offloaded hardware path between PCIe and Ethernet on the LS1021A. You will need to develop an RT application in software that listens for packets on the WiFi interface and routes them to a switch interface, while adding a VLAN tag where the PCP decides what is its timeslot in the Qbv schedule.

liu-jeff, 2019-11-25:
> There is no offloaded hardware path between PCIe and Ethernet on the LS1021A. You will need to develop an RT application in software that listens for packets on the WiFi interface and routes them to [...]

Thank you very much,but  if this ,I don't think the delay can guarantee it well.

liu-jeff, 2019-11-25:
> There is no offloaded hardware path between PCIe and Ethernet on the LS1021A. You will need to develop an RT application in software that listens for packets on the WiFi interface and routes them to [...]

I also want to know if the LS1028ARDB is also a SJA1105 network card?

vladimiroltean, 2019-11-25:
No, it has a Vitesse/Microsemi/Microchip VSC9959 switch called "Felix" which is embedded inside the SoC. This does not, however, mean that it has PCIe acceleration.

liu-jeff, 2019-11-25:
> No, it has a Vitesse/Microsemi/Microchip VSC9959 switch called "Felix" which is embedded inside the SoC. This does not, however, mean that it has PCIe acceleration.

Does it also support the wifi module？does its wifi still not communicate with Felix? So if it is LS1028, if you want wifi packet to implement qbv, still need to write an application to grab the wifi packet, and then route to Felix?

vladimiroltean, 2019-11-25:
Same situation as with the LS1021A-TSN in this regard.

liu-jeff, 2019-11-25:
> Same situation as with the LS1021A-TSN in this regard.

So if I want to synchronize the wifi card, do you have any good suggestions?Is it necessary to have a wireless network card at a rate like Felix?

liu-jeff, 2019-11-25:
> Same situation as with the LS1021A-TSN in this regard.

I know that wireless has its own synchronization mechanism, but the accuracy does not meet the TSN requirements, then does it still need hardware support?

vladimiroltean, 2019-11-25:
> So if I want to synchronize the wifi card, do you have any good suggestions?

No.

liu-jeff, 2019-11-25:
> > So if I want to synchronize the wifi card, do you have any good suggestions?
> 
> No.
OK，Thank you very much. if you test a Compex WLE900VX ,please tell me ,thanks again！

liu-jeff, 2019-11-27:
> > > So if I want to synchronize the wifi card, do you have any good suggestions?
> > 
> > 
> > No.
> > OK，Thank you very much. if you test a Compex WLE900VX ,please tell me ,thanks again！
Did you test wifi？

vladimiroltean, 2019-11-27:
I said " I will try to test a Compex WLE900VX in a few weeks." 2 days ago.

liu-jeff, 2019-11-28:
> I said " I will try to test a Compex WLE900VX in a few weeks." 2 days ago.

I am sorry, thank you very much!