nxp-archive/openil #38: For LS1028ARDB, how to set VLAN priority and gate mapping

liu-jeff, 2019-11-26:
For LS1028ARDB, how to set VLAN priority and gate mapping

vladimiroltean, 2019-11-26:
```
[ -d /sys/class/net/br0 ] && ip link del dev br0
ip link add dev br0 type bridge vlan_filtering 1
ip link set dev swp0 master br0
ip link set dev swp1 master br0
ip link set dev swp2 master br0
ip link set dev swp3 master br0
ip link set dev swp4 master br0
ip link set dev swp5 master br0
ip addr add 192.168.1.1/24 dev br0
bridge vlan add dev swp0 vid 100
bridge vlan add dev swp1 vid 100 untagged
bridge vlan add dev swp2 vid 100 pvid
bridge vlan add dev swp3 vid 100 pvid untagged
```

By adding the switch ports to a bridge with VLAN awareness enabled, the switch is configured automatically to map the VLAN PCP of ingress frames to a traffic class, both on ingress and egress. Priority remapping is not yet configurable.
The 4 "bridge vlan add" commands are there to illustrate the types of VLANs you can add on each port:
- No flags: accept this VLAN on the ingress port, send it as tagged on this egress port
- `untagged`: accept this VLAN on the ingress port, send it as untagged on this egress port
- `pvid`: accept this VLAN on the ingress port and assign it internally to all untagged traffic, send all frames that are tagged with this VLAN as tagged on this egress port
- `pvid untagged`: accept this VLAN on the ingress port and assign it internally to all untagged traffic, send all frames that are tagged with this VLAN as untagged on this egress port

If the `vlan_filtering` property of the bridge is not enabled, then the switch ignores all VLAN information.

liu-jeff, 2019-11-26:
> ```
> [ -d /sys/class/net/br0 ] && ip link del dev br0
> ip link add dev br0 type bridge vlan_filtering 1
> ip link set dev swp0 master br0
> ip link set dev swp1 master br0
> ip link set dev swp2 master br0 [...]

Can't it be vlan tag on the entrance for untagged traffic like LS1021?

liu-jeff, 2019-11-26:
> ```
> [ -d /sys/class/net/br0 ] && ip link del dev br0
> ip link add dev br0 type bridge vlan_filtering 1
> ip link set dev swp0 master br0
> ip link set dev swp1 master br0
> ip link set dev swp2 master br0 [...]

Automatic configuration, if it is priority 5, then it is mapped to gate 5?

vladimiroltean, 2019-11-26:
> Can't it be added vlan tag on the entrance for untagged traffic like LS1021?

Yes it can, "pvid" means "port-based VLAN". The default pvid in Linux is 1, which means that VLAN 1 is added to all frames. But VLAN 1 is also untagged by default, which means it's removed so you don't see it.
Run `bridge vlan` for more info.

> Automatic configuration, if it is priority 5, then it is mapped to gate 5?

Yes.

liu-jeff, 2019-11-27:
> > Can't it be added vlan tag on the entrance for untagged traffic like LS1021?
> 
> Yes it can, "pvid" means "port-based VLAN". The default pvid in Linux is 1, which means that VLAN 1 is added to all frames. But VLAN 1 is also untagged by default, which means it's removed so you don't see it.
> Run `bridge vlan` for more info, and make sure to see [man bridge](http://man7.org/linux/man-pages/man8/bridge.8.html).
> 
> > Automatic configuration, if it is priority 5, then it is mapped to gate 5?
> 
> Yes.

What should I do if I want to modify the port vlan priority?

vladimiroltean, 2019-11-27:
This is not configurable through standard Linux tools at the moment.

You can do the following:

1. Save the following to a script text file (say `felix.sh`). It is work in progress and at the moment does only a single thing. You can put more stuff from the hardware reference manual into it though:
```
#!/bin/sh

set -e -u -o pipefail

port=
block=
op=

while [ $# -ne 0 ]; do
	case "$1" in
	-d | --device)
		case "$2" in
		port0 | port1 | port2 | port3 | port4 | port5 | port6)
			# Trim "port" prefix from input argument and leave only
			# the number in the ${port} variable
			port=${2#port}
			;;
		*)
			echo "Invalid device specified: $2!"
			echo "Try one of: port0 | port1 | port2 | port3 | port4 | port5 | port6"
			return 1
			;;
		esac
		shift
		;;
	-b | --block)
		# Taken from /proc/iomem
		case "$2" in
		sys | rew | qs | ptp | port | qsys | ana)
			block="$2"
			;;
		*)
			echo "Invalid block specified: $2!"
			echo "Try one of: sys | rew | qs | ptp | port | qsys | ana"
			;;
		esac
		shift
		;;
	-g | --group)
		# Block-specific handlers need to do further sanity checking on this
		group="$2"
		shift
		;;
	-r | --reg)
		# Block/group-specific handlers need to do further sanity checking on this
		reg="$2"
		shift
		;;
	-h | --help)
		echo "Usage: tsntool-switch-raw --device <port_number> --block <block>"
		return 0
		;;
	read)
		op="read"
		;;
	write)
		op="write"
		reg_value="$2"
		shift
		;;
	*)
		echo "Invalid argument $1!"
		return 1
		;;
	esac
	shift
done

if [ "${op}" = "" ]; then
	echo "Please specify operation (read or write)!"
	return 1
fi

if [ "${port}" = "" ]; then
	echo "Please input --device parameter!"
	return 1
fi

if [ "${block}" = "" ]; then
	echo "Please input --block parameter!"
	return 1
fi

if [ "${group}" = "" ]; then
	echo "Please input --group parameter!"
	return 1
fi

if [ "${reg}" = "" ]; then
	echo "Please input --reg parameter!"
	return 1
fi

felix_read32() {
	local addr="$1"
	printf "0x%04x\n" "$(busybox devmem ${addr})"
}

felix_write32() {
	local addr="$1"
	local value="$2"
	busybox devmem "${addr}" 32 "${value}"
}

handle_es_ana() {
	local es_ana_bar=$(cat /proc/iomem | awk -F- '/: ana$/ { gsub(/ /, "", $0); print "0x"$1 }')
	local es_ana_port_offset="0x7800" # LS1028ARM.pdf 16.3.3 ANA
	local es_ana_port_spacing="0x100"
	local es_ana_port_base="$((${es_ana_bar} + ${es_ana_port_offset} + ${port} * ${es_ana_port_spacing}))"

	case "${group}" in
	ana)
		echo "Not implemented!"; return 1
		;;
	pgid)
		echo "Not implemented!"; return 1
		;;
	ana_tables)
		echo "Not implemented!"; return 1
		;;
	sg_access)
		echo "Not implemented!"; return 1
		;;
	sg_config)
		echo "Not implemented!"; return 1
		;;
	sg_status)
		echo "Not implemented!"; return 1
		;;
	port)
		local reg_offset=

		case "${reg}" in
		vlan_cfg)
			reg_offset="0x0"
			;;
		port_cfg)
			reg_offset="0x4"
			;;
		qos_cfg)
			reg_offset="0x8"
			;;
		vcap_cfg)
			reg_offset="0xc"
			;;
		*)
			echo "Invalid register ${reg}!"
			echo "Please specify one of: vlan_cfg | port_cfg | qos_cfg | vcap_cfg"
			return 1
			;;
		esac

		reg_addr="$((${es_ana_port_base} + ${reg_offset}))"
		;;
	pfc)
		echo "Not implemented!"; return 1
		;;
	common)
		echo "Not implemented!"; return 1
		;;
	pol)
		echo "Not implemented!"; return 1
		;;
	pol_misc)
		echo "Not implemented!"; return 1
		;;
	*)
		echo "Invalid group ${group}!"
		echo "Please specify one of: ana | pgid | ana_tables | sg_access | sg_config | sg_status | port | pfc | common | pol | pol_misc"
		;;
	esac
}

case "${block}" in
ana)
	handle_es_ana
	;;
*)
	echo "Block ${block} not yet supported!"
	;;
esac

case ${op} in
read)
	felix_read32 "${reg_addr}"
	;;
write)
	felix_write32 "${reg_addr}" "${reg_value}"
	;;
esac
```

2. Usage example for setting ANA:PORT:QOS_CFG.QOS_DEFAULT_VAL to 7 for port 0:
```
# ./felix.sh read --device port0 --block ana --group port --reg qos_cfg
0x0000
# ./felix.sh write 0xe0 --device port0 --block ana --group port --reg qos_cfg
# ./felix.sh read --device port0 --block ana --group port --reg qos_cfg
0x00e0
```

liu-jeff, 2019-11-28:
> This is not configurable through standard Linux tools at the moment.
> 
> You can do the following:
> 
> 1. Save the following to a script text file (say `felix.sh`). It is work in progress and at the moment does only a single thing. You can put more stuff from the hardware reference manual into it though:
> 
> ```
> #!/bin/sh
> 
> set -e -u -o pipefail
> 
> port=
> block=
> op=
> 
> while [ $# -ne 0 ]; do
> 	case "$1" in
> 	-d | --device)
> 		case "$2" in
> 		port0 | port1 | port2 | port3 | port4 | port5 | port6)
> 			# Trim "port" prefix from input argument and leave only
> 			# the number in the ${port} variable
> 			port=${2#port}
> 			;;
> 		*)
> 			echo "Invalid device specified: $2!"
> 			echo "Try one of: port0 | port1 | port2 | port3 | port4 | port5 | port6"
> 			return 1
> 			;;
> 		esac
> 		shift
> 		;;
> 	-b | --block)
> 		# Taken from /proc/iomem
> 		case "$2" in
> 		sys | rew | qs | ptp | port | qsys | ana)
> 			block="$2"
> 			;;
> 		*)
> 			echo "Invalid block specified: $2!"
> 			echo "Try one of: sys | rew | qs | ptp | port | qsys | ana"
> 			;;
> 		esac
> 		shift
> 		;;
> 	-g | --group)
> 		# Block-specific handlers need to do further sanity checking on this
> 		group="$2"
> 		shift
> 		;;
> 	-r | --reg)
> 		# Block/group-specific handlers need to do further sanity checking on this
> 		reg="$2"
> 		shift
> 		;;
> 	-h | --help)
> 		echo "Usage: tsntool-switch-raw --device <port_number> --block <block>"
> 		return 0
> 		;;
> 	read)
> 		op="read"
> 		;;
> 	write)
> 		op="write"
> 		reg_value="$2"
> 		shift
> 		;;
> 	*)
> 		echo "Invalid argument $1!"
> 		return 1
> 		;;
> 	esac
> 	shift
> done
> 
> if [ "${op}" = "" ]; then
> 	echo "Please specify operation (read or write)!"
> 	return 1
> fi
> 
> if [ "${port}" = "" ]; then
> 	echo "Please input --device parameter!"
> 	return 1
> fi
> 
> if [ "${block}" = "" ]; then
> 	echo "Please input --block parameter!"
> 	return 1
> fi
> 
> if [ "${group}" = "" ]; then
> 	echo "Please input --group parameter!"
> 	return 1
> fi
> 
> if [ "${reg}" = "" ]; then
> 	echo "Please input --reg parameter!"
> 	return 1
> fi
> 
> felix_read32() {
> 	local addr="$1"
> 	printf "0x%04x\n" "$(busybox devmem ${addr})"
> }
> 
> felix_write32() {
> 	local addr="$1"
> 	local value="$2"
> 	busybox devmem "${addr}" 32 "${value}"
> }
> 
> handle_es_ana() {
> 	local es_ana_bar=$(cat /proc/iomem | awk -F- '/: ana$/ { gsub(/ /, "", $0); print "0x"$1 }')
> 	local es_ana_port_offset="0x7800" # LS1028ARM.pdf 16.3.3 ANA
> 	local es_ana_port_spacing="0x100"
> 	local es_ana_port_base="$((${es_ana_bar} + ${es_ana_port_offset} + ${port} * ${es_ana_port_spacing}))"
> 
> 	case "${group}" in
> 	ana)
> 		echo "Not implemented!"; return 1
> 		;;
> 	pgid)
> 		echo "Not implemented!"; return 1
> 		;;
> 	ana_tables)
> 		echo "Not implemented!"; return 1
> 		;;
> 	sg_access)
> 		echo "Not implemented!"; return 1
> 		;;
> 	sg_config)
> 		echo "Not implemented!"; return 1
> 		;;
> 	sg_status)
> 		echo "Not implemented!"; return 1
> 		;;
> 	port)
> 		local reg_offset=
> 
> 		case "${reg}" in
> 		vlan_cfg)
> 			reg_offset="0x0"
> 			;;
> 		port_cfg)
> 			reg_offset="0x4"
> 			;;
> 		qos_cfg)
> 			reg_offset="0x8"
> 			;;
> 		vcap_cfg)
> 			reg_offset="0xc"
> 			;;
> 		*)
> 			echo "Invalid register ${reg}!"
> 			echo "Please specify one of: vlan_cfg | port_cfg | qos_cfg | vcap_cfg"
> 			return 1
> 			;;
> 		esac
> 
> 		reg_addr="$((${es_ana_port_base} + ${reg_offset}))"
> 		;;
> 	pfc)
> 		echo "Not implemented!"; return 1
> 		;;
> 	common)
> 		echo "Not implemented!"; return 1
> 		;;
> 	pol)
> 		echo "Not implemented!"; return 1
> 		;;
> 	pol_misc)
> 		echo "Not implemented!"; return 1
> 		;;
> 	*)
> 		echo "Invalid group ${group}!"
> 		echo "Please specify one of: ana | pgid | ana_tables | sg_access | sg_config | sg_status | port | pfc | common | pol | pol_misc"
> 		;;
> 	esac
> }
> 
> case "${block}" in
> ana)
> 	handle_es_ana
> 	;;
> *)
> 	echo "Block ${block} not yet supported!"
> 	;;
> esac
> 
> case ${op} in
> read)
> 	felix_read32 "${reg_addr}"
> 	;;
> write)
> 	felix_write32 "${reg_addr}" "${reg_value}"
> 	;;
> esac
> ```
> 
> 1. Usage example for setting ANA:PORT:QOS_CFG.QOS_DEFAULT_VAL to 7 for port 0:
> 
> ```
> # ./felix.sh read --device port0 --block ana --group port --reg qos_cfg
> 0x0000
> # ./felix.sh write 0xe0 --device port0 --block ana --group port --reg qos_cfg
> # ./felix.sh read --device port0 --block ana --group port --reg qos_cfg
> 0x00e0
> ```

Ok, thank you very much!

liu-jeff, 2019-12-17:
> This is not configurable through standard Linux tools at the moment.
> 
> You can do the following:
> 
> 1. Save the following to a script text file (say `felix.sh`). It is work in progress and at the moment does only a single thing. You can put more stuff from the hardware reference manual into it though:
> 
> ```
> #!/bin/sh
> 
> set -e -u -o pipefail
> 
> port=
> block=
> op=
> 
> while [ $# -ne 0 ]; do
> 	case "$1" in
> 	-d | --device)
> 		case "$2" in
> 		port0 | port1 | port2 | port3 | port4 | port5 | port6)
> 			# Trim "port" prefix from input argument and leave only
> 			# the number in the ${port} variable
> 			port=${2#port}
> 			;;
> 		*)
> 			echo "Invalid device specified: $2!"
> 			echo "Try one of: port0 | port1 | port2 | port3 | port4 | port5 | port6"
> 			return 1
> 			;;
> 		esac
> 		shift
> 		;;
> 	-b | --block)
> 		# Taken from /proc/iomem
> 		case "$2" in
> 		sys | rew | qs | ptp | port | qsys | ana)
> 			block="$2"
> 			;;
> 		*)
> 			echo "Invalid block specified: $2!"
> 			echo "Try one of: sys | rew | qs | ptp | port | qsys | ana"
> 			;;
> 		esac
> 		shift
> 		;;
> 	-g | --group)
> 		# Block-specific handlers need to do further sanity checking on this
> 		group="$2"
> 		shift
> 		;;
> 	-r | --reg)
> 		# Block/group-specific handlers need to do further sanity checking on this
> 		reg="$2"
> 		shift
> 		;;
> 	-h | --help)
> 		echo "Usage: "
> 		echo "felix.sh read --device <port_number> --block <block> --group <group> --reg <reg>"
> 		echo "felix.sh write --device <port_number> --block <block> --group <group> --reg <reg> <val>"
> 		return 0
> 		;;
> 	read)
> 		op="read"
> 		;;
> 	write)
> 		op="write"
> 		reg_value="$2"
> 		shift
> 		;;
> 	*)
[... 327 lines cut ...]
> *)
> 	echo "Block ${block} not yet supported!"
> 	;;
> esac
> 
> case ${op} in
> read)
> 	felix_read32 "${reg_addr}"
> 	;;
> write)
> 	felix_write32 "${reg_addr}" "${reg_value}"
> 	;;
> esac
> ```
> 
> 1. Usage example for setting ANA:PORT:QOS_CFG.QOS_DEFAULT_VAL to 7 for port 0:
> 
> ```
> # ./felix.sh read --device port0 --block ana --group port --reg qos_cfg
> 0x0000
> # ./felix.sh write 0xe0 --device port0 --block ana --group port --reg qos_cfg
> # ./felix.sh read --device port0 --block ana --group port --reg qos_cfg
> 0x00e0
> ```
Hello，I want to know how port0-port6 corresponds to swp0-swp5

vladimiroltean, 2019-12-17:
Please ignore port6. The portN notation should match the swpN notation of the Linux interfaces.

liu-jeff, 2019-12-17:
> Please ignore port6. The portN notation should match the swpN notation of the Linux interfaces.

Thank you very much, I am doing qbv test with LS1028 and found that the tutorial is very simple. It does not mention how to set the  time slot length of each gate.

vladimiroltean, 2019-12-17:
You can do something like this:

```
# Given @frame_len bytes, @count frames and @link_speed in Mbps,
# returns the minimum number of nanoseconds required to keep a Qbv gate open
# to transmit that.
qbv_window() {
	local frame_len=$1
	local count=$2
	local link_speed=$3
	local bit_time=$((1000 / ${link_speed}))
	# 7 bytes preamble, 1 byte SFD, 4 bytes FCS and 12 bytes IFG
	local overhead=24
	local octets=$((${frame_len} + ${overhead}))

	echo "$((${octets} * 8 * ${bit_time}))"
}

	# PTP
	t1=$(qbv_window 200 1 $speed_mbps)
	# raw-l2-send
	t2=$(qbv_window $length 1 $speed_mbps)
	# Best effort
	t0=$(($cycle_time - $t1 - $t2))

	cat > qbv0.txt <<-EOF
		t0 01011111 $t0 # best effort
		t1 10000000 $t1 # ptp4l
		t2 00100000 $t2 # raw-l2-send
	EOF
	cat qbv0.txt
	tsntool qbvset --device "${iface}" --entryfile qbv0.txt --enable \
		--basetime "${mac_base_time}"
```
Source: https://github.com/vladimiroltean/tsn-scripts/blob/master/time-test.sh#L576
```

liu-jeff, 2019-12-17:
> You can do something like this:
>
> ```
> # Given @frame_len bytes, @count frames and @link_speed in Mbps,
> # returns the minimum number of nanoseconds required to keep a Qbv gate open
> # to transmit that. [...]

> You can do something like this:
>
> ```
> # Given @frame_len bytes, @count frames and @link_speed in Mbps,
> # returns the minimum number of nanoseconds required to keep a Qbv gate open
> # to transmit that. [...]
Thank you very much, but I'm stupid. Is there a simpler tutorial like LS1021?

vladimiroltean, 2019-12-17:
The example above creates a file named `qbv0.txt` with each line defining a timeslot.
The syntax for this file is:
```
t0 00100000 100
```
Which means "first timeslot should open gate for traffic class 5 and close all other gates. This action should last for 100 ns after which the hardware should execute the next gate action". There is no next gate action (t1) so the schedule starts back from the beginning, with another t0 in this example.
`cycle_time` is a variable holding the length of time that it should take from one t0 to the next. In my post above it is equal to t0+t1+t2. Because I didn't care exactly how long t2 should take, only how long the cycle should be, I calculated t2 to be the remaining time after bandwidth for t0 and t1 was reserved.

liu-jeff, 2019-12-17:
> The example above creates a file named `qbv0.txt` with each line defining a timeslot.
> The syntax for this file is:
> 
> ```
> t0 00100000 100
> ```
> 
> Which means "first timeslot should open gate for traffic class 5 and close all other gates. This action should last for 100 ns after which the hardware should execute the next gate action". There is no next gate action (t1) so the schedule starts back from the beginning, with another t0 in this example.
> `cycle_time` is a variable holding the length of time that it should take from one t0 to the next t0. In my post above it is equal to t0+t1+t2. It is not an explicit input to the file. Because I didn't care exactly how long t2 should take, only how long the cycle should be, I calculated t2 to be the remaining time after bandwidth for t0 and t1 was reserved.
> The generated file is passed to the NXP-specific `tsntool` program which reads it and talks to the Ethernet kernel driver telling it to apply the configuration.
> `mac_base_time` is a variable holding the future time that the entire network agreed to align the schedule to. All clocks need to be synchronized for it to have any meaning.

for t0 01011111,Do 6 gates open at the same time in one time slot, so will it not cause conflict?

vladimiroltean, 2019-12-17:
An open gate just means that packets from that traffic class are allowed to be sent. In case there is contention on the wire, the strict priority scheduling will select the packets from the higher priority traffic class first.

liu-jeff, 2019-12-17:
> This is not configurable through standard Linux tools at the moment.
>
> You can do the following:
>
> 1. Save the following to a script text file (say `felix.sh`). It is work in progress and at the [...]
Bother you again，Through this configuration file I can modify the vlan priority of the port, so I want to know:
1) 0x00e0 corresponds to priority 7, what about other priority correspondence rules?
2) I want to modify the vlan id of the port, what should I do?

vladimiroltean, 2019-12-17:
1. ES_ANA_PORT_QOS_CFG register description:

* Bit 8: DP_DEFAULT_VAL (Default drop precedence level.)
* Bits 7-5: QOS_DEFAULT_VAL (Default QoS class.)
* Bit 4: QOS_DSCP_ENA (If set, the DP level and QoS class can be based on DSCP values.)
* Bit 3: QOS_PCP_ENA (If set, DP level and QoS class can be based on the PCP and DEI bits for tagged frames.)
* Bit 2: DSCP_TRANSLATE_ENA (Set if the DSCP value must be translated before using the DSCP value. If set, the translated DSCP value is given from DSCP_CFG[DSCP].DSCP_TRANSLATE_VAL.)
* Bits 1:0: DSCP_REWR_CFG (Configure which DSCP values to rewrite based on DP level and QoS class. If the DSCP value is to berewritten, then the new DSCP = DSCP_REWR_CFG[8*DP level + QoS class].DSCP_QOS_REWR_VAL.)

So 0xe0 means that QOS_DEFAULT_VAL is 0b111. In retrospect I think I should have enabled the QOS_PCP_ENA bit as well (which means the correct value for PCP 7 would have been 0xe8). For e.g. PCP 5 it would be 0xa8.

2.
```
ip link set dev br0 type bridge vlan_filtering 1
# Delete the existing pvid
bridge vlan del dev swp2 vid 1
bridge vlan add dev swp2 vid 100 pvid untagged
```
The "pvid" keyword is what does the trick.

liu-jeff, 2019-12-17:
> 1. ES_ANA_PORT_QOS_CFG register description:
> 
> * Bit 8: DP_DEFAULT_VAL (Default drop precedence level.)
> * Bits 7-5: QOS_DEFAULT_VAL (Default QoS class.)
> * Bit 4: QOS_DSCP_ENA (If set, the DP level and QoS class can be based on DSCP values.)
> * Bit 3: QOS_PCP_ENA (If set, DP level and QoS class can be based on the PCP and DEI bits for tagged frames.)
> * Bit 2: DSCP_TRANSLATE_ENA (Set if the DSCP value must be translated before using the DSCP value. If set, the translated DSCP value is given from DSCP_CFG[DSCP].DSCP_TRANSLATE_VAL.)
> * Bits 1:0: DSCP_REWR_CFG (Configure which DSCP values to rewrite based on DP level and QoS class. If the DSCP value is to berewritten, then the new DSCP = DSCP_REWR_CFG[8*DP level + QoS class].DSCP_QOS_REWR_VAL.)
> 
> So 0xe0 means that QOS_DEFAULT_VAL is 0b111. In retrospect I think I should have enabled the QOS_PCP_ENA bit as well (which means the correct value for PCP 7 would have been 0xe8). For e.g. PCP 5 it would be 0xa8.
> 
> 
> ```
> ip link set dev br0 type bridge vlan_filtering 1
> # Delete the existing pvid
> bridge vlan del dev swp2 vid 1
> bridge vlan add dev swp2 vid 100 pvid untagged
> ```
> 
> The "pvid" keyword is what does the trick.
bridge vlan add dev swp0 vid 100 pvid untagged
After entering this command, the host connected to swp0 cannot communicate normally (the host connected to swp1 cannot be pinged)

liu-jeff, 2019-12-17:
bridge vlan add dev swp0 vid 1 pvid untagged，after entering this ,It's normal again

vladimiroltean, 2019-12-17:
> bridge vlan add dev swp0 vid 1 pvid untagged，after entering this ,It's normal again

It's a switch, so it has an ingress port and an egress port. The VLAN needs to be installed on the CPU port as well: on swp4, if you ping through eno2, or on swp5, if you ping through swp0 (yes this is not a mistake).

liu-jeff, 2019-12-17:
> It's a switch, so it has an ingress port and an egress port. The VLAN needs to be installed on the CPU port as well: on swp4, if you ping through eno2, or on swp5, if you ping through swp0 (yes this [...]

Ok, thank you very much,already solved. Now I have three 1028 boards. Do you have a demo that implements synchronous qbv?

vladimiroltean, 2019-12-17:
No, I don't have any.

liu-jeff, 2019-12-17:
> It's a switch, so it has an ingress port and an egress port. The VLAN needs to be installed on the CPU port as well: on swp4, if you ping through eno2, or on swp5, if you ping through swp0 (yes this [...]

I am so sorry to bother you again, after bridge vlan add dev swp4 vid 100 pvid untagged，the host connected to swp0 cannot communicate the LS1028 still ，

vladimiroltean, 2019-12-17:
Please share the output of `bridge vlan`.

liu-jeff, 2019-12-17:
> Please share the output of `bridge vlan`.

[image]
[image]

vladimiroltean, 2019-12-17:
No, just `bridge vlan`, aka `bridge vlan show`.

liu-jeff, 2019-12-17:
[image]

vladimiroltean, 2019-12-17:
I don't think it is a problem with the VLANs. Do you have an IP address on eno2?

liu-jeff, 2019-12-17:
[image]
I ping the br0

vladimiroltean, 2019-12-17:
Should still be fine nonetheless. Can you share statistics counters?
`ethtool -S swp0 | grep -v ': 0'`
`ethtool -S swp5 | grep -v ': 0'`
`ethtool -S eno3 | grep -v ': 0'`

Preferably before and after a failed ping.

liu-jeff, 2019-12-17:
[image]
[image]
[image]
after failed ping 
[image]
[image]
[image]

liu-jeff, 2019-12-17:
Do you have some suggestion

vladimiroltean, 2020-08-06:
Please open another ticket if this issue still reproduces on the master branch (probably not).

closed 2020-08-06 by vladimiroltean