nxp-archive/openil #40: Can the scheduling model of the output port be changed?

liu-jeff, 2019-12-01:
Hello,  Can the scheduling model of the output port be changed? If I want to modify it, what should I do?

vladimiroltean, 2019-12-01:
What does this mean?

liu-jeff, 2019-12-01:
> What does this mean?
Can the code in fpga be changed?like  Qbv

liu-jeff, 2019-12-01:
> What does this mean?

Because the CPU only configures the TSN function and cannot be changed. If I want to change the TSN function, what should I do?

vladimiroltean, 2019-12-01:
Assuming that you are asking about sja1105, felix or enetc, the silicon implementation is fixed in all cases. There is no FPGA.

liu-jeff, 2019-12-01:
> Assuming that you are asking about sja1105, felix or enetc, the silicon implementation is fixed in all cases. There is no FPGA or firmware.

I don't quite follow you. I want to know whether the function (like QBV) of ls1021 or ls1028TSN can be modified? Because the kernel only does configuration,You mean it can't change it, it can only configure ?

vladimiroltean, 2019-12-01:
Please provide a more concrete example by what you mean with change vs configure.

liu-jeff, 2019-12-01:
> Please provide a more concrete example of what you mean with change vs configure.

For ls1021, the TSN function is in sja1105 chip, but the code in sja1105 cannot be modified. For example, I want to change the number of QBV gates (8 to 9). Can I change it? In fact, I just want to know whether the TSN function of ls1021 or 1028 can be developed again

vladimiroltean, 2019-12-01:
Ok, see the answer above then.

liu-jeff, 2019-12-01:
> Ok, see the answer above then. The hardware implementation is fixed.
Ok,thank you very much!

closed 2020-08-06 by vladimiroltean