nxp-archive/openil #41: 802.11AS, how much the link delay exceeds the threshold, synchronization cannot be performed

liu-jeff, 2019-12-05:
802.11AS, how much the link delay exceeds the threshold, synchronization cannot be performed

vladimiroltean, 2019-12-05:
> 802.11AS, how much the link delay exceeds the threshold, synchronization cannot be performed

The 802.11 task force refers to Wi-Fi standards, and 802.11AS doesn't exist. You probably mean 802.1AS?

Is this a question?

liu-jeff, 2019-12-05:
> > 802.11AS, how much the link delay exceeds the threshold, synchronization cannot be performed
> 
> The 802.11 task force refers to Wi-Fi standards, and 802.11AS doesn't exist. You probably mean 802.1AS?
> 
> Is this a question?

yes  it is 802.1AS

vladimiroltean, 2019-12-05:
And what issue have you encountered and on what board?

liu-jeff, 2019-12-05:
> And what issue have you encountered and on what board?

For  LS1021,I commented out neighborPropDelayThresh 800 of gPTP.cfg and found that it can be synchronized with other tsn devices

vladimiroltean, 2019-12-05:
On which port and what is the reported path delay? (also what link partner)

liu-jeff, 2019-12-05:
> On which port and what is the reported path delay? (also what link partner)
Cisco IE4000 series, and ADI's test suite,Port is ETH2,I do not  know how to look path delay

vladimiroltean, 2019-12-05:
> I do not know how to look path delay

How did you know to increase neighborPropDelayThresh then?

liu-jeff, 2019-12-05:
> > I do not know how to look path delay
> 
> How did you know to increase neighborPropDelayThresh then?
> 
> ```
> journalctl -b -u linuxptp -f
> ptp4l[14991.500]: master offset         32 s2 freq  -20537 path delay       746
> ptp4l[14992.500]: master offset         21 s2 freq  -20538 path delay       746
> ptp4l[14993.499]: master offset         -3 s2 freq  -20556 path delay       746
> ```
> 
> Of course the path delay is large if you synchronize the eth2 port directly. You are supposed to take timestamps at the sja1105 MACs. This will compensate for the switching delays.
So how  to take timetamps at the sja1105 MACs

liu-jeff, 2019-12-05:
> How did you know to increase neighborPropDelayThresh then?
>
> ```
> journalctl -b -u linuxptp -f
> ptp4l[14991.500]: master offset         32 s2 freq  -20537 path delay       746
> ptp4l[14992.500]: master [...]

My sync command is ptp4l -i eth2 -p /dev/ptp0 -f gPTP.cfg -m

vladimiroltean, 2019-12-05:
> My sync command is ptp4l -i eth2 -p /dev/ptp0 -f gPTP.cfg -m

And the output?

liu-jeff, 2019-12-05:
> > My sync command is ptp4l -i eth2 -p /dev/ptp0 -f gPTP.cfg -m
> 
> And the output?

[image]

I also want to know what is the maximum neighborPropDelayThresh, synchronization cannot be achieved

vladimiroltean, 2019-12-05:
I mean what is the output when it works (when you remove the neighborPropDelayThresh setting from gPTP.cfg)...

The 802.1AS spec says:

> gPTP requires that all Bridges and end-stations in the gPTP domain be time-aware-systems, i.e., the protocol does not transfer timing over “ordinary Bridges” (those that meet the requirements of IEEE Std 802.1D-2004 or IEEE Std 802.1Q-2005, but do NOT meet the requirements of this standard). A time-aware system uses the peer delay mechanism on each port to determine if an “ordinary Bridge” is at the other end of the link or in between itself and the Pdelay responder. If, on sending Pdelay_Req
> a) no response is received,
> b) multiple responses are received, or
> c) the measured propagation delay exceeds a specified threshold, then
>
> the protocol concludes that an “ordinary Bridge” or end-to-end TC (see IEEE Std 1588-2008) is present. In this case, the link attached to the port is deemed not capable of running gPTP, and BMCA ignores it.

I think this paragraph is pretty clear. The neighborPropDelayThresh is put there to detect if you are running gPTP over a dumb switch, and it is quite successful at detecting that.

Due to the yet still unsolved timestamp reordering issue with the sja1105 kernel driver, I don't recommend you to use that either, at the moment.

So my current suggestion is to either remove the threshold, or set it to be larger than the real path delay between eth2 and the link partner.

liu-jeff, 2019-12-05:
> I mean what is the output when it works (when you remove the neighborPropDelayThresh setting from gPTP.cfg)...
>
> The 802.1AS spec says:
>
>
> I think this paragraph is pretty clear. The [...]

It is the output after work！
[image]

Is there an upper limit for this specified threshold?Can I synchronize if my actual link latency is high?

vladimiroltean, 2019-12-05:
> Is there an upper limit for this specified threshold?

I couldn't find out where the value of 800 ns from gPTP.cfg comes from. I don't see the threshold having a specified value in the 802.1AS standard.

> Can I synchronize if my actual link latency is high?

As long as it's symmetric, yes.

liu-jeff, 2019-12-05:
> > Is there an upper limit for this specified threshold?
> 
> I couldn't find out where the value of 800 ns from gPTP.cfg comes from. I don't see the threshold having a specified value in the 802.1AS standard.
> 
> > Can I synchronize if my actual link latency is high?
> 
> As long as it's symmetric, yes.
> (within a certain limit)
How to understand symmetry？

vladimiroltean, 2019-12-05:
> How to understand symmetry?

Path delay in one direction is equal to path delay in the other.

liu-jeff, 2019-12-05:
> I couldn't find out where the value of 800 ns from gPTP.cfg comes from. I don't see the threshold having a specified value in the 802.1AS standard.
>
>
> As long as it's symmetric, yes.
> (within a [...]
[image]
This is a screenshot from the ADI test suite. The link delay is too large to synchronize.

liu-jeff, 2019-12-05:
> > How to understand symmetry?
> 
> Path delay in one direction is equal to path delay in the other.

If the delay jitter on both sides of the link is relatively large, is it impossible to synchronize?

vladimiroltean, 2019-12-05:
> If the delay jitter on both sides of the link is relatively large, is it impossible to synchronize?

There exist various PDV filtering algorithms, but largely yes.

liu-jeff, 2019-12-05:
> > If the delay jitter on both sides of the link is relatively large, is it impossible to synchronize?
> 
> There exist various PDV filtering algorithms, but largely yes.
Can you have a look at my picture above

liu-jeff, 2019-12-05:
> > If the delay jitter on both sides of the link is relatively large, is it impossible to synchronize?
> 
> There exist various PDV filtering algorithms, but largely yes.
On the board of ADI, it can be seen that the master slave clock has an error of more than 20 ms. Is this the reason for the failure of synchronization

vladimiroltean, 2019-12-05:
> Can you have a look at my picture above

Yes, the line delay on port 2 is 1060 ns. It is high, but not high enough to be the cause of loss of sync. The cause is more than likely the delay jitter, which is not visible in your screenshot.

20 ms sync offset over a dumb switch does not surprise me. The correct way to approach this, as I said, would be to take the PTP timestamps from the switch ports. This would ensure the delta only contains 2x the PHY delays, which is more or less constant.

> So how to take timetamps at the sja1105 MACs

One way would be to use ptp4l with the mainline DSA kernel driver for sja1105, for example.

liu-jeff, 2019-12-05:
> Yes, the line delay on port 2 is 1060 ns. It is high, but not high enough to be the cause of loss of sync. The cause is more than likely the delay jitter, which is not visible in your screenshot. [...]

So how do you make delay jitter smaller

vladimiroltean, 2019-12-05:
> So how do you make delay jitter smaller

The path delay is the difference between the RX timestamp at the destination MAC, and the TX timestamp at the source MAC.

```
+--------------------------------------------------------+
|                                                        |
|                                                        |
|               +------------- eth2                      |
|               |                                        |
|               |                                        |
|              cpu                                       |
|   +--------+      +--------+      +--------+           |
|   |        |      |        |      |        |           |
|   |  swp5  |      |  swp3  |      |  eth1  |           |
|   |        |      |        |      |        |           |
|   +--------+      +--------+      +--------+           |
|   |        |      |        |      |        |           |
|   |  swp4  |      |  swp2  |      |  eth0  |           |
|   |        |      |        |      |        |           |
+---+--------+------+--------+------+--------+-----------+
```

The more processing there is between the TX and the RX timestamps were taken, the more jittery the diff is going to be.

At the moment you are taking timestamps on eth2. This means the entire switch queuing delays are going to be part of the diff.

If you have a mechanism to take the timestamps from the switch ports (swp2, ... swp5), then the diff will only contain the cable delay to your Cisco device.

The mainline kernel driver gives you this option.

liu-jeff, 2019-12-05:
> One way would be to use ptp4l with the mainline DSA kernel driver for sja1105, for example.

> Yes, the line delay on port 2 is 1060 ns. It is high, but not high enough to be the cause of loss of sync. The cause is more than likely the delay jitter, which is not visible in your screenshot. [...]

> > So how do you make delay jitter smaller
> 
> The path delay is the difference between the RX timestamp at the destination MAC, and the TX timestamp at the source MAC.
> 
> ```
> +--------------------------------------------------------+
> |                                                        |
> |                                                        |
> |               +------------- eth2                      |
> |               |                                        |
> |               |                                        |
> |          cpu port                                      |
> |   +--------+      +--------+      +--------+           |
> |   |        |      |        |      |        |           |
> |   |  swp5  |      |  swp3  |      |  eth1  |           |
> |   |        |      |        |      |        |           |
> |   +--------+      +--------+      +--------+           |
> |   |        |      |        |      |        |           |
> |   |  swp4  |      |  swp2  |      |  eth0  |           |
> |   |        |      |        |      |        |           |
> +---+--------+------+--------+------+--------+-----------+
> ```
> 
> The more processing there is between the TX and the RX timestamps were taken, the more jittery the diff is going to be.
> 
> At the moment you are taking timestamps on eth2. This means the entire switch queuing delays are going to be part of the diff.
> 
> If you have a mechanism to take the timestamps from the switch ports (swp2, ... swp5), then the diff will only contain the cable delay to your Cisco device.
> 
> The mainline kernel driver gives you this option.

Thank you very much ,but Can you give me a more detailed tutorial

vladimiroltean, 2019-12-05:
> Can you give me a more detailed tutorial

Isn't this other ticket opened by you as well?
https://github.com/openil/openil/issues/35#issuecomment-560351458

In short:
compile github.com/vladimiroltean/openil-community
ptp4l -i swp2 -i swp3 -i swp4 -i swp5 -f gPTP.cfg -m

> It is found that when performing 802.1AS synchronization with other tsn devices, there is indeed instability and sometimes it stops

The "sometimes it stops" behavior affects the mainline kernel driver, isn't this what you were talking about?

I am actively trying to find and fix the root issue behind it, but if you open new questions non-stop then when am I supposed to do that?

liu-jeff, 2019-12-06:
> > Can you give me a more detailed tutorial
> 
> Isn't this other ticket opened by you as well?
> [#35 (comment)](https://github.com/openil/openil/issues/35#issuecomment-560351458)
> 
> In short:
> compile github.com/vladimiroltean/openil-community
> ptp4l -i swp2 -i swp3 -i swp4 -i swp5 -f gPTP.cfg -m
> 
> > It is found that when performing 802.1AS synchronization with other tsn devices, there is indeed instability and sometimes it stops
> 
> The "sometimes it stops" behavior affects the mainline kernel driver, isn't this what you were talking about?
> 
> I am actively trying to find and fix the root issue behind it, but if you open new questions non-stop then when am I supposed to do that?

I am so sorry to bother you!

liu-jeff, 2019-12-07:
> Isn't this other ticket opened by you as well?
> [#35 (comment)](https://github.com/openil/openil/issues/35#issuecomment-560351458)
>
> In short:
> compile github.com/vladimiroltean/openil-community
> ptp4l [...]
I'm sorry to disturb you again, but I still want to ask you a question, I want to know what the meaning of rms max freq delay in the picture below
[image]

vladimiroltean, 2020-08-06:
Please open another ticket if this problem still reproduces in the latest master branch.

closed 2020-08-06 by vladimiroltean