nxp-archive/openil #43: How to change the ip address in LS1021ATSN using openil-community

sabarishsubramanian, 2019-12-06:
I have done the initial steps by loading the files in sd card and inserting in the board and booting it and paused it and i got => in the boot loader command.

Now how will proceed to change the ip address?

vladimiroltean, 2019-12-06:
```
setenv ipaddr 192.168.1.2
setenv serverip 192.168.1.1
ping $serverip
```
Is this what you want? To ping from U-Boot?

sabarishsubramanian, 2019-12-06:
want to change the MAC addresse and ip address of the  boards by executing these U-boot commands

vladimiroltean, 2019-12-06:
To change the MAC address, follow the [official U-Boot steps to reprogram the EEPROM ](https://gitlab.denx.de/u-boot/u-boot/tree/master/board/freescale/ls1021atsn)and reboot.

sabarishsubramanian, 2019-12-09:
its not working 
=> setenv ethaddr  00:04:9f:ef:00:00
=> setenv eth1addr 00:04:9f:ef:01:01
=> setenv eth2addr 00:04:9f:ef:02:02

do i need to use this command for changing ip ?

if yes do i need to change any address of it and wht is that?

sabarishsubramanian, 2019-12-09:
i got it

 i have another question 

where can i find the network folder or interfaces 

because am not getting in etc folde r when i give 

/etc/network/interfaces 

this command

closed 2019-12-09 by sabarishsubramanian