nxp-archive/openil #44: how to find the network folder or interfaces?

sabarishsubramanian, 2019-12-09:
In openil we can change the ethernet address in the boot menu and also change the ip address of the tsn boards and VLAN by changing few lines in /etc/network/interfaces files.

But i dont see such files existing in my community kernal.

So please let me know how to do the changes in the folder?

sabarishsubramanian, 2019-12-09:
# /etc/network/interfaces – configuration file for ifup(8), ifdown(8)
# The loopback interface
auto lo
iface loinet loopback
auto eth2
iface eth2 inet static
address 172.15.0.1
netmask 255.255.255.0
auto eth1
iface eth1 inet dhcp
auto eth0i
face eth0 inet static
address 192.168.15.1
netmask 255.255.255.0

I want to edit in the above file. But i cant find any folder name interfaces in the above mentioned path.

Please help me getting it.

vladimiroltean, 2019-12-09:
The "community" version of OpenIL for the LS1021A-TSN uses systemd-networkd as a network manager, in its default configuration (it can be changed, of course). Therefore, the legacy paths for configuring network interfaces are not used. The configuration files are in `/etc/systemd/network/`.
You can use the [Freedesktop web page](https://www.freedesktop.org/software/systemd/man/systemd.network.html) for general reference documentation on how to configure the interfaces.

sabarishsubramanian, 2019-12-09:
thank you for your info i can found it now but the problem is where should i edit it i mean which folder i could see 5 different files inside network folder can you please specify exactly which file i need to edit

vladimiroltean, 2019-12-09:
The VLAN setup is not the same in the community version and you should take some time to [understand what that means](https://www.kernel.org/doc/html/latest/networking/dsa/sja1105.html#traffic-support).
By default in the DSA driver, 8 VLAN IDs private to the kernel are configured on the switch ports so the kernel can do stuff such as ping on individual ports and not just through eth2.
Then with a DSA/switchdev driver, VLAN awareness is handled through the bridge net device `vlan_filtering` property and `bridge vlan` commands. But the kernel driver can't handle at the moment [0] both (a) VLANs requested by the user, and (b) private VLANs necessary for decoding the switch source port. So, while theoretically you can edit `/etc/systemd/network/br0.netdev` and add:

```
[Bridge]
VLANFiltering=no
```

and install VLANs to switch ports using commands as below:

```
bridge vlan add dev swp2 vid 100
```

but doing so **will disable regular traffic through sockets opened on the swp2, swp3, swp4, swp5 net devices** (except management traffic: STP, PTP, LLDP etc). In `vlan_filtering=1` mode you go back to sending traffic in a similar way as in the official OpenIL distribution (aka over the eth2 net device).

This answer is mainly just describing the aspects strictly related to the sja1105 driver here. You can choose to use systemd or something else.

The systemd-specific ways for creating a VLAN sub-interface are:
- put the following in `/etc/systemd/network/eth2.network`:

```
[Match]
Name=eth2

[Network]
DHCP=ipv4
VLAN=eth2.100
```

- put the following in `/etc/systemd/network/eth2.100.netdev`:

```
[NetDev]
Name=eth2.100
Kind=vlan

[VLAN]
Id=100
```

and for adding a VLAN to the port membership table of a switch port (add the following to `/etc/systemd/network/swp.network`):

```
[BridgeVLAN]
VLAN=100
```

I don't know if this is useful to your setup. It would help if you could explain what VLAN setup you want to perform, and I can suggest how the system can be configured for that.

[0] There is [a patch for that](https://www.spinics.net/lists/netdev/msg613722.html), but I'm not yet convinced it is the best solution, so it's not integrated in the community OpenIL.

sabarishsubramanian, 2019-12-09:
actually am trying to work first with single tsn board and i want to change the ip address and mac address of it.

later i want to change for 3 boards parallely without changing the usb wire from the host pc after the connection setup.

vladimiroltean, 2019-12-09:
MAC address should be a non-volatile property of each board, stored in EEPROM and read out by U-Boot. It is a shame that official OpenIL puts fixed MAC addresses in the default U-Boot environment, as they will collide as soon as you have more than one board in the same network. So as long as the EEPROM is set up and the U-Boot environment has the `ethaddr`, `eth1addr` etc variables unset, each board should be able to use its own MAC address.

As for IP address assignment, that is not an intrinsically board-specific property but depends on the network it is part of. So there are various ways of doing it, depending on what you want to achieve (DHCP, DHCP reservations based on MAC address, static IP addresses). What do you want to achieve, and are the steps in the systemd-networkd documentation not clear with regard to that?

sabarishsubramanian, 2019-12-09:
i need to have a static address for ethernet how should i proceed for one tsn board

vladimiroltean, 2019-12-09:
Put the following in `/etc/systemd/network/eth0.network`:

```
[Match]
Name=eth0

[Network]
Address=192.168.1.100/24
Gateway=192.168.1.1
```

Based on an example taken from the [Arch wiki](https://wiki.archlinux.org/index.php/Systemd-networkd#Wired_adapter_using_a_static_IP).

sabarishsubramanian, 2019-12-09:
i did for my host ip address i tried to do for eth 1 and eth 2 it didnt change
eth0      Link encap:Ethernet  HWaddr 00:04:9F:EF:00:00  
          inet addr:192.168.30.21  Bcast:192.168.30.255  Mask:255.255.255.0
          inet6 addr: fe80::204:9fff:feef:0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:40 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:6476 (6.3 KiB)
          Base address:0xa000 

eth1      Link encap:Ethernet  HWaddr 00:04:9F:EF:01:01  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Base address:0xe000 

eth2      Link encap:Ethernet  HWaddr 00:04:9F:EF:02:02  
          inet6 addr: fe80::204:9fff:feef:202/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1504  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:25 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:3655 (3.5 KiB)
          Base address:0xc000

vladimiroltean, 2019-12-09:
IP addresses for bridged interfaces won't stick. If an interface is enslaved to br0 then you should put the IP address on br0, otherwise put the IP address on it.

vladimiroltean, 2019-12-09:
And can you use the tooling from iproute2 instead of ifconfig please (ip addr, ip link, etc), ifconfig has been deprecated for at least 10 years now and iproute2 is more informative (e.g. the output of "ip a" or "bridge link" would have told me which interfaces are bridged).

sabarishsubramanian, 2019-12-10:
cant able to give ip address to eth2 port 

eth0      Link encap:Ethernet  HWaddr 00:04:9F:EF:00:00  
          inet addr:192.168.30.21  Bcast:192.168.30.255  Mask:255.255.255.0
          inet6 addr: fe80::204:9fff:feef:0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:53 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:7949 (7.7 KiB)
          Base address:0xa000 

eth1      Link encap:Ethernet  HWaddr 00:04:9F:EF:01:01  
          inet addr:192.168.1.130  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Base address:0xe000 

eth2      Link encap:Ethernet  HWaddr 00:04:9F:EF:02:02  
          inet6 addr: fe80::204:9fff:feef:202/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1504  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:29 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:4034 (3.9 KiB)
          Base address:0xc000

vladimiroltean, 2019-12-10:
Please share the contents of your `/etc/systemd/network/eth2.network` file.

sabarishsubramanian, 2019-12-10:
i got cleared it 

my next step is connecting the tsn wirelessly and am experiencing this 

ssh: connect to host 192.168.30.66 port 22: No route to host

vladimiroltean, 2019-12-10:
"No route to host" means exactly what it says: the PC you are trying to SSH from has no IP route towards 192.168.30.66. You either put the PC in the same subnet (since it's probably a /24 network, that means 192.168.30.x), or you have a default gateway that knows how to find the IP.
On another note, you should first try to ping before you attempt to SSH into the box. Root SSH login is not enabled by default, so you'll have extra trouble configuring that. There is a command called `passwordless-ssh-login` that you can run for one-time easy setup of root SSH access with no password.

sabarishsubramanian, 2019-12-10:
ssh: connect to host 192.168.30.61 port 22: Connection refused

now its getting refused

closed 2019-12-10 by sabarishsubramanian