nxp-archive/openil #48: Time sychronization

sabarishsubramanian, 2019-12-12:
Hello

I am trying to have a time sychronization of three TSN boards. I am trying to do in openil- community part.

The pdf which i have is of openil where they use  SJA1105 tools and all. where in community part there is no more  SJA1105 tools exits.

Could someone please help me how to start and proceed further?

Regards
sabarish s

sabarishsubramanian, 2019-12-12:
i have changed my tsn boards ip address and all.

I have a host pc, 3 tsn boards and a switch. The connection is done according to the pdf of openil.

Now i wanna time synchronise it. How shall i start and proceed further since in openil community there are no more SJA1105 tools exits.

Please someone help me figure this out.

Thank you

vladimiroltean, 2019-12-12:
Hi Sabarish,

I will post a detailed guide here but at the moment I need to take care of some other pressing issues, so it will take me a day or so.

Thanks,
-Vladimir

sabarishsubramanian, 2019-12-12:
can u just guide me by then how should i start this so that i can try it out till that time.

sabarishsubramanian, 2019-12-12:
Ok no problem i will wait your reply and documentation soon.

Thank you

Regards
sabarish subramanian

sabarishsubramanian, 2019-12-13:
hey @vladimiroltean sorry to disturb you. Waiting for your documentation soon.

thank tou

vladimiroltean, 2019-12-13:
Yes, I am working on it. It is taking a while.

vladimiroltean, 2019-12-13:
Setup
-----

It is recommended that you update the openil-community image on all boards to
at least [the following
commit](https://github.com/vladimiroltean/openil-community/commit/c3a7ef8b0a0d6e1d5cf6e8b64014f9eafc75abee)(which
at the time of writing is the most recent).

Follow the [official U-Boot steps to reprogram the
EEPROM](https://gitlab.denx.de/u-boot/u-boot/tree/master/board/freescale/ls1021atsn)
to make sure that all boards have unique MAC addresses.

For the synchronized 802.1Qbv demo, the following setup is advised:

```
+------------------------------------------------------------+
|                                          LS1021A-TSN       |
|                         gianfar            Board 1         |
|                 +-------- eth2                             |
|                 |      DSA master                          |
|                 |                                          |
|              cpu port                                      |
|   +--------+         +--------+         +--------+         |
|   |        |         |        |         |        |         |
|   |  swp5  |         |  swp3  |         |  eth1  | gianfar |
|   |        |         |        |         |        |         |
|   +--------+ sja1105 +--------+         +--------+         |
|   |        |         |        |         |        |         |
|   |  swp4  |         |  swp2  |         |  eth0  | gianfar |
|   |        |         |        |         |        |         |
+---+--------+---------+--------+---------+--------+---------+
                                              |
                                              |
                                      +-------+
                                      |
+-------------------------------------|----------------------+
|                                     |    LS1021A-TSN       |
|                         gianfar     |      Board 2         |
|                 +-------- eth2      |                      |
|                 |      DSA master   |                      |
|                 |                   |                      |
|              cpu port               |                      |
|   +--------+         +--------+     |   +--------+         |
|   |        |         |        |     |   |        |         |
|   |  swp5  |         |  swp3  |<----+   |  eth1  | gianfar |
|   |        |         |        |         |        |         |
|   +--------+ sja1105 +--------+         +--------+         |
|   |        |         |        |         |        |         |
|   |  swp4  |         |  swp2  |-----+   |  eth0  | gianfar |
|   |        |         |        |     |   |        |         |
+---+--------+---------+--------+-----|---+--------+---------+
                                      |
                                      |
                                      |
+-------------------------------------|----------------------+
|                                     |    LS1021A-TSN       |
|                         gianfar     |       Board 3        |
|                 +-------- eth2      |                      |
|                 |      DSA master   |                      |
|                 |                   |                      |
|              cpu port               |                      |
|   +--------+         +--------+     |   +--------+         |
|   |        |         |        |     |   |        |         |
|   |  swp5  |         |  swp3  |     |   |  eth1  | gianfar |
|   |        |         |        |     |   |        |         |
|   +--------+ sja1105 +--------+     |   +--------+         |
|   |        |         |        |     |   |        |         |
|   |  swp4  |         |  swp2  |     +-->|  eth0  | gianfar |
|   |        |         |        |         |        |         |
+---+--------+---------+--------+---------+--------+---------+
[... 315 lines cut ...]
# PTP: create a window for 1 packet of 200 bytes at the current link speed
t1=$(qbv_window 200 1 $speed_mbps)
# ping: create a window for 1 packet of 64 bytes
t2=$(qbv_window 64 1 $speed_mbps)
# Best effort: the rest of the time up to 30 ms
t0=$(($cycle_time - $t1 - $t2))

tc qdisc add dev swp2 parent root handle 100 taprio \
	num_tc 8 \
	map 0 1 2 3 4 5 6 7 \
	queues 1@0 1@1 1@2 1@3 1@4 1@5 1@6 1@7 \
	base-time ${base_time} \
	sched-entry S $(gatemask 7) $t0 \
	sched-entry S $(gatemask 5) $t1 \
	sched-entry S $(gatemask "0 1 2 3 4 6") $t2 \
	flags 2
```

More complete documentation on the tc-taprio offload command for the sja1105 is
available
[here](https://www.kernel.org/doc/html/latest/networking/dsa/sja1105.html#time-aware-scheduling).

Please note that at the time of writing I did not have access to 3 LS1021A-TSN
boards, so boards 1 and 3 were LS1028A-RDB during my testing. I will keep
updating this post if I find any issues with the steps.

sabarishsubramanian, 2019-12-16:
Thank you @vladimiroltean am trying to do the steps according to the above mentioned. If i get stuck with something will let you know .

sabarishsubramanian, 2019-12-17:
Hello

I tried the steps which was given by you.
Everything works fine, even the pinging happens between the board.

Am now experiencing a problem in time synchronization and it is as follows:
BOARD1:
[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Thu 2019-10-10 12:56:03 UTC. --
Oct 10 13:05:01 OpenIL ptp4l[180]: [544.942] port 1: rogue peer delay response
Oct 10 13:05:01 OpenIL ptp4l[180]: [544.947] port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Oct 10 13:05:01 OpenIL ptp4l[180]: [545.040] selected local clock 00049f.fffe.ef0000 as best master
Oct 10 13:05:01 OpenIL ptp4l[180]: [545.041] port 1: assuming the grand master role
Oct 10 13:05:17 OpenIL ptp4l[180]: [561.124] port 1: FAULTY to LISTENING on INIT_COMPLETE
Oct 10 13:05:21 OpenIL ptp4l[180]: [564.727] port 1: announce timeout
Oct 10 13:05:21 OpenIL ptp4l[180]: [564.728] bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 13:05:21 OpenIL ptp4l[180]: [564.728] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 13:05:21 OpenIL ptp4l[180]: [564.728] selected local clock 00049f.fffe.ef0000 as best master
Oct 10 13:05:21 OpenIL ptp4l[180]: [564.728] port 1: assuming the grand master role

BOARD 2:
[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Tue 2019-09-03 09:27:20 UTC. --
Sep 03 09:29:03 OpenIL ptp4l[165]: [108.362] port 4: assuming the grand master role
Sep 03 09:29:03 OpenIL ptp4l[165]: [108.367] selected local clock 00049f.fffe.ef0505 as best master
Sep 03 09:29:03 OpenIL ptp4l[165]: [108.367] port 1: assuming the grand master role
Sep 03 09:29:03 OpenIL ptp4l[165]: [108.367] port 2: assuming the grand master role
Sep 03 09:29:03 OpenIL ptp4l[165]: [108.367] port 3: assuming the grand master role
Sep 03 09:29:03 OpenIL ptp4l[165]: [108.367] port 4: assuming the grand master role
Sep 03 09:30:09 OpenIL systemd[1]: Stopping Precision Time Protocol daemon...
Sep 03 09:30:09 OpenIL systemd[1]: linuxptp.service: Succeeded.
Sep 03 09:30:09 OpenIL systemd[1]: Stopped Precision Time Protocol daemon.
Sep 03 09:30:09 OpenIL systemd[1]: linuxptp.service: Consumed 2.282s CPU time.

BOARD3 :
[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Thu 2019-10-10 12:56:03 UTC. --
Oct 10 13:04:52 OpenIL ptp4l[180]: [534.990] port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Oct 10 13:04:52 OpenIL ptp4l[180]: [535.070] selected local clock 00049f.fffe.ef0606 as best master
Oct 10 13:04:52 OpenIL ptp4l[180]: [535.071] port 1: assuming the grand master role
Oct 10 13:05:08 OpenIL ptp4l[180]: [551.151] port 1: FAULTY to LISTENING on INIT_COMPLETE
Oct 10 13:05:08 OpenIL ptp4l[180]: [551.361] clockcheck: clock jumped backward or running slower than expected!
Oct 10 13:05:12 OpenIL ptp4l[180]: [555.091] port 1: announce timeout
Oct 10 13:05:12 OpenIL ptp4l[180]: [555.091] bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 13:05:12 OpenIL ptp4l[180]: [555.091] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 13:05:12 OpenIL ptp4l[180]: [555.091] selected local clock 00049f.fffe.ef0606 as best master
Oct 10 13:05:12 OpenIL ptp4l[180]: [555.091] port 1: assuming the grand master role

According to this it seems like the boards are not communicating the time information within themself. But the ping to other boards works fine.  How can i solve this issue?

Thanks in advance.

sabarishsubramanian, 2019-12-18:
hey 

Today i reflshed everything and i did it from the start and today am again stuck in synchronization and its output is this

Board 1:
[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Thu 2019-10-10 12:56:03 UTC. --
Oct 10 13:16:40 OpenIL ptp4l[262]: [1243.241] port 1: INITIALIZING to LISTENING on INIT_COMPLETE
Oct 10 13:16:40 OpenIL ptp4l[262]: [1243.244] port 0: INITIALIZING to LISTENING on INIT_COMPLETE
Oct 10 13:16:43 OpenIL ptp4l[262]: [1246.837] port 1: announce timeout
Oct 10 13:16:43 OpenIL ptp4l[262]: [1246.837] bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 13:16:43 OpenIL ptp4l[262]: [1246.837] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 13:16:43 OpenIL ptp4l[262]: [1246.837] selected local clock 00049f.fffe.ef0000 as best master
Oct 10 13:16:43 OpenIL ptp4l[262]: [1246.838] port 1: assuming the grand master role
Oct 10 13:16:45 OpenIL ptp4l[262]: [1248.628] port 1: new foreign master 00049f.fffe.ef0505-2
Oct 10 13:16:45 OpenIL ptp4l[262]: [1248.629] selected best master clock 00049f.fffe.ef0505
Oct 10 13:16:45 OpenIL ptp4l[262]: [1248.629] port 1: assuming the grand master role
^C
[root@OpenIL ~] # 

Board 2
[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Tue 2019-09-03 09:27:20 UTC. --
Sep 03 10:32:25 OpenIL ptp4l[263]: [3910.300] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:32:26 OpenIL ptp4l[263]: [3911.052] rms   69 max   84 freq -25211 +/-  41 delay   635 +/-   0
Sep 03 10:32:27 OpenIL ptp4l[263]: [3912.054] rms   32 max   42 freq -25223 +/-   8 delay   635 +/-   0
Sep 03 10:32:28 OpenIL ptp4l[263]: [3913.057] rms    9 max   17 freq -25207 +/-   8 delay   635 +/-   0
Sep 03 10:32:29 OpenIL ptp4l[263]: [3914.059] rms    6 max    8 freq -25191 +/-   6 delay   635 +/-   0
Sep 03 10:32:29 OpenIL ptp4l[263]: [3914.811] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:32:29 OpenIL ptp4l[263]: [3914.811] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:32:30 OpenIL ptp4l[263]: [3914.937] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:32:30 OpenIL ptp4l[263]: [3915.062] rms 3037000498 max 8589934586 freq -25190 +/-  11 delay   634 +/-   0
Sep 03 10:32:30 OpenIL ptp4l[263]: [3915.062] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:32:31 OpenIL ptp4l[263]: [3916.065] rms    7 max   13 freq -25199 +/-   7 delay   634 +/-   0
Sep 03 10:32:32 OpenIL ptp4l[263]: [3916.942] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:32:32 OpenIL ptp4l[263]: [3916.942] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:32:32 OpenIL ptp4l[263]: [3917.067] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:32:32 OpenIL ptp4l[263]: [3917.068] rms 3037000498 max 8589934585 freq -25187 +/-   4 delay   635 +/-   0
Sep 03 10:32:32 OpenIL ptp4l[263]: [3917.193] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:32:33 OpenIL ptp4l[263]: [3918.071] rms   21 max   35 freq -25152 +/-  22 delay   634 +/-   0
Sep 03 10:32:34 OpenIL ptp4l[263]: [3919.073] rms   29 max   35 freq -25197 +/-   7 delay   634 +/-   0
Sep 03 10:32:34 OpenIL ptp4l[263]: [3919.326] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:32:34 OpenIL ptp4l[263]: [3919.575] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:32:35 OpenIL ptp4l[263]: [3920.076] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:32:35 OpenIL ptp4l[263]: [3920.076] rms 4294967288 max 8589934577 freq -25198 +/-  12 delay   634 +/-   0
Sep 03 10:32:35 OpenIL ptp4l[263]: [3920.076] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:32:35 OpenIL ptp4l[263]: [3920.201] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:32:35 OpenIL ptp4l[263]: [3920.327] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:32:36 OpenIL ptp4l[263]: [3921.078] rms   17 max   25 freq -25177 +/-  13 delay   634 +/-   0
Sep 03 10:32:36 OpenIL ptp4l[263]: [3921.204] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:32:36 OpenIL ptp4l[263]: [3921.204] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:32:36 OpenIL ptp4l[263]: [3921.323] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:32:36 OpenIL ptp4l[263]: [3921.455] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:32:37 OpenIL ptp4l[263]: [3922.081] rms 3037000508 max 8589934616 freq -25192 +/-  13 delay   634 +/-   0
Sep 03 10:32:37 OpenIL ptp4l[263]: [3922.323] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:32:37 OpenIL ptp4l[263]: [3922.332] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:32:37 OpenIL ptp4l[263]: [3922.457] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:32:37 OpenIL ptp4l[263]: [3922.583] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:32:38 OpenIL ptp4l[263]: [3923.084] rms   12 max   22 freq -25207 +/-  14 delay   634 +/-   0
Sep 03 10:32:39 OpenIL ptp4l[263]: [3924.087] rms   11 max   24 freq -25199 +/-  12 delay   634 +/-   0
Sep 03 10:32:39 OpenIL ptp4l[263]: [3924.838] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:32:39 OpenIL ptp4l[263]: [3924.839] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
[... cut ...]
Oct 10 14:01:05 OpenIL ptp4l[272]: [3908.312] bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 14:01:05 OpenIL ptp4l[272]: [3908.312] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 14:01:05 OpenIL ptp4l[272]: [3908.312] selected local clock 00049f.fffe.ef0606 as best master
Oct 10 14:01:05 OpenIL ptp4l[272]: [3908.313] port 1: assuming the grand master role
^C
[root@OpenIL ~] # 

Kindly please help me how to solve this issue

Thank you
regards
sabarish s

vladimiroltean, 2019-12-18:
This is strange, sorry for the inconvenience.
I think this is what is going on:
When you run the `ip link set dev br0 type bridge vlan_filtering 1` command, the driver does this:
```
[  379.897744] sja1105 spi0.1: Reset switch and programmed static config. Reason: VLAN filtering
[  379.908409] sja1105 spi0.1: Disabled switch tagging
```
The switch reset introduces a momentary interruption in the synchronization process.
The PTP slaves then adjust their clocks to a sudden spike in PTP time, which makes their frequency adjustment stuck to a high value from which they don't recover. They keep printing the clockcheck messages.
I reproduced this issue, and by restarting the ptp4l process with `-l 7` for debugging output, I see:
```
ptp4l[103607.121]: port 1: drop erroneous nratio 0.898000, max offset 0.000200
```
So all further frequency adjustments are too large and get dropped.

As a workaround until I figure out how to recover automatically from this, can you please run on all 3 boards, in whatever order:
```
systemctl stop linuxptp
phc_ctl /dev/ptp0 freq 0 && phc_ctl /dev/ptp1 freq 0
systemctl start linuxptp
```
The above resets the frequency adjustment for the eTSEC PTP hardware clock (/dev/ptp0) and for the sja1105 PTP hardware clock (/dev/ptp1). It works fine afterwards for me.

sabarishsubramanian, 2019-12-18:
systemctl stop linuxptp
phc_ctl /dev/ptp0 freq 0 && phc_ctl /dev/ptp1 freq 0
systemctl start linuxptp

i ran this above commands 

and gave  $ journalctl -b -u linuxptp -f

BOARD 1

[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Thu 2019-10-10 12:56:03 UTC. --
Oct 10 13:59:57 OpenIL ptp4l[327]: [3841.097] port 1: announce timeout
Oct 10 13:59:57 OpenIL ptp4l[327]: [3841.097] bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 13:59:57 OpenIL ptp4l[327]: [3841.097] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 13:59:57 OpenIL ptp4l[327]: [3841.097] selected local clock 00049f.fffe.ef0000 as best master
Oct 10 13:59:57 OpenIL ptp4l[327]: [3841.097] port 1: assuming the grand master role
Oct 10 13:59:59 OpenIL ptp4l[327]: [3842.756] port 1: new foreign master 00049f.fffe.ef0505-2
Oct 10 13:59:59 OpenIL ptp4l[327]: [3842.756] selected best master clock 00049f.fffe.ef0505
Oct 10 13:59:59 OpenIL ptp4l[327]: [3842.756] port 1: assuming the grand master role
Oct 10 14:00:03 OpenIL ptp4l[327]: [3846.758] selected best master clock 00049f.fffe.ef0505
Oct 10 14:00:03 OpenIL ptp4l[327]: [3846.758] port 1: assuming the grand master role

BOARD 2
[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Tue 2019-09-03 09:27:20 UTC. --
Sep 03 10:53:47 OpenIL ptp4l[307]: [5192.617] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:53:47 OpenIL ptp4l[307]: [5192.617] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:53:47 OpenIL ptp4l[307]: [5192.743] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:53:47 OpenIL ptp4l[307]: [5192.868] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:53:48 OpenIL ptp4l[307]: [5193.119] rms 3037000489 max 8589934561 freq -25266 +/-  32 delay   626 +/-   0
Sep 03 10:53:49 OpenIL ptp4l[307]: [5193.996] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:53:49 OpenIL ptp4l[307]: [5193.996] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:53:49 OpenIL ptp4l[307]: [5194.121] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:53:49 OpenIL ptp4l[307]: [5194.122] rms 3037000483 max 8589934544 freq -25252 +/-  13 delay   626 +/-   0
Sep 03 10:53:49 OpenIL ptp4l[307]: [5194.247] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:53:50 OpenIL ptp4l[307]: [5195.124] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:53:50 OpenIL ptp4l[307]: [5195.124] rms 3037000506 max 8589934610 freq -25218 +/-  21 delay   626 +/-   0
Sep 03 10:53:50 OpenIL ptp4l[307]: [5195.125] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:53:50 OpenIL ptp4l[307]: [5195.249] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:53:50 OpenIL ptp4l[307]: [5195.375] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:53:51 OpenIL ptp4l[307]: [5196.128] rms   12 max   18 freq -25235 +/-  11 delay   626 +/-   0
Sep 03 10:53:51 OpenIL ptp4l[307]: [5196.754] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:53:51 OpenIL ptp4l[307]: [5196.754] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:53:51 OpenIL ptp4l[307]: [5196.881] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:53:52 OpenIL ptp4l[307]: [5197.005] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:53:52 OpenIL ptp4l[307]: [5197.130] rms 3796250625 max 10737418240 freq -25241 +/-   5 delay   626 +/-   0
Sep 03 10:53:53 OpenIL ptp4l[307]: [5198.133] rms    8 max   11 freq -25250 +/-   5 delay   625 +/-   0
Sep 03 10:53:53 OpenIL ptp4l[307]: [5198.258] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:53:53 OpenIL ptp4l[307]: [5198.259] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:53:53 OpenIL ptp4l[307]: [5198.383] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:53:53 OpenIL ptp4l[307]: [5198.509] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:53:54 OpenIL ptp4l[307]: [5199.136] rms 3796250625 max 10737418240 freq -25212 +/-  32 delay   624 +/-   0
Sep 03 10:53:55 OpenIL ptp4l[307]: [5200.138] rms   37 max   41 freq -25253 +/-  13 delay   625 +/-   0
Sep 03 10:53:55 OpenIL ptp4l[307]: [5200.891] clockcheck: clock jumped backward or running slower than expected!
Sep 03 10:53:55 OpenIL ptp4l[307]: [5200.892] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 10:53:56 OpenIL ptp4l[307]: [5201.141] clockcheck: clock jumped forward or running faster than expected!
Sep 03 10:53:56 OpenIL ptp4l[307]: [5201.141] rms 8547888651 max 13958643725 freq -25254 +/-  15 delay   624 +/-   0
Sep 03 10:53:56 OpenIL ptp4l[307]: [5201.392] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 10:53:57 OpenIL ptp4l[307]: [5202.145] rms 4935125805 max 13958643692 freq -25255 +/-   5 delay   623 +/-   0
Sep 03 10:53:58 OpenIL ptp4l[307]: [5203.147] rms    7 max   12 freq -25245 +/-   5 delay   623 +/-   0
Sep 03 10:53:59 OpenIL ptp4l[307]: [5204.151] rms    6 max   13 freq -25241 +/-   5 delay   623 +/-   0
Sep 03 10:54:00 OpenIL ptp4l[307]: [5205.153] rms    4 max    8 freq -25245 +/-   6 delay   623 +/-   0
Sep 03 10:54:01 OpenIL ptp4l[307]: [5206.156] rms    5 max    8 freq -25247 +/-   7 delay   623 +/-   0
[... cut ...]
Oct 10 14:03:51 OpenIL ptp4l[288]: [4074.739] rms 11863250 max 33554338 freq  -5007 +/- 678 delay   636 +/-   0
Oct 10 14:03:52 OpenIL ptp4l[288]: [4075.744] rms 3037000465 max 8589934493 freq  -4057 +/- 175 delay   635 +/-   0
Oct 10 14:03:53 OpenIL ptp4l[288]: [4076.747] rms 11863246 max 33554328 freq  -4322 +/- 102 delay   635 +/-   0
Oct 10 14:03:54 OpenIL ptp4l[288]: [4077.751] rms 4294967298 max 8589934693 freq  -4228 +/-  63 delay   635 +/-   0
^C

i can still see its not yet synchronized

vladimiroltean, 2019-12-18:
Can you do the following:
Boot the board acting as a switch (2) first. Keep the ptp4l logging open.
Boot the board acting as a GM (1). Watch the ptp4l process on board 2 while it synchronizes to the GM.
Boot the board acting as slave (3). Watch it again until it synchronizes through the switch to the GM.
Everything should be fine at this point.
Then, while you run the setup commands on board 2, can you monitor the other two, to see which one appears to break the synchronization?

sabarishsubramanian, 2019-12-19:
Hello @vladimiroltean 

I tried to do the above steps 

after doing the first two steps only i got this in each board

Board 1
[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Thu 2019-10-10 12:56:03 UTC. --
Oct 10 12:56:15 OpenIL ptp4l[180]: [18.741] port 1: link up
Oct 10 12:56:15 OpenIL ptp4l[180]: [18.862] port 1: FAULTY to LISTENING on INIT_COMPLETE
Oct 10 12:56:18 OpenIL ptp4l[180]: [22.218] port 1: announce timeout
Oct 10 12:56:18 OpenIL ptp4l[180]: [22.219] bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 12:56:18 OpenIL ptp4l[180]: [22.219] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Oct 10 12:56:18 OpenIL ptp4l[180]: [22.219] selected local clock 00049f.fffe.ef0000 as best master
Oct 10 12:56:18 OpenIL ptp4l[180]: [22.219] port 1: assuming the grand master role
Oct 10 12:56:20 OpenIL ptp4l[180]: [23.537] port 1: new foreign master 00049f.fffe.ef0505-2
Oct 10 12:56:20 OpenIL ptp4l[180]: [23.537] selected best master clock 00049f.fffe.ef0505
Oct 10 12:56:20 OpenIL ptp4l[180]: [23.537] port 1: assuming the grand master role
^C

Board 2
[root@OpenIL ~] # journalctl -b -u linuxptp -f
-- Logs begin at Tue 2019-09-03 09:27:20 UTC. --
Sep 03 09:27:28 OpenIL ptp4l[166]: [13.505] port 2: LISTENING to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Sep 03 09:27:28 OpenIL ptp4l[166]: [13.572] port 3: link down
Sep 03 09:27:28 OpenIL ptp4l[166]: [13.573] port 3: LISTENING to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Sep 03 09:27:29 OpenIL ptp4l[166]: [13.671] port 4: link down
Sep 03 09:27:29 OpenIL ptp4l[166]: [13.672] port 4: LISTENING to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Sep 03 09:27:29 OpenIL ptp4l[166]: [13.750] selected local clock 00049f.fffe.ef0505 as best master
Sep 03 09:27:29 OpenIL ptp4l[166]: [13.750] port 1: assuming the grand master role
Sep 03 09:27:29 OpenIL ptp4l[166]: [13.750] port 2: assuming the grand master role
Sep 03 09:27:29 OpenIL ptp4l[166]: [13.750] port 3: assuming the grand master role
Sep 03 09:27:29 OpenIL ptp4l[166]: [13.751] port 4: assuming the grand master role
Sep 03 09:28:38 OpenIL ptp4l[166]: [83.229] port 2: link up
Sep 03 09:28:38 OpenIL ptp4l[166]: [83.341] port 2: FAULTY to LISTENING on INIT_COMPLETE
Sep 03 09:28:42 OpenIL ptp4l[166]: [87.320] port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Sep 03 09:28:42 OpenIL ptp4l[166]: [87.320] selected local clock 00049f.fffe.ef0505 as best master
Sep 03 09:28:42 OpenIL ptp4l[166]: [87.321] port 1: assuming the grand master role
Sep 03 09:28:42 OpenIL ptp4l[166]: [87.321] port 2: assuming the grand master role
Sep 03 09:28:42 OpenIL ptp4l[166]: [87.321] port 3: assuming the grand master role
Sep 03 09:28:42 OpenIL ptp4l[166]: [87.321] port 4: assuming the grand master role
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.336] port 2: link down
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.336] port 2: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.380] selected local clock 00049f.fffe.ef0505 as best master
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.380] port 1: assuming the grand master role
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.381] port 2: assuming the grand master role
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.381] port 3: assuming the grand master role
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.381] port 4: assuming the grand master role
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.382] selected local clock 00049f.fffe.ef0505 as best master
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.382] port 1: assuming the grand master role
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.382] port 2: assuming the grand master role
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.382] port 3: assuming the grand master role
Sep 03 09:29:07 OpenIL ptp4l[166]: [112.382] port 4: assuming the grand master role
Sep 03 09:29:13 OpenIL ptp4l[166]: [118.583] port 2: link up
Sep 03 09:29:14 OpenIL ptp4l[166]: [118.661] port 2: FAULTY to LISTENING on INIT_COMPLETE
Sep 03 09:29:17 OpenIL ptp4l[166]: [121.702] port 2: new foreign master 00049f.fffe.ef0000-1
Sep 03 09:29:17 OpenIL ptp4l[166]: [122.013] port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Sep 03 09:29:17 OpenIL ptp4l[166]: [122.013] selected local clock 00049f.fffe.ef0505 as best master
Sep 03 09:29:17 OpenIL ptp4l[166]: [122.013] port 1: assuming the grand master role
Sep 03 09:29:17 OpenIL ptp4l[166]: [122.015] port 2: assuming the grand master role
Sep 03 09:29:17 OpenIL ptp4l[166]: [122.016] port 3: assuming the grand master role
Sep 03 09:29:17 OpenIL ptp4l[166]: [122.016] port 4: assuming the grand master role
Sep 03 09:29:19 OpenIL ptp4l[166]: [123.702] selected best master clock 00049f.fffe.ef0000
Sep 03 09:29:19 OpenIL ptp4l[166]: [123.702] port 2: MASTER to UNCALIBRATED on RS_SLAVE
Sep 03 09:29:20 OpenIL ptp4l[166]: [124.963] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 09:29:21 OpenIL ptp4l[166]: [125.590] clockcheck: clock jumped forward or running faster than expected!
Sep 03 09:29:21 OpenIL ptp4l[166]: [125.592] rms 47816127266 max 95523692514 freq -22833 +/- 8630 delay   626 +/-   0
Sep 03 09:29:21 OpenIL ptp4l[166]: [125.593] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 09:29:21 OpenIL ptp4l[166]: [125.715] clockcheck: clock jumped backward or running slower than expected!
[... 12 lines cut ...]
Sep 03 09:29:25 OpenIL ptp4l[166]: [129.727] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
[... cut ...]
Sep 03 09:29:27 OpenIL ptp4l[166]: [131.858] port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
Sep 03 09:29:27 OpenIL ptp4l[166]: [131.983] clockcheck: clock jumped forward or running faster than expected!
Sep 03 09:29:27 OpenIL ptp4l[166]: [132.109] port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
Sep 03 09:29:28 OpenIL ptp4l[166]: [132.612] rms 6074000969 max 17179869097 freq -25965 +/-  38 delay   625 +/-   0

So i didnt proceed to step 3 and later 

can you please help me out of it 

Thanks in advance

Regards
sabarish s

vladimiroltean, 2019-12-19:
Ok, can you please stop the systemd services and start the synchronization manually?
Can you share the output of:

Board 1:
```
systemctl stop linuxptp
phc_ctl /dev/ptp0 freq 0
phc_ctl /dev/ptp0 get
phc_ctl /dev/ptp0 set
phc_ctl /dev/ptp0 get
cat /etc/linuxptp.cfg
ptp4l -f /etc/linuxptp.cfg -m
```

Board 2:
```
systemctl stop linuxptp
phc_ctl /dev/ptp1 freq 0
phc_ctl /dev/ptp1 get
phc_ctl /dev/ptp1 set
phc_ctl /dev/ptp1 get
cat /etc/linuxptp.cfg
ptp4l -f /etc/linuxptp.cfg -m
```

I get the following output:

Board 1:
```
[root@OpenIL ~] # systemctl stop linuxptp
[root@OpenIL ~] # phc_ctl /dev/ptp0 freq 0
phc_ctl[332.313]: adjusted clock frequency offset to 0.000000ppb
[root@OpenIL ~] # phc_ctl /dev/ptp0 get
phc_ctl[337.348]: clock time is 400.134314377 or Thu Jan  1 00:06:40 1970

[root@OpenIL ~] # phc_ctl /dev/ptp0 set
phc_ctl[344.947]: set clock time to 1570712503.312964160 or Thu Oct 10 13:01:43 2019

[root@OpenIL ~] # phc_ctl /dev/ptp0 get
phc_ctl[350.978]: clock time is 1570712509.343109795 or Thu Oct 10 13:01:49 2019

[root@OpenIL ~] # cat /etc/linuxptp.cfg
#
# 802.1AS example configuration containing those attributes which
# differ from the defaults.  See the file, default.cfg, for the
# complete list of available options.
#
[global]
gmCapable               1
priority1               248
priority2               248
logAnnounceInterval     0
logSyncInterval         -3
syncReceiptTimeout      3
neighborPropDelayThresh 800
min_neighbor_prop_delay -20000000
assume_two_step         1
path_trace_enabled      1
follow_up_info          1
transportSpecific       0x1
ptp_dst_mac             01:80:C2:00:00:0E
network_transport       L2
delay_mechanism         P2P
tx_timestamp_timeout    20
step_threshold          0.00002

[eth0]
[root@OpenIL ~] # ptp4l -f /etc/linuxptp.cfg -m
ptp4l[452.249]: selected /dev/ptp0 as PTP clock
ptp4l[452.340]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
[... 89 lines cut ...]
ptp4l[526.414]: port 4: LISTENING to GRAND_MASTER on RS_GRAND_MASTER
ptp4l[526.598]: port 1: announce timeout
ptp4l[526.599]: bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[526.599]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[526.599]: selected best master clock 001f7b.fffe.630307
ptp4l[526.599]: port 1: assuming the grand master role
ptp4l[526.599]: port 4: assuming the grand master role
ptp4l[526.620]: port 2: announce timeout
ptp4l[526.620]: bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[526.620]: port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[526.620]: selected best master clock 001f7b.fffe.630307
ptp4l[526.620]: port 1: assuming the grand master role
ptp4l[526.620]: port 2: assuming the grand master role
ptp4l[526.620]: port 4: assuming the grand master role
ptp4l[526.766]: port 3: announce timeout
ptp4l[526.766]: bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[526.766]: port 3: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[526.767]: selected best master clock 001f7b.fffe.630307
ptp4l[526.767]: port 1: assuming the grand master role
ptp4l[526.767]: port 2: assuming the grand master role
ptp4l[526.767]: port 3: assuming the grand master role
ptp4l[526.767]: port 4: assuming the grand master role
```

If you don't reproduce this output, can you share the output of `systemctl status linuxptp-system-clock` and `journalctl -b -u linuxptp-system-clock -f`? This service should not work at the moment, so it shouldn't interfere, just want to make sure that it's not attempting to discipline the PTP clock simultaneously with ptp4l.

sabarishsubramanian, 2019-12-19:
so i did the above i got this

Board 1:

[root@OpenIL ~] # systemctl stop linuxptp
[root@OpenIL ~] # phc_ctl /dev/ptp0 freq 0
phc_ctl[3092.961]: adjusted clock frequency offset to 0.000000ppb
[root@OpenIL ~] # phc_ctl /dev/ptp0 get
phc_ctl[3100.154]: clock time is 3100.153122445 or Thu Jan  1 00:51:40 1970

[root@OpenIL ~] # phc_ctl /dev/ptp0 set
phc_ctl[3109.020]: set clock time to 1570715265.815823920 or Thu Oct 10 13:47:45 2019

[root@OpenIL ~] # phc_ctl /dev/ptp0 get
phc_ctl[3115.251]: clock time is 1570715272.048420735 or Thu Oct 10 13:47:52 2019

[root@OpenIL ~] # cat /etc/linuxptp.cfg
#
# 802.1AS example configuration containing those attributes which
# differ from the defaults.  See the file, default.cfg, for the
# complete list of available options.
#
[global]
gmCapable		1
priority1		248
priority2		248
logAnnounceInterval	0
logSyncInterval		-3
syncReceiptTimeout	3
neighborPropDelayThresh	800
min_neighbor_prop_delay	-20000000
assume_two_step		1
path_trace_enabled	1
follow_up_info		1
transportSpecific	0x1
ptp_dst_mac		01:80:C2:00:00:0E
network_transport	L2
delay_mechanism		P2P
tx_timestamp_timeout	20
step_threshold		0.00002

[eth0]
[root@OpenIL ~] # ptp4l -f /etc/linuxptp.cfg -m
ptp4l[3176.767]: selected /dev/ptp0 as PTP clock
ptp4l[3176.861]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[3176.863]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[3180.611]: port 1: announce timeout
ptp4l[3180.612]: bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[3180.612]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[3180.613]: selected local clock 00049f.fffe.ef0000 as best master
ptp4l[3180.613]: port 1: assuming the grand master role
ptp4l[3182.138]: port 1: new foreign master 00049f.fffe.ef0505-2
ptp4l[3182.139]: selected best master clock 00049f.fffe.ef0505
ptp4l[3182.140]: port 1: assuming the grand master role
ptp4l[3276.944]: selected best master clock 00049f.fffe.ef0505
ptp4l[3276.945]: port 1: assuming the grand master role

Board 2 :
[root@OpenIL ~] # systemctl stop linuxptp
[root@OpenIL ~] # phc_ctl /dev/ptp1 freq 0
phc_ctl[3385.851]: adjusted clock frequency offset to 0.000000ppb
[root@OpenIL ~] # phc_ctl /dev/ptp1 get
phc_ctl[3391.813]: clock time is 3235.005164008 or Thu Jan  1 00:53:55 1970

[root@OpenIL ~] # phc_ctl /dev/ptp1 set
phc_ctl[3397.256]: set clock time to 1567506232.519933120 or Tue Sep  3 10:23:52 2019

[root@OpenIL ~] # phc_ctl /dev/ptp1 get
phc_ctl[3402.555]: clock time is 1567506237.818777096 or Tue Sep  3 10:23:57 2019

[... 159 lines cut ...]

In both boards
[root@OpenIL ~] # journalctl -b -u linuxptp-system-clock -f
-- Logs begin at Thu 2019-10-10 12:56:03 UTC. --
Oct 10 13:54:42 OpenIL phc2sys[181]: [3525.325] Waiting for ptp4l...
Oct 10 13:54:42 OpenIL phc2sys[181]: [3525.326] uds: sendto failed: No such file or directory
Oct 10 13:54:43 OpenIL phc2sys[181]: [3526.327] Waiting for ptp4l...
Oct 10 13:54:43 OpenIL phc2sys[181]: [3526.328] uds: sendto failed: No such file or directory
Oct 10 13:54:44 OpenIL phc2sys[181]: [3527.329] Waiting for ptp4l...
Oct 10 13:54:44 OpenIL phc2sys[181]: [3527.330] uds: sendto failed: No such file or directory
Oct 10 13:54:45 OpenIL phc2sys[181]: [3528.331] Waiting for ptp4l...
Oct 10 13:54:45 OpenIL phc2sys[181]: [3528.332] uds: sendto failed: No such file or directory
Oct 10 13:54:46 OpenIL phc2sys[181]: [3529.333] Waiting for ptp4l...
Oct 10 13:54:46 OpenIL phc2sys[181]: [3529.333] uds: sendto failed: No such file or directory
Oct 10 13:54:47 OpenIL phc2sys[181]: [3530.335] Waiting for ptp4l...
Oct 10 13:54:47 OpenIL phc2sys[181]: [3530.335] uds: sendto failed: No such file or directory
Oct 10 13:54:48 OpenIL phc2sys[181]: [3531.337] Waiting for ptp4l...
Oct 10 13:54:48 OpenIL phc2sys[181]: [3531.337] uds: sendto failed: No such file or directory
Oct 10 13:54:49 OpenIL phc2sys[181]: [3532.338] Waiting for ptp4l...
Oct 10 13:54:49 OpenIL phc2sys[181]: [3532.339] uds: sendto failed: No such file or directory
Oct 10 13:54:50 OpenIL phc2sys[181]: [3533.340] Waiting for ptp4l...
Oct 10 13:54:50 OpenIL phc2sys[181]: [3533.341] uds: sendto failed: No such file or directory
Oct 10 13:54:51 OpenIL phc2sys[181]: [3534.342] Waiting for ptp4l...
Oct 10 13:54:51 OpenIL phc2sys[181]: [3534.343] uds: sendto failed: No such file or directory
^C

vladimiroltean, 2019-12-19:
Does the other board act the same if you swap roles board 1 <-> 2?

sabarishsubramanian, 2019-12-19:
yes it shows the same outputs even if i swap

vladimiroltean, 2019-12-19:
If you stop ptp4l on all boards, do you see any other PTP traffic on the ports in tcpdump? You can recognize it by the Ethertype of 0x88f7.

vladimiroltean, 2019-12-19:
Does the synchronization work directly between boards 1 and 3 (eth0 and eth0)?

sabarishsubramanian, 2019-12-19:
am getting 

Board 1
[root@OpenIL ~] # tcpdump
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
14:17:30.639457 ARP, Request who-has 1.0.0.1 tell 169.254.121.79, length 46
14:17:31.680888 ARP, Request who-has 1.1.1.1 tell 169.254.121.79, length 46
14:17:32.719555 ARP, Request who-has 1.1.1.1 tell 169.254.121.79, length 46
14:17:33.759442 ARP, Request who-has 1.1.1.1 tell 169.254.121.79, length 46
14:17:34.800812 ARP, Request who-has 8.8.8.8 tell 169.254.121.79, length 46
14:17:35.839522 ARP, Request who-has 8.8.8.8 tell 169.254.121.79, length 46
14:17:36.879520 ARP, Request who-has 8.8.8.8 tell 169.254.121.79, length 46
14:17:37.804574 ARP, Request who-has 1.0.0.1 tell 169.254.121.79, length 46
^C
8 packets captured
8 packets received by filter
0 packets dropped by kernel

Board 2
[root@OpenIL ~] # tcpdump
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on br0, link-type EN10MB (Ethernet), capture size 262144 bytes

sabarishsubramanian, 2019-12-19:
> Does the synchronization work directly between boards 1 and 3 (eth0 and eth0)?

I have the connection same which was mentioned by you 

Just for an interface am using a switch so that i can connect it wirelessly through ssh at eth1 in all boards

But rest of the connections are same as you mentioned above.

vladimiroltean, 2019-12-19:
The question was, if you temporarily remove board 2 from the setup and just connect board 1 with 3, do they synchronize properly or do they trigger the same clockcheck messages?

sabarishsubramanian, 2019-12-19:
YES THEY ARE

BOARD 1
[root@OpenIL ~] # ptp4l -f /etc/linuxptp.cfg -m
ptp4l[5448.262]: selected /dev/ptp0 as PTP clock
ptp4l[5448.341]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[5448.344]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[5451.527]: port 1: announce timeout
ptp4l[5451.528]: bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[5451.529]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[5451.531]: selected local clock 00049f.fffe.ef0000 as best master
ptp4l[5451.532]: port 1: assuming the grand master role

BOARD 3
[root@OpenIL ~] # ptp4l -f /etc/linuxptp.cfg -m
ptp4l[5450.221]: selected /dev/ptp0 as PTP clock
ptp4l[5450.311]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[5450.313]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[5453.734]: port 1: new foreign master 00049f.fffe.ef0000-1
ptp4l[5453.734]: selected best master clock 00049f.fffe.ef0000
ptp4l[5453.734]: port 1: LISTENING to SLAVE on RS_SLAVE
ptp4l[5455.241]: rms 316374926 max 632750120 freq  -4784 +/- 1879 delay   780 +/-   0
ptp4l[5456.244]: rms  168 max  254 freq  -4778 +/- 232 delay   781 +/-   0
ptp4l[5457.247]: rms  276 max  289 freq  -4266 +/-  80 delay   781 +/-   0
ptp4l[5458.251]: rms  177 max  229 freq  -4163 +/-   9 delay   781 +/-   0
ptp4l[5459.253]: rms   58 max   94 freq  -4214 +/-  19 delay   781 +/-   0
ptp4l[5460.255]: rms   10 max   14 freq  -4274 +/-  12 delay   781 +/-   0
ptp4l[5461.258]: rms   15 max   20 freq  -4299 +/-   6 delay   780 +/-   0
ptp4l[5462.261]: rms   12 max   21 freq  -4308 +/-   6 delay   781 +/-   0
ptp4l[5463.264]: rms    3 max    6 freq  -4302 +/-   4 delay   781 +/-   0
ptp4l[5464.267]: rms    3 max    6 freq  -4303 +/-   4 delay   781 +/-   0
ptp4l[5465.269]: rms    3 max    6 freq  -4302 +/-   5 delay   781 +/-   0
ptp4l[5466.272]: rms    4 max    5 freq  -4299 +/-   4 delay   780 +/-   0
ptp4l[5467.276]: rms    4 max   10 freq  -4299 +/-   5 delay   780 +/-   0
ptp4l[5468.279]: rms    4 max    6 freq  -4301 +/-   5 delay   781 +/-   0
ptp4l[5469.282]: rms    4 max    6 freq  -4302 +/-   5 delay   781 +/-   0
ptp4l[5470.284]: rms    3 max    7 freq  -4302 +/-   5 delay   782 +/-   0
ptp4l[5471.287]: rms    3 max    3 freq  -4298 +/-   3 delay   782 +/-   0
ptp4l[5472.291]: rms    4 max    7 freq  -4297 +/-   5 delay   782 +/-   0
ptp4l[5473.293]: rms    4 max    8 freq  -4297 +/-   6 delay   782 +/-   0
ptp4l[5474.296]: rms    4 max    8 freq  -4294 +/-   6 delay   782 +/-   0
ptp4l[5475.299]: rms    4 max    7 freq  -4297 +/-   6 delay   782 +/-   0
ptp4l[5476.302]: rms    2 max    3 freq  -4296 +/-   3 delay   782 +/-   0
ptp4l[5477.305]: rms    3 max    7 freq  -4298 +/-   4 delay   782 +/-   0
ptp4l[5478.308]: rms    5 max    7 freq  -4302 +/-   6 delay   782 +/-   0
^C

vladimiroltean, 2019-12-19:
In all cases of lost synchronization that you shared, the max offset to the GM is close to 8589934592 ns, which divided by 8 ns (the duration of a sja1105 clock tick) is 0x40000000, or 2^30. So I assume it is an integer overflow error, sign extend error, or similar, during the process of reconstructing the timestamps or retrieving them. I will review the code closely and see if I find any path where this can occur.
What I find very strange is that I do not reproduce this issue on my boards.

sabarishsubramanian, 2019-12-19:
Thank you @vladimiroltean  even am trying to rectify. I hope this problem solves soon.

sabarishsubramanian, 2019-12-19:
NOW even the board  and 1 connectd to eth0 have problem

Board 1

[root@OpenIL ~] # ptp4l -f /etc/linuxptp.cfg -m
ptp4l[1942.821]: selected /dev/ptp0 as PTP clock
ptp4l[1942.901]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1942.904]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1946.443]: port 1: new foreign master 00049f.fffe.ef0606-1
ptp4l[1946.445]: selected best master clock 00049f.fffe.ef0606
ptp4l[1946.446]: port 1: assuming the grand master role
ptp4l[1946.447]: port 1: LISTENING to GRAND_MASTER on RS_GRAND_MASTER
ptp4l[1946.906]: port 1: rogue peer delay response
ptp4l[1946.907]: port 1: GRAND_MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[1946.980]: selected local clock 00049f.fffe.ef0000 as best master
ptp4l[1946.981]: port 1: assuming the grand master role
ptp4l[1963.061]: port 1: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[1966.199]: port 1: announce timeout
ptp4l[1966.200]: bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1966.200]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1966.201]: selected local clock 00049f.fffe.ef0000 as best master
ptp4l[1966.201]: port 1: assuming the grand master role
ptp4l[1980.071]: port 1: rogue peer delay response
ptp4l[1980.072]: port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[1980.150]: selected local clock 00049f.fffe.ef0000 as best master
ptp4l[1980.151]: port 1: assuming the grand master role
^C[root@OpenIL ~] # 

Board 3
[root@OpenIL ~] # ptp4l -f /etc/linuxptp.cfg -m
ptp4l[1946.644]: selected /dev/ptp0 as PTP clock
ptp4l[1946.741]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1946.745]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1950.251]: port 1: announce timeout
ptp4l[1950.253]: bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1950.254]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1950.256]: selected local clock 00049f.fffe.ef0606 as best master
ptp4l[1950.257]: port 1: assuming the grand master role
ptp4l[1966.196]: port 1: new foreign master 00049f.fffe.ef0000-1
ptp4l[1966.197]: selected best master clock 00049f.fffe.ef0000
ptp4l[1966.200]: port 1: MASTER to SLAVE on RS_SLAVE
ptp4l[1967.698]: rms  462 max  614 freq  -4659 +/- 312 delay   764 +/-   0
ptp4l[1968.701]: rms   79 max  116 freq  -4402 +/- 111 delay   764 +/-   0
ptp4l[1969.704]: rms  131 max  141 freq  -4153 +/-  39 delay   764 +/-   0
ptp4l[1970.707]: rms   84 max  111 freq  -4103 +/-   4 delay   764 +/-   0
ptp4l[1971.710]: rms   29 max   46 freq  -4126 +/-  11 delay   764 +/-   0
ptp4l[1972.713]: rms    4 max    6 freq  -4151 +/-   6 delay   764 +/-   0
ptp4l[1973.716]: rms    7 max   14 freq  -4162 +/-   7 delay   764 +/-   0
ptp4l[1974.719]: rms    5 max    9 freq  -4165 +/-   5 delay   764 +/-   0
ptp4l[1975.722]: rms    3 max    6 freq  -4163 +/-   5 delay   764 +/-   0
ptp4l[1976.725]: rms    4 max    6 freq  -4160 +/-   6 delay   764 +/-   0
ptp4l[1977.727]: rms    4 max    5 freq  -4166 +/-   5 delay   765 +/-   0
ptp4l[1978.730]: rms    4 max    5 freq  -4164 +/-   5 delay   765 +/-   0
ptp4l[1979.733]: rms    2 max    5 freq  -4166 +/-   3 delay   765 +/-   0
ptp4l[1980.359]: port 1: rx sync timeout
ptp4l[1980.360]: bc_event 2683: returning EV_ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1980.360]: port 1: SLAVE to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1980.361]: selected local clock 00049f.fffe.ef0606 as best master
ptp4l[1980.361]: port 1: assuming the grand master role

sabarishsubramanian, 2019-12-20:
Hey

There is synchronization happening between board 1 and 3 when i connect the ETH0  board 1 to ETH0 of board 3.

But later i tried to do with board 1 and 2 ETH0 OF BAORD 1 TO eth 3 of board 2   

and board 2 and 3 ETh 2 of board 2 to eth0 of board 3.

In both the cases board 2 is not synchronising.

Is there any problem with the bridge of board 2 because i dont think there is problem with the boards.

Kindly help me out.

Thanks 

regards
sabarish s

vladimiroltean, 2019-12-20:
Why did you reduce the SPI frequency?
What other changes have you made?

sabarishsubramanian, 2019-12-20:
I changed because 12mhzwas not working for me also so i one of the thread i saw u asked to lower the fgreq to 6mhz so i did it 

just that one change i did nothing else

vladimiroltean, 2019-12-20:
What do you mean "was not working"? What was not working?

sabarishsubramanian, 2019-12-20:
https://github.com/openil/openil/issues/27#issuecomment-550352870

I am continuing this project and i got the same error so is saw you asked to reduce spi freq

vladimiroltean, 2019-12-20:
Yes but that was just to understand if the SPI bus is the problem or not. It is not a proper solution to reduce the SPI frequency to half.
I also never heard back on that thread.
There is a lot of SPI communication going on for PTP and it needs to happen pretty fast for reliable synchronization.
I will test the boards at 6 MHz and see if I reproduce your issues.
Do all 3 boards need the reduced SPI frequency.

sabarishsubramanian, 2019-12-20:
yes all three boards at 6mhz

vladimiroltean, 2019-12-20:
All 3 boards fail at 12?

sabarishsubramanian, 2019-12-20:
yes

vladimiroltean, 2019-12-20:
Does any intermediary frequency between 6 and 12 work?

sabarishsubramanian, 2019-12-20:
i didnt check on that

vladimiroltean, 2019-12-20:
In retrospect, even the 0x40000000 offsets sound like bit flips. There is no CRC for PTP timestamps over SPI.
if you share the kernel boot log, is there any retry when uploading the sja1105 static config?

sabarishsubramanian, 2019-12-20:
where can i find the kernal boot log

vladimiroltean, 2019-12-20:
> where can i find the kernal boot log

dmesg

vladimiroltean, 2019-12-20:
Is the power plug grounded?
Is the microUSB cable plugged in?

sabarishsubramanian, 2019-12-20:
# dmesg
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 5.5.0-rc1 (ssubramanian@ii-tsn2) (gcc version 8.3.0 (Buildroot 2019.11-git-g2b2c0eb694)) #1 SMP Mon Dec 16 14:22:29 CET 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: NXP LS1021A-TSN Board
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 64 MiB at 0xbc000000
[    0.000000] On node 0 totalpages: 262144
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 21 pages/cpu s54056 r8192 d23768 u86016
[    0.000000] pcpu-alloc: s54056 r8192 d23768 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 260416
[    0.000000] Kernel command line: console=ttyS0,115200n8 root=/dev/mmcblk0p2 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 933352K/1048576K available (16384K kernel code, 2077K rwdata, 11500K rodata, 2048K init, 6604K bss, 49688K reserved, 65536K cma-reserved, 196608K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Running RCU self tests
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU lockdep checking is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=16 to nr_cpu_ids=2.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x33c/0x508 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 12.50MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2e2049cda, max_idle_ns: 440795202628 ns
[    0.000006] sched_clock: 56 bits at 12MHz, resolution 80ns, wraps every 4398046511080ns
[    0.000024] Switching to timer-based delay loop, resolution 80ns
[    0.002147] Console: colour dummy device 80x30
[    0.002195] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.002208] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.002222] ... MAX_LOCK_DEPTH:          48
[    0.002234] ... MAX_LOCKDEP_KEYS:        8192
[    0.002248] ... CLASSHASH_SIZE:          4096
[    0.002261] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.002274] ... MAX_LOCKDEP_CHAINS:      65536
[    0.002287] ... CHAINHASH_SIZE:          32768
[    0.002300]  memory used by lock dependency info: 3997 kB
[    0.002313]  memory used for stack traces: 2112 kB
[    0.002326]  per task-struct memory footprint: 1536 bytes
[    0.002411] Calibrating delay loop (skipped), value calculated using timer frequency.. 25.00 BogoMIPS (lpj=125000)
[    0.002433] pid_max: default: 32768 minimum: 301
[    0.002812] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002842] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.006086] CPU: Testing write buffer coherency: ok
[    0.007438] CPU0: update cpu_capacity 1024
[    0.007458] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.009446] Setting up static identity map for 0x80300000 - 0x803000ac
[    0.010816] rcu: Hierarchical SRCU implementation.
[    0.018290] EFI services will not be available.
[    0.019630] smp: Bringing up secondary CPUs ...
[    0.022062] CPU1: update cpu_capacity 1024
[    0.022073] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.022905] smp: Brought up 1 node, 2 CPUs
[    0.022936] SMP: Total of 2 processors activated (50.00 BogoMIPS).
[    0.022950] CPU: All CPU(s) started in HYP mode.
[... 272 lines cut ...]
[   11.630109] br0: port 4(swp5) entered blocking state
[   11.635245] br0: port 4(swp5) entered disabled state
[   11.673604] device swp5 entered promiscuous mode
[   12.156152] fsl-gianfar soc:ethernet@2d90000 eth2: Link is Up - 1Gbps/Full - flow control off
[   12.196029] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[   12.221199] device eth2 entered promiscuous mode
[   12.229026] sja1105 spi0.1 swp2: configuring for phy/rgmii-id link mode
[   12.346815] sja1105 spi0.1 swp5: configuring for phy/rgmii-id link mode
[   12.384144] sja1105 spi0.1 swp4: configuring for phy/rgmii-id link mode
[   12.392814] sja1105 spi0.1 swp3: configuring for phy/rgmii-id link mode
[   14.504842] random: crng init done
[   14.508700] random: 7 urandom warning(s) missed due to ratelimiting
[   15.124517] fsl-gianfar soc:ethernet@2d50000 eth1: Link is Up - 1Gbps/Full - flow control off
[... cut ...]
[  618.004494] fsl-gianfar soc:ethernet@2d10000 eth0: Link is Up - 1Gbps/Full - flow control off
[ 1645.524380] fsl-gianfar soc:ethernet@2d10000 eth0: Link is Down
[ 1651.764475] fsl-gianfar soc:ethernet@2d10000 eth0: Link is Up - 1Gbps/Full - flow control off
[ 3252.324342] fsl-gianfar soc:ethernet@2d10000 eth0: Link is Down
[ 3312.564490] fsl-gianfar soc:ethernet@2d50000 eth1: Link is Up - 1Gbps/Full - flow control off
[ 3443.684624] fsl-gianfar soc:ethernet@2d10000 eth0: Link is Up - 1Gbps/Full - flow control off

vladimiroltean, 2019-12-20:
This boot log looks fine, but is also inconclusive.
Can you run this 10-20 times in a row:
```
ip link set dev br0 type bridge vlan_filtering 1
ip link set dev br0 type bridge vlan_filtering 0
```

The following output is normal:
```
[root@OpenIL ~] # ip link set dev br0 type bridge vlan_filtering 1
[   51.217901] sja1105 spi0.1: Reset switch and programmed static config. Reason: VLAN filtering
[   51.229189] sja1105 spi0.1: Disabled switch tagging
[root@OpenIL ~] # ip link set dev br0 type bridge vlan_filtering 0
[   51.957707] sja1105 spi0.1: Reset switch and programmed static config. Reason: VLAN filtering
[   51.968789] sja1105 spi0.1: Enabled switch tagging
```

Any of the following is not:
```
Switch reported invalid local CRC on the uploaded config, retrying...
Switch reported invalid global CRC on the uploaded config, retrying...
Switch reported that configuration is invalid, retrying...
```

sabarishsubramanian, 2019-12-20:
in all 3 boards??

vladimiroltean, 2019-12-20:
No, just one. Assuming the error is the same on all 3.

vladimiroltean, 2019-12-20:
> Is the power plug grounded?
> Is the microUSB cable plugged in?

Can you please also answer these questions? Are you logged into the boards through ssh or through console?

sabarishsubramanian, 2019-12-20:
Am getting nothing
[root@OpenIL ~] # ip link set dev br0 type bridge vlan_filtering 1
[root@OpenIL ~] # ip link set dev br0 type bridge vlan_filtering 0
[root@OpenIL ~] # ip link set dev br0 type bridge vlan_filtering 1
[root@OpenIL ~] # ip link set dev br0 type bridge vlan_filtering 0
[root@OpenIL ~] # ip link set dev br0 type bridge vlan_filtering 1
[root@OpenIL ~] # ip link set dev br0 type bridge vlan_filtering 0

vladimiroltean, 2019-12-20:
That's what I suspected. Can you please repeat all tests, including SPI at 12 MHz, through console, please? I'll explain later if it works.

sabarishsubramanian, 2019-12-20:
okay i will repeat all test making spi frequency 12mhz 

but how can i access to all boards at the same time through console?

vladimiroltean, 2019-12-20:
USB hub?

vladimiroltean, 2019-12-20:
A simpler experiment may be to plug the microUSB cable into the board's own USB port.

sabarishsubramanian, 2019-12-20:
I have 3 boards i used to flash the memory card and insert in boards and then i connect a usb through my pc to the board and use minicom and assign  ip and mac address to it so once done for board 1 i disconnect the usb and connect the board through ssh and work since i have only one usb 

and i proceed to do for later boards.

so how can i use a console for it?

vladimiroltean, 2019-12-20:
Just test one board at 12 MHz connected to your computer. I would like to know if there is any issue with the SPI bus at full frequncy, that's all.

sabarishsubramanian, 2019-12-20:
okay i ll change the frq of 1 board to 12 mhz and 
run
ip link set dev br0 type bridge vlan_filtering 1
ip link set dev br0 type bridge vlan_filtering 0
 this command n check

sabarishsubramanian, 2019-12-20:
dev br0 type bridge vlan_filtering 1pi@2100000/ethernet-switch@1] # ip link setd
[ 8307.271170] sja1105 spi0.1: Reset switch and programmed static config. Reasog
[ 8307.289835] sja1105 spi0.1: Disabled switch tagging                          
dev br0 type bridge vlan_filtering 0pi@2100000/ethernet-switch@1] # ip link setd
[ 8331.235157] sja1105 spi0.1: Reset switch and programmed static config. Reasog
[ 8331.252582] sja1105 spi0.1: Enabled switch tagging     

I got this at board 1

vladimiroltean, 2019-12-20:
At 12 MHz? Ok, nice.
Can you please keep the microUSB cable plugged in?
As I said, it's not necessary that you use the console, so you can plug it into one of the board's own USB A ports (yes, I know this looks silly).

sabarishsubramanian, 2019-12-20:
yes i have kept it connected

vladimiroltean, 2019-12-20:
Ok please let me know if synchronization works now.

sabarishsubramanian, 2019-12-20:
you mean i should put the commands in console??

if yes what all commands ??

[39 more replies]

closed 2020-08-06 by vladimiroltean