nxp-archive/openil #50: In LS1021ATSN, How to make the sja1105 switch learn the source MAC of the frame that matches the virtual link.

niceniu, 2020-01-02:
In LS1021ATSN, How to make the sja1105 switch learn the source MAC of the frame that matches the virtual link.

vladimiroltean, 2020-01-09:
It does.
To prove this, we need an environment where we can actually see the learned entries in the FDB table of the switch. I have prepared a branch based on the DSA driver with a small demonstration of frame trapping using virtual links (actually it is the same demonstration as here: https://github.com/openil/openil/issues/39).
The branch is here: https://github.com/vladimiroltean/ls1021atsn-linux/tree/virtual-links

The test I have made using the DSA driver for sja1105, which for the purpose of this demo is hardcoded to trap the DMAC of 01:02:03:04:05:06, is:
```
# Check the current FDB table of the source port
bridge fdb show dev swp4
00:01:02:03:04:03 master br0 
00:04:9f:05:04:03 master br0 
00:01:02:03:04:05 self 
00:04:9f:05:de:06 self

# On the sender system (Host A), transmit frames with a SMAC of aa:bb:cc:dd:ee:ff and a
# DMAC of 01:02:03:04:05:06. The expectation is that the SMAC is learned.
# The raw-l2-send program is from here:
# https://github.com/vladimiroltean/tsn-scripts/blob/master/raw-l2-send.c
raw-l2-send -i eth0 --smac 00:01:02:03:04:05 --dmac 01:02:03:04:05:06 --vid 1 -p 0 -b 0 -c 0.2 -n 20000 -s 100

# Check on swp4 that frames are trapped
tcpdump -i swp4 -e -n -Q in
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on swp4, link-type EN10MB (Ethernet), capture size 262144 bytes
23:10:53.199327 aa:bb:cc:dd:04:03 > 01:02:03:04:05:06, ethertype 802.1Q (0x8100), length 100: vlan 1, p 0, ethertype 0x22f0, 
        0x0000:  15e8 5ac0 c645 2600 005a beef dead beef  ..Z..E&..Z......
        0x0010:  dead beef dead beef dead beef dead beef  ................
        0x0020:  dead beef dead beef dead beef dead beef  ................
        0x0030:  dead beef dead beef dead beef dead beef  ................
        0x0040:  dead beef dead beef dead beef dead beef  ................
        0x0050:  dead

# Now check that frames are not received by Host B

# Then examine the FDB table again:
bridge fdb show dev swp4
aa:bb:cc:dd:04:03 master br0 
00:01:02:03:04:03 master br0 
00:04:9f:05:04:03 master br0 
00:01:02:03:04:05 self 
aa:bb:cc:dd:ee:ff self 
00:04:9f:05:de:06 self
```

niceniu, 2020-01-10:
thank you for your reply.
>
>The branch is here: https://github.com/vladimiroltean/ls1021atsn-linux/tree/virtual-links
>
How to compile this branch？
I use the command
export LINUX_OVERRIDE_SRCDIR = / path / to /ls1021atsn-linux-virtual-links
make linux-rebuild
But compilation fails

vladimiroltean, 2020-01-10:
How does it fail?

niceniu, 2020-01-10:
dtc-parser.y: In function ‘yyparse’:
dtc-parser.y:208:4: error: too few arguments to function ‘build_node’
In file included from dtc-parser.y:24:0:
scripts/dtc/dtc.h:202:14: note: declared here
 struct node *build_node(struct property *proplist, struct node *children,
              ^
dtc-parser.y:226:4: error: too few arguments to function ‘build_property’
In file included from dtc-parser.y:24:0:
scripts/dtc/dtc.h:196:18: note: declared here
 struct property *build_property(char *name, struct data val,
                  ^
scripts/dtc/dtc.h:89:29: error: too few arguments to function ‘build_property’
 #define empty_data ((struct data){ 0 /* all .members = 0 or NULL */ })
                             ^
dtc-parser.y:230:60: note: in expansion of macro ‘empty_data’
scripts/dtc/dtc.h:196:18: note: declared here
 struct property *build_property(char *name, struct data val,
                  ^
dtc-parser.y:505:4: error: too few arguments to function ‘build_node_delete’
In file included from dtc-parser.y:24:0:
scripts/dtc/dtc.h:204:14: note: declared here
 struct node *build_node_delete(struct srcpos *srcpos);

vladimiroltean, 2020-01-10:
Strange. I didn't attempt building the 5.5 kernel in the official OpenIL buildroot though. I used this [forked repository](https://github.com/vladimiroltean/openil-community) and I changed

```
BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/vladimiroltean/ls1021atsn-linux.git"
BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="virtual-links"
```

in the `configs/nxp_ls1021atsn_defconfig` file.

Alternatively, you can build the kernel outside of OpenIL:

```
export ARCH=arm
export CC=arm-linux-gnueabi-
export PATH=/path/to/cross/compiler/bin/
make defconfig && make lsdk.config
LOADADDR=0x80008000 make -j 8 uImage dtbs
```

then transfer the `arch/arm/boot/uImage` and `arch/arm/boot/dts/ls1021a-tsn.dtb` file to the `mmblk0p1` VFAT partition of the SD card.

niceniu, 2020-01-15:
> Strange. I didn't attempt building the 5.5 kernel in the official OpenIL buildroot though. I used this [forked repository](https://github.com/vladimiroltean/openil-community) and I changed
>
> ``` [...]
I compiled the openil you said, but I still get an error:
```
/home/aad/openil-community/output/build/systemd-244.1//build
The Meson build system
Version: 0.52.1
Source dir: /home/aad/openil-community/output/build/systemd-244.1
Build dir: /home/aad/openil-community/output/build/systemd-244.1/build
Build type: cross build
Project name: systemd
Project version: 244
C compiler for the build machine: cc (gcc 4.8.4 "cc (Ubuntu 4.8.4-2ubuntu1~14.04.4) 4.8.4")
C linker for the build machine: GNU ld.bfd 2.24
C compiler for the host machine: /home/aad/openil-community/output/host/bin/arm-buildroot-linux-gnueabihf-gcc (gcc 8.3.0 "arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot 2020.02-git-g2fa26e2-dirty) 8.3.0")
C linker for the host machine: GNU ld.bfd 2.32
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: arm
Host machine cpu: cortex-a7
Target machine cpu family: arm
Target machine cpu: cortex-a7

meson.build:33:0: ERROR: Program or command 'realpath' not found or not executable

A full log can be found at /home/aad/openil-community/output/build/systemd-244.1/build/meson-logs/meson-log.txt
make: *** [/home/aad/openil-community/output/build/systemd-244.1/.stamp_configured] error 1
```

niceniu, 2020-01-16:
I encountered a problem when compiling openil in the branch you provided, and the compilation prompts DSA patch conflicts:
```
>>> linux virtual-links Patching

Applying 0001-gianfar-Fix-TX-timestamping-with-stacked-DSA-and-PHY.patch using patch: 
Error: duplicate filename '0001-gianfar-Fix-TX-timestamping-with-stacked-DSA-and-PHY.patch'
Conflicting files are:
  already applied: /home/aad/openil-community/board/nxp/ls1021atsn/patches//linux/0001-gianfar-Fix-TX-timestamping-with-stacked-DSA-and-PHY.patch
  to be applied  : /home/aad/openil-community/board/nxp/ls1021atsn/patches//linux/0001-gianfar-Fix-TX-timestamping-with-stacked-DSA-and-PHY.patch
make: *** [/home/aad/openil-community/output/build/linux-virtual-links/.stamp_patched] error 1
```

vladimiroltean, 2020-01-16:
You can delete the patches in `board/nxp/ls1021atsn/patches/linux` that don't apply. The `virtual-links` branch is more recent than the kernel base for which those patches are for, and most of them have been integrated already.

niceniu, 2020-01-16:
I don't know which patches are not applicable. I deleted the conflicting patch mentioned above, and I get an new error. Can I delete all the patches?
```
>>> linux virtual-links Patching

Applying 0002-spi-spi-fsl-dspi-Simplify-bytes_per_word-gymnastics.patch using patch: 
patching file drivers/spi/spi-fsl-dspi.c

Applying 0003-spi-spi-fsl-dspi-Fix-16-bit-word-order-in-32-bit-XSP.patch using patch: 
patching file drivers/spi/spi-fsl-dspi.c

Applying 0004-net-dsa-sja1105-Increase-SPI-word-size-to-32-bits.patch using patch: 
patching file drivers/net/dsa/sja1105/sja1105_main.c
Hunk #1 succeeded at 2087 (offset 7 lines).
patching file drivers/net/dsa/sja1105/sja1105_static_config.c

Applying 0005-net-dsa-sja1105-give-the-SPI-kernel-thread-real-time.patch using patch: 
patching file drivers/net/dsa/sja1105/sja1105_main.c
Hunk #1 succeeded at 2088 (offset 7 lines).

Applying 0006-ptp-introduce-ptp_cancel_worker_sync.patch using patch: 
patching file drivers/ptp/ptp_clock.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/ptp/ptp_clock.c.rej
patching file include/linux/ptp_clock_kernel.h
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file include/linux/ptp_clock_kernel.h.rej
```

vladimiroltean, 2020-01-16:
Yes, you can delete all the patches.

niceniu, 2020-01-16:
Thank you very much for your guidance, this method is very useful.

vladimiroltean, 2020-08-06:
This is now supported through the tc mirred action: https://www.kernel.org/doc/html/latest/networking/dsa/sja1105.html#routing-actions-redirect-trap-drop

closed 2020-08-06 by vladimiroltean