nxp-archive/openil #54: LS1021ATSN - Steps to run Qbv

Norman03, 2020-04-02:
Hello, 
I have connected two PC's to the LS1021ATSN switch in switched Ethernet ports(eth2 and eth3). The LS1021ATSN switch running openIL community kernel. Below is the network and configuration

PTP master: PC1
PTP slave: PC2 and LS1021ATSN switch
Now I need a steps to run a 802.1 Qbv(synchronized traffic). As I followed the steps from openil guide (https://www.openil.org/news/files/OpenIL_User_Guide_Rev1.2.pdf) I couldn't repeat the same.

**Inferences:**
1. The output of the ptp synchronization is not stable after some time and observed a huge offset in ptp as well as PHC2SYS running in the PC's 
2. QBV traffic is not scheduled by the switch

Could you please guide me with right document for the above configuration? Since I bought this switch a year back, still I couldn't find right documentation in place to repeat and test.

vladimiroltean, 2020-04-07:
Hi Norman,

The revised driver for the LS1021A-TSN switch will be included in OpenIL 1.8. It includes some PTP synchronization fixes which improve the stability. For now, the driver is available in the community kernel. I would recommend that you pick the latest commit from the master branch.

I tried to put together some steps below. They will be included in official documentation when it comes out.

Topology
------------

For demonstrating the SJA1105 TSN features, the following topology is required:
* 1 LS1021A-TSN board, acting as a TSN switch
* 1 generic host (can be a PC or another board) capable of PTP hardware timestamping, acting as a sender of latency-sensitive traffic
* 1 generic host (can be a PC or another board), acting as a sender of high-bandwidth traffic
* 1 generic host (can be a PC or another board) capable of PTP hardware timestamping, acting as receiver for the latency-sensitive and for the high-bandwidth traffic

The required software packages for the generic hosts are:
* ptp4l, phc2sys and phc_ctl from the linuxptp package: https://github.com/openil/linuxptp
* iperf3
* isochron from the tsn-scripts package: https://github.com/vladimiroltean/tsn-scripts/tree/isochron

The generic hosts are assumed to be connected to the LS1021A-TSN board through an interface called eth0.
This topology is depicted in the following figure.
[image]

SJA1105 Linux support
-------------------------------

The SJA1105 switch is supported in the OpenIL Linux kernel using the Distributed Switch Architecture (DSA) framework (an overview of which can be found at https://netdevconf.info/2.1/papers/distributed-switch-architecture.pdf).
The following kernel configuration options are available for controlling its features:
*	CONFIG_NET_DSA_SJA1105: enables base support for probing the SJA1105 ports as 4 standalone net devices capable of sending and receiving traffic
*	CONFIG_NET_DSA_SJA1105_PTP: enables additional support for the PTP Hardware Clock (PHC), visible in /dev/ptp1 on the LS1021A-TSN board, and for PTP timestamping on the SJA1105 ports
*	CONFIG_NET_DSA_SJA1105_TAS: enables additional support for the Time-Aware Scheduler (TAS), which is configured via the tc-taprio qdisc offload

The documentation for this kernel driver is available at https://www.kernel.org/doc/html/latest/networking/dsa/sja1105.html. Below is a listing of several driver features.
The LS1021A-TSN device tree (arch/arm/boot/dts/ls1021a-tsn.dts) defines the sja1105 port names as swp2, swp3, swp4 and swp5. The numbers have a direct correspondence with the chassis labels ETH2, ETH3, ETH4 and ETH5. The ETH2 chassis label (represented in Linux by the swp2 net device) should not be confused with the eth2 net device, which represents the LS1021A host port for this switch (called DSA master).
On the LS1021A-TSN board, network management is done by the systemd-networkd daemon, whose configuration files are located in /etc/systemd/network/. On this board, the following configuration files for systemd-networkd are present by default:
* br0.netdev: Creates a bridge net device with VLAN filtering disabled, STP disabled and MVRP disabled
* br0.network: Configures the net devices enslaved to br0 to request an IPv4 address via DHCP
* eth0.network, eth1.network, swp.network: Configures all 6 ports of the LS1021A-TSN board to be part of the same br0 bridge (4 ports are bridged in hardware, 2 ports are bridged in software)
* eth2.network: Configures the DSA master port to come up automatically, and assigns it a dummy link-local IP address. Having the DSA master interface up is a requirement for using the switch net devices.

Although all ports are configured for L2 forwarding by default (and therefore the only IP address for this board should be assigned to br0), this can be changed by removing the "Bridge=br0" line from the files in /etc/systemd/network/ and then running "systemctl restart systemd-networkd".
In standalone mode, each SJA1105 port is able of acquiring an IP address and transferring general purpose packets to/from the kernel. This is internally supported by the kernel driver by repurposing the VLAN tagging functionality for switch port separation and identification. Therefore the ability to support general purpose traffic I/O only works as long as the user does not request VLAN tagging, via the bridge vlan_filtering option. When this happens, the switch driver goes to a reduced functionality mode, where the swpN net devices are no longer capable of sending and receiving general packets to/from the kernel. This is a driver limitation that currently does not have a solution.
Actually there is a second mechanism of frame tagging, which works for STP and PTP traffic and does not rely on VLAN tagging. Therefore, the STP and PTP protocols remain operational on the sja1105 driver even when the ports are enslaved to a bridge with vlan_filtering=1.
When VLAN awareness is disabled, the sja1105 ports perform no checks on VLAN port membership or PCP, and performs no alteration to the VLAN tags. For these operations, the following command is necessary:

```
ip link set dev br0 type bridge vlan_filtering 1
```

Once VLAN filtering is enabled, the VLAN table of each switch port can be inspected and modified using the "bridge vlan" commands from the iproute2 package.
The STP state machine can be started on the bridge using the following command:

```
ip link set dev br0 type bridge stp_state 1
ip link set dev br0 down
ip link set dev br0 up
```

[... cut ...] ies in the sender's operating system. Generally speaking, the jitter of the sender is eliminated by the first switch upon packet admission into the TSN network. The effect is that the receiver sees a packet stream with low jitter.
The path delay can be reduced by decreasing the advance time. It is configured in such a way that the packets arrive on the switch prior to the gate opening, which depends on the jitter of the sender. Minimizing the TX jitter is outside the scope of this demonstration.

Norman03, 2020-04-26:
@vladimiroltean Thanks. 
Is this the [community kernel](https://github.com/vladimiroltean/ls1021atsn-linux) you mentioned above?

**Steps I have performed:**
1. In have cloned [openil](https://github.com/openil/openil) and changed the below in ls1021 config.

2. Chnaged below in ls1021_defconfig file
BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/vladimiroltean/ls1021atsn-linux.git"
BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="master"

Am I compiling the kernel right?

vladimiroltean, 2020-04-26:
Hi Norman,
By "community" I mean the master branch of https://github.com/vladimiroltean/openil-community. That is because some aspects regarding LS1021A-TSN user space support in OpenIL are not yet finalized (such as systemd and a recent enough tc binary).
If you look there, it uses just the net-next kernel: https://github.com/vladimiroltean/openil-community/blob/master/configs/nxp_ls1021atsn_defconfig#L53 (with no other patches).
At the moment, the ls1021atsn-linux kernel repository does not contain any significant feature which is not available upstream, so I don't see reasons why you should use it.
Hope this helps,
-Vladimir

Norman03, 2020-05-29:
@Vladimir, I tried running the tsn-script in host1(send) and host2(recv) command respectively as mentioned above.

Command in host 1:
```
isochron send --interface eth0 --dmac 00:04:9f:05:de:06 --priority 6 --vid 0 \
        --base-time 0 --cycle-time 400000 --shift-time 50000 --advance-time 90000 \
        --num-frames 10000 --frame-size 64 --client 192.168.1.2 --quiet
```
Command in host 2:
`isochron rcv --interface eth0 --quiet`

Below is the error, I'm facing in sender(host 1) and no terminal prints in host2. 
[image]

From the error I understood that, connect() api in send.c(line 384) return the error code 111: which states a port number error. I checked the ports list, but the port in which the sender should send the frames are properly closed and opened. 
This error is not reproducible. Few times the sender application get crashed with error stating "PHC offset is greater than 1 seconds when compared to system clock".

vladimiroltean, 2020-06-01:
Hi Norman,
Is your eth0 interface capable of performing hardware timestamping of packets with an EtherType of 0x22f0? From the "Timed out waiting for TX timestamp: -11 (Resource temporarily unavailable)" message, it looks like it isn't.
The LS1021A-TSN board is capable of timestamping these packets on ports eth0 and eth1.

Norman03, 2020-06-12:
Hi, Can you provide me the exact ptpconfig file or config parameters used in the host PC and LS1021ATSN switch?

Below is the configuration I'm using:
**Host 1:**
gmCapable		1
priority1		248
priority2		248
logAnnounceInterval	0
logSyncInterval		-3
syncReceiptTimeout	3
neighborPropDelayThresh	800
min_neighbor_prop_delay	-20000000
assume_two_step		1
path_trace_enabled	1
follow_up_info		1
transportSpecific	0x1
ptp_dst_mac		01:80:C2:00:00:0E
network_transport	L2
delay_mechanism		P2P

**Host 2:**
gmCapable		1
priority1		248
priority2		248
logAnnounceInterval	0
logSyncInterval		-3
syncReceiptTimeout	3
neighborPropDelayThresh	800
min_neighbor_prop_delay	-20000000
assume_two_step		1
path_trace_enabled	1
follow_up_info		1
transportSpecific	0x1
ptp_dst_mac		01:80:C2:00:00:0E
network_transport	L2
delay_mechanism		P2P

LS1021ATSN:
1. No special commands or scripts or configuration is used. By default the PTP service is running.
2. What is the PTP version used in above demo?

vladimiroltean, 2020-06-12:
For the endpoints, official linuxptp version 2.0 and [the standard gPTP.cfg](https://github.com/richardcochran/linuxptp/blob/master/configs/gPTP.cfg) should be enough.
For the switch, the linuxptp version that is used is the one from OpenIL, since it has been [patched for 802.1AS time-aware bridge support](https://github.com/openil/linuxptp/commit/087580e4ae3d3e8cd57c8cb5c990baf4bfec7011).
If you use the ptp4l systemd service (which reads /etc/linuxptp.cfg), you don't need to do any other modification to it (the only required modification to gPTP.cfg, the `tx_timestamp_timeout 20` line, has already been made).
```
[root@LS1021ATSN ~] # cat /usr/lib/systemd/system/ptp4l.service 
[Unit]
Description=Precision Time Protocol daemon
After=syslog.target network.target
Before=time-sync.target
Wants=time-sync.target
Wants=phc2sys.service

[Service]
ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp.cfg
Restart=always

[Install]
WantedBy=multi-user.target
```
What are the problems you're facing?

Norman03, 2020-06-12:
In host 1:
[image]

In LS1021ATSN:
[image]

What am I doing wrong here? The host 1 is connected to ETH3 (switch chassis label) and host 1 uses below config

Host 1:
gmCapable 1
priority1 248
priority2 248
logAnnounceInterval 0
logSyncInterval -3
syncReceiptTimeout 3
neighborPropDelayThresh 800
min_neighbor_prop_delay -20000000
assume_two_step 1
path_trace_enabled 1
follow_up_info 1
transportSpecific 0x1
ptp_dst_mac 01:80:C2:00:00:0E
network_transport L2
delay_mechanism P2P

vladimiroltean, 2020-06-12:
You're not doing anything wrong.
Could you please perform the following debugging steps:
Increase the propagation delay threshold to a large value so that we can avoid false positives of non-802.1AS devices present on the network:
```
[root@LS1021ATSN ~] # vim /etc/linuxptp.cfg
neighborPropDelayThresh 2000
```
(do this for the other device as well)

Increase log level for the ptp4l systemd service:
```
[root@LS1021ATSN ~] # vim /usr/lib/systemd/system/ptp4l.service
ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp.cfg -l 7
[root@LS1021ATSN ~] # systemctl daemon-reload
[root@LS1021ATSN ~] # systemctl restart ptp4l
```
If under debug, you are seeing these messages:
```
[root@LS1021ATSN ~] # journalctl -b -u ptp4l -f
Jun 12 14:49:13 LS1021ATSN ptp4l[4080]: [254028.244] port 1: drop erroneous nratio 1.031944, max offset 0.000200
```
then it may happen that due to some condition, the frequency of the master clock is stuck too high or too low. But in your setup, I'm not sure which device will negotiate as master and which as slave, so just temporarily stop the ptp4l service, reset the PHC frequency to zero, and restart it:
```
[root@LS1021ATSN ~] # systemctl stop ptp4l
[root@LS1021ATSN ~] # phc_ctl /dev/ptp1 freq 0
phc_ctl[254044.961]: adjusted clock frequency offset to 0.000000ppb
[root@LS1021ATSN ~] # systemctl restart ptp4l
```
(do the same thing on the other device).

If you still fail to synchronize, please post logs collected with "-l 7" from the 2 devices, in textual form.

Thanks,
-Vladimir

Norman03, 2020-06-12:
After performing below step:
[root@LS1021ATSN ~] # vim /usr/lib/systemd/system/ptp4l.service
ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp.cfg -l 7
[root@LS1021ATSN ~] # systemctl daemon-reload
[root@LS1021ATSN ~] # systemctl restart ptp4l

result: I think log level does not exists
[image]

vladimiroltean, 2020-06-12:
Please post the whole contents of the file. There was just 1 line which needed to be modified. It should look in the end like this:
```
[Unit]
Description=Precision Time Protocol daemon
After=syslog.target network.target
Before=time-sync.target
Wants=time-sync.target
Wants=phc2sys.service

[Service]
ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp.cfg -l 7
Restart=always

[Install]
WantedBy=multi-user.target
```

Norman03, 2020-06-12:
> Please post the whole contents of the file.
No contents available in that file.

Norman03, 2020-06-12:
I rebooted the switch, now the host 1 becomes a ptp slave and switch becomes a master. Now connected host 2 in switch and ran ptp4l with defaultcfg. Now both switch and host2 becomes master. Why it is so?

vladimiroltean, 2020-06-12:
> Now both switch and host2 becomes master. Why it is so?

Please check my debugging suggestions again for running `phc_ctl freq 0`. Did you try it?

Norman03, 2020-06-12:
Yes, Still facing the same issue.

vladimiroltean, 2020-06-12:
Please share logs with -l 7, as requested.

Norman03, 2020-06-12:
Error log when using ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp.cfg -l 7

[image]

vladimiroltean, 2020-06-12:
So you are running into the exact `drop erroneous nratio` error condition that I talked about earlier. What was the exact phc_ctl command that you used to reset the PHC frequency on the master? Please note that I can't spell this command out to you because I don't know what your master device is. You can specify to phc_ctl not only a PHC name, but also an Ethernet interface name and it will make the association automatically.

Norman03, 2020-06-12:
> phc_ctl /dev/ptp0 freq 0
I used above command in host 1 and host 2. Later, it was observed that host1 and host2 start to sync
[image]

But in LS1021ATSN switch, the ptp4l logs observed to have "rougue peer delay response time"
[image]

vladimiroltean, 2020-06-12:
I saw some discussion here about the fact that there might be peer delay requests that are thrown away when the GM changes, and then the peer delay responses are unexpected, but no patch has been sent to fix it:
https://sourceforge.net/p/linuxptp/mailman/message/36769331/
I'd try to see exactly what Richard's suggestion of marking the requests as stale entails.

But if I'm right and the reason why you see the rogue peer delay responses is because the GM has changed, then restarting ptp4l again should be enough for the errors to go away, right?

Norman03, 2020-06-13:
No, it is not the case. When I rebooted the switch/ptp4l, only host2 which is connected to the eth3(chassis label - LS1021ATSN switch) able to change its state to slave and start to sync with master(TSN switch); on the other hand host1 connected in eth2(chassis label - LS1021ATSN switch) assume it as a GM. 

In this case, below is the error log in LS1021ATSN switch;
[image]

vladimiroltean, 2020-06-14:
Please provide a pcap PTP packet recording taken on ports swp3 and swp4 with tcpdump.

Norman03, 2020-06-14:
[swp_traces_LS1021ATSN.zip](https://github.com/openil/openil/files/4775897/swp_traces_LS1021ATSN.zip)
Test configuration: (same in guide shared above in this issue)
Taprio: taprio loaded in swp2 with vlan bridge interface as '1'
Host 2 (ptp4l) - connected to swp2: 
[image]
Host 1 (ptp4l) - connected to swp3: Its working as expected, because its connected in swp3 - since no taprio is loaded

I have captured a trace in LS1021ATSN switch on swp2,3,4 interfaces. During capturing the trace in switch, I noticed that few packets are dropped in the interfaces.
[image]

vladimiroltean, 2020-06-14:
Congratulations, you found a bug.
Please take the opportunity and also upgrade your board image to official OpenIL 1.8, then apply the patch below (which you can save to a `sja1105-ptp-taprio-bugfix.patch` file):

```diff
From 85c00a595ecf5d59d01e31187fd612375e14aca8 Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Mon, 15 Jun 2020 00:30:35 +0300
Subject: [PATCH] board/ls1021atsn: add kernel bugfix for sja1105 PTP with
 tc-taprio

This fixes a bug reported here:
https://github.com/openil/openil/issues/54#issuecomment-643720241

Patch was also submitted here:
https://lkml.org/lkml/2020/6/14/254

---
 ...fix-PTP-timestamping-with-large-tc-t.patch | 83 +++++++++++++++++++
 configs/nxp_ls1021atsn_defconfig              |  2 +
 2 files changed, 85 insertions(+)
 create mode 100644 board/nxp/ls1021atsn/patches/linux/0001-net-dsa-sja1105-fix-PTP-timestamping-with-large-tc-t.patch

diff --git a/board/nxp/ls1021atsn/patches/linux/0001-net-dsa-sja1105-fix-PTP-timestamping-with-large-tc-t.patch b/board/nxp/ls1021atsn/patches/linux/0001-net-dsa-sja1105-fix-PTP-timestamping-with-large-tc-t.patch
new file mode 100644
index 000000000000..1dd11671e050
--- /dev/null
+++ b/board/nxp/ls1021atsn/patches/linux/0001-net-dsa-sja1105-fix-PTP-timestamping-with-large-tc-t.patch
@@ -0,0 +1,83 @@
+From 69a8bffaebd417e7bebf5dac43cdb2cb564c7a91 Mon Sep 17 00:00:00 2001
+From: Vladimir Oltean <vladimir.oltean@nxp.com>
+Date: Sun, 14 Jun 2020 23:22:30 +0300
+Subject: [PATCH net] net: dsa: sja1105: fix PTP timestamping with large
+ tc-taprio cycles
+
+It isn't actually described clearly at all in UM10944.pdf, but on TX of
+a management frame (such as PTP), this needs to happen:
+
+- The destination MAC address (i.e. 01-80-c2-00-00-0e), along with the
+  desired destination port, need to be installed in one of the 4
+  management slots of the switch, over SPI.
+- The host can poll over SPI for that management slot's ENFPORT field.
+  That gets unset when the switch has matched the slot to the frame.
+
+And therein lies the problem. ENFPORT does not mean that the packet has
+been transmitted. Just that it has been received over the CPU port, and
+that the mgmt slot is yet again available.
+
+This is relevant because of what we are doing in sja1105_ptp_txtstamp_skb,
+which is called right after sja1105_mgmt_xmit. We are in a hard
+real-time deadline, since the hardware only gives us 24 bits of TX
+timestamp, so we need to read the full PTP clock to reconstruct it.
+Because we're in a hurry (in an attempt to make sure that we have a full
+64-bit PTP time which is as close as possible to the actual transmission
+time of the frame, to avoid 24-bit wraparounds), first we read the PTP
+clock, then we poll for the TX timestamp to become available.
+
+But of course, we don't know for sure that the frame has been
+transmitted when we read the full PTP clock. We had assumed that ENFPORT
+means it has, but the assumption is incorrect. And while in most
+real-life scenarios this has never been caught due to software delays,
+nowhere is this fact more obvious than with a tc-taprio offload, where
+PTP traffic gets a small timeslot very rarely (example: 1 packet per 10
+ms). In that case, we will be reading the PTP clock for timestamp
+reconstruction too early (before the packet has been transmitted), and
+this renders the reconstruction procedure incorrect (see the assumptions
+described in the comments found on function sja1105_tstamp_reconstruct).
+So the PTP TX timestamps will be off by 1<<24 clock ticks, or 135 ms
+(1 tick is 8 ns).
+
+So fix this case of premature optimization by simply reordering the
+sja1105_ptpegr_ts_poll and the sja1105_ptpclkval_read function calls. It
+turns out that in practice, the 135 ms hard deadline for PTP timestamp
+wraparound is not so hard, since even the most bandwidth-intensive PTP
+profiles, such as 802.1AS-2011, have a sync frame interval of 125 ms.
+So if we couldn't deliver a timestamp in 135 ms (which we can), we're
+toast and have much bigger problems anyway.
+
+Fixes: 47ed985e97f5 ("net: dsa: sja1105: Add logic for TX timestamping")
+Signed-off-by: Vladimir Oltean <vladimir.oltean@nxp.com>
+---
+ drivers/net/dsa/sja1105/sja1105_ptp.c | 8 ++++----
+ 1 file changed, 4 insertions(+), 4 deletions(-)
+
+diff --git a/drivers/net/dsa/sja1105/sja1105_ptp.c b/drivers/net/dsa/sja1105/sja1105_ptp.c
+index bc0e47c1dbb9..177134596458 100644
+--- a/drivers/net/dsa/sja1105/sja1105_ptp.c
++++ b/drivers/net/dsa/sja1105/sja1105_ptp.c
+@@ -891,16 +891,16 @@ void sja1105_ptp_txtstamp_skb(struct dsa_switch *ds, int port,
+ 
+ 	mutex_lock(&ptp_data->lock);
+ 
+-	rc = sja1105_ptpclkval_read(priv, &ticks, NULL);
++	rc = sja1105_ptpegr_ts_poll(ds, port, &ts);
+ 	if (rc < 0) {
+-		dev_err(ds->dev, "Failed to read PTP clock: %d\n", rc);
++		dev_err(ds->dev, "timed out polling for tstamp\n");
+ 		kfree_skb(skb);
+ 		goto out;
+ 	}
+ 
+-	rc = sja1105_ptpegr_ts_poll(ds, port, &ts);
++	rc = sja1105_ptpclkval_read(priv, &ticks, NULL);
+ 	if (rc < 0) {
+-		dev_err(ds->dev, "timed out polling for tstamp\n");
++		dev_err(ds->dev, "Failed to read PTP clock: %d\n", rc);
+ 		kfree_skb(skb);
+ 		goto out;
+ 	}
+-- 
+2.25.1
+
diff --git a/configs/nxp_ls1021atsn_defconfig b/configs/nxp_ls1021atsn_defconfig
index 961c905213d0..5a5353aa771f 100644
--- a/configs/nxp_ls1021atsn_defconfig
+++ b/configs/nxp_ls1021atsn_defconfig
@@ -46,6 +46,8 @@ BR2_LINUX_KERNEL_UIMAGE=y
 BR2_LINUX_KERNEL_UIMAGE_LOADADDR="0x80008000"
 BR2_LINUX_KERNEL_NEEDS_HOST_OPENSSL=y
 
+BR2_GLOBAL_PATCH_DIR="board/nxp/ls1021atsn/patches"
+
 # Serial port config
 BR2_TARGET_GENERIC_GETTY=y
 BR2_TARGET_GENERIC_GETTY_PORT="ttyS0"
-- 
2.25.1

```

as follows:

```
git clone https://github.com/openil/openil.git
cd openil
vim sja1105-ptp-taprio-bugfix.patch
# copy-paste
git am sja1105-ptp-taprio-bugfix.patch
make nxp_ls1021atsn_defconfig
make
```

vladimiroltean, 2020-06-16:
Please note that the patch has been applied on the master branch now (which contains some more bug fixes). You can get it by changing the default kernel branch which gets built:

```diff
diff --git a/linux/Config.in b/linux/Config.in
index 438674f20ef2..8c8f24c9f289 100644
--- a/linux/Config.in
+++ b/linux/Config.in
@@ -117,7 +117,7 @@ config BR2_LINUX_KERNEL_CUSTOM_REPO_URL

 config BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION
        string "Custom repository version"
-       default "OpenIL-v1.8-linux-202005"
+       default "linux-5.4.y"
        help
          Revision to use in the typical format used by
          Git/Mercurial/Subversion E.G. a sha id, a tag, branch, ..
```

Then rebuild the sdcard.img with the master branch kernel as follows:

```
make nxp_ls1021atsn_defconfig
make linux-dirclean world
```

vladimiroltean, 2020-08-06:
Assuming problem solved with the `sja1105-ptp-taprio-bugfix.patch` which is also integrated into the master branch.

closed 2020-08-06 by vladimiroltean

Waner98, 2021-05-06:
Hi @vladimiroltean ,

Sorry to reopen this topic, but I saw your explanations concerning vlan_filtering with VLAN tag conservation. Does it mean that we cannot at the same time, classify our packets by priority in egress and keep their VLAN tag intact ? We have to make a choice between both solutions ?

vladimiroltean, 2021-05-06:
What do you mean by "VLAN tag conservation"?
The switch carries a VLAN ID internally with each packet. When vlan_filtering is turned on, the internal VLAN ID and PCP are derived from the frame headers, otherwise they are set to some default values.
Every VLAN configured by the bridge is installed on one or more ports. Every VLAN installed on a port has certain flags (egress-untagged, pvid). If you want to see a VLAN ID on egress, just don't make that VLAN ID be "untagged" in the "bridge vlan add" command.

Waner98, 2021-05-06:
Well, OK I understand a little bit better whats happening inside. But in the case of a "non static VLAN attribution". For example, if I have a talker (following some queuing discipline etc ...) and for which its priority is traduced by a VLAN PCP before going in egress of the talker. The frame will arrive in my TSN switch with its own VLAN tag. What is happeing in this case ? When I do not want a "static VLAN attribution by port " I mean.

vladimiroltean, 2021-05-06:
If ingress port is swp2 and egress port is swp3, and the sender transmits with VLAN ID 100, then:
```
ip link add br0 type bridge vlan_filtering 1
ip link set swp2 master br0
ip link set swp3 master br0
bridge vlan add dev swp2 vid 100
bridge vlan add dev swp3 vid 100
```
You might need to delete the bridge created by default by systemd-networkd (depending on what changed you've already made to the system).

Waner98, 2021-05-06:
Ok thanks for your answer, and so the SJA1105  will by itself get PCP from incoming frames on swp2 and classify them in egress of the switch on swp3 ?

Waner98, 2021-05-06:
[image]
following this logic

vladimiroltean, 2021-05-06:
Yes, the driver automatically creates a one-to-one mapping between VLAN PCP and traffic class, which cannot be changed.

Waner98, 2021-05-06:
Ok, Thanks for your answers

Waner98, 2021-06-10:
Hi @vladimiroltean I followed your advices but I had some troubles ,

I try to put a little example in place to prove the efficiency of tc qdisc taprio offloaded on hardware of LS1021 TSN switch. But I have some troubles to get results that I want, It seems that the qdisc does not have any effect on my flows. 

My Set-up : 

One talker sends Ethernet frames on swp5 port of TSN switch via its VLAN (VID : 8, PCP : 1).
One talker sends Ethernet frames on swp2 port of TSN switch via its VLAN (VID : 8, PCP : 0). 
One listener is linked to swp3 port of TSN switch and sniff data tcpdump (and write a .pcap file for wireshark). 

TSN switch configuration: 

As I want to use VLAN tags that are already present in my frames, I allow VLAN tag conservation with : 
> ip link set br0 type bridge vlan_filtering 1
> ip link set swp2 master br0
> ip link set swp3 master br0
> ip link set swp5 master br0
> bridge vlan add dev swp2 vip 8
> bridge vlan add dev swp3 vip 8
> bridge vlan add dev swp5 vip 8

TAPRIO qdisc : 

I want to install my qdisc in hardware, and I want to see its effect on my frames. So I chose to put some high time for each gate state to see the effect of my queuing discipline on wireshark. (the period of sending data for both talker is 250µs). 

> tc qdisc replace dev swp3 parent root handle 100 taprio num_tc 8 map 0 1 2 3 4 5 6 7 queues 1@0 1@1 1@2 1@3 1@4 1@5 1@6 1@7 sched-entry S 0x01 50000000 sched-entry S 0x02 50000000 sched-entry S 0x00 50000000 sched-entry S 0x03 50000000 flags 2 

Results : 

When I look at wireshark my differents flows, this is just like no qdisc was installed. It does not stop any flows during my 50ms but I do not understand why. 

(I already checked that the qdisc was well installed with tc qdisc show dev swp3). 

Do you have any idea ? Do you know it there is a way to check the content of SHARINDX in order to know where my frames pass to be forwarded ? Because what my talker send are packets on multicast adress/port with priority inside my VLAN tag

Waner98, 2021-06-29:
Hi @vladimiroltean, did you have the time to take a look at this ?

liing0228, 2022-06-09:
> Hi @vladimiroltean I followed your advices but I had some troubles ,
>
> I try to put a little example in place to prove the efficiency of tc qdisc taprio offloaded on hardware of LS1021 TSN switch. [...]

HI sir,
i think i face the same issue .
Did you know how to fix it or what is the point ?
I will be grateful if you can tell me pls.

vladimiroltean, 2022-06-09:
You need to run PTP for the time aware shaper to start. Check the kernel log through the dmesg command and look for messages from the sja1105 driver to see when it becomes active.