nxp-archive/openil #56: SJA1105 PTP E2E Transparent Clock

RionelMessi, 2020-04-18:
Hello,

I'm currently working on LS1021ATSN boards with SJA1105 switch on it using the OpenIL 201908 release build image.
My setup is as follows: I have 2 host PC's each connected to one of the SJA1105 ports so that the hosts could synchronize using PTP while the switch should be working as an E2E Transparent Clock.
If I send the PTP packets over raw ethernet (ptp4l -2) it works perfectly well as I can see the correction field being updated by the switch as expected. However, if I use UDP transport with IPv4 (ptp4l -4) the correction field is not being updated at all and remains zero.

My Question is: Does the SJA1105 only support raw ethernet transport of PTP packages if running as E2E Transparent Clock? Is there a possibiliy to configure it in a way that it provides the residence time of PTP packages sent over UDP transport as well?

Btw I configured the SJA1105 as free running E2E Transparent Clock according to **sja1105-ptp-free-tc.sh** script provided by the linuxptp implementation of OpenIL:

> General Parameters Table
sja1105-tool config modify general-parameters-table switchid 0x0
sja1105-tool config modify general-parameters-table host_port 0x4
sja1105-tool config modify general-parameters-table send_meta0 0x0
sja1105-tool config modify general-parameters-table send_meta1 0x0
sja1105-tool config modify general-parameters-table mac_fltres0 0x000000000000
sja1105-tool config modify general-parameters-table mac_flt0 0xffffffffffff
sja1105-tool config modify general-parameters-table mac_fltres1 0x000000000000
sja1105-tool config modify general-parameters-table mac_flt1 0xffffffffffff
sja1105-tool config modify general-parameters-table ignore2stf 0x1
>

vladimiroltean, 2020-04-18:
The SJA1105 does not timestamp other PTP messages than those with raw Ethernet transport.

RionelMessi, 2020-04-19:
Okay, thank you very much for the quick response.

closed 2020-04-19 by RionelMessi