nxp-archive/openil #57: aarch64-linux-gnu-gcc: ERROR: unsafe header/library path used in cross-compilation: '-I/usr/local/include'

razr, 2020-05-22:
I'm trying to build OpenIL 1.8 for the NXP LS1028A, so I did

git clone https://github.com/openil/openil.git
cd openil
git checkout OpenIL-v1.8-202005
make nxp_ls1028ardb-64b_ubuntu_full_defconfig
make 2>&1 | tee build.log

After a while I got

/home/akholodn/github/openil/openil/output/host/bin/aarch64-linux-gnu-gcc   timemaster.o print.o rtnl.o sk.o util.o version.o  -lm -lrt  -o timemaster
/home/akholodn/github/openil/openil/output/host/bin/aarch64-linux-gnu-gcc  -Wall -DVER=2.0  -D_GNU_SOURCE -DHAVE_CLOCK_ADJTIME -DHAVE_POSIX_SPAWN -DHAVE_ONESTEP_SYNC    -DHAVE_NET_SNMP -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -g -O2 -fdebug-prefix-map=/build/net-snmp-ry1JTp/net-snmp-5.7.3+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security -DNETSNMP_USE_INLINE -Ulinux -Dlinux=linux -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -I/usr/lib/x86_64-linux-gnu/perl/5.26/CORE -Wdate-time -D_FORTIFY_SOURCE=2 -I. -I/usr/include -c snmp4lptp.c
aarch64-linux-gnu-gcc: ERROR: unsafe header/library path used in cross-compilation: '-I/usr/local/include'
makefile:92: recipe for target 'snmp4lptp.o' failed
make[2]: *** [snmp4lptp.o] Error 1
package/pkg-generic.mk:266: recipe for target '/home/akholodn/github/openil/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/.stamp_built' failed
make[1]: *** [/home/akholodn/github/openil/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/.stamp_built] Error 2
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2

razr, 2020-05-23:
I have grabbed also the latest release from here https://github.com/openil/openil/releases/tag/OpenIL-v1.8-202005 and it is the same error.

I'm using Ubuntu 18.04 as a build host

vladimiroltean, 2020-05-23:
Hi there,

Please apply this patch to OpenIL:
```
From 1dd4d67a5aa30b203216c1a3dcc87a9ab978f5ea Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Sat, 23 May 2020 12:48:24 +0300
Subject: [PATCH] package/qoriq-ptp4l: pull upstream patch for removing
 snmp4lptp

Currently, the snmpflg.sh script searches for include paths that contain
net-snmp-agent-includes.h. If the host environment has this header
installed through the package manager, then the linuxptp makefile will
try to bring in this dependency, which will make cross-compilation fail,
as seen below:

output/host/bin/aarch64-linux-gnu-gcc timemaster.o print.o rtnl.o sk.o util.o version.o -lm -lrt -o timemaster
output/host/bin/aarch64-linux-gnu-gcc -Wall -DVER=2.0 -D_GNU_SOURCE
-DHAVE_CLOCK_ADJTIME -DHAVE_POSIX_SPAWN -DHAVE_ONESTEP_SYNC
-DHAVE_NET_SNMP -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -g -O2
-fdebug-prefix-map=/build/net-snmp-ry1JTp/net-snmp-5.7.3+dfsg=.
-fstack-protector-strong -Wformat -Werror=format-security
-DNETSNMP_USE_INLINE -Ulinux -Dlinux=linux -D_REENTRANT -D_GNU_SOURCE
-DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include
-D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64
-I/usr/lib/x86_64-linux-gnu/perl/5.26/CORE -Wdate-time
-D_FORTIFY_SOURCE=2 -I. -I/usr/include -c snmp4lptp.c
aarch64-linux-gnu-gcc: ERROR: unsafe header/library path used in cross-compilation: '-I/usr/local/include'
makefile:92: recipe for target 'snmp4lptp.o' failed
make[2]: *** [snmp4lptp.o] Error 1
package/pkg-generic.mk:266: recipe for target 'output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/.stamp_built' failed
make[1]: *** [output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/.stamp_built] Error 2
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2

But the SNMP code was removed from mainline linuxptp, so the bug will
never get resolved. So remove it from our downstream linuxptp too.

---
 ...0001-Remove-the-unfinished-SNMP-code.patch | 470 ++++++++++++++++++
 1 file changed, 470 insertions(+)
 create mode 100644 package/nxp/qoriq-ptp4l/0001-Remove-the-unfinished-SNMP-code.patch

diff --git a/package/nxp/qoriq-ptp4l/0001-Remove-the-unfinished-SNMP-code.patch b/package/nxp/qoriq-ptp4l/0001-Remove-the-unfinished-SNMP-code.patch
new file mode 100644
index 000000000000..9b321bfa9714
--- /dev/null
+++ b/package/nxp/qoriq-ptp4l/0001-Remove-the-unfinished-SNMP-code.patch
@@ -0,0 +1,470 @@
+From 01db4c884998246e6599d78004cd7cf06161c30e Mon Sep 17 00:00:00 2001
+From: Richard Cochran <richardcochran@gmail.com>
+Date: Sun, 29 Dec 2019 11:20:10 -0800
+Subject: [PATCH] Remove the unfinished SNMP code.
+
+Unfortunately the SNMP code still has issues like not passing the
+valgrind test, and no one is able to finish this up right now.  This
+patch removes the SNMP program so that the upcoming release does not
+contain unfinished work, potentially misleading end users about the
+scope and completeness of the features.
+
+Signed-off-by: Richard Cochran <richardcochran@gmail.com>
+Reviewed-by: Jacob Keller <jacob.e.keller@intel.com>
+---
+ makefile        |  14 ----
+ snmp4lptp.8     | 119 ------------------------------
+ snmp4lptp.c     | 192 ------------------------------------------------
+ snmp4lptp_mib.h |  30 --------
+ snmpflg.sh      |  42 -----------
+ 5 files changed, 397 deletions(-)
+ delete mode 100644 snmp4lptp.8
[... 438 lines cut ...]
+-				printf " -DHAVE_NET_SNMP `net-snmp-config --cflags`"
+-				break 2
+-			fi
+-		done
+-	done
+-}
+-
+-echo "$(snmp_flags)"
+-- 
+2.25.1
+
-- 
2.25.1

```

and then do the following:
```
rm -rf output/build/qoriq-ptp4l*
rm -rf dl/qoriq-ptp4l*
make
```

Thanks,
-Vladimir

razr, 2020-05-23:
```
akholodn@razilix:~/github/openil/openil$ git am snmp4lptp.patch
Applying: package/qoriq-ptp4l: pull upstream patch for removing snmp4lptp
.git/rebase-apply/patch:48: trailing whitespace.
.git/rebase-apply/patch:59: trailing whitespace.
.git/rebase-apply/patch:61: trailing whitespace.
.git/rebase-apply/patch:70: trailing whitespace.
.git/rebase-apply/patch:99: trailing whitespace.
-or from an SNMP-specific form.
warning: squelched 2 whitespace errors
warning: 7 lines add whitespace errors.
akholodn@razilix:~/github/openil/openil$ rm -rf output/build/qoriq-ptp4l*
akholodn@razilix:~/github/openil/openil$ rm -rf dl/qoriq-ptp4l*
akholodn@razilix:~/github/openil/openil$ make
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Downloading
Initialized empty Git repository in /home/akholodn/github/openil/openil/dl/qoriq-ptp4l/git/.git/
Fetching all references
remote: Enumerating objects: 805, done.
remote: Counting objects: 100% (805/805), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 5133 (delta 794), reused 785 (delta 785), pack-reused 4328
Receiving objects: 100% (5133/5133), 1.04 MiB | 1.22 MiB/s, done.
Resolving deltas: 100% (3868/3868), done.
From https://github.com/openil/linuxptp
 * [new branch]      master     -> origin/master
remote: Enumerating objects: 20, done.
remote: Counting objects: 100% (20/20), done.
remote: Total 152 (delta 20), reused 20 (delta 20), pack-reused 132
Receiving objects: 100% (152/152), 51.69 KiB | 433.00 KiB/s, done.
Resolving deltas: 100% (88/88), completed with 7 local objects.
From https://github.com/openil/linuxptp
 * [new tag]         OpenIL-linuxptp-201705 -> OpenIL-linuxptp-201705
 * [new tag]         OpenIL-linuxptp-201708 -> OpenIL-linuxptp-201708
 * [new tag]         OpenIL-linuxptp-201712 -> OpenIL-linuxptp-201712
warning: refname 'eee6d0fb995dddf27acde2addfb1fe89b71ab82d' is ambiguous.
Git normally never creates a ref that ends with 40 hex characters
because it will be ignored when you just specify 40-hex. These refs
may be created by mistake. For example,

  git checkout -b $br $(git rev-parse ...)

where "$br" is somehow empty and a 40-hex ref is created. Please
examine these refs and maybe delete them. Turn this message off by
running "git config advice.objectNameWarning false"
WARNING: no hash file for qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d.tar.gz
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Extracting
gzip -d -c /home/akholodn/github/openil/openil/dl/qoriq-ptp4l/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d.tar.gz | tar --strip-components=1 -C /home/akholodn/github/openil/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d   -xf -
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Patching

Applying 0001-Remove-the-unfinished-SNMP-code.patch using patch:
patching file makefile
Hunk #1 succeeded at 47 (offset 9 lines).
Hunk #2 FAILED at 64.
1 out of 2 hunks FAILED -- saving rejects to file makefile.rej
patching file snmp4lptp.8
patching file snmp4lptp.c
patching file snmp4lptp_mib.h
patching file snmpflg.sh
package/pkg-generic.mk:228: recipe for target '/home/akholodn/github/openil/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/.stamp_patched' failed
make[1]: *** [/home/akholodn/github/openil/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/.stamp_patched] Error 1
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2

akholodn@razilix:~/github/openil/openil$ cat ./output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/makefile.rej
--- makefile
+++ makefile
@@ -64,13 +57,6 @@ hwstamp_ctl: hwstamp_ctl.o version.o

 phc_ctl: phc_ctl.o phc.o sk.o util.o clockadj.o sysoff.o print.o version.o

-snmp4lptp: config.o hash.o interface.o msg.o phc.o pmc_common.o print.o sk.o \
- snmp4lptp.o tlv.o $(TRANSP) util.o
-       $(CC) $^ $(LDFLAGS) $(LOADLIBES) $(LDLIBS) $(snmplib) -o $@
-
-snmp4lptp.o: snmp4lptp.c
-       $(CC) $(CPPFLAGS) $(CFLAGS) $(snmpflg) -c $<
-
 timemaster: phc.o print.o rtnl.o sk.o timemaster.o util.o version.o

 version.o: .version version.sh $(filter-out version.d,$(DEPEND))

```

vladimiroltean, 2020-05-23:
I fixed up the patch, please copy and try again.

razr, 2020-05-23:
yeap, it works. Thanks @vladimiroltean

closed 2020-05-23 by vladimiroltean

razr, 2020-05-27:
hello @vladimiroltean, did you commit your patch?

razr, 2020-05-27:
ah, I see you did it here :)
https://github.com/openil/openil/commit/e15c5f088c904354311bb349add1932ddbdeaa7c

Bibo000001, 2023-12-18:
what are talking about?