nxp-archive/openil #58: rsync: chgrp "$HOME/github/openil/openil/output/build/buildroot-fs/ext2/target/bin/nc" failed: Operation not permitted (1)

razr, 2020-05-27:
I did:
user@host:~/github/openil/openil$ make
user@host:~/github/openil/openil$ rm -rf output/build/linux-OpenIL-v1.8-linux-202005/
user@host:~/github/openil/openil$ make

then I get 
>>>   Generating filesystem image rootfs.ext2
mkdir -p /home/akholodn/github/openil/openil/output/images
rm -rf /home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2
mkdir -p /home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2
rsync -auH --exclude=/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM /home/akholodn/github/openil/openil/output/target/ /home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2/target
rsync: chgrp "/home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2/target/bin/nc" failed: Operation not permitted (1)
rsync: chgrp "/home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2/target/bin/.bzip2.L5m09c" failed: Operation not permitted (1)
rsync: chgrp "/home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2/target/usr/bin/.perl5.26.1.QFiaZe" failed: Operation not permitted (1)
rsync: chgrp "/home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2/target/usr/bin/.zipinfo.ci8bpe" failed: Operation not permitted (1)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
make: *** [fs/ext2/ext2.mk:50: /home/akholodn/github/openil/openil/output/images/rootfs.ext2] Error 23

vladimiroltean, 2020-05-27:
This looks like an issue on your side. Did you run any command as root?

razr, 2020-05-27:
nc points to /etc/alternatives/nc

akholodn@windix:~/github/openil/openil$ ls -la /home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2/target/bin/nc
lrwxrwxrwx 1 akholodn wruser 20 Mai 27 22:29 /home/akholodn/github/openil/openil/output/build/buildroot-fs/ext2/target/bin/nc -> /etc/alternatives/nc

vladimiroltean, 2020-05-27:
This is strange. I'm building from scratch to see if I can reproduce.
I think nc is provided by the busybox package. Could you run "make busybox-dirclean busybox world | tee build.lo" and show the full output?

razr, 2020-05-27:
```
akholodn@windix:~/github/openil/openil$ make busybox-dirclean
rm -Rf /home/akholodn/github/openil/openil/output/build/busybox-1.31.1

akholodn@windix:~/github/openil/openil$ make busybox
busybox-1.31.1.tar.bz2: OK (sha256: d0f940a72f648943c1f2211e0e3117387c31d765137d92bd8284a3fb9752a998)
>>> busybox 1.31.1 Extracting
bzcat /home/akholodn/github/openil/openil/dl/busybox/busybox-1.31.1.tar.bz2 | /home/akholodn/github/openil/openil/output/host/bin/tar --strip-components=1 -C /home/akholodn/github/openil/openil/output/build/busybox-1.31.1   -xf -
>>> busybox 1.31.1 Patching

Applying 0001-networking-libiproute-use-linux-if_packet.h-instead-.patch using patch: 
patching file networking/libiproute/iplink.c

Applying 0002-Makefile.flags-strip-non-l-arguments-returned-by-pkg.patch using patch: 
patching file Makefile.flags
/usr/bin/install: missing destination file operand after '/home/akholodn/github/openil/openil/output/build/busybox-1.31.1/.config'
Try '/usr/bin/install --help' for more information.
make: *** [package/busybox/busybox.mk:375: /home/akholodn/github/openil/openil/output/build/busybox-1.31.1/.stamp_dotconfig] Error 1
```

vladimiroltean, 2020-05-27:
I just completed a full build and I couldn't reproduce this. Not entirely sure why this portion fails for you:
```
Applying 0002-Makefile.flags-strip-non-l-arguments-returned-by-pkg.patch using patch: 
patching file Makefile.flags
touch /opt/openil/output-ls1028ardb/build/busybox-1.31.1/.stamp_patched
/usr/bin/install -m 0644 -D package/busybox/busybox.config /opt/openil/output-ls1028ardb/build/busybox-1.31.1/.config
support/kconfig/merge_config.sh -m -O /opt/openil/output-ls1028ardb/build/busybox-1.31.1/ /opt/openil/output-ls1028ardb/build/busybox-1.31.1/.config  board/nxp/common/ls/busybox.config
Using /opt/openil/output-ls1028ardb/build/busybox-1.31.1/.config as base
Merging board/nxp/common/ls/busybox.config
```

Is your git directory clean? Could you share the output of "git status"?

razr, 2020-05-27:
Just to make sure: 
I'm on OpenIL-v1.8-202005 and  
make nxp_ls1028ardb-64b_ubuntu_full_defconfig

```
akholodn@windix:~/github/openil/openil$ git status
HEAD detached from OpenIL-v1.8-202005
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	build.log

nothing added to commit but untracked files present (use "git add" to track)

akholodn@windix:~/github/openil/openil$ git log
commit 20c54692a89fbffe71fe05a06e73da61b45db8cc (HEAD)
Author: Vladimir Oltean <vladimir.oltean@nxp.com>
Date:   Sat May 23 12:48:24 2020 +0300
```

vladimiroltean, 2020-05-27:
Ah, ok, I was building the plain nxp_ls1028ardb-64b_defconfig.
Looking at ubuntu_full while it's building, I do see these lines at the end of the file:
```
# busybox setting disable
BR2_PACKAGE_BUSYBOX=n
```
I wonder why it's picked up by the build system, when it shouldn't be?
If you run just:
```
make busybox-dirclean world
```
do you get any other errors?
By any chance have you been mixing defconfigs in the same output directory?

razr, 2020-05-27:
I'm rebuilding everything from scratch and let you know tomorrow.

vladimiroltean, 2020-05-27:
Just a hint, you can use the O= variable to build in separate output directories.
Example:
```
make nxp_ls1028ardb-64b_defconfig O=output-ls1028ardb
make O=output-ls1028ardb | tee output-ls1028ardb/build.log
```

razr, 2020-05-28:
@vladimiroltean I have rebuilt everything, and the error stays the same. 
This is how it looks on my target
```
root@LS1028ARDB-Ubuntu:~# which nc                                                                                                       
/bin/nc                                                                                                                                  
root@LS1028ARDB-Ubuntu:~# ls -la /bin/nc                                                                                                 
lrwxrwxrwx 1 root root 20 May 28  2020 /bin/nc -> /etc/alternatives/nc   
```

vladimiroltean, 2020-05-28:
Which package is providing nc in your build?
```
find output -name nc
```

razr, 2020-05-28:
```
akholodn@windix:~/github/openil/openil$ ls -la output/build/buildroot-fs/ext2/target/etc/alternatives/nc
lrwxrwxrwx 1 akholodn wruser 15 Mai 28 09:51 output/build/buildroot-fs/ext2/target/etc/alternatives/nc -> /bin/nc.openbsd
```

it should be this one
```
./package/netcat-openbsd
./package/netcat-openbsd/netcat-openbsd.mk
./package/netcat-openbsd/netcat-openbsd.hash
```

vladimiroltean, 2020-05-28:
I just noticed that we're not on the same git HEAD. And I just can't seem to understand what it going on, or reproduce this. Before rebuilding from scratch, have you updated your git tree to the current master? If not, could you please do so?
```
git fetch origin && git reset --hard origin/master
```

razr, 2020-05-28:
Yes, I'm on the officially released _OpenIL-v1.8-202005_ .
Uff, actually I wanted to enable Intel Wifi 8265 M.2 module, so I have reconfigured my kernel and then figured out that _make_ does not work properly. 
Let me run it on master

vladimiroltean, 2020-05-28:
Save your kernel config changes to the board/nxp/ls1028ardb/linux.config file, that way you won't lose them next time.

razr, 2020-05-28:
[build.log](https://github.com/openil/openil/files/4698324/build.log)

make asks me two times for the sudo password, the second time somewhere before the rsync

vladimiroltean, 2020-05-28:
Huh, that's a good one.
Looks like it's expected, judging from [our own README](https://github.com/openil/openil/blob/master/README.OpenIL#L17)?
```
You do not need to be root to build or run OpenIL.
"sudo" permission is required when building ubuntu RFS.
```

razr, 2020-05-28:
maybe :), but it fails if I do
```
user@host:~/github/openil/openil$ make
user@host:~/github/openil/openil$ rm -rf output/build/linux-OpenIL-v1.8-linux-202005/
user@host:~/github/openil/openil$ make
```
I'd expect it asks for the sudo somewhere again.

razr, 2020-05-28:
@vladimiroltean, do you know by any chance why it asks for sudo? It should not actually do it. I don't want that your build script erases my / folder :)

vladimiroltean, 2020-05-28:
Yes, I don't want that either.
It asks for sudo here, during do_distrorfs_first_stage, so that's why further invocations of 'make' don't need sudo again, since the skeleton rootfs was already made:
https://github.com/openil/openil/blob/master/board/nxp/common/ls/post-custom-skeleton-ubuntu-base-18.04.sh#L4
by the way, yes, if there's any sort of bug in this script, which would result in RFSDIR being unset, then script would delete your /dev/ and your /etc/apt/apt.conf. So you have valid concerns, I suppose. This is new to me, we didn't use to need this. So if you don't have a secure environment where no data would be lost (such as docker), then I suppose you should only build the plain Buildroot rootfs. I'm going to bring this up with the team and see what our options are.

mingkaihu, 2020-05-30:
HI razr, there're three parts in the script with sudo.

1. To install the needed package on the host machine.
The script is checking if the packages have been install on host machine, if yes, it will not install anything. We've list all the needed packages in system requirement chapter to avoid the installation.

2. to setup http_proxy on host machine.
remove it and leave it to customer to setup proxy.

3. to install package in the target ubuntu system.
this doesn't change anything on host machine.

does it make sense?

razr, 2020-05-30:
@mingkaihu, I think it is a good idea to do things like
https://github.com/openil/openil/blob/master/board/nxp/common/ls/post-custom-skeleton-ubuntu-base-18.04.sh#L75
it is a modification of the host system.
```
sudo tee -a /etc/wgetrc
```

And this I prefer to install explicitly by myself, I don't want it to be hidden in the script
```
    for pkg in binfmt-support qemu-system-common qemu-user-static debootstrap; do
	if ! dpkg-query -l $pkg | grep ii 1>/dev/null; then
	    echo installing $pkg
	    sudo apt-get -y install $pkg
        fi
    done
```

This part can be done without sudo permissions as well, see e.g. https://wiki.debian.org/fakechroot
```
	sudo debootstrap --arch=$1 --foreign $4 $RFSDIR
	echo "installing for second-stage ..."
	DEBIAN_FRONTEND=noninteractive DEBCONF_NONINTERACTIVE_SEEN=true LC_ALL=C LANGUAGE=C LANG=C \
	sudo chroot $RFSDIR /debootstrap/debootstrap  --second-stage
	echo "configure ... "
	DEBIAN_FRONTEND=noninteractive DEBCONF_NONINTERACTIVE_SEEN=true LC_ALL=C LANGUAGE=C LANG=C \
	sudo chroot $RFSDIR dpkg --configure -a
```

razr, 2020-05-30:
It is a very interesting discussion, but could someone take a look into this defect, please :)? As I have mentioned after a reconfiguring Linux kernel I can't do _make_ anymore. It fails with the error I have described in the title. I need to do _sudo make_