nxp-archive/openil #61: Query about phc2sys

diegotxegp, 2020-06-18:
Hi there,

After synchronising my network (slave-master-slave) with IEEE 802.1AS gPTP, acting my LS1021ATSN platform as master. My question is: What phc2sys instructions should I use on the slaves?

I tried to use this:   sudo phc2sys -a -rr -m

But a message appears saying: "waiting for ptp4l". But ptp4l is already working.

Another command I tried was:   sudo phc2sys -c /ptp/ptp0 -s CLOCK_REALTIME -O 0 -m
With this command, the node tries to synchronise CLOCK_REALTIME to its PHC but it appears an offset high.

My question is: What instruction for phc2sys do you recommend me?
And another question is: Must I run phc2sys also in the master?

Thank you very much for reading me.

Kind regards,
Diego

vladimiroltean, 2020-06-18:
Did you notice the phc2sys systemctl service that comes preinstalled?
```
[root@LS1021ATSN ~] # systemctl status phc2sys.service 
● phc2sys.service - Precision Time Protocol system clock synchronization
     Loaded: loaded (/usr/lib/systemd/system/phc2sys.service; disabled; vendor preset: enabled)
     Active: active (running) since Thu 2020-06-18 10:40:42 UTC; 6h ago
   Main PID: 824 (phc2sys)
      Tasks: 1 (limit: 2047)
     Memory: 408.0K
     CGroup: /system.slice/phc2sys.service
             └─824 /usr/sbin/phc2sys -a -rr -f /etc/linuxptp-system-clock.cfg

Jun 18 17:32:59 LS1021ATSN phc2sys[824]: [77650.736] swp5 sys offset     -1923 s2 freq  -22809 delay 161679
Jun 18 17:33:00 LS1021ATSN phc2sys[824]: [77651.737] swp5 sys offset        72 s2 freq  -21391 delay 163039
```
Inside `/etc/linuxptp-system-clock.cfg` you will find:
```
transportSpecific       0x1
```
which is needed if ptp4l is operating in 802.1AS mode, otherwise ptp4l and phc2sys will not be able to communicate through the UNIX domain socket.

vladimiroltean, 2020-06-18:
> And another question is: Must I run phc2sys also in the master?

If you are using the LS1021A-TSN board as a master, you need to hook it up to a valid time reference. On the board itself there is not even a battery-backed RTC, so after each boot, the system time will start from Jan 1st 1970. So, something like GPS, NTP, etc.
But nonetheless, phc2sys is happy to provide CLOCK_REALTIME into the PTP domain, and it is configured to do that by default (-rr).

diegotxegp, 2020-06-19:
> Did you notice the phc2sys systemctl service that comes preinstalled?
>
> ```
> [root@LS1021ATSN ~] # systemctl status phc2sys.service
> ● phc2sys.service - Precision Time Protocol system clock [...]

I get this result when I write "systemctl status phc2sys.service":

[root@LS1021ATSN ~] # systemctl status phc2sys.service
● phc2sys.service - Precision Time Protocol system clock synchronization
     Loaded: loaded (/usr/lib/systemd/system/phc2sys.service; disabled; vendor preset: enabled)
     Active: inactive (dead)

And inside "/etc/linuxptp-system-clock.cfg", yes, I have "transportSpecific       0x1".
[root@LS1021ATSN /etc] # cat linuxptp-system-clock.cfg

# Configuration file for phc2sys

[global]
step_threshold		0.00002
first_step_threshold	0.00002
transportSpecific	0x1

diegotxegp, 2020-06-19:
> If you are using the LS1021A-TSN board as a master, you need to hook it up to a valid time reference. On the board itself there is not even a battery-backed RTC, so after each boot, the system time [...]

I do not need a current time. I do not mind if it is 1970. What I want is simply that time from LS1021ATSN is synchronised in the slaves.

So, with you answer,  I understand that there is no problem to keep the time in 1970 every time I reboot the LS1021ATSN platform. In general, the network will work perfectly synchronised, regardless if it is a current time.

diegotxegp, 2020-06-19:
I tried to enable phc2sys with "systemctl enable phc2sys --now" but it does not work.

vladimiroltean, 2020-06-19:
> but it does not work

Helpful.
How does it not work? What does `journalctl -b -u phc2sys -f` show?

diegotxegp, 2020-06-19:
Sorry Vladimir.

Executing "journalctl -b -u phc2sys -f" at the LS1021ATSN platform I get this result:

[root@LS1021ATSN ~] # journalctl -b -u phc2sys -f
-- Logs begin at Fri 2020-02-07 21:55:08 UTC. --
Feb 08 00:45:40 LS1021ATSN phc2sys[507]: [32090.164] swp5 sys offset       160 s2 freq  -20537 delay 166640
Feb 08 00:45:41 LS1021ATSN phc2sys[507]: [32091.165] swp5 sys offset      -672 s2 freq  -21321 delay 169760
Feb 08 00:45:42 LS1021ATSN phc2sys[507]: [32092.166] swp5 sys offset       664 s2 freq  -20187 delay 166160
Feb 08 00:45:43 LS1021ATSN phc2sys[507]: [32093.168] swp5 sys offset      -640 s2 freq  -21291 delay 167760
Feb 08 00:45:44 LS1021ATSN phc2sys[507]: [32094.169] swp5 sys offset      1128 s2 freq  -19715 delay 165040
Feb 08 00:45:45 LS1021ATSN phc2sys[507]: [32095.171] swp5 sys offset      -800 s2 freq  -21305 delay 167200
Feb 08 00:45:46 LS1021ATSN phc2sys[507]: [32096.172] swp5 sys offset       224 s2 freq  -20521 delay 167280
Feb 08 00:45:47 LS1021ATSN phc2sys[507]: [32097.173] swp5 sys offset      -584 s2 freq  -21262 delay 167040
Feb 08 00:45:48 LS1021ATSN phc2sys[507]: [32098.175] swp5 sys offset     -1096 s2 freq  -21949 delay 169280
Feb 08 00:45:49 LS1021ATSN phc2sys[507]: [32099.176] swp5 sys offset     -1352 s2 freq  -22534 delay 172080
Feb 08 00:45:50 LS1021ATSN phc2sys[507]: [32100.178] swp5 sys offset      3800 s2 freq  -17787 delay 165280
Feb 08 00:45:51 LS1021ATSN phc2sys[507]: [32101.179] swp5 sys offset       640 s2 freq  -19807 delay 165280
Feb 08 00:45:52 LS1021ATSN phc2sys[507]: [32102.181] swp5 sys offset     -1808 s2 freq  -22063 delay 167520

But if I run it on one of the nodes, it does not show anything. As I said to you, when I type "sudo phc2sys -a -rr", a message with "Waiting for ptp4l..." appears.

What I want it synchronise my Linux system time to the PHC of the nodes. Once, ptp4l has synchronise all the PHC of the domain to the Grandmaster, I want to transfer that time to the Linux system time.
As I said to you, I do not mind that that time is in 1970. Neither I have a GPS as a reference clock.

Thank you very much for your help and your quick response.

Diego

vladimiroltean, 2020-06-19:
> Executing "journalctl -b -u phc2sys -f" at the LS1021ATSN platform I get this result:

So, it works. Quite a big difference. Now, granted, the offsets aren't the best and I'll need to look into why the kernel headers aren't picked up properly (again), but nonetheless, the procedure is correct and what should happen happens.

> But if I run it on one of the nodes, it does not show anything. As I said to you, when I type "sudo phc2sys -a -rr", a message with "Waiting for ptp4l..." appears.

I'm forced to repeat myself:

> Inside /etc/linuxptp-system-clock.cfg you will find:
> 
> transportSpecific       0x1
> 
> which is needed if ptp4l is operating in 802.1AS mode, otherwise ptp4l and phc2sys will not be able to communicate through the UNIX domain socket.

If we take what I said and apply it to the situation at hand, it would read like this:
"Since you are using the linuxptp stack not only on the LS1021A-TSN but also on your other nodes, everything protocol-related which stands true for the LS1021A-TSN also stands true for your other nodes. So, if you are using the gPTP.cfg profile on another node, then you should also use phc2sys with the `--transportSpecific 1` option on that node, if you want to involve the CLOCK_REALTIME in clock synchronization".

diegotxegp, 2020-06-22:
Following your comments, I got these results on my nodes:

nodo2@nodo2-N-A:~$ sudo phc2sys -a -rr --transportSpecific 0x1 -m
[sudo] password for nodo2: 
phc2sys[1797.065]: reconfiguring after port state change
phc2sys[1797.067]: selecting CLOCK_REALTIME for synchronization
phc2sys[1797.067]: selecting enp3s0 as the master clock
phc2sys[1797.068]: CLOCK_REALTIME phc offset 1592814778765002635 s0 freq      +0 delay   5388
phc2sys[1798.068]: failed to step clock: Invalid argument
phc2sys[1798.069]: CLOCK_REALTIME phc offset 1592814778765013231 s1 freq  +10589 delay   5616
phc2sys[1799.069]: CLOCK_REALTIME phc offset 1592814778765013304 s2 freq +100000000 delay   5364
phc2sys[1800.070]: CLOCK_REALTIME phc offset 1592814778654804391 s2 freq +100000000 delay   5033
phc2sys[1801.070]: CLOCK_REALTIME phc offset 1592814778543662571 s2 freq +100000000 delay   5043
phc2sys[1802.070]: CLOCK_REALTIME phc offset 1592814778432524145 s2 freq +100000000 delay   4849
phc2sys[1803.071]: CLOCK_REALTIME phc offset 1592814778321390847 s2 freq +100000000 delay   5043
phc2sys[1804.071]: CLOCK_REALTIME phc offset 1592814778210259226 s2 freq +100000000 delay   5043
phc2sys[1805.071]: CLOCK_REALTIME phc offset 1592814778099125324 s2 freq +100000000 delay   5044
phc2sys[1806.072]: CLOCK_REALTIME phc offset 1592814777987992639 s2 freq +100000000 delay   4742
phc2sys[1807.072]: CLOCK_REALTIME phc offset 1592814777876858160 s2 freq +100000000 delay   5033
phc2sys[1808.072]: CLOCK_REALTIME phc offset 1592814777765722789 s2 freq +100000000 delay   5022

I am not sure it works properly because of two reasons; it appears this message "phc2sys[1798.068]: failed to step clock: Invalid argument" and, as you can see, offset is quite big.

How could I solve it? Thank you

Diego

vladimiroltean, 2020-06-22:
Here is where having a CLOCK_REALTIME in 1970 might become a bit of a problem.
Does it work if you manually configure your master in 2020?
```
phc_ctl CLOCK_REALTIME set 1592817646.744752137
```

diegotxegp, 2020-06-22:
After the previous results I sent you, I paused the "sudo phc2sys -a -rr --transportSpecific 0x1 -m" and got these different results:

nodo2@nodo2-N-A:~$ sudo phc2sys -a -rr --transportSpecific 0x1 -m
[sudo] password for nodo2: 
phc2sys[4334.473]: reconfiguring after port state change
phc2sys[4334.475]: selecting CLOCK_REALTIME for synchronization
phc2sys[4334.475]: selecting enp3s0 as the master clock
phc2sys[4334.477]: CLOCK_REALTIME phc offset 1592814497082320705 s0 freq +100000000 delay   4741
phc2sys[4335.478]: CLOCK_REALTIME phc offset 1592814496970978619 s1 freq   +9561 delay   5032
phc2sys[4336.478]: CLOCK_REALTIME phc offset   -599364 s2 freq -589803 delay   5592
phc2sys[4337.479]: CLOCK_REALTIME phc offset     -2200 s2 freq -172448 delay   5344
phc2sys[4338.479]: CLOCK_REALTIME phc offset    181478 s2 freq  +10570 delay   5605
phc2sys[4339.479]: CLOCK_REALTIME phc offset    180911 s2 freq  +64446 delay   5340
phc2sys[4340.480]: CLOCK_REALTIME phc offset    125944 s2 freq  +63753 delay   5376
phc2sys[4341.480]: CLOCK_REALTIME phc offset     71483 s2 freq  +47075 delay   5616
phc2sys[4342.481]: CLOCK_REALTIME phc offset     33688 s2 freq  +30725 delay   5376
phc2sys[4343.481]: CLOCK_REALTIME phc offset     12392 s2 freq  +19535 delay   5604
phc2sys[4344.481]: CLOCK_REALTIME phc offset      2256 s2 freq  +13117 delay   5628
phc2sys[4345.482]: CLOCK_REALTIME phc offset     -1487 s2 freq  +10050 delay   5604
phc2sys[4346.482]: CLOCK_REALTIME phc offset     -2049 s2 freq   +9042 delay   5364
phc2sys[4347.482]: CLOCK_REALTIME phc offset     -1617 s2 freq   +8860 delay   5616
phc2sys[4348.483]: CLOCK_REALTIME phc offset      -986 s2 freq   +9006 delay   5579
phc2sys[4349.483]: CLOCK_REALTIME phc offset      -598 s2 freq   +9098 delay   5592
phc2sys[4350.484]: CLOCK_REALTIME phc offset      -200 s2 freq   +9316 delay   5604
phc2sys[4351.484]: CLOCK_REALTIME phc offset       -48 s2 freq   +9408 delay   5376
phc2sys[4352.484]: CLOCK_REALTIME phc offset       -78 s2 freq   +9364 delay   5556
phc2sys[4353.485]: CLOCK_REALTIME phc offset        60 s2 freq   +9479 delay   5604
phc2sys[4354.485]: CLOCK_REALTIME phc offset        -1 s2 freq   +9436 delay   5604
phc2sys[4355.486]: CLOCK_REALTIME phc offset      -106 s2 freq   +9330 delay   5220
phc2sys[4356.486]: CLOCK_REALTIME phc offset        68 s2 freq   +9472 delay   5364
phc2sys[4357.486]: CLOCK_REALTIME phc offset        27 s2 freq   +9452 delay   5628
phc2sys[4358.487]: CLOCK_REALTIME phc offset       -38 s2 freq   +9395 delay   5604
phc2sys[4359.487]: CLOCK_REALTIME phc offset      -100 s2 freq   +9322 delay   5603
phc2sys[4360.488]: CLOCK_REALTIME phc offset        59 s2 freq   +9451 delay   5616
phc2sys[4361.488]: CLOCK_REALTIME phc offset        49 s2 freq   +9458 delay   5604
phc2sys[4362.488]: CLOCK_REALTIME phc offset       -59 s2 freq   +9365 delay   5604
phc2sys[4363.489]: CLOCK_REALTIME phc offset       117 s2 freq   +9523 delay   5604
phc2sys[4364.489]: CLOCK_REALTIME phc offset        97 s2 freq   +9538 delay   5580
phc2sys[4365.490]: CLOCK_REALTIME phc offset       -25 s2 freq   +9445 delay   5376
phc2sys[4366.490]: CLOCK_REALTIME phc offset       -88 s2 freq   +9375 delay   5580
phc2sys[4367.491]: CLOCK_REALTIME phc offset      -145 s2 freq   +9292 delay   5604
phc2sys[4368.491]: CLOCK_REALTIME phc offset        11 s2 freq   +9404 delay   5592
phc2sys[4369.491]: CLOCK_REALTIME phc offset       -52 s2 freq   +9344 delay   5604
phc2sys[4370.492]: CLOCK_REALTIME phc offset       150 s2 freq   +9531 delay   5580
phc2sys[4371.492]: CLOCK_REALTIME phc offset         0 s2 freq   +9426 delay   5352
phc2sys[4372.493]: CLOCK_REALTIME phc offset       -32 s2 freq   +9394 delay   5592

Much smaller. Not touching anything. Just I stopped and run it again. I noticed that Linux system time (the clock as you see on your desktop) changed and the date changed to 1970. That is why I asked you why the previous results were so big, if I knew that the node clock must to synchronise to master and then take the exact time as the master.

Using ptp4l and phc2sys for IEEE 1588 I got really small offset such as -15,15 ns for ptp4l on the nodes and similar on the nodes for phc2sys.

Changing the master time to 2020, I got these results running phc2sys on my node, but this time, the Linux system time is kept in 1970.

nodo2@nodo2-N-A:~$ sudo phc2sys -a -rr --transportSpecific 0x1 -m
phc2sys[4900.297]: reconfiguring after port state change
phc2sys[4900.300]: selecting CLOCK_REALTIME for synchronization
phc2sys[4900.300]: selecting enp3s0 as the master clock
phc2sys[4900.300]: CLOCK_REALTIME phc offset    -59024 s0 freq   +9441 delay   5364
phc2sys[4901.301]: CLOCK_REALTIME phc offset    -59156 s1 freq   +9309 delay   5639
phc2sys[4902.301]: CLOCK_REALTIME phc offset       174 s2 freq   +9483 delay   5640
phc2sys[4903.301]: CLOCK_REALTIME phc offset       133 s2 freq   +9494 delay   5592
phc2sys[4904.302]: CLOCK_REALTIME phc offset       108 s2 freq   +9509 delay   5580
phc2sys[4905.302]: CLOCK_REALTIME phc offset        47 s2 freq   +9481 delay   5604
phc2sys[4906.303]: CLOCK_REALTIME phc offset        29 s2 freq   +9477 delay   5592
phc2sys[4907.303]: CLOCK_REALTIME phc offset        12 s2 freq   +9468 delay   5604
phc2sys[4908.303]: CLOCK_REALTIME phc offset         6 s2 freq   +9466 delay   5220
phc2sys[4909.304]: CLOCK_REALTIME phc offset       -22 s2 freq   +9440 delay   5544
[... cut ...]
phc2sys[4919.308]: CLOCK_REALTIME phc offset        65 s2 freq   +9498 delay   5604
phc2sys[4920.308]: CLOCK_REALTIME phc offset        13 s2 freq   +9466 delay   5640
phc2sys[4921.309]: CLOCK_REALTIME phc offset        16 s2 freq   +9473 delay   5615
phc2sys[4922.309]: CLOCK_REALTIME phc offset         5 s2 freq   +9466 delay   5604
phc2sys[4923.310]: CLOCK_REALTIME phc offset        -6 s2 freq   +9457 delay   5604
phc2sys[4924.310]: CLOCK_REALTIME phc offset       -12 s2 freq   +9449 delay   5604

Diego

vladimiroltean, 2020-06-22:
> Much smaller. Not touching anything. Just I stopped and run it again

Maybe just enough time has passed from the beginning of 1970 that clock stepping doesn't error out any longer?
There are bugs surrounding the clock stepping of CLOCK_REALTIME to a value close to 0 (Jan 1st 1970), and nobody is actively looking into fixing them, because nobody is in 1970 anymore.

vladimiroltean, 2020-06-22:
> Changing the master time to 2020, I got these results running phc2sys on my node, but this time, the Linux system time is kept in 1970.

How do you know that the Linux system time is kept in 1970? What does `phc_ctl CLOCK_REALTIME get` say?

diegotxegp, 2020-06-22:
As you said to me, I update LS1021ATSN time to current time in 2020:

**LS1021ATSN**

[root@LS1021ATSN ~] # phc_ctl CLOCK_REALTIME get
phc_ctl[8940.338]: clock time is 1592821706.513017528 or Mon Jun 22 10:28:26 2020

But the node is kept in 1970:
nodo2@nodo2-N-A:~$ phc_ctl CLOCK_REALTIME get
phc_ctl[8609.960]: clock time is 8876.479821394 or Thu Jan  1 03:27:56 1970

And the synchronisation with ptp4l between the LS1021ATSN platform and the node, and the phc2sys of the node show these results:

[root@LS1021ATSN ~] # ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg -m --tx_timestamp_timeout 20
ptp4l[5072.712]: selected /dev/ptp1 as PTP clock
ptp4l[5072.800]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[5072.880]: port 2: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[5072.882]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[5075.898]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[5075.898]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[5075.898]: assuming the grand master role
ptp4l[5076.603]: port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[5076.603]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[5076.604]: assuming the grand master role
ptp4l[5076.604]: assuming the grand master role
ptp4l[5084.536]: port 2: new foreign master aabbcc.fffe.00094a-1
ptp4l[5084.536]: selected best master clock aabbcc.fffe.00094a
ptp4l[5084.537]: assuming the grand master role
ptp4l[5084.537]: assuming the grand master role

**Node**

nodo2@nodo2-N-A:~$ sudo ptp4l -i enp3s0 -f /home/nodo2/linuxptp-2.0/configs/gPTP.cfg -m --tx_timestamp_timeout 20
[sudo] password for nodo2: 
ptp4l[4775.805]: selected /dev/ptp0 as PTP clock
ptp4l[4775.844]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[4775.844]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[4779.748]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[4779.750]: selected local clock aabbcc.fffe.00094a as best master
ptp4l[4779.752]: assuming the grand master role
ptp4l[4779.824]: port 1: new foreign master 00049f.fffe.ef0808-2
ptp4l[4781.825]: selected best master clock 00049f.fffe.ef0808
ptp4l[4781.825]: port 1: MASTER to UNCALIBRATED on RS_SLAVE
ptp4l[4782.097]: port 1: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[4782.849]: rms 1794 max 2384 freq +12407 +/- 1246 delay   234 +/-   0
ptp4l[4783.850]: rms  337 max  518 freq +13511 +/- 450 delay   233 +/-   0
ptp4l[4784.855]: rms  563 max  588 freq +14519 +/- 149 delay   233 +/-   0
ptp4l[4785.857]: rms  351 max  470 freq +14706 +/-  17 delay   234 +/-   0
ptp4l[4786.858]: rms  109 max  183 freq +14590 +/-  42 delay   234 +/-   0
ptp4l[4787.860]: rms   21 max   31 freq +14469 +/-  25 delay   235 +/-   0
ptp4l[4788.861]: rms   31 max   36 freq +14418 +/-   8 delay   236 +/-   0
ptp4l[4789.862]: rms   18 max   30 freq +14412 +/-   5 delay   236 +/-   0
ptp4l[4790.864]: rms    9 max   14 freq +14413 +/-   6 delay   236 +/-   0
ptp4l[4791.865]: rms    4 max    6 freq +14414 +/-   3 delay   236 +/-   0
ptp4l[4792.867]: rms    4 max    7 freq +14423 +/-   4 delay   236 +/-   0
ptp4l[4793.868]: rms    3 max    6 freq +14425 +/-   3 delay   236 +/-   0
ptp4l[4794.869]: rms    3 max    5 freq +14424 +/-   4 delay   236 +/-   0
ptp4l[4795.871]: rms    3 max    5 freq +14420 +/-   4 delay   236 +/-   0
ptp4l[4796.872]: rms    2 max    4 freq +14421 +/-   3 delay   236 +/-   0
ptp4l[4797.873]: rms    4 max    8 freq +14426 +/-   4 delay   236 +/-   0
ptp4l[4798.875]: rms    5 max    7 freq +14420 +/-   6 delay   236 +/-   0
ptp4l[4799.876]: rms    3 max    6 freq +14418 +/-   3 delay   236 +/-   0
ptp4l[4800.878]: rms    3 max    4 freq +14422 +/-   3 delay   236 +/-   0
ptp4l[4801.879]: rms    5 max    8 freq +14426 +/-   6 delay   235 +/-   0
ptp4l[4802.880]: rms    4 max    6 freq +14425 +/-   5 delay   235 +/-   0
ptp4l[4803.882]: rms    3 max    4 freq +14422 +/-   3 delay   235 +/-   0
ptp4l[4804.883]: rms    4 max    6 freq +14419 +/-   4 delay   236 +/-   0
ptp4l[4805.884]: rms    3 max    4 freq +14421 +/-   4 delay   236 +/-   0
ptp4l[4806.886]: rms    3 max    5 freq +14423 +/-   4 delay   236 +/-   0
ptp4l[4807.887]: rms    3 max    3 freq +14423 +/-   4 delay   236 +/-   0
[... 276 lines cut ...]
nodo2@nodo2-N-A:~$ sudo phc2sys -a -rr --transportSpecific 0x1 -m
[sudo] password for nodo2: 
phc2sys[8585.491]: reconfiguring after port state change
phc2sys[8585.493]: selecting CLOCK_REALTIME for synchronization
phc2sys[8585.495]: selecting enp3s0 as the master clock
phc2sys[8585.496]: CLOCK_REALTIME phc offset 1592814823157685387 s0 freq   +9442 delay   5364
phc2sys[8586.498]: CLOCK_REALTIME phc offset 1592814823157685530 s1 freq   +9585 delay   5628
phc2sys[8587.498]: CLOCK_REALTIME phc offset       -12 s2 freq   +9573 delay   5364
phc2sys[8588.499]: CLOCK_REALTIME phc offset        42 s2 freq   +9623 delay   5604
phc2sys[8589.499]: CLOCK_REALTIME phc offset         7 s2 freq   +9601 delay   5604
phc2sys[8590.500]: CLOCK_REALTIME phc offset        24 s2 freq   +9620 delay   5640
phc2sys[8591.500]: CLOCK_REALTIME phc offset       -22 s2 freq   +9581 delay   5604
phc2sys[8592.501]: CLOCK_REALTIME phc offset        11 s2 freq   +9607 delay   5604
phc2sys[8593.501]: CLOCK_REALTIME phc offset       -19 s2 freq   +9581 delay   5244
phc2sys[8594.501]: CLOCK_REALTIME phc offset       -12 s2 freq   +9582 delay   5076
[... cut ...]
phc2sys[8598.502]: CLOCK_REALTIME phc offset       -27 s2 freq   +9572 delay   5592
phc2sys[8599.503]: CLOCK_REALTIME phc offset         8 s2 freq   +9599 delay   5604
phc2sys[8600.503]: CLOCK_REALTIME phc offset         0 s2 freq   +9594 delay   5604

According what I know, both ptp4l results and phc2sys results look well, however master is in 2020 and the node in 1970.

Take a look this line in ptp4l of the node "ptp4l[5081.012]: subscriber 000000.0000.000000-1 timed out". It appears suddenly in ptp4l of the node, just once.

vladimiroltean, 2020-06-22:
Assuming you have phc2sys and ptp4l on all nodes, the propagation chain of the time is like this:
Master's CLOCK_REALTIME -> master's PHC (/dev/ptp0 or similar) -> slave's PHC -> slave's CLOCK_REALTIME.
So, if one of your nodes has a CLOCK_REALTIME is in 1970, it means one of 2 things:
- You are not synchronizing its CLOCK_REALTIME to the PHC, or
- The PHC is in 1970.

If the PHC is in 1970, it means one of 2 things:
- You are not synchronizing it to the master using ptp4l
- The master is in 1970.

I hope you can see where this is going. So the clock propagation tree is interrupted somewhere.

vladimiroltean, 2020-06-22:
> Take a look this line in ptp4l of the node "ptp4l[5081.012]: subscriber 000000.0000.000000-1 timed out". It appears suddenly in ptp4l of the node, just once.

It means ptp4l has detected that one phc2sys process has been closed.

diegotxegp, 2020-06-22:
> > Take a look this line in ptp4l of the node "ptp4l[5081.012]: subscriber 000000.0000.000000-1 timed out". It appears suddenly in ptp4l of the node, just once.
> 
> It means ptp4l has detected that one phc2sys process has been closed.

Yes, it makes sense. I stopped it.

diegotxegp, 2020-06-22:
> Assuming you have phc2sys and ptp4l on all nodes, the propagation chain of the time is like this:
> Master's CLOCK_REALTIME -> master's PHC (/dev/ptp0 or similar) -> slave's PHC -> slave's [...]

I understand. Let me check it.

diegotxegp, 2020-06-22:
Apart from the current time of 2020 is missed somewhere. Do you think that values of ptp4l and phc2sys in LS1021ATSN and the node are acceptable? RMS, delays, phc2sys offset...

vladimiroltean, 2020-06-22:
I don't like the phc2sys offsets, and now I see why.

Could you please apply this patch:

```diff
commit 09cf2f44b2c68b8e2c830b7375d5a36c45a9566e
Author: Vladimir Oltean <vladimir.oltean@nxp.com>
Date:   Mon Jun 22 14:52:43 2020 +0300

    package/nxp/qoriq-ptp4l: depend on Linux kernel
    
    As explained in commit 24f79f6ddb05 ("package/nxp/qoriq-ptp4l: use
    compiled kernel headers"), linuxptp needs to use the compiled kernel
    headers, and for that, the kernel needs to be compiled first. So,
    linuxptp should have the kernel as a dependency.
    
    Not sure how come this dependency got missed.
    
    Fixes: 24f79f6ddb05 ("package/nxp/qoriq-ptp4l: use compiled kernel headers")
    Signed-off-by: Vladimir Oltean <vladimir.oltean@nxp.com>

diff --git a/package/nxp/qoriq-ptp4l/qoriq-ptp4l.mk b/package/nxp/qoriq-ptp4l/qoriq-ptp4l.mk
index 0856bdbd8cfc..3908b825a65f 100644
--- a/package/nxp/qoriq-ptp4l/qoriq-ptp4l.mk
+++ b/package/nxp/qoriq-ptp4l/qoriq-ptp4l.mk
@@ -9,6 +9,7 @@ QORIQ_PTP4L_SITE = $(call qstrip,$(BR2_PACKAGE_QORIQ_PTP4L_SITE))
 QORIQ_PTP4L_SITE_METHOD = git
 QORIQ_PTP4L_LICENSE = GPL2
 QORIQ_PTP4L_LICENSE_FILES = COPYING
+QORIQ_PTP4L_DEPENDENCIES = linux
 
 QORIQ_PTP4L_MAKE_OPTS = \
 	prefix=/usr \
```

And then rebuild:
```
make nxp_ls1021atsn_defconfig
make qoriq-ptp4l-dirclean world
```

diegotxegp, 2020-06-22:
Ok. If compiling it again, we solve those offset, fine.

diegotxegp, 2020-06-22:
Sorry Vladimir, but I do not find your commit 09cf2f44b2c68b8e2c830b7375d5a36c45a9566e

vladimiroltean, 2020-06-22:
The procedure to apply a patch is:
1. Go to the openil folder.
2. Copy the patch content from here, paste it to a new text file called `phc2sys-kernel-headers-bugfix.patch`.
3. Apply it using `git am phc2sys-kernel-headers-bugfix.patch`.
4. Rebuild the linuxptp package using the steps above.

diegotxegp, 2020-06-22:
Once compiling done, I have these results for LS1021ATSN and one node with ptp4l and phc2sys.

**LS1021ATSN**

[root@LS1021ATSN ~] # ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg -m --tx_timestamp_timeout 20
ptp4l[1212.010]: selected /dev/ptp1 as PTP clock
ptp4l[1212.090]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1212.170]: port 2: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1212.171]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1215.366]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1215.367]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[1215.367]: assuming the grand master role
ptp4l[1215.654]: port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1215.654]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[1215.654]: assuming the grand master role
ptp4l[1215.654]: assuming the grand master role
ptp4l[1248.812]: port 2: new foreign master aabbcc.fffe.00094a-1
ptp4l[1248.813]: selected best master clock aabbcc.fffe.00094a
ptp4l[1248.813]: assuming the grand master role
ptp4l[1248.813]: assuming the grand master role

[root@LS1021ATSN ~] # phc2sys -a -rr --transportSpecific 0x1 -m
phc2sys[1223.256]: reconfiguring after port state change
phc2sys[1223.261]: selecting swp3 for synchronization
phc2sys[1223.262]: skipping swp2: swp3 has the same clock and is already selected
phc2sys[1223.263]: selecting CLOCK_REALTIME as the master clock
phc2sys[1223.265]: swp3 sys offset      4064 s0 freq  -24839 delay 170400
phc2sys[1224.268]: swp3 sys offset      5760 s2 freq  -23148 delay 158720
phc2sys[1225.269]: swp3 sys offset      3256 s2 freq  -19892 delay 162240
phc2sys[1226.271]: swp3 sys offset      3880 s2 freq  -18291 delay 157440
phc2sys[1227.272]: swp3 sys offset      4136 s2 freq  -16871 delay 151760
phc2sys[1228.274]: swp3 sys offset      -368 s2 freq  -20134 delay 160560
phc2sys[1229.275]: swp3 sys offset      -880 s2 freq  -20756 delay 159520
phc2sys[1230.276]: swp3 sys offset      -648 s2 freq  -20788 delay 161840
phc2sys[1231.278]: swp3 sys offset       352 s2 freq  -19983 delay 164400
phc2sys[1232.279]: swp3 sys offset      1648 s2 freq  -18581 delay 161200
phc2sys[1233.281]: swp3 sys offset      3912 s2 freq  -15823 delay 157840
phc2sys[1234.282]: swp3 sys offset      2040 s2 freq  -16521 delay 161040
phc2sys[1235.284]: swp3 sys offset      2480 s2 freq  -15469 delay 159920
phc2sys[1236.285]: swp3 sys offset      4656 s2 freq  -12549 delay 158240
phc2sys[1237.286]: swp3 sys offset      4112 s2 freq  -11696 delay 161920
phc2sys[1238.288]: swp3 sys offset      7704 s2 freq   -6871 delay 158160
phc2sys[1239.289]: swp3 sys offset      3104 s2 freq   -9160 delay 162720
phc2sys[1240.291]: swp3 sys offset      4536 s2 freq   -6796 delay 158880
phc2sys[1241.292]: swp3 sys offset      1616 s2 freq   -8356 delay 164480
phc2sys[1242.294]: swp3 sys offset      4768 s2 freq   -4719 delay 158640
phc2sys[1243.295]: swp3 sys offset      7416 s2 freq    -640 delay 156720
phc2sys[1244.297]: swp3 sys offset      4712 s2 freq   -1120 delay 160400
phc2sys[1245.298]: swp3 sys offset      4608 s2 freq    +190 delay 160560
phc2sys[1246.299]: swp3 sys offset     16248 s2 freq  +13212 delay 142320
phc2sys[1247.301]: swp3 sys offset      3856 s2 freq   +5695 delay 158320
phc2sys[1248.302]: swp3 sys offset    -24768 s2 freq  -21772 delay 162960
phc2sys[1249.304]: swp3 sys offset    -22168 s2 freq  -26603 delay 160160
phc2sys[1250.305]: swp3 sys offset     13416 s2 freq   +2331 delay 159360
phc2sys[1251.306]: swp3 sys offset     10792 s2 freq   +3732 delay 158240
phc2sys[1252.308]: swp3 sys offset      6528 s2 freq   +2705 delay 161280
phc2sys[1253.309]: swp3 sys offset      6144 s2 freq   +4280 delay 162400

**Node**

nodo2@nodo2-N-A:~$ sudo ptp4l -i enp3s0 -f /home/nodo2/linuxptp-2.0/configs/gPTP.cfg -m --tx_timestamp_timeout 20
ptp4l[1222.534]: selected /dev/ptp0 as PTP clock
ptp4l[1222.592]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1222.593]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[1225.720]: port 1: new foreign master 00049f.fffe.ef0808-2
ptp4l[1225.869]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1225.869]: selected local clock aabbcc.fffe.00094a as best master
ptp4l[1225.869]: assuming the grand master role
ptp4l[1227.720]: selected best master clock 00049f.fffe.ef0808
ptp4l[1227.721]: port 1: MASTER to UNCALIBRATED on RS_SLAVE
[... 23 lines cut ...]
phc2sys[1238.969]: CLOCK_REALTIME phc offset     11201 s2 freq   +7406 delay   5388
phc2sys[1239.969]: CLOCK_REALTIME phc offset    -48488 s2 freq  -48923 delay   5604
phc2sys[1240.969]: CLOCK_REALTIME phc offset    -22678 s2 freq  -37659 delay   5268
phc2sys[1241.970]: CLOCK_REALTIME phc offset      9430 s2 freq  -12355 delay   5388
phc2sys[1242.970]: CLOCK_REALTIME phc offset     18952 s2 freq      -4 delay   5556
phc2sys[1243.970]: CLOCK_REALTIME phc offset     41491 s2 freq  +28221 delay   4908
phc2sys[1244.971]: CLOCK_REALTIME phc offset     15434 s2 freq  +14611 delay   5388
phc2sys[1245.971]: CLOCK_REALTIME phc offset    -60072 s2 freq  -56265 delay   5088
phc2sys[1246.971]: CLOCK_REALTIME phc offset    -29844 s2 freq  -44058 delay   5556
phc2sys[1247.971]: CLOCK_REALTIME phc offset     10365 s2 freq  -12803 delay   5388
phc2sys[1248.972]: CLOCK_REALTIME phc offset     50983 s2 freq  +30925 delay   5556
phc2sys[1249.972]: CLOCK_REALTIME phc offset     25907 s2 freq  +21144 delay   5616
[... cut ...]
phc2sys[1255.974]: CLOCK_REALTIME phc offset     15132 s2 freq  +15520 delay   5388
phc2sys[1256.975]: CLOCK_REALTIME phc offset    -60804 s2 freq  -55877 delay   5556
phc2sys[1257.975]: CLOCK_REALTIME phc offset    -29323 s2 freq  -42637 delay   5628
phc2sys[1258.976]: CLOCK_REALTIME phc offset      7040 s2 freq  -15071 delay   5256
phc2sys[1259.976]: CLOCK_REALTIME phc offset     49934 s2 freq  +29935 delay   5244
phc2sys[1260.976]: CLOCK_REALTIME phc offset     27125 s2 freq  +22106 delay   4968

They have similar values as before compiling.

vladimiroltean, 2020-06-22:
Yes, nothing improved. Do you still have the build log?

diegotxegp, 2020-06-22:
user@user-VirtualBox:~/openil/output/images$ git log --oneline -10
118b7c5 Merge pull request #144 in DNIND/openil from ~NXA08424/openil:2020.02-dev-0529 to 2020.02-dev
94f52db ubuntu: remove to the proxy for wget and check the rootfs path
30713d5 Merge pull request #142 in DNIND/openil from ~NXA08424/openil:2020.02-dev-0526 to 2020.02-dev
ccc212c package/nxp/qoriq-tsntool: bump to version OpenIL-v1.8-202005
858285a Merge pull request #139 in DNIND/openil from ~NXA08424/openil:2020.02-dev-0524 to 2020.02-dev
e15c5f0 package/qoriq-ptp4l: pull upstream patch for removing snmp4lptp
87fdad7 Merge pull request #138 in DNIND/openil from ~NXF46132/openil:2020.02-dev to 2020.02-dev
192f03d package/qoriq-libnfc-nci: modify the version of libnfc-nci to R2.4
47d22e5 Merge pull request #137 in DNIND/openil from ~NXF24402/openil:2020.02-dev-0522 to 2020.02-dev
a3ab31e Add file sysrepo-tsn.service

vladimiroltean, 2020-06-22:
And the build log?

diegotxegp, 2020-06-22:
Here your are the log.

user@user-VirtualBox:~/openil$ make qoriq-ptp4l-dirclean world
rm -Rf /home/diegotxe/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d
/usr/bin/make -j1 O=/home/diegotxe/openil/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
WARNING: no hash file for qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d.tar.gz
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Extracting
gzip -d -c /home/diegotxe/openil/dl/qoriq-ptp4l/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d.tar.gz | tar --strip-components=1 -C /home/diegotxe/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d   -xf -
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Patching

Applying 0001-Remove-the-unfinished-SNMP-code.patch using patch: 
patching file makefile
patching file snmp4lptp.8
patching file snmp4lptp.c
patching file snmp4lptp_mib.h
patching file snmpflg.sh
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Configuring
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Building
export PKG_CONFIG_SYSROOT_DIR="/home/diegotxe/openil/output/host/arm-buildroot-linux-gnueabihf/sysroot"; export KBUILD_OUTPUT="/home/diegotxe/openil/output/build/linux-"OpenIL-v1.8-linux-202005""; PATH="/home/diegotxe/openil/output/host/bin:/home/diegotxe/openil/output/host/sbin:/home/diegotxe/bin:/home/diegotxe/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" /usr/bin/make -j1 prefix=/usr CC="/home/diegotxe/openil/output/host/bin/arm-linux-gnueabihf-gcc" CROSS_COMPILE="/home/diegotxe/openil/output/host/bin/arm-linux-gnueabihf-" -C /home/diegotxe/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d
DEPEND timemaster.c
DEPEND sysoff.c
DEPEND pmc_common.c
DEPEND pmc.c
DEPEND phc_ctl.c
DEPEND phc2sys.c
DEPEND nsm.c
DEPEND hwstamp_ctl.c
DEPEND version.c
DEPEND util.c
DEPEND unicast_service.c
DEPEND unicast_fsm.c
DEPEND unicast_client.c
DEPEND uds.c
DEPEND udp6.c
DEPEND udp.c
DEPEND tsproc.c
DEPEND transport.c
DEPEND tlv.c
DEPEND telecom.c
DEPEND tc.c
DEPEND stats.c
DEPEND sk.c
DEPEND servo.c
DEPEND rtnl.c
DEPEND raw.c
DEPEND p2p_tc.c
DEPEND ptp4l.c
DEPEND print.c
DEPEND pqueue.c
DEPEND port_signaling.c
DEPEND port.c
DEPEND pi.c
DEPEND phc.c
DEPEND nullf.c
DEPEND ntpshm.c
DEPEND msg.c
DEPEND mmedian.c
DEPEND mave.c
DEPEND linreg.c
DEPEND hash.c
DEPEND fsm.c
DEPEND filter.c
DEPEND fault.c
DEPEND e2e_tc.c
DEPEND designated_fsm.c
DEPEND config.c
DEPEND clockcheck.c
DEPEND clockadj.c
DEPEND clock.c
DEPEND bmc.c
[... 190 lines cut ...]
printf '   	(cd /home/diegotxe/openil/output/build/buildroot-fs/tar/target; find -print0 | LC_ALL=C sort -z | tar  --pax-option=exthdr.name=%%d/PaxHeaders/%%f,atime:=0,ctime:=0 -cf /home/diegotxe/openil/output/images/rootfs.tar --null --xattrs-include='\''*'\'' --no-recursion -T - --numeric-owner)\n' >> /home/diegotxe/openil/output/build/buildroot-fs/tar/fakeroot
chmod a+x /home/diegotxe/openil/output/build/buildroot-fs/tar/fakeroot
PATH="/home/diegotxe/openil/output/host/bin:/home/diegotxe/openil/output/host/sbin:/home/diegotxe/bin:/home/diegotxe/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" FAKEROOTDONTTRYCHOWN=1 /home/diegotxe/openil/output/host/bin/fakeroot -- /home/diegotxe/openil/output/build/buildroot-fs/tar/fakeroot
rootdir=/home/diegotxe/openil/output/build/buildroot-fs/tar/target
table='/home/diegotxe/openil/output/build/buildroot-fs/full_devices_table.txt'
ln -snf /home/diegotxe/openil/output/host/arm-buildroot-linux-gnueabihf/sysroot /home/diegotxe/openil/output/staging
>>>   Executing post-image script board/nxp/ls1021atsn/post-image.sh

INFO: cmd: "mkdir -p "/home/diegotxe/openil/output/build/genimage.tmp"" (stderr):
INFO: cmd: "rm -rf "/home/diegotxe/openil/output/build/genimage.tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/home/diegotxe/openil/output/build/genimage.tmp"" (stderr):
INFO: cmd: "cp -a "/home/diegotxe/openil/output/target" "/home/diegotxe/openil/output/build/genimage.tmp/root"" (stderr):
INFO: cmd: "find '/home/diegotxe/openil/output/build/genimage.tmp/root' -depth -type d -printf '%P\0' | xargs -0 -I {} touch -r '/home/diegotxe/openil/output/target/{}' '/home/diegotxe/openil/output/build/genimage.tmp/root/{}'" (stderr):
INFO: cmd: "mkdir -p "/home/diegotxe/openil/output/images"" (stderr):
INFO: vfat(boot.vfat): cmd: "dd if=/dev/zero of="/home/diegotxe/openil/output/images/boot.vfat" seek=268435456 count=0 bs=1 2>/dev/null" (stderr):
INFO: vfat(boot.vfat): cmd: "mkdosfs   '/home/diegotxe/openil/output/images/boot.vfat'" (stderr):
INFO: vfat(boot.vfat): adding file 'ls1021a-tsn.dtb' as 'ls1021a-tsn.dtb' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/diegotxe/openil/output/images/boot.vfat' '/home/diegotxe/openil/output/images/ls1021a-tsn.dtb' '::'" (stderr):
INFO: vfat(boot.vfat): adding file 'uImage' as 'uImage' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/diegotxe/openil/output/images/boot.vfat' '/home/diegotxe/openil/output/images/uImage' '::'" (stderr):
INFO: hdimage(sdcard.img): adding partition 'u-boot' from 'u-boot-with-spl-pbl.bin' ...
INFO: hdimage(sdcard.img): adding partition 'u-boot-environment' from 'uboot-env.bin' ...
INFO: hdimage(sdcard.img): adding partition 'boot' (in MBR) from 'boot.vfat' ...
INFO: hdimage(sdcard.img): adding partition 'rootfs' (in MBR) from 'rootfs.ext2' ...
INFO: hdimage(sdcard.img): writing MBR

vladimiroltean, 2020-06-22:
The build log looks fine.
Dumb question: what does `ls -la /usr/sbin/phc2sys` show on the board? Is it a recent timestamp (corresponding to a recently rebuilt binary) or an old timestamp?

diegotxegp, 2020-06-22:
> Dumb question: what does `ls -la /usr/sbin/phc2sys` show on the board?

nodo2@nodo2-N-A:~$ ls -la /usr/sbin/phc2sys
ls: cannot access '/usr/sbin/phc2sys': No such file or directory

>Is it a recent timestamp (corresponding to a recently rebuilt binary) or an old timestamp?

If you are talking to me if my SD card image is recent, yes. I compiled the 1.8 OpenIL version.

vladimiroltean, 2020-06-22:
> nodo2@nodo2-N-A:~$ ls -la /usr/sbin/phc2sys
> ls: cannot access '/usr/sbin/phc2sys': No such file or directory

Why are you giving me this, I don't care about phc2sys about the Intel PC, I care about phc2sys on the OpenIL board. The point was to check if this binary got updated after rebuild or not.

diegotxegp, 2020-06-22:
Oh yes, yes. I was wrong.

[root@LS1021ATSN ~] # ls -la /usr/sbin/phc2sys
-rwxr-xr-x    1 root     root        119368 Jun 22  2020 /usr/sbin/phc2sys

vladimiroltean, 2020-06-22:
Will come back to you a little bit later (a few hours or so).

diegotxegp, 2020-06-22:
OK. Thank you

diegotxegp, 2020-06-23:
> Assuming you have phc2sys and ptp4l on all nodes, the propagation chain of the time is like this:
> Master's CLOCK_REALTIME -> master's PHC (/dev/ptp0 or similar) -> slave's PHC -> slave's [...]

I saw my error. I was not taking time from CLOCK_REALTIME from master with phc2sys. That is why I got a time in 1970 in the slave, because the PHC was in 1970 after reboot.

Apart from that, CLOCK_REALTIME was on February 7th 2020 after reboot. I thought that both CLOCK_REALTIME and PHC come back to 1970 after every reboot. Not only reboot. Even I unplugged the platform from energy. It seemed to me curious.

A second question is: How can I keep a new IP address at br0 at the platform? I prefer connecting to the LS1021ATSN platform via SSH, but initially br0 IP address is empty. So, I connect first via PuTTy for changing br0 to an IP address and then I can connect via SSH.

And also I would want to ask you about if you could see something to fix those bif offsets on phc2sys.

Thank you very much, Vladimir, for your help. I appreciate it.

Diego

vladimiroltean, 2020-06-23:
I honestly don't know why the patch worked for me, because it _wasn't_ supposed to work....
So currently linuxptp searching for compiled kernel headers is a bit of a mess, please try this patch instead:
```
From 2686f1a2d773af778cd9fe90b149441aaf8f8db8 Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <olteanv@gmail.com>
Date: Tue, 23 Jun 2020 12:14:52 +0300
Subject: [PATCH RFC] phc2sys: provide missing kernel headers for sysoff
 functionality

Currently it is very finicky to deploy linuxptp in an automated build
system and make KBUILD_OUTPUT pick up the output of "make
headers_install" in order for the application to make full use of the
features exposed by the runtime kernel. And the toolchain/libc will
almost certainly never contain recent enough kernel headers to be of any
use here. And there's no good reason for that: the application can probe
at runtime for the sysoff methods supported by the kernel anyway.

So let's provide the kernel definitions for sysoff, sysoff_precise and
sysoff_extended, such that SYSOFF_COMPILE_TIME_MISSING is not something
that will bother us any longer.

---
 missing.h | 48 ++++++++++++++++++++++++++++++++++++++++++++++++
 sysoff.c  | 27 +--------------------------
 sysoff.h  |  2 +-
 3 files changed, 50 insertions(+), 27 deletions(-)

diff --git a/missing.h b/missing.h
index bc708cbbbe8a..c77d381e76b4 100644
--- a/missing.h
+++ b/missing.h
@@ -97,6 +97,54 @@ struct compat_ptp_clock_caps {
 
 #endif /*LINUX_VERSION_CODE < 5.8*/
 
+#ifndef PTP_SYS_OFFSET
+
+#define PTP_SYS_OFFSET     _IOW(PTP_CLK_MAGIC, 5, struct ptp_sys_offset)
+
+struct ptp_sys_offset {
+	unsigned int n_samples; /* Desired number of measurements. */
+	unsigned int rsv[3];    /* Reserved for future use. */
+	/*
+	 * Array of interleaved system/phc time stamps. The kernel
+	 * will provide 2*n_samples + 1 time stamps, with the last
+	 * one as a system time stamp.
+	 */
+	struct ptp_clock_time ts[2 * PTP_MAX_SAMPLES + 1];
+};
+
+#endif /* PTP_SYS_OFFSET */
+
+#ifndef PTP_SYS_OFFSET_PRECISE
+
+#define PTP_SYS_OFFSET_PRECISE \
+	_IOWR(PTP_CLK_MAGIC, 8, struct ptp_sys_offset_precise)
+
+struct ptp_sys_offset_precise {
+	struct ptp_clock_time device;
+	struct ptp_clock_time sys_realtime;
+	struct ptp_clock_time sys_monoraw;
+	unsigned int rsv[4];    /* Reserved for future use. */
+};
+
+#endif /* PTP_SYS_OFFSET_PRECISE */
+
+#ifndef PTP_SYS_OFFSET_EXTENDED
+
+#define PTP_SYS_OFFSET_EXTENDED \
+	_IOWR(PTP_CLK_MAGIC, 9, struct ptp_sys_offset_extended)
+
+struct ptp_sys_offset_extended {
+	unsigned int n_samples; /* Desired number of measurements. */
+	unsigned int rsv[3];    /* Reserved for future use. */
+	/*
+	 * Array of [system, phc, system] time stamps. The kernel will provide
+	 * 3*n_samples time stamps.
+	 */
+	struct ptp_clock_time ts[PTP_MAX_SAMPLES][3];
+};
+
+#endif /* PTP_SYS_OFFSET_EXTENDED */
+
 #ifndef PTP_PIN_SETFUNC
 
 enum ptp_pin_function {
diff --git a/sysoff.c b/sysoff.c
index 05d2ed617780..03cdb87dcab2 100644
--- a/sysoff.c
+++ b/sysoff.c
@@ -27,8 +27,6 @@
 
 #define NS_PER_SEC 1000000000LL
 
-#ifdef PTP_SYS_OFFSET
-
 static int64_t pctns(struct ptp_clock_time *t)
 {
 	return t->sec * NS_PER_SEC + t->nsec;
@@ -42,7 +40,6 @@ static struct {
 
 static int sysoff_precise(int fd, int64_t *result, uint64_t *ts)
 {
-#ifdef PTP_SYS_OFFSET_PRECISE
 	struct ptp_sys_offset_precise pso;
 	memset(&pso, 0, sizeof(pso));
 	if (ioctl(fd, PTP_SYS_OFFSET_PRECISE, &pso)) {
@@ -52,9 +49,6 @@ static int sysoff_precise(int fd, int64_t *result, uint64_t *ts)
 	*result = pctns(&pso.sys_realtime) - pctns(&pso.device);
 	*ts = pctns(&pso.sys_realtime);
 	return SYSOFF_PRECISE;
-#else
-	return SYSOFF_COMPILE_TIME_MISSING;
-#endif
 }
 
 static void insertion_sort(int length, int64_t interval, int64_t offset, uint64_t ts)
@@ -100,7 +94,6 @@ static int64_t sysoff_estimate(struct ptp_clock_time *pct, int extended,
 static int sysoff_extended(int fd, int n_samples,
 			   int64_t *result, uint64_t *ts, int64_t *delay)
 {
-#ifdef PTP_SYS_OFFSET_EXTENDED
 	struct ptp_sys_offset_extended pso;
 	memset(&pso, 0, sizeof(pso));
 	pso.n_samples = n_samples;
@@ -110,9 +103,6 @@ static int sysoff_extended(int fd, int n_samples,
 	}
 	*result = sysoff_estimate(&pso.ts[0][0], 1, n_samples, ts, delay);
 	return SYSOFF_EXTENDED;
-#else
-	return SYSOFF_COMPILE_TIME_MISSING;
-#endif
 }
 
 static int sysoff_basic(int fd, int n_samples,
@@ -141,7 +131,7 @@ int sysoff_measure(int fd, int method, int n_samples,
 	case SYSOFF_BASIC:
 		return sysoff_basic(fd, n_samples, result, ts, delay);
 	}
-	return SYSOFF_COMPILE_TIME_MISSING;
+	return SYSOFF_RUN_TIME_MISSING;
 }
 
 int sysoff_probe(int fd, int n_samples)
@@ -165,18 +155,3 @@ int sysoff_probe(int fd, int n_samples)
 
 	return SYSOFF_RUN_TIME_MISSING;
 }
-
-#else /* !PTP_SYS_OFFSET */
-
-int sysoff_measure(int fd, int method, int n_samples,
-		   int64_t *result, uint64_t *ts, int64_t *delay)
-{
-	return SYSOFF_COMPILE_TIME_MISSING;
-}
-
-int sysoff_probe(int fd, int n_samples)
-{
-	return SYSOFF_COMPILE_TIME_MISSING;
-}
-
-#endif /* PTP_SYS_OFFSET */
diff --git a/sysoff.h b/sysoff.h
index 79d22908ce48..e4de91948231 100644
--- a/sysoff.h
+++ b/sysoff.h
@@ -19,9 +19,9 @@
  */
 
 #include <stdint.h>
+#include "missing.h"
 
 enum {
-	SYSOFF_COMPILE_TIME_MISSING = -2,
 	SYSOFF_RUN_TIME_MISSING = -1,
 	SYSOFF_PRECISE,
 	SYSOFF_EXTENDED,
-- 
2.25.1

```
Save it in `package/nxp/qoriq-ptp4l/0001-phc2sys-provide-missing-kernel-headers-for-sysoff-fu.patch` and then rebuild using the same `make qoriq-ptp4l-dirclean world` command as before.
Thanks,
-Vladimir

vladimiroltean, 2020-06-23:
> A second question is: How can I keep a new IP address at br0 at the platform? I prefer connecting to the LS1021ATSN platform via SSH, but initially br0 IP address is empty. So, I connect first via PuTTy for changing br0 to an IP address and then I can connect via SSH.

In `/etc/systemd/network/br0.network` on the LS1021A-TSN board, you can see that it has this content:
```
DHCP=ipv4
```
So it gets an IP over DHCP.
You have 2 options:
- If you have a DHCP server, record the MAC address of the br0 interface, and create a DHCP reservation for it, such that the IP address given by DHCP is constant.
- Assign a static IP address by following the systemd-networkd documentation available here: https://www.freedesktop.org/software/systemd/man/systemd.network.html.

diegotxegp, 2020-06-23:
> I honestly don't know why the patch worked for me, because it _wasn't_ supposed to work....
> So currently linuxptp searching for compiled kernel headers is a bit of a mess, please try this patch instead:
> 
> ```diff
> From 2686f1a2d773af778cd9fe90b149441aaf8f8db8 Mon Sep 17 00:00:00 2001
> From: Vladimir Oltean <olteanv@gmail.com>
> Date: Tue, 23 Jun 2020 12:14:52 +0300
> Subject: [PATCH RFC] phc2sys: provide missing kernel headers for sysoff
>  functionality
> 
> Currently it is very finicky to deploy linuxptp in an automated build
> system and make KBUILD_OUTPUT pick up the output of "make
> headers_install" in order for the application to make full use of the
> features exposed by the runtime kernel. And the toolchain/libc will
> almost certainly never contain recent enough kernel headers to be of any
> use here. And there's no good reason for that: the application can probe
> at runtime for the sysoff methods supported by the kernel anyway.
> 
> So let's provide the kernel definitions for sysoff, sysoff_precise and
> sysoff_extended, such that SYSOFF_COMPILE_TIME_MISSING is not something
> that will bother us any longer.
> 
> Signed-off-by: Vladimir Oltean <olteanv@gmail.com>
> ---
>  missing.h | 48 ++++++++++++++++++++++++++++++++++++++++++++++++
>  sysoff.c  | 27 +--------------------------
>  sysoff.h  |  2 +-
>  3 files changed, 50 insertions(+), 27 deletions(-)
> 
> diff --git a/missing.h b/missing.h
> index bc708cbbbe8a..c77d381e76b4 100644
> --- a/missing.h
> +++ b/missing.h
> @@ -97,6 +97,54 @@ struct compat_ptp_clock_caps {
>  
>  #endif /*LINUX_VERSION_CODE < 5.8*/
>  
> +#ifndef PTP_SYS_OFFSET
> +
> +#define PTP_SYS_OFFSET     _IOW(PTP_CLK_MAGIC, 5, struct ptp_sys_offset)
> +
> +struct ptp_sys_offset {
> +	unsigned int n_samples; /* Desired number of measurements. */
> +	unsigned int rsv[3];    /* Reserved for future use. */
> +	/*
> +	 * Array of interleaved system/phc time stamps. The kernel
> +	 * will provide 2*n_samples + 1 time stamps, with the last
> +	 * one as a system time stamp.
> +	 */
> +	struct ptp_clock_time ts[2 * PTP_MAX_SAMPLES + 1];
> +};
> +
> +#endif /* PTP_SYS_OFFSET */
> +
> +#ifndef PTP_SYS_OFFSET_PRECISE
> +
> +#define PTP_SYS_OFFSET_PRECISE \
> +	_IOWR(PTP_CLK_MAGIC, 8, struct ptp_sys_offset_precise)
> +
> +struct ptp_sys_offset_precise {
> +	struct ptp_clock_time device;
> +	struct ptp_clock_time sys_realtime;
> +	struct ptp_clock_time sys_monoraw;
> +	unsigned int rsv[4];    /* Reserved for future use. */
> +};
> +
> +#endif /* PTP_SYS_OFFSET_PRECISE */
> +
> +#ifndef PTP_SYS_OFFSET_EXTENDED
> +
[... 122 lines cut ...]
/usr/bin/make -j1 O=/home/diegotxe/openil/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
WARNING: no hash file for qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d.tar.gz
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Extracting
gzip -d -c /home/diegotxe/openil/dl/qoriq-ptp4l/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d.tar.gz | tar --strip-components=1 -C /home/diegotxe/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d   -xf -
>>> qoriq-ptp4l eee6d0fb995dddf27acde2addfb1fe89b71ab82d Patching

Applying 0001-Remove-the-unfinished-SNMP-code.patch using patch: 
patching file makefile
patching file snmp4lptp.8
patching file snmp4lptp.c
patching file snmp4lptp_mib.h
patching file snmpflg.sh

Applying 0001-phc2sys-provide-missing-kernel-headers-for-sysoff-fu.patch using patch: 
patching file missing.h
Hunk #1 FAILED at 97.
1 out of 1 hunk FAILED -- saving rejects to file missing.h.rej
patching file sysoff.c
Hunk #7 FAILED at 155.
1 out of 7 hunks FAILED -- saving rejects to file sysoff.c.rej
patching file sysoff.h
package/pkg-generic.mk:228: fallo en las instrucciones para el objetivo '/home/diegotxe/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/.stamp_patched'
make[1]: *** [/home/diegotxe/openil/output/build/qoriq-ptp4l-eee6d0fb995dddf27acde2addfb1fe89b71ab82d/.stamp_patched] Error 1
Makefile:84: fallo en las instrucciones para el objetivo '_all'
make: *** [_all] Error 2`

vladimiroltean, 2020-06-23:
Yes, sorry, I had made the patch against mainline linuxptp and not against the OpenIL version, here it is again:
```diff
From 9338815d09ce73aa8d038682fd5e218c9b9bde84 Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <olteanv@gmail.com>
Date: Tue, 23 Jun 2020 12:14:52 +0300
Subject: [PATCH OpenIL] phc2sys: provide missing kernel headers for sysoff
 functionality

Currently it is very finicky to deploy linuxptp in an automated build
system and make KBUILD_OUTPUT pick up the output of "make
headers_install" in order for the application to make full use of the
features exposed by the runtime kernel. And the toolchain/libc will
almost certainly never contain recent enough kernel headers to be of any
use here. And there's no good reason for that: the application can probe
at runtime for the sysoff methods supported by the kernel anyway.

So let's provide the kernel definitions for sysoff, sysoff_precise and
sysoff_extended, such that SYSOFF_COMPILE_TIME_MISSING is not something
that will bother us any longer.

---
 missing.h | 52 +++++++++++++++++++++++++++++++++++++++++++++++++++-
 sysoff.c  | 27 +--------------------------
 sysoff.h  |  2 +-
 3 files changed, 53 insertions(+), 28 deletions(-)

diff --git a/missing.h b/missing.h
index 8f92079d4b3c..49f390bdf6a0 100644
--- a/missing.h
+++ b/missing.h
@@ -23,9 +23,11 @@
 #ifndef HAVE_MISSING_H
 #define HAVE_MISSING_H
 
-#include <time.h>
+#include <linux/ptp_clock.h>
+#include <linux/version.h>
 #include <sys/syscall.h>
 #include <sys/timex.h>
+#include <time.h>
 #include <unistd.h>
 
 #ifndef ADJ_TAI
@@ -60,6 +62,54 @@ enum {
 };
 #endif
 
+#ifndef PTP_SYS_OFFSET
+
+#define PTP_SYS_OFFSET     _IOW(PTP_CLK_MAGIC, 5, struct ptp_sys_offset)
+
+struct ptp_sys_offset {
+	unsigned int n_samples; /* Desired number of measurements. */
+	unsigned int rsv[3];    /* Reserved for future use. */
+	/*
+	 * Array of interleaved system/phc time stamps. The kernel
+	 * will provide 2*n_samples + 1 time stamps, with the last
+	 * one as a system time stamp.
+	 */
+	struct ptp_clock_time ts[2 * PTP_MAX_SAMPLES + 1];
+};
+
+#endif /* PTP_SYS_OFFSET */
+
+#ifndef PTP_SYS_OFFSET_PRECISE
+
+#define PTP_SYS_OFFSET_PRECISE \
+	_IOWR(PTP_CLK_MAGIC, 8, struct ptp_sys_offset_precise)
+
+struct ptp_sys_offset_precise {
+	struct ptp_clock_time device;
+	struct ptp_clock_time sys_realtime;
+	struct ptp_clock_time sys_monoraw;
+	unsigned int rsv[4];    /* Reserved for future use. */
+};
+
+#endif /* PTP_SYS_OFFSET_PRECISE */
+
+#ifndef PTP_SYS_OFFSET_EXTENDED
+
+#define PTP_SYS_OFFSET_EXTENDED \
+	_IOWR(PTP_CLK_MAGIC, 9, struct ptp_sys_offset_extended)
+
+struct ptp_sys_offset_extended {
+	unsigned int n_samples; /* Desired number of measurements. */
+	unsigned int rsv[3];    /* Reserved for future use. */
+	/*
+	 * Array of [system, phc, system] time stamps. The kernel will provide
+	 * 3*n_samples time stamps.
+	 */
+	struct ptp_clock_time ts[PTP_MAX_SAMPLES][3];
+};
+
+#endif /* PTP_SYS_OFFSET_EXTENDED */
+
 #ifndef LIST_FOREACH_SAFE
 #define	LIST_FOREACH_SAFE(var, head, field, tvar)			\
 	for ((var) = LIST_FIRST((head));				\
diff --git a/sysoff.c b/sysoff.c
index b993ee98bb41..03cdb87dcab2 100644
--- a/sysoff.c
+++ b/sysoff.c
@@ -27,8 +27,6 @@
 
 #define NS_PER_SEC 1000000000LL
 
-#ifdef PTP_SYS_OFFSET
-
 static int64_t pctns(struct ptp_clock_time *t)
 {
 	return t->sec * NS_PER_SEC + t->nsec;
@@ -42,7 +40,6 @@ static struct {
 
 static int sysoff_precise(int fd, int64_t *result, uint64_t *ts)
 {
-#ifdef PTP_SYS_OFFSET_PRECISE
 	struct ptp_sys_offset_precise pso;
 	memset(&pso, 0, sizeof(pso));
 	if (ioctl(fd, PTP_SYS_OFFSET_PRECISE, &pso)) {
@@ -52,9 +49,6 @@ static int sysoff_precise(int fd, int64_t *result, uint64_t *ts)
 	*result = pctns(&pso.sys_realtime) - pctns(&pso.device);
 	*ts = pctns(&pso.sys_realtime);
 	return SYSOFF_PRECISE;
-#else
-	return SYSOFF_COMPILE_TIME_MISSING;
-#endif
 }
 
 static void insertion_sort(int length, int64_t interval, int64_t offset, uint64_t ts)
@@ -100,7 +94,6 @@ static int64_t sysoff_estimate(struct ptp_clock_time *pct, int extended,
 static int sysoff_extended(int fd, int n_samples,
 			   int64_t *result, uint64_t *ts, int64_t *delay)
 {
-#ifdef PTP_SYS_OFFSET_EXTENDED
 	struct ptp_sys_offset_extended pso;
 	memset(&pso, 0, sizeof(pso));
 	pso.n_samples = n_samples;
@@ -110,9 +103,6 @@ static int sysoff_extended(int fd, int n_samples,
 	}
 	*result = sysoff_estimate(&pso.ts[0][0], 1, n_samples, ts, delay);
 	return SYSOFF_EXTENDED;
-#else
-	return SYSOFF_COMPILE_TIME_MISSING;
-#endif
 }
 
 static int sysoff_basic(int fd, int n_samples,
@@ -141,7 +131,7 @@ int sysoff_measure(int fd, int method, int n_samples,
 	case SYSOFF_BASIC:
 		return sysoff_basic(fd, n_samples, result, ts, delay);
 	}
-	return SYSOFF_COMPILE_TIME_MISSING;
+	return SYSOFF_RUN_TIME_MISSING;
 }
 
 int sysoff_probe(int fd, int n_samples)
@@ -165,18 +155,3 @@ int sysoff_probe(int fd, int n_samples)
 
 	return SYSOFF_RUN_TIME_MISSING;
 }
-
-#else /* !PTP_SYS_OFFSET */
-
-int sysoff_measure(int fd, int n_samples,
-		   int64_t *result, uint64_t *ts, int64_t *delay)
-{
-	return SYSOFF_COMPILE_TIME_MISSING;
-}
-
-int sysoff_probe(int fd, int n_samples)
-{
-	return SYSOFF_COMPILE_TIME_MISSING;
-}
-
-#endif /* PTP_SYS_OFFSET */
diff --git a/sysoff.h b/sysoff.h
index 79d22908ce48..e4de91948231 100644
--- a/sysoff.h
+++ b/sysoff.h
@@ -19,9 +19,9 @@
  */
 
 #include <stdint.h>
+#include "missing.h"
 
 enum {
-	SYSOFF_COMPILE_TIME_MISSING = -2,
 	SYSOFF_RUN_TIME_MISSING = -1,
 	SYSOFF_PRECISE,
 	SYSOFF_EXTENDED,
-- 
2.25.1

```

> I was not allowed "git am"

This is a different kind of patch. It is not a patch against the openil/buildroot build system, therefore you shouldn't apply it to openil's git tree. Instead, it is a package patch. It gets applied on the linuxptp source code after it gets downloaded. So, you did the right thing.
Please modify your local patch with this one and recompile qoriq-ptp4l again. Sorry for the trouble.

diegotxegp, 2020-06-23:
Do not worry, Vladimir.

I have already built the new OpenIL with your patch and these are the results.

**LS1021ATSN**

[root@LS1021ATSN ~] # ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg -m --tx_timestamp_timeout 20
ptp4l[195.315]: selected /dev/ptp1 as PTP clock
ptp4l[195.400]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[195.480]: port 2: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[195.482]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[199.401]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[199.402]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[199.402]: assuming the grand master role
ptp4l[199.439]: port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[199.440]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[199.441]: assuming the grand master role
ptp4l[199.442]: assuming the grand master role
ptp4l[360.900]: port 2: new foreign master aabbcc.fffe.00094a-1
ptp4l[360.901]: selected best master clock aabbcc.fffe.00094a
ptp4l[360.901]: assuming the grand master role
ptp4l[360.901]: assuming the grand master role

[root@LS1021ATSN ~] # phc2sys -a -rr --transportSpecific 0x1 -m
phc2sys[296.589]: reconfiguring after port state change
phc2sys[296.592]: selecting swp3 for synchronization
phc2sys[296.594]: skipping swp2: swp3 has the same clock and is already selected
phc2sys[296.596]: selecting CLOCK_REALTIME as the master clock
phc2sys[296.598]: swp3 sys offset       -16 s0 freq  -20906 delay   6640
phc2sys[297.601]: swp3 sys offset         0 s2 freq  -20890 delay   6800
phc2sys[298.602]: swp3 sys offset       -24 s2 freq  -20914 delay   6720
phc2sys[299.604]: swp3 sys offset       -16 s2 freq  -20913 delay   6640
phc2sys[300.605]: swp3 sys offset       -48 s2 freq  -20950 delay   6720
phc2sys[301.606]: swp3 sys offset       -32 s2 freq  -20948 delay   6640
phc2sys[302.608]: swp3 sys offset       -32 s2 freq  -20958 delay   6560
phc2sys[303.609]: swp3 sys offset       -16 s2 freq  -20952 delay   6640
phc2sys[304.611]: swp3 sys offset       -96 s2 freq  -21036 delay   6640
phc2sys[305.612]: swp3 sys offset         8 s2 freq  -20961 delay   6640
phc2sys[306.614]: swp3 sys offset         8 s2 freq  -20959 delay   6720
phc2sys[307.615]: swp3 sys offset       -64 s2 freq  -21028 delay   6640
phc2sys[308.616]: swp3 sys offset       -80 s2 freq  -21064 delay   6720
phc2sys[309.618]: swp3 sys offset       -64 s2 freq  -21072 delay   6720
phc2sys[310.619]: swp3 sys offset       -40 s2 freq  -21067 delay   6640
phc2sys[311.621]: swp3 sys offset       -48 s2 freq  -21087 delay   6640
phc2sys[312.622]: swp3 sys offset        -8 s2 freq  -21061 delay   6720
phc2sys[313.624]: swp3 sys offset       -40 s2 freq  -21096 delay   6720
phc2sys[314.625]: swp3 sys offset       -40 s2 freq  -21108 delay   6640
phc2sys[315.628]: swp3 sys offset       -16 s2 freq  -21096 delay   6800
phc2sys[316.629]: swp3 sys offset      -104 s2 freq  -21188 delay   6560
phc2sys[317.631]: swp3 sys offset      -112 s2 freq  -21228 delay   6640
phc2sys[318.632]: swp3 sys offset       -88 s2 freq  -21237 delay   6560
phc2sys[319.633]: swp3 sys offset       -48 s2 freq  -21224 delay   6640
phc2sys[320.635]: swp3 sys offset      -104 s2 freq  -21294 delay   6640
phc2sys[321.636]: swp3 sys offset       -80 s2 freq  -21301 delay   6640
phc2sys[322.638]: swp3 sys offset       -80 s2 freq  -21325 delay   6640
phc2sys[323.639]: swp3 sys offset      -128 s2 freq  -21397 delay   6640
phc2sys[324.640]: swp3 sys offset      -128 s2 freq  -21436 delay   6640
phc2sys[325.642]: swp3 sys offset       -80 s2 freq  -21426 delay   6640
phc2sys[326.643]: swp3 sys offset       -80 s2 freq  -21450 delay   6720
phc2sys[327.645]: swp3 sys offset      -184 s2 freq  -21578 delay   6640
phc2sys[328.646]: swp3 sys offset      -160 s2 freq  -21609 delay   6640
phc2sys[329.648]: swp3 sys offset      -168 s2 freq  -21665 delay   6560
phc2sys[330.649]: swp3 sys offset      -176 s2 freq  -21724 delay   6560
phc2sys[331.651]: swp3 sys offset      -248 s2 freq  -21848 delay   6560
phc2sys[332.652]: swp3 sys offset      -184 s2 freq  -21859 delay   6640
phc2sys[333.654]: swp3 sys offset      -216 s2 freq  -21946 delay   6640
phc2sys[334.655]: swp3 sys offset      -304 s2 freq  -22099 delay   6560
phc2sys[335.656]: swp3 sys offset      -240 s2 freq  -22126 delay   6640
phc2sys[336.658]: swp3 sys offset      -336 s2 freq  -22294 delay   6640
phc2sys[337.659]: swp3 sys offset      -328 s2 freq  -22387 delay   6720
[... 262 lines cut ...]
phc2sys[416.753]: CLOCK_REALTIME phc offset    112292 s2 freq +115723 delay   4919
phc2sys[417.754]: CLOCK_REALTIME phc offset    -22249 s2 freq  +14870 delay   5196
phc2sys[418.755]: CLOCK_REALTIME phc offset    -70266 s2 freq  -39822 delay   5232
phc2sys[419.755]: CLOCK_REALTIME phc offset    -55440 s2 freq  -46076 delay   5220
phc2sys[420.756]: CLOCK_REALTIME phc offset    -24295 s2 freq  -31563 delay   5184
phc2sys[421.757]: CLOCK_REALTIME phc offset     -3580 s2 freq  -18136 delay   4932
phc2sys[422.757]: CLOCK_REALTIME phc offset   -138810 s2 freq -154440 delay   5592
phc2sys[423.758]: CLOCK_REALTIME phc offset    -28012 s2 freq  -85285 delay   4945
phc2sys[424.758]: CLOCK_REALTIME phc offset    228884 s2 freq +163207 delay   5413
phc2sys[425.759]: CLOCK_REALTIME phc offset    113522 s2 freq +116511 delay   5592
phc2sys[426.760]: CLOCK_REALTIME phc offset    -21503 s2 freq  +15542 delay   5543
[... cut ...]
phc2sys[432.764]: CLOCK_REALTIME phc offset    -30127 s2 freq  -87342 delay   5569
phc2sys[433.764]: CLOCK_REALTIME phc offset    229804 s2 freq +163551 delay   5148
phc2sys[434.764]: CLOCK_REALTIME phc offset    115672 s2 freq +118361 delay   5159
phc2sys[435.765]: CLOCK_REALTIME phc offset    -21138 s2 freq  +16252 delay   5375

It seems quite similar from my point of view. Big offsets. Both ptp4l and phc2sys. What do you think?
But did you say that you get better results in your office? What am I doing bad?

Diego

vladimiroltean, 2020-06-23:
> But did you say that you get better results in your office?

Yes, quite different results.

```
[root@LS1021ATSN ~] # journalctl -b -u phc2sys -f
-- Logs begin at Tue 2020-06-23 10:31:52 UTC. --
Jun 23 15:41:29 LS1021ATSN phc2sys[626]: [27502.186] swp5 sys offset         3 s2 freq  -20380 delay   6720
Jun 23 15:41:30 LS1021ATSN phc2sys[626]: [27503.187] swp5 sys offset       -80 s2 freq  -20462 delay   6720
Jun 23 15:41:31 LS1021ATSN phc2sys[626]: [27504.189] swp5 sys offset       -19 s2 freq  -20425 delay   6720
Jun 23 15:41:32 LS1021ATSN phc2sys[626]: [27505.190] swp5 sys offset        57 s2 freq  -20355 delay   6800
Jun 23 15:41:33 LS1021ATSN phc2sys[626]: [27506.192] swp5 sys offset       -19 s2 freq  -20413 delay   6720
Jun 23 15:41:34 LS1021ATSN phc2sys[626]: [27507.193] swp5 sys offset        56 s2 freq  -20344 delay   6720
Jun 23 15:41:35 LS1021ATSN phc2sys[626]: [27508.195] swp5 sys offset       -48 s2 freq  -20431 delay   6720
Jun 23 15:41:36 LS1021ATSN phc2sys[626]: [27509.196] swp5 sys offset        50 s2 freq  -20348 delay   6720
Jun 23 15:41:37 LS1021ATSN phc2sys[626]: [27510.197] swp5 sys offset       -48 s2 freq  -20431 delay   6640
Jun 23 15:41:38 LS1021ATSN phc2sys[626]: [27511.199] swp5 sys offset       -19 s2 freq  -20416 delay   6640
Jun 23 15:41:39 LS1021ATSN phc2sys[626]: [27512.200] swp5 sys offset       -29 s2 freq  -20432 delay   6720
Jun 23 15:41:40 LS1021ATSN phc2sys[626]: [27513.201] swp5 sys offset       -18 s2 freq  -20430 delay   6640
Jun 23 15:41:41 LS1021ATSN phc2sys[626]: [27514.203] swp5 sys offset        48 s2 freq  -20369 delay   6720
```

> What am I doing bad?

I don't know yet.
Your phc2sys delays are much lower now, which means we finally got past the issue we were trying to solve (we have PTP_SYS_OFFSET_EXTENDED now). But we have a different problem.
Could you add `--step_threshold 0.00002 --first_step_threshold 0.00002` to the arguments of `phc2sys` and do the test again?

diegotxegp, 2020-06-23:
**LS1021ATSN**

[root@LS1021ATSN ~] # ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg -m --tx_timestamp_timeout 20
ptp4l[195.315]: selected /dev/ptp1 as PTP clock
ptp4l[195.400]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[195.480]: port 2: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[195.482]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[199.401]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[199.402]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[199.402]: assuming the grand master role
ptp4l[199.439]: port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[199.440]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[199.441]: assuming the grand master role
ptp4l[199.442]: assuming the grand master role
ptp4l[360.900]: port 2: new foreign master aabbcc.fffe.00094a-1
ptp4l[360.901]: selected best master clock aabbcc.fffe.00094a
ptp4l[360.901]: assuming the grand master role
ptp4l[360.901]: assuming the grand master role
ptp4l[1769.318]: port 2: rogue peer delay response
ptp4l[1769.319]: port 2: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[1769.390]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[1769.390]: assuming the grand master role
ptp4l[1769.390]: assuming the grand master role
ptp4l[1785.470]: port 2: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[1789.215]: port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1789.215]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[1789.215]: assuming the grand master role
ptp4l[1789.215]: assuming the grand master role
ptp4l[1805.662]: port 2: new foreign master aabbcc.fffe.00094a-1
ptp4l[1805.662]: selected best master clock aabbcc.fffe.00094a
ptp4l[1805.662]: assuming the grand master role
ptp4l[1805.663]: assuming the grand master role

[root@LS1021ATSN ~] # phc2sys -a -rr --transportSpecific 0x1 -m --step_threshold 0.00002 --first_step_threshold 0.00002
phc2sys[1739.458]: reconfiguring after port state change
phc2sys[1739.459]: selecting swp3 for synchronization
phc2sys[1739.460]: skipping swp2: swp3 has the same clock and is already selected
phc2sys[1739.460]: selecting CLOCK_REALTIME as the master clock
phc2sys[1739.463]: swp3 sys offset        80 s0 freq  -20920 delay   6880
phc2sys[1740.466]: swp3 sys offset       -40 s2 freq  -21040 delay   6720
phc2sys[1741.467]: swp3 sys offset        64 s2 freq  -20976 delay   6720
phc2sys[1742.469]: swp3 sys offset        24 s2 freq  -20996 delay   6640
phc2sys[1743.470]: swp3 sys offset         0 s2 freq  -21013 delay   6640
phc2sys[1744.472]: swp3 sys offset        40 s2 freq  -20973 delay   6720
phc2sys[1745.473]: swp3 sys offset        48 s2 freq  -20953 delay   6720
phc2sys[1746.475]: swp3 sys offset         0 s2 freq  -20987 delay   6640
phc2sys[1747.476]: swp3 sys offset        -8 s2 freq  -20995 delay   6640
phc2sys[1748.478]: swp3 sys offset       -32 s2 freq  -21021 delay   6640
phc2sys[1749.479]: swp3 sys offset       -48 s2 freq  -21047 delay   6640
phc2sys[1750.481]: swp3 sys offset       -64 s2 freq  -21077 delay   6640
phc2sys[1751.482]: swp3 sys offset       -24 s2 freq  -21056 delay   6640
phc2sys[1752.484]: swp3 sys offset       -72 s2 freq  -21112 delay   6640
phc2sys[1753.485]: swp3 sys offset       -32 s2 freq  -21093 delay   6640
phc2sys[1754.487]: swp3 sys offset       -56 s2 freq  -21127 delay   6640
phc2sys[1755.488]: swp3 sys offset       -40 s2 freq  -21128 delay   6640
phc2sys[1756.490]: swp3 sys offset       -72 s2 freq  -21172 delay   6640
phc2sys[1757.491]: swp3 sys offset       -64 s2 freq  -21185 delay   6720
phc2sys[1758.493]: swp3 sys offset      -128 s2 freq  -21268 delay   6720
phc2sys[1759.494]: swp3 sys offset       -96 s2 freq  -21275 delay   6640
phc2sys[1760.496]: swp3 sys offset       -88 s2 freq  -21296 delay   6640
phc2sys[1761.498]: swp3 sys offset      -104 s2 freq  -21338 delay   6720
phc2sys[1762.499]: swp3 sys offset       -88 s2 freq  -21353 delay   6720
phc2sys[1763.500]: swp3 sys offset      -168 s2 freq  -21460 delay   6720
phc2sys[1764.502]: swp3 sys offset      -176 s2 freq  -21518 delay   6640
phc2sys[1765.503]: swp3 sys offset      -176 s2 freq  -21571 delay   6640
phc2sys[1766.505]: swp3 sys offset      -152 s2 freq  -21600 delay   6720
phc2sys[1767.506]: swp3 sys offset      -168 s2 freq  -21661 delay   6640
phc2sys[1768.508]: swp3 sys offset      -176 s2 freq  -21720 delay   6640
phc2sys[1769.508]: port 00049f.fffe.ef0808-2 changed state
phc2sys[1769.509]: reconfiguring after port state change
[... 155 lines cut ...]
phc2sys[1807.025]: enp3s0 sys offset      2424 s2 freq +62499999 delay   5595
phc2sys[1808.025]: enp3s0 sys offset      2548 s2 freq +62499999 delay   5895
phc2sys[1809.026]: enp3s0 sys offset      2770 s2 freq +62499999 delay   5880
phc2sys[1810.027]: enp3s0 sys offset      2939 s2 freq +62499999 delay   5895
phc2sys[1811.027]: enp3s0 sys offset      3054 s2 freq +62499999 delay   5025
phc2sys[1812.027]: port aabbcc.fffe.00094a-1 changed state
phc2sys[1812.028]: reconfiguring after port state change
phc2sys[1812.028]: selecting enp3s0 for synchronization
phc2sys[1812.028]: selecting CLOCK_REALTIME as the master clock
phc2sys[1812.028]: enp3s0 sys offset      3230 s2 freq +62499999 delay   5235
phc2sys[1813.029]: enp3s0 sys offset      3416 s2 freq +62499999 delay   6405
phc2sys[1814.029]: enp3s0 sys offset      3579 s2 freq +62499999 delay   6405
[... cut ...]
phc2sys[1820.032]: enp3s0 sys offset      4707 s2 freq +62499999 delay   4935
phc2sys[1821.033]: enp3s0 sys offset      4868 s2 freq +62499999 delay   6405
phc2sys[1822.033]: enp3s0 sys offset      5067 s2 freq +62499999 delay   5655
phc2sys[1823.034]: enp3s0 sys offset      5253 s2 freq +62499999 delay   5955
phc2sys[1824.034]: enp3s0 sys offset      5455 s2 freq +62499999 delay   6360
phc2sys[1825.035]: enp3s0 sys offset      5595 s2 freq +62499999 delay   5895

Suddenly, node stopped synchronisation and assumed its clock as a master.

diegotxegp, 2020-06-23:
My network is like I explained you at another Github issue about ptp4l. A LS1021ATSN platform acts as a master, and identical slave nodes are connected to ETH2 and ETH3 ports of LS1021ATSN. OpenIL version is 1.8 (May 2020)

Tell me what exact instructions you use to run this network in 802.1AS mode. Just what you do. Maybe I skipped some step accidentally. If not, I do not understand what is happening.

vladimiroltean, 2020-06-23:
> Tell me what exact instructions you use to run this network in 802.1AS mode

Yes, the only difference in my setup is that I'm using the master branch of OpenIL (plus the patch I gave you for fixing PTP_SYS_OFFSET_EXTENDED with phc2sys and sja1105), while you're using the 1.8 release tag. But, between the release tag and the master branch, another very similar `rogue peer delay response` bug report has been submitted (and fixed, on the master branch only):
https://github.com/openil/openil/issues/54

So, (while I do understand that it is rather annoying to reflash the SD card again), I think it is worth a shot to fetch and checkout the openil master branch, then `make nxp_ls1021atsn_defconfig && make` again.
The point with using the openil master branch is that it includes this patch: https://github.com/openil/openil/commit/3556336e79d56c7cbd023c66fb4df2fae19611ac which will compile the linux kernel from its respective master branch (linux-5.4.y), which is going to contain this fix: https://github.com/openil/linux/commit/adc93ad9ad0e51c8c10700c2d846ffeb3a1a2ced

diegotxegp, 2020-06-23:
Sorry for the question. Could you explain me the steps to follow to compile the master branch of OpenIL? I am afraid I will confuse instead.

Thank you in advance, Vladimir, and sorry.

vladimiroltean, 2020-06-23:
```
git fetch origin
git reset --hard origin/master # This will lose any changes you had. If you want to keep something, commit it and rebase it on top.
# Now there are 2 options, either rm -rf output, or use the existing build artefacts. This will depend on your own judgement based on what were the changes that were done. In this particular case, it looks ok to not delete the output folder. So just:
make nxp_ls1021atsn_defconfig
make 2&>1 | tee output/build.log
```
If everything goes as intended, linux will be compiled from the linux-5.4.y branch this time.

diegotxegp, 2020-06-24:
Vladimir. I follow your steps and nothing was compiled. The sdcard.img is the same, dated yesterday. /output/build.log is empty.

vladimiroltean, 2020-06-24:
Sorry, I had a typo. Instead of "make 2&>1" it should be "make 2>&1", to redirect stderr to stdout.

diegotxegp, 2020-06-24:
I compiled it and theses are the results.

**LS1021ATSN**

```
[root@LS1021ATSN ~] # ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg -m --tx_timestamp_timeout 20
ptp4l[200.432]: selected /dev/ptp1 as PTP clock
ptp4l[200.540]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[200.620]: port 2: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[200.622]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[203.810]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[203.811]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[203.811]: assuming the grand master role
ptp4l[203.871]: port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[203.871]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[203.871]: assuming the grand master role
ptp4l[203.871]: assuming the grand master role
ptp4l[890.991]: port 2: rogue peer delay response
ptp4l[890.991]: port 2: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[891.070]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[891.070]: assuming the grand master role
ptp4l[891.070]: assuming the grand master role
ptp4l[907.150]: port 2: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[910.591]: port 2: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[910.592]: selected local clock 00049f.fffe.ef0808 as best master
ptp4l[910.592]: assuming the grand master role
ptp4l[910.592]: assuming the grand master role
ptp4l[926.883]: port 2: new foreign master aabbcc.fffe.00094a-1
ptp4l[926.883]: selected best master clock aabbcc.fffe.00094a
ptp4l[926.883]: assuming the grand master role
ptp4l[926.883]: assuming the grand master role
ptp4l[949.276]: selected best master clock aabbcc.fffe.00094a
ptp4l[949.276]: assuming the grand master role
ptp4l[949.276]: assuming the grand master role
ptp4l[986.940]: selected best master clock aabbcc.fffe.00094a
ptp4l[986.941]: assuming the grand master role
ptp4l[986.941]: assuming the grand master role

[root@LS1021ATSN ~] # phc2sys -a -rr --transportSpecific 0x1 --step_threshold 0.0002 --first_step_threshold 0.0002 -m
phc2sys[1242.741]: reconfiguring after port state change
phc2sys[1242.743]: selecting swp3 for synchronization
phc2sys[1242.744]: skipping swp2: swp3 has the same clock and is already selected
phc2sys[1242.744]: selecting CLOCK_REALTIME as the master clock
phc2sys[1242.745]: swp3 sys offset        -8 s0 freq  -20696 delay   6640
phc2sys[1243.747]: swp3 sys offset        32 s2 freq  -20656 delay   6720
phc2sys[1244.749]: swp3 sys offset       -40 s2 freq  -20696 delay   6720
phc2sys[1245.750]: swp3 sys offset       -40 s2 freq  -20708 delay   6720
phc2sys[1246.752]: swp3 sys offset       -40 s2 freq  -20720 delay   6640
phc2sys[1247.753]: swp3 sys offset        40 s2 freq  -20652 delay   6720
phc2sys[1248.754]: port 00049f.fffe.ef0808-2 changed state
phc2sys[1248.754]: reconfiguring after port state change
phc2sys[1248.755]: selecting swp3 for synchronization
phc2sys[1248.755]: skipping swp2: swp3 has the same clock and is already selected
phc2sys[1248.755]: selecting CLOCK_REALTIME as the master clock
phc2sys[1248.757]: swp3 sys offset        72 s2 freq  -20608 delay   6880
phc2sys[1249.758]: swp3 sys offset      -104 s2 freq  -20762 delay   6720
phc2sys[1250.759]: swp3 sys offset       -72 s2 freq  -20762 delay   6640
phc2sys[1251.761]: swp3 sys offset         8 s2 freq  -20703 delay   6640
phc2sys[1252.762]: swp3 sys offset        -8 s2 freq  -20717 delay   6720
phc2sys[1253.764]: swp3 sys offset        16 s2 freq  -20695 delay   6720
phc2sys[1254.765]: swp3 sys offset        32 s2 freq  -20674 delay   6720
phc2sys[1255.767]: swp3 sys offset       -32 s2 freq  -20729 delay   6720
phc2sys[1256.768]: swp3 sys offset        80 s2 freq  -20626 delay   6800
phc2sys[1257.770]: swp3 sys offset         0 s2 freq  -20682 delay   6720
phc2sys[1258.771]: swp3 sys offset        32 s2 freq  -20650 delay   6800
phc2sys[1259.773]: swp3 sys offset       -72 s2 freq  -20745 delay   6640
phc2sys[1260.774]: swp3 sys offset        64 s2 freq  -20630 delay   6720
phc2sys[1261.776]: swp3 sys offset       -16 s2 freq  -20691 delay   6720
phc2sys[1262.777]: swp3 sys offset       -48 s2 freq  -20728 delay   6720
phc2sys[1263.778]: swp3 sys offset        16 s2 freq  -20678 delay   6720
[... 158 lines cut ...]
phc2sys[1271.420]: CLOCK_REALTIME phc offset       -40 s2 freq  -10773 delay   4908
phc2sys[1272.421]: CLOCK_REALTIME phc offset        37 s2 freq  -10708 delay   5592
phc2sys[1273.421]: CLOCK_REALTIME phc offset         4 s2 freq  -10730 delay   5544
phc2sys[1274.422]: CLOCK_REALTIME phc offset         3 s2 freq  -10730 delay   5592
phc2sys[1275.423]: CLOCK_REALTIME phc offset       -37 s2 freq  -10769 delay   4932
phc2sys[1276.423]: CLOCK_REALTIME phc offset       -37 s2 freq  -10780 delay   5376
phc2sys[1277.424]: CLOCK_REALTIME phc offset        -3 s2 freq  -10757 delay   5364
phc2sys[1278.424]: CLOCK_REALTIME phc offset         5 s2 freq  -10750 delay   5340
phc2sys[1279.425]: CLOCK_REALTIME phc offset       -14 s2 freq  -10768 delay   5196
phc2sys[1280.426]: CLOCK_REALTIME phc offset        46 s2 freq  -10712 delay   5328
phc2sys[1281.426]: CLOCK_REALTIME phc offset       -48 s2 freq  -10792 delay   5568
phc2sys[1282.427]: CLOCK_REALTIME phc offset        12 s2 freq  -10747 delay   5124
phc2sys[1283.428]: CLOCK_REALTIME phc offset        33 s2 freq  -10722 delay   5712
[... cut ...]
phc2sys[1286.430]: CLOCK_REALTIME phc offset      -141 s2 freq  -10828 delay   5148
phc2sys[1287.431]: CLOCK_REALTIME phc offset       -73 s2 freq  -10803 delay   5592
phc2sys[1288.431]: CLOCK_REALTIME phc offset       -23 s2 freq  -10775 delay   5364
```

If you notice, ptp4l offsets of the node are increasing each time.

I used the commands you purposed in section 8.2.3 "Synchronized 802.1Qbv demo" or the last recommendations about commands; `--transportSpecific 0x1 --step_threshold 0.0002 --first_step_threshold 0.0002`

It seems it does not improve. Curious.

diegotxegp, 2020-06-25:
Hello Vladimir.

I came back to OpenIL 1.8 release, and I run these commands:

For **LS1021ATSN**:
ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg -m --tx_timestamp_timeout 20
**phc2sys** no because, as I said to you, I do not mind the time was in 1970 or 2054. Just the same for the whole network. Then, I avoid that problem you said. Perfect.

For **node**:
ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg -m --tx_timestamp_timeout 20
sudo phc2sys -a -rr --transportSpecific 0x1 --step_threshold 0.0002 --first_step_threshold 0.0002 -m

And it works perfectly from my point of view. Low offset both on ptp4l and phc2sys.

nodo2@nodo2-N-A:~$ sudo ptp4l -i enp3s0 -f /home/nodo2/linuxptp-2.0/configs/gP.cfg --tx_timestamp_timeout 20 -m
ptp4l[575.901]: selected /dev/ptp0 as PTP clock
ptp4l[575.944]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[575.945]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[579.263]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[579.265]: selected local clock aabbcc.fffe.00094a as best master
ptp4l[579.266]: assuming the grand master role
ptp4l[579.847]: port 1: new foreign master 00049f.fffe.ef0808-2
ptp4l[581.847]: selected best master clock 00049f.fffe.ef0808
ptp4l[581.847]: port 1: MASTER to UNCALIBRATED on RS_SLAVE
ptp4l[582.107]: port 1: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[582.858]: rms   22 max   37 freq +14246 +/-  23 delay   233 +/-   0
ptp4l[583.859]: rms   22 max   29 freq +14286 +/-  14 delay   233 +/-   0
ptp4l[584.861]: rms   19 max   26 freq +14306 +/-   5 delay   233 +/-   0
ptp4l[585.862]: rms    8 max   17 freq +14302 +/-   6 delay   233 +/-   0
ptp4l[586.863]: rms    3 max    5 freq +14299 +/-   4 delay   233 +/-   0
ptp4l[587.865]: rms    4 max    7 freq +14293 +/-   4 delay   233 +/-   0
ptp4l[588.866]: rms    3 max    7 freq +14292 +/-   4 delay   233 +/-   0
ptp4l[589.867]: rms    3 max    6 freq +14296 +/-   4 delay   234 +/-   0
ptp4l[590.869]: rms    4 max    7 freq +14294 +/-   5 delay   234 +/-   0
ptp4l[591.870]: rms    3 max    5 freq +14294 +/-   4 delay   234 +/-   0
ptp4l[592.871]: rms    4 max    9 freq +14294 +/-   6 delay   236 +/-   0
ptp4l[593.873]: rms    3 max    5 freq +14294 +/-   4 delay   236 +/-   0
ptp4l[594.874]: rms    4 max    7 freq +14295 +/-   5 delay   236 +/-   0
ptp4l[595.875]: rms    4 max    7 freq +14293 +/-   5 delay   236 +/-   0
ptp4l[596.876]: rms    4 max    8 freq +14293 +/-   5 delay   236 +/-   0
ptp4l[597.878]: rms    3 max    5 freq +14293 +/-   4 delay   236 +/-   0
ptp4l[598.879]: rms    4 max    8 freq +14296 +/-   5 delay   236 +/-   0
ptp4l[599.880]: rms    4 max    7 freq +14295 +/-   5 delay   236 +/-   0
ptp4l[600.882]: rms    4 max    8 freq +14293 +/-   5 delay   236 +/-   0
ptp4l[601.883]: rms    4 max    8 freq +14293 +/-   6 delay   236 +/-   0
ptp4l[602.885]: rms    4 max    7 freq +14293 +/-   5 delay   236 +/-   0
ptp4l[603.886]: rms    3 max    5 freq +14293 +/-   4 delay   236 +/-   0
ptp4l[604.887]: rms    4 max    8 freq +14293 +/-   5 delay   236 +/-   0
ptp4l[605.889]: rms    3 max    6 freq +14293 +/-   5 delay   236 +/-   0
ptp4l[606.890]: rms    3 max    5 freq +14294 +/-   4 delay   236 +/-   0
ptp4l[607.892]: rms    2 max    4 freq +14292 +/-   3 delay   235 +/-   0
ptp4l[608.893]: rms    4 max    7 freq +14292 +/-   6 delay   234 +/-   0
ptp4l[609.894]: rms    2 max    4 freq +14292 +/-   3 delay   233 +/-   0
ptp4l[610.895]: rms    4 max    8 freq +14294 +/-   5 delay   233 +/-   0
ptp4l[611.897]: rms    4 max    8 freq +14297 +/-   5 delay   233 +/-   0
ptp4l[612.898]: rms    4 max    8 freq +14291 +/-   5 delay   234 +/-   0
ptp4l[613.900]: rms    3 max    6 freq +14290 +/-   4 delay   235 +/-   0
ptp4l[614.901]: rms    3 max    6 freq +14292 +/-   5 delay   234 +/-   0
ptp4l[615.903]: rms    4 max    6 freq +14292 +/-   5 delay   234 +/-   0
ptp4l[616.905]: rms    3 max    6 freq +14292 +/-   5 delay   233 +/-   0
ptp4l[617.906]: rms    3 max    6 freq +14295 +/-   4 delay   233 +/-   0
ptp4l[618.908]: rms    4 max    7 freq +14293 +/-   5 delay   233 +/-   0

And phc2sys in the node, similar. I do not have the values of this.

Now, I was going to write you to explain these good results, and repeat the steps, and I get these results in phc2sys on the node. The previous one, the CLOCK_REALTIME changed automatically its value to take the master's PHC value (1970). This time no.

 nodo2@nodo2-N-A:~$ sudo phc2sys -a -rr --transportSpecific 0x1 --step_threshold 0.0002 --first_step_threshold 0.0002 -m
phc2sys[811.347]: reconfiguring after port state change
phc2sys[811.348]: selecting CLOCK_REALTIME for synchronization
phc2sys[811.348]: selecting enp3s0 as the master clock
phc2sys[811.348]: CLOCK_REALTIME phc offset 1593085251906565919 s0 freq   +9237 delay   5412
phc2sys[812.349]: failed to step clock: Invalid argument
phc2sys[812.349]: CLOCK_REALTIME phc offset 1593085251906565885 s1 freq   +9203 delay   5628
phc2sys[813.349]: CLOCK_REALTIME phc offset 1593085251906565884 s0 freq   +9203 delay   5256
phc2sys[814.350]: CLOCK_REALTIME phc offset 1593085251906565896 s0 freq   +9203 delay   5604
phc2sys[815.350]: failed to step clock: Invalid argument
phc2sys[815.350]: CLOCK_REALTIME phc offset 1593085251906565914 s1 freq   +9221 delay   5604
phc2sys[816.350]: clockcheck: clock jumped forward or running faster than expected!
phc2sys[816.351]: CLOCK_REALTIME phc offset 1593085251906565902 s0 freq   +9221 delay   5616
[... cut ...] ep clock: Invalid argument
phc2sys[817.351]: CLOCK_REALTIME phc offset 1593085251906565894 s1 freq   +9213 delay   5616
phc2sys[818.352]: clockcheck: clock jumped forward or running faster than expected!
phc2sys[818.352]: CLOCK_REALTIME phc offset 1593085251906565879 s0 freq   +9213 delay   5400

I will repeat this procedure some more time because I think it happened to me using IEEE 1588 as well. Like CLOCK_REALTIME does not take well the time, and if you try the same operation later, it works.

vladimiroltean, 2020-06-25:
> phc2sys[815.350]: failed to step clock: Invalid argument

As I said before:

> There are bugs surrounding the clock stepping of CLOCK_REALTIME to a value close to 0 (Jan 1st 1970), and nobody is actively looking into fixing them, because nobody is in 1970 anymore.

So, you would need to manually move the PTP clock of your master (that is the LS1021A-TSN, according to your logs) to a different time, at each boot:

```
phc_ctl /dev/ptp1 set 1592817646.744752137
```

And only then start ptp4l.

diegotxegp, 2020-06-26:
Yes. It is true. I set it and it worked fine. My question is how do I keep this time permanent at /dev/ptp1?

vladimiroltean, 2020-06-26:
You could add the command as `ExecStartPre` in the `ptp4l` service. Like this:
```
vim /usr/lib/systemd/system/ptp4l.service
[Unit]
Description=Precision Time Protocol daemon
After=syslog.target network.target
Before=time-sync.target
Wants=time-sync.target
Wants=phc2sys.service

[Service]
ExecStartPre=/usr/sbin/phc_ctl /dev/ptp1 set 1592817646.744752137
ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp.cfg
Restart=always

[Install]
WantedBy=multi-user.target
```

diegotxegp, 2020-06-29:
I did that and reset, but /dev/ptp1 kept in 1970. Maybe it is, as you could see some messages behind, that ptp4l.service is inactive when LS1021ATSN platforms boots.

vladimiroltean, 2020-06-29:
Have you tried enabling the ptp4l service with systemctl?

diegotxegp, 2020-06-30:
Yes, once I enable ptp4l service "systemctl enable --now ptp4l", /dev/ptp1 comes to 2020. On the other thread, I undestood to you that if I wriite the command "ptp4l ...", activating ptp4l service would not be necessary. It works as well.

Apart from that, I have a new doubt for you. Since last week, the lights of the occupied ports ETH2 and ETH3 is blinking when LS1021ATSN boots, not just when I use ptp4l. I have not touched anything on its configuration.

vladimiroltean, 2020-06-30:
You can check with tcpdump what traffic is being transmitted. If it has an EtherType of 0x88f7, it's PTP.

vladimiroltean, 2020-06-30:
> On the other thread, I undestood to you that if I wriite the command "ptp4l ...", activating ptp4l service would not be necessary

And of course that is true. But if you run ptp4l manually, you should also run phc_ctl manually. I thought your question is how to do it automatically...

diegotxegp, 2020-06-30:
Ah. Another thing in case someone wants to compile OpenIL. Yesterday, I tried to compile it again, to discard that it is some corrupt data in SD, and an error raised. After "make" command, it was compiling as usual but after some hours, an error raised with the following message:

diegotxegp, 2020-06-30:
* [new tag]         OpenIL-u-boot-201805 -> OpenIL-u-boot-201805
 * [new tag]         OpenIL-u-boot-201810 -> OpenIL-u-boot-201810
 * [new tag]         OpenIL-v1.7-Baremetal-202001 -> OpenIL-v1.7-Baremetal-202001
 * [new tag]         OpenIL-v1.7-u-boot-202001 -> OpenIL-v1.7-u-boot-202001
remote: Enumerating objects: 36, done.
remote: Counting objects: 100% (36/36), done.
remote: Total 43 (delta 36), reused 36 (delta 36), pack-reused 7
Unpacking objects: 100% (43/43), done.
De https://github.com/openil/u-boot
 * [new tag]         OpenIL-u-boot-201901 -> OpenIL-u-boot-201901
 * [new tag]         OpenIL-u-boot-201904 -> OpenIL-u-boot-201904
 * [new tag]         OpenIL-u-boot-201908 -> OpenIL-u-boot-201908
Could not fetch special ref 'master'; assuming it is not special.
Commit 'master' does not exist in this repository.
--2020-06-29 17:22:26--  http://sources.buildroot.net/uboot/uboot-master.tar.gz
Resolviendo sources.buildroot.net (sources.buildroot.net)... 104.26.0.37, 172.67.72.56, 104.26.1.37, ...
Conectando con sources.buildroot.net (sources.buildroot.net)[104.26.0.37]:80... conectado.
Petición HTTP enviada, esperando respuesta... 404 Not Found
2020-06-29 17:22:26 ERROR 404: Not Found.

--2020-06-29 17:22:26--  http://sources.buildroot.net/uboot-master.tar.gz
Resolviendo sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 172.67.72.56, 104.26.0.37, ...
Conectando con sources.buildroot.net (sources.buildroot.net)[104.26.1.37]:80... conectado.
Petición HTTP enviada, esperando respuesta... 404 Not Found
2020-06-29 17:22:26 ERROR 404: Not Found.

package/pkg-generic.mk:167: fallo en las instrucciones para el objetivo '/home/diegotxe/openil/output/build/uboot-master/.stamp_downloaded'
make[1]: *** [/home/diegotxe/openil/output/build/uboot-master/.stamp_downloaded] Error 1
Makefile:84: fallo en las instrucciones para el objetivo '_all'
make: *** [_all] Error 2

[10 more replies]

closed 2020-07-02 by diegotxegp