nxp-archive/openil #62: Synchronization issue between the slave and the master

Maku-chan, 2020-06-29:
Hello,

I am new to synchronization and i configured the ptp on both slave and master and getting following errors at the slave end.

ptp4l[42247.991]: timed out while polling for tx timestamp
ptp4l[42247.991]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[42247.992]: port 1: send sync failed
ptp4l[42247.992]: port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[42248.024]: clockcheck: clock jumped backward or running slower than expected!
ptp4l[42248.026]: selected best master clock 001b21.fffe.ddf274
ptp4l[42248.026]: port 2: LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[42248.040]: port 2: rogue peer delay response
ptp4l[42248.040]: port 2: UNCALIBRATED to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[42264.056]: port 1: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[42264.104]: port 2: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[42265.360]: port 2: new foreign master 001b21.fffe.ddf274-1
ptp4l[42269.360]: selected best master clock 001b21.fffe.ddf274
ptp4l[42269.360]: port 2: LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[42271.996]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[42271.996]: selected best master clock 001b21.fffe.ddf274
ptp4l[42273.122]: timed out while polling for tx timestamp
ptp4l[42273.123]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[42273.124]: port 1: send sync failed
ptp4l[42273.124]: port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[42273.164]: clockcheck: clock jumped backward or running slower than expected!
ptp4l[42273.397]: clockcheck: clock jumped forward or running faster than expected!
ptp4l[42280.587]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42285.566]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42288.447]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42289.196]: port 1: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[42291.427]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42295.547]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42296.243]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[42296.243]: selected best master clock 001b21.fffe.ddf274
ptp4l[42297.369]: timed out while polling for tx timestamp
ptp4l[42297.369]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[42297.369]: port 1: send sync failed
ptp4l[42297.369]: port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[42297.404]: clockcheck: clock jumped backward or running slower than expected!
ptp4l[42297.404]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42297.427]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42297.809]: clockcheck: clock jumped forward or running faster than expected!
ptp4l[42298.407]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42298.419]: port 2: rogue peer delay response
ptp4l[42298.419]: port 2: UNCALIBRATED to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[42313.444]: port 1: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[42314.488]: port 2: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[42315.219]: port 2: new foreign master 001b21.fffe.ddf274-1
ptp4l[42319.219]: selected best master clock 001b21.fffe.ddf274
ptp4l[42319.219]: port 2: LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[42319.660]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[42319.660]: selected best master clock 001b21.fffe.ddf274
ptp4l[42320.786]: timed out while polling for tx timestamp
ptp4l[42320.786]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[42320.786]: port 1: send sync failed
ptp4l[42320.786]: port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[42320.836]: clockcheck: clock jumped backward or running slower than expected!
ptp4l[42321.055]: clockcheck: clock jumped forward or running faster than expected!
ptp4l[42323.346]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42327.340]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42331.326]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42335.427]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42336.876]: port 1: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[42338.306]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42343.286]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42343.985]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[42343.985]: selected best master clock 001b21.fffe.ddf274
ptp4l[42345.111]: timed out while polling for tx timestamp
ptp4l[42345.111]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[42345.111]: port 1: send sync failed
[... 114 lines cut ...]
ptp4l[42514.820]: clockcheck: clock jumped forward or running faster than expected!
ptp4l[42514.945]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42515.824]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42516.804]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42519.925]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42524.905]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42527.784]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42530.596]: port 1: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[42531.764]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42535.764]: port 2: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[42536.793]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[42536.794]: selected best master clock 001b21.fffe.ddf274
ptp4l[42537.920]: timed out while polling for tx timestamp
ptp4l[42537.920]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[42537.921]: port 1: send sync failed
ptp4l[42537.921]: port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[42537.960]: clockcheck: clock jumped backward or running slower than expected!
ptp4l[42537.961]: port 2: SLAVE to UNCALIBRATED on SYNCHRONIZATION_FAULT
ptp4l[42537.977]: port 2: rogue peer delay response
ptp4l[42537.977]: port 2: UNCALIBRATED to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[42554.004]: port 1: FAULTY to LISTENING on INIT_COMPLETE

I am really lost and i dont know how to fix this error. 

-Maku_chan

vladimiroltean, 2020-06-29:
Hi Maku-chan,
I lack the information about which OpenIL board you are using, and which OpenIL version. I will infer it is the LS1021A-TSN board running OpenIL 1.8, please feel free to correct me.
If that is the case, I believe you are running a ptp4l command by hand instead of using the ptp4l systemctl service, could you share what that ptp4l command is? Whatever it is, I think it's missing `--tx_timestamp_timeout 20`.
You might also be interested in reading the discussion from this topic: https://github.com/openil/openil/issues/59
Thanks,
-Vladimir

Maku-chan, 2020-06-29:
Hi,

I am using Mstn soc-e board. The command to run the ptp service is mentioned below which i am using
 sudo ptp4l -2 -A -i eth0 -l6 -m -f ptp4l.conf

The tx_timestamp_timeout i have mentioned in the ptp41.conf file

-Maku_chan

vladimiroltean, 2020-06-30:
> I am using Mstn soc-e board

What does that have to do with OpenIL? Please consult with your board vendor, as this seems to me as either a driver misconfiguration, or a ptp4l.conf (contents of which you didn't share) misconfiguration.

Thanks,
-Vladimir

Maku-chan, 2020-06-30:
Hello,

I tried to find some information regarding this board but couldnt find any proper forum to ask it. I have the configuration for both the master and slave below I hope that will be useful.

Master
[global]
#
# Default Data Set
#
twoStepFlag		1
slaveOnly		0
priority1		42	
priority2		42
domainNumber		0
utc_offset		0
clockClass		248
clockAccuracy		0xFE
offsetScaledLogVariance	0xFFFF
free_running		0
freq_est_interval	1
dscp_event		0
dscp_general		0
#
# Port Data Set
#
logAnnounceInterval	1
logSyncInterval		0
logMinDelayReqInterval	0
logMinPdelayReqInterval	0
announceReceiptTimeout	3
syncReceiptTimeout	3
delayAsymmetry		0
fault_reset_interval	4
neighborPropDelayThresh	20000000
#
# Run time options
#
assume_two_step		1
logging_level		6
path_trace_enabled	0
follow_up_info		0
hybrid_e2e		0
tx_timestamp_timeout	3
use_syslog		1
verbose			0
summary_interval	0
kernel_leap		1
check_fup_sync		0
#
# Servo Options
#
pi_proportional_const	0.0
pi_integral_const	0.0
pi_proportional_scale	0.0
pi_proportional_exponent	-0.3
pi_proportional_norm_max	0.7
pi_integral_scale	0.0
pi_integral_exponent	0.4
pi_integral_norm_max	0.3
step_threshold		0.0
first_step_threshold	0.00002
max_frequency		900000000
clock_servo		pi
sanity_freq_limit	200000000
ntpshm_segment		0
#
# Transport options
#
transportSpecific	0x0
ptp_dst_mac		01:1B:19:00:00:00
p2p_dst_mac		01:80:C2:00:00:0E
udp_ttl			1
udp6_scope		0x0E
uds_address		/var/run/ptp4l
#
# Default interface options
#
network_transport	L2
delay_mechanism		E2E
time_stamping		hardware
tsproc_mode		filter
delay_filter		moving_median
delay_filter_length	10
egressLatency		0
ingressLatency		0
boundary_clock_jbod	0
#
# Clock description
#
productDescription	;;
revisionData		;;
manufacturerIdentity	00:00:00
userDescription		;
timeSource		0xA0

Slave
[global]

#
# Configurable parameters
#
neighborPropDelayThresh	20000000
min_neighbor_prop_delay	-20000000
priority1		255
priority2		255
domainNumber	0
clockClass		248
clockAccuracy		0xFE
logAnnounceInterval	1
logSyncInterval		-3
logMinPdelayReqInterval	0
announceReceiptTimeout	3
syncReceiptTimeout	0
delayAsymmetry    0
tx_timestamp_timeout	4
use_syslog		1
verbose			0
logging_level		6

#
# 802.1AS specific
#	
assume_two_step		1
path_trace_enabled	0
follow_up_info		0
transportSpecific	0x0
ptp_dst_mac		01:1B:19:00:00:00
network_transport	L2
delay_mechanism		P2P

#
# Servo Options
#
pi_proportional_const	0.0
pi_integral_const	0.0
pi_proportional_scale	0.0
pi_proportional_exponent	-0.3
pi_proportional_norm_max	0.7
pi_integral_scale	0.0
pi_integral_exponent	0.4
pi_integral_norm_max	0.3
step_threshold		0.00002
first_step_threshold	0.00002
max_frequency		900000000
clock_servo		pi
sanity_freq_limit	200000000
ntpshm_segment		0

#
# Implementation specific
#
mgmt_iface  eth0
[PORT.0]
[PORT.1]
[PORT.2]
[PORT.3]

vladimiroltean, 2020-06-30:
> Basically the slave is originally running on AS 2011

None of the configuration files you shared is using 802.1AS.

> so the slave script is to make it work properly for 1588 to be compatible with the master.

What script?

Maku-chan, 2020-06-30:
Hi

The board is running by default on AS 2011. These scripts are to run 1588 standard on both the devices.

vladimiroltean, 2020-06-30:
Do the boards work without any of your changes, in 802.1AS mode?

closed 2020-08-06 by vladimiroltean