nxp-archive/openil #63: I can not synchronise two boards with a LS1021ATSN platform as a bridge

diegotxegp, 2020-07-15:
My issue is that I can not synchronise two boards with a LS1021ATSN platform as a bridge. The standard I want to use is IEEE 802.1AS.

My network is one board as a master, another board as a slave and a LS1021ATSN platform as a bridge in the middle. The porst used are ETH2 (swp2) for board master and ETH3 (swp3) for board slave.

I followed the steps I saw in the OpenIL 1.8 manual, in the section 8.2.3, but each board acts like it is alone in the network. Using the command "journalctl -b -u ptp4l -f", there is not ptp4l communication through the LS1021ATSN platform.

Could you help me with this?

Thank you.

Diego

vladimiroltean, 2020-07-15:
Hi Diego,
Are you using the 1.8 release tag? Would it be possible to try with the master branch?
If you run ptp4l with "-l 7", do you see any messages that should not be there?
Thanks,
-Vladimir

diegotxegp, 2020-07-15:
Hello Vladimir,

Yes, I am using the 1.8 release tag because that time when we talked, we found the way to work with this release, instead of the master branch.

I run ptp4l on both boards with "-l 7" and these are the results:

*Node 1:* Master

nodo2@nodo2-N-A:~$ sudo ptp4l -i enp3s0 -f /home/nodo2/linuxptp-2.0/configs/gPTP.cfg --tx_timestamp_timeout 20 -m -l 7
ptp4l[614.806]: config item (null).assume_two_step is 1
ptp4l[614.807]: config item (null).check_fup_sync is 0
ptp4l[614.807]: config item (null).tx_timestamp_timeout is 20
ptp4l[614.808]: config item (null).clock_servo is 0
ptp4l[614.808]: config item (null).clock_type is 32768
ptp4l[614.809]: config item (null).clock_servo is 0
ptp4l[614.809]: config item (null).clockClass is 248
ptp4l[614.810]: config item (null).clockAccuracy is 254
ptp4l[614.810]: config item (null).offsetScaledLogVariance is 65535
ptp4l[614.811]: config item (null).productDescription is ';;'
ptp4l[614.811]: config item (null).revisionData is ';;'
ptp4l[614.812]: config item (null).userDescription is ''
ptp4l[614.812]: config item (null).manufacturerIdentity is '00:00:00'
ptp4l[614.813]: config item (null).domainNumber is 0
ptp4l[614.813]: config item (null).slaveOnly is 0
ptp4l[614.814]: config item (null).gmCapable is 1
ptp4l[614.814]: config item (null).gmCapable is 1
ptp4l[614.815]: config item (null).G.8275.defaultDS.localPriority is 128
ptp4l[614.815]: config item (null).time_stamping is 1
ptp4l[614.815]: config item (null).twoStepFlag is 1
ptp4l[614.815]: config item (null).twoStepFlag is 1
ptp4l[614.815]: config item (null).time_stamping is 1
ptp4l[614.815]: config item (null).priority1 is 248
ptp4l[614.815]: config item (null).priority2 is 248
ptp4l[614.815]: interface index 2 is up
ptp4l[614.819]: config item (null).free_running is 0
ptp4l[614.819]: selected /dev/ptp0 as PTP clock
ptp4l[614.819]: config item (null).uds_address is '/var/run/ptp4l'
ptp4l[614.819]: section item /var/run/ptp4l.announceReceiptTimeout now 0
ptp4l[614.819]: section item /var/run/ptp4l.delay_mechanism now 0
ptp4l[614.819]: section item /var/run/ptp4l.network_transport now 0
ptp4l[614.819]: section item /var/run/ptp4l.delay_filter_length now 1
ptp4l[614.821]: config item (null).free_running is 0
ptp4l[614.821]: config item (null).freq_est_interval is 1
ptp4l[614.821]: config item (null).gmCapable is 1
ptp4l[614.821]: config item (null).kernel_leap is 1
ptp4l[614.821]: config item (null).utc_offset is 37
ptp4l[614.821]: config item (null).timeSource is 160
ptp4l[614.821]: config item (null).pi_proportional_const is 0.000000
ptp4l[614.821]: config item (null).pi_integral_const is 0.000000
ptp4l[614.821]: config item (null).pi_proportional_scale is 0.000000
ptp4l[614.821]: config item (null).pi_proportional_exponent is -0.300000
ptp4l[614.821]: config item (null).pi_proportional_norm_max is 0.700000
ptp4l[614.821]: config item (null).pi_integral_scale is 0.000000
ptp4l[614.821]: config item (null).pi_integral_exponent is 0.400000
ptp4l[614.821]: config item (null).pi_integral_norm_max is 0.300000
ptp4l[614.821]: config item (null).step_threshold is 0.000000
ptp4l[614.821]: config item (null).first_step_threshold is 0.000020
ptp4l[614.821]: config item (null).max_frequency is 900000000
ptp4l[614.821]: config item (null).dataset_comparison is 0
ptp4l[614.821]: config item (null).delay_filter_length is 10
ptp4l[614.821]: config item (null).delay_filter is 1
ptp4l[614.821]: config item (null).tsproc_mode is 0
ptp4l[614.821]: config item (null).initial_delay is 0
ptp4l[614.821]: config item (null).summary_interval is 0
ptp4l[614.821]: config item (null).sanity_freq_limit is 200000000
ptp4l[614.822]: PI servo: sync interval 1.000 kp 0.700 ki 0.300000
ptp4l[614.822]: config item /var/run/ptp4l.boundary_clock_jbod is 0
ptp4l[614.822]: config item /var/run/ptp4l.network_transport is 0
ptp4l[614.822]: config item /var/run/ptp4l.delayAsymmetry is 0
ptp4l[614.822]: config item /var/run/ptp4l.follow_up_info is 1
[... 67 lines cut ...]
ptp4l[616.882]: port 1: delay timeout
ptp4l[617.883]: port 1: delay timeout
ptp4l[618.837]: port 1: announce timeout
ptp4l[618.838]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[618.838]: selected local clock aabbcc.fffe.00094a as best master
ptp4l[618.838]: assuming the grand master role
ptp4l[618.839]: port 1: master tx announce timeout
ptp4l[618.884]: port 1: delay timeout
ptp4l[618.964]: port 1: master sync timeout
ptp4l[619.090]: port 1: master sync timeout
ptp4l[619.215]: port 1: master sync timeout
ptp4l[619.341]: port 1: master sync timeout
ptp4l[619.467]: port 1: master sync timeout
ptp4l[619.592]: port 1: master sync timeout
ptp4l[619.718]: port 1: master sync timeout
ptp4l[619.840]: port 1: master tx announce timeout
ptp4l[619.844]: port 1: master sync timeout
ptp4l[619.884]: port 1: delay timeout
ptp4l[619.969]: port 1: master sync timeout

*Node 2* Slave: (the same)

If you want me to try the master branch, please, remind me the steps to build it. The way I always do is clonning the openil branch (official) and then compiling.

Diego

vladimiroltean, 2020-07-15:
Nothing looks out of the ordinary. Does pinging the switch work?

diegotxegp, 2020-07-15:
Yes, it does.

nodo2@nodo2-N-A:~$ ping 172.15.0.1
PING 172.15.0.1 (172.15.0.1) 56(84) bytes of data.
64 bytes from 172.15.0.1: icmp_seq=1 ttl=64 time=1.14 ms
64 bytes from 172.15.0.1: icmp_seq=2 ttl=64 time=0.360 ms
64 bytes from 172.15.0.1: icmp_seq=3 ttl=64 time=0.348 ms
64 bytes from 172.15.0.1: icmp_seq=4 ttl=64 time=0.380 ms
64 bytes from 172.15.0.1: icmp_seq=5 ttl=64 time=0.373 ms

vladimiroltean, 2020-07-15:
There is too little information. Don't just provide ptp4l verbose logs on the nodes, provide them on the switch as well.

diegotxegp, 2020-07-15:
> Using the command "journalctl -b -u ptp4l -f", there is not ptp4l communication through the LS1021ATSN platform.

I typed "journalctl -b -u ptp4l -f" but nothing appears. The LS1021ATSN platform would act as a bridge, and one node woulb be a master and the other slave. All the synhcronisation messages should cross through the LS1021ATSN.

Do you want me to try another command apart from "journalctl -b -u ptp4l -f"? It is what I saw in the manual to check ptp4l communication.

vladimiroltean, 2020-07-15:
Yes, that doesn't help me too much.
Please edit `/etc/linuxptp.cfg`, and insert this line immediately after `[global]`:
```
logging_level 7
```
Then `systemctl restart ptp4l` and watch the logs with journalctl again.

diegotxegp, 2020-07-16:
It seems that now there is communication. You can see it.

[root@LS1021ATSN ~] # systemctl restart ptp4l
[root@LS1021ATSN ~] # journalctl -b -u ptp4l -f
-- Logs begin at Fri 2020-02-07 15:50:54 UTC. --
Feb 07 15:56:13 LS1021ATSN ptp4l[453]: [323.642] port 1: master tx announce timeout
Feb 07 15:56:13 LS1021ATSN ptp4l[453]: [323.687] port 1: master sync timeout
Feb 07 15:56:13 LS1021ATSN ptp4l[453]: [323.702] port 2: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [323.813] port 1: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [323.828] port 2: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [323.914] port 2: master tx announce timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [323.938] port 1: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [323.953] port 2: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.063] port 1: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.078] port 2: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.189] port 1: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.204] port 2: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.314] port 1: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.329] port 2: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.365] port 1: delay timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.367] delay   filtered        233   raw        235
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.439] port 1: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.448] port 2: delay timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.450] delay   filtered        230   raw        234
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.454] port 2: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.565] port 1: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.579] port 2: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.642] port 1: master tx announce timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.690] port 1: master sync timeout
Feb 07 15:56:14 LS1021ATSN ptp4l[453]: [324.705] port 2: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [324.815] port 1: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [324.830] port 2: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [324.915] port 2: master tx announce timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [324.940] port 1: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [324.956] port 2: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.066] port 1: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.081] port 2: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.191] port 1: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.206] port 2: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.316] port 1: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.331] port 2: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.365] port 1: delay timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.367] delay   filtered        236   raw        237
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.442] port 1: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.448] port 2: delay timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.452] delay   filtered        230   raw        229
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.457] port 2: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.567] port 1: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.582] port 2: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.642] port 1: master tx announce timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.692] port 1: master sync timeout
Feb 07 15:56:15 LS1021ATSN ptp4l[453]: [325.707] port 2: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [325.818] port 1: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [325.833] port 2: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [325.915] port 2: master tx announce timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [325.943] port 1: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [325.958] port 2: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.068] port 1: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.083] port 2: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.194] port 1: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.209] port 2: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.319] port 1: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.334] port 2: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.365] port 1: delay timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.367] delay   filtered        238   raw        240
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.444] port 1: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.448] port 2: delay timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.451] delay   filtered        231   raw        232
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.459] port 2: master sync timeout
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.570] port 1: master sync timeout
[... 3 lines cut ...]
Feb 07 15:56:16 LS1021ATSN ptp4l[453]: [326.710] port 2: master sync timeout
Feb 07 15:56:17 LS1021ATSN ptp4l[453]: [326.820] port 1: master sync timeout
[... cut ...]
Feb 07 15:56:17 LS1021ATSN ptp4l[453]: [327.462] port 2: master sync timeout
Feb 07 15:56:17 LS1021ATSN ptp4l[453]: [327.572] port 1: master sync timeout
Feb 07 15:56:17 LS1021ATSN ptp4l[453]: [327.587] port 2: master sync timeout
Feb 07 15:56:17 LS1021ATSN ptp4l[453]: [327.643] port 1: master tx announce timeout

Tell me if you think it is correct.

Another thing. In this way, LS1021ATSN platform is again the master. How can I force the network to use one board as a master, and the other as a slave, just using the LS1021 as a time-aware bridge on IEEE 802.1AS?

diegotxegp, 2020-07-17:
Could you check it?

vladimiroltean, 2020-07-17:
In the last log you shared, the LS1021A-TSN board is indeed acting as master. I can see the delay calculations taking place, so that part should be fine, but I don't understand, from your previous messages, if the nodes are also acting as masters or not. If everybody is acting as master, that's a problem and I don't see, in the logs you've shared, why it happens. Have you made any other changes on the LS1021A board, like enabling VLAN, things like that?

To disable grandmaster functionality for 802.1AS in a board, edit its ptp4l configuration file and set gmCapable to 0.

diegotxegp, 2020-07-20:
> In the last log you shared, the LS1021A-TSN board is indeed acting as master. I can see the delay calculations taking place, so that part should be fine, but I don't understand, from your previous [...]
My doubt was how to disable LS1021ATSN not to be grandmaster anymore. Following the steps shown in the section 8.2.3 of the OpenIL 1.8 manual, LS1021ATSN converts automatically in grandmaster.

> To disable grandmaster functionality for 802.1AS in a board, edit its ptp4l configuration file and set gmCapable to 0.

I have done it, even before you said it to me, and in that case, each node acts as a master, without recognize to one another. Pinging they can see one another.

diegotxegp, 2020-07-20:
Vladimir. Do not worry. It is already solved.

I realised that ptp4l.service was disabled. Typing "systemctl enable ptp4l" and rebooting, it acts as a time-aware bridge, and the nodes as master and slave, respectively.

Previously, when I changed gmCapable from 1 to 0, I forgot to enable the ptp4l.service.

I hope this is useful for others.

vladimiroltean, 2020-07-20:
So, what was the problem, exactly?

diegotxegp, 2020-07-20:
The problem was that changing "gmCapable" to 0 at the LS1021ATSN platform, both nodes that supossedly were going to be master and slave, respectively, they both kept as a master. They understood that there is no more PTP clocks to negociate which is master and which slave. That means, that there is no communication between them (although they could ping each other). So, I enable ptp4l.service and then, one was selected as a master, and the other as a slave. How I was looking for. I do not know whether you understood it.

Once that is solved, I realised another thing. If I enable ptp4l.service at LS1021ATSN and I keep gmCapable to 1, if I try run ptp4l between LS1021ATSN and a node, the ptp4l offset is quite bigger (2000-3000ns) than using just the command "ptp4l -i swp2 -i swp3..." at the LS1021ATSN (70ns as much).

vladimiroltean, 2020-07-20:
> If I enable ptp4l.service at LS1021ATSN and I keep gmCapable to 1, if I try run ptp4l between LS1021ATSN and a node, the ptp4l offset is quite bigger (2000-3000ns) than using just the command "ptp4l -i swp2 -i swp3..." at the LS1021ATSN (70ns as much).

What's different between the command you run on the command line and the command invoked by `/usr/lib/systemd/system/ptp4l.service`?

diegotxegp, 2020-07-20:
My command is this: "ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg --tx_timestamp_timeout 20 -m"

Thanks to you I knew I needed "--tx_timestamp_timeout 20".

vladimiroltean, 2020-07-20:
```
[root@LS1021ATSN ~] # diff /etc/linuxptp.cfg /etc/ptp4l_cfg/gPTP.cfg 
--- /etc/linuxptp.cfg
+++ /etc/ptp4l_cfg/gPTP.cfg
@@ -19,10 +19,3 @@
 ptp_dst_mac            01:80:C2:00:00:0E
 network_transport      L2
 delay_mechanism                P2P
-tx_timestamp_timeout   20
-step_threshold         0.00002
-
-[swp2]
-[swp3]
-[swp4]
-[swp5]
```

What happens if you add `--step_threshold 0.00002` to your command-line invocation?

diegotxegp, 2020-07-20:
This is the result of one of the nodes when LS1021ATSN is master and the ptp4l.service is enable.

ptp4l[14277.479]: selected /dev/ptp0 as PTP clock
ptp4l[14277.509]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[14277.509]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[14278.512]: port 1: rogue peer delay response
ptp4l[14278.512]: port 1: LISTENING to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
phc2sys
phc2sys[14288.515]: reconfiguring after port state change
phc2sys[14288.516]: selecting enp3s0 for synchronization
phc2sys[14288.516]: selecting CLOCK_REALTIME as the master clock
phc2sys[14288.516]: enp3s0 sys offset -2425579431045313 s0 freq      +0 delay   5360
phc2sys[14289.516]: enp3s0 sys offset -2425579431040912 s1 freq   +4400 delay   5360
phc2sys[14290.516]: enp3s0 sys offset     -4857 s2 freq    -457 delay   5264
phc2sys[14291.516]: enp3s0 sys offset       -28 s2 freq   +2915 delay   5312
---> Ve al otro NODO y ejecuta: 'sudo tuned -d' y 'sudo ./ptp4l-phc2sys.sh'
nodo2@nodo2-N-A:~/Experimentos-Junio-2020$ phc2sys[14292.517]: enp3s0 sys offset      1424 s2 freq   +4359 delay   5296
phc2sys[14293.517]: enp3s0 sys offset      1512 s2 freq   +4874 delay   5296
phc2sys[14294.517]: enp3s0 sys offset      1058 s2 freq   +4873 delay   5312
ptp4l[14294.584]: port 1: FAULTY to LISTENING on INIT_COMPLETE
phc2sys[14295.517]: enp3s0 sys offset       600 s2 freq   +4733 delay   5312
phc2sys[14296.518]: enp3s0 sys offset       303 s2 freq   +4616 delay   5264
phc2sys[14297.518]: enp3s0 sys offset        91 s2 freq   +4495 delay   5280
ptp4l[14298.053]: port 1: new foreign master 00049f.fffe.ef0808-2
phc2sys[14298.518]: enp3s0 sys offset        48 s2 freq   +4479 delay   5264
ptp4l[14298.539]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[14298.539]: selected local clock aabbcc.fffe.00094a as best master
ptp4l[14298.539]: assuming the grand master role
phc2sys[14299.518]: port aabbcc.fffe.00094a-1 changed state
phc2sys[14299.518]: reconfiguring after port state change
phc2sys[14299.519]: selecting enp3s0 for synchronization
phc2sys[14299.519]: selecting CLOCK_REALTIME as the master clock
phc2sys[14299.519]: enp3s0 sys offset       -39 s0 freq   +4479 delay   5280
ptp4l[14300.054]: selected best master clock 00049f.fffe.ef0808
ptp4l[14300.054]: port 1: MASTER to UNCALIBRATED on RS_SLAVE
phc2sys[14300.519]: port aabbcc.fffe.00094a-1 changed state
phc2sys[14300.519]: reconfiguring after port state change
phc2sys[14300.519]: master clock not ready, waiting...
ptp4l[14300.921]: port 1: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
phc2sys[14301.519]: port aabbcc.fffe.00094a-1 changed state
phc2sys[14301.520]: reconfiguring after port state change
phc2sys[14301.520]: selecting CLOCK_REALTIME for synchronization
phc2sys[14301.520]: selecting enp3s0 as the master clock
phc2sys[14301.520]: CLOCK_REALTIME phc offset 14152709052929264 s0 freq      +0 delay   4944
ptp4l[14301.548]: rms 7076354526472290 max 14152709052945848 freq -17768 +/- 8024 delay   234 +/-   0
phc2sys[14302.520]: CLOCK_REALTIME phc offset 14152709052919672 s1 freq   -9591 delay   4908
ptp4l[14302.550]: rms 9027 max 9859 freq  -3024 +/- 3508 delay   233 +/-   0
phc2sys[14303.520]: CLOCK_REALTIME phc offset      6419 s2 freq   -3172 delay   4908
ptp4l[14303.553]: rms 6184 max 8237 freq  +1277 +/- 441 delay   233 +/-   0
phc2sys[14304.520]: CLOCK_REALTIME phc offset      4076 s2 freq   -3589 delay   4944
ptp4l[14304.555]: rms 1599 max 2926 freq  -1768 +/- 1197 delay   231 +/-   0
phc2sys[14305.521]: CLOCK_REALTIME phc offset     -2131 s2 freq   -8573 delay   4896
ptp4l[14305.558]: rms 1859 max 2913 freq  -6113 +/- 1487 delay   231 +/-   0
phc2sys[14306.521]: CLOCK_REALTIME phc offset     -7894 s2 freq  -14975 delay   4920
ptp4l[14306.560]: rms 2716 max 3302 freq  -9968 +/- 708 delay   232 +/-   0
phc2sys[14307.521]: CLOCK_REALTIME phc offset     -4592 s2 freq  -14042 delay   4932
ptp4l[14307.562]: rms  650 max  883 freq  -6720 +/- 712 delay   232 +/-   0
phc2sys[14308.521]: CLOCK_REALTIME phc offset     -1679 s2 freq  -12506 delay   4896
ptp4l[14308.565]: rms 1081 max 2207 freq  -7315 +/- 1512 delay   232 +/-   0
phc2sys[14309.521]: CLOCK_REALTIME phc offset     -5692 s2 freq  -17023 delay   4896
ptp4l[14309.567]: rms 2683 max 3435 freq -12355 +/- 866 delay   233 +/-   0
phc2sys[14310.522]: CLOCK_REALTIME phc offset     -2571 s2 freq  -15610 delay   4920
ptp4l[14310.570]: rms  832 max 1084 freq  -8827 +/- 838 delay   233 +/-   0
phc2sys[14311.522]: CLOCK_REALTIME phc offset       837 s2 freq  -12973 delay   4920
ptp4l[14311.573]: rms  988 max 1405 freq  -7523 +/- 559 delay   235 +/-   0
phc2sys[14312.523]: CLOCK_REALTIME phc offset      3701 s2 freq   -9858 delay   4920
ptp4l[14312.574]: rms 1589 max 1625 freq  -5285 +/- 515 delay   236 +/-   0
phc2sys[14313.523]: CLOCK_REALTIME phc offset      4973 s2 freq   -7475 delay   4908
ptp4l[14313.577]: rms 1425 max 1567 freq  -3921 +/- 376 delay   236 +/-   0
phc2sys[14314.523]: CLOCK_REALTIME phc offset      4949 s2 freq   -6008 delay   4896
[... 82 lines cut ...]
phc2sys[14355.541]: CLOCK_REALTIME phc offset      -184 s2 freq  -10372 delay   4944
ptp4l[14355.684]: rms   76 max  124 freq  -5815 +/- 104 delay   233 +/-   0
phc2sys[14356.541]: CLOCK_REALTIME phc offset      -367 s2 freq  -10610 delay   4920
ptp4l[14356.686]: rms  150 max  172 freq  -6086 +/-  65 delay   233 +/-   0
[... cut ...]
phc2sys[14364.543]: CLOCK_REALTIME phc offset      1744 s2 freq  -11155 delay   4909
ptp4l[14364.706]: rms 2262 max 4511 freq  -5069 +/- 2535 delay   237 +/-   0
phc2sys[14365.543]: CLOCK_REALTIME phc offset     10428 s2 freq   -1948 delay   4920
ptp4l[14365.708]: rms 4716 max 6193 freq  +2493 +/- 1955 delay   237 +/-   0
phc2sys[14366.543]: CLOCK_REALTIME phc offset      4038 s2 freq   -5209 delay   4908
ptp4l[14366.710]: rms 2281 max 3148 freq  -5648 +/- 1395 delay   237 +/-   0
phc2sys[14367.544]: CLOCK_REALTIME phc offset      -299 s2 freq   -8335 delay   4932

diegotxegp, 2020-07-20:
Here you are the results I said before. The LS1021ATSN platform is gmCapable (1) and ptp4l.service is enable. As you can see, rms is high.

nodo2@nodo2-N-A:~$ sudo ptp4l -i enp3s0 -f /home/*/linuxptp-2.0/configs/gPTP.c --tx_timestamp_timeout 20 -m
ptp4l[320.920]: selected /dev/ptp0 as PTP clock
ptp4l[320.957]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[320.957]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[324.165]: port 1: new foreign master 00049f.fffe.ef0808-2
ptp4l[324.186]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[324.188]: selected local clock aabbcc.fffe.00094a as best master
ptp4l[324.189]: assuming the grand master role
ptp4l[326.166]: selected best master clock 00049f.fffe.ef0808
ptp4l[326.167]: port 1: MASTER to UNCALIBRATED on RS_SLAVE
ptp4l[327.176]: port 1: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[327.928]: rms 7610 max 10822 freq  +5907 +/- 6878 delay   234 +/-   0
ptp4l[328.930]: rms 4496 max 6007 freq  -4235 +/- 3408 delay   234 +/-   0
ptp4l[329.933]: rms 5175 max 6072 freq -11023 +/- 531 delay   234 +/-   0
ptp4l[330.935]: rms 1787 max 3051 freq  -9422 +/- 1107 delay   234 +/-   0
ptp4l[331.937]: rms  781 max 1040 freq  -6450 +/- 313 delay   235 +/-   0
ptp4l[332.940]: rms  285 max  377 freq  -6590 +/- 132 delay   236 +/-   0
ptp4l[333.942]: rms  617 max  893 freq  -5775 +/- 446 delay   236 +/-   0
ptp4l[334.944]: rms  746 max  960 freq  -4814 +/- 190 delay   236 +/-   0
ptp4l[335.947]: rms  109 max  137 freq  -5571 +/- 142 delay   236 +/-   0
ptp4l[336.949]: rms  450 max  774 freq  -6184 +/- 448 delay   236 +/-   0
ptp4l[337.951]: rms  953 max 1030 freq  -7715 +/- 379 delay   236 +/-   0
ptp4l[338.954]: rms  679 max  997 freq  -7971 +/- 536 delay   236 +/-   0
ptp4l[339.956]: rms  481 max  702 freq  -6583 +/- 258 delay   237 +/-   0
ptp4l[340.958]: rms  495 max  959 freq  -6741 +/- 678 delay   237 +/-   0
ptp4l[341.961]: rms 1623 max 2077 freq  -3890 +/- 954 delay   237 +/-   0
ptp4l[342.964]: rms 1483 max 2104 freq  -2879 +/- 1193 delay   237 +/-   0
ptp4l[343.966]: rms 1811 max 2141 freq  -7317 +/- 1020 delay   237 +/-   0
ptp4l[344.968]: rms 1558 max 2123 freq  -8803 +/- 459 delay   237 +/-   0
ptp4l[345.971]: rms  970 max 1680 freq  -6248 +/- 1115 delay   239 +/-   0
ptp4l[346.973]: rms 1385 max 1940 freq  -4270 +/- 773 delay   237 +/-   0
ptp4l[347.977]: rms 1532 max 2290 freq  -7913 +/- 1292 delay   237 +/-   0
ptp4l[348.982]: rms 1753 max 2428 freq  -9996 +/- 892 delay   238 +/-   0
ptp4l[349.983]: rms 1348 max 1889 freq  -6244 +/- 1060 delay   239 +/-   0
ptp4l[350.986]: rms 1492 max 1935 freq  -4362 +/- 361 delay   240 +/-   0
ptp4l[351.988]: rms  267 max  397 freq  -5875 +/- 360 delay   240 +/-   0
ptp4l[352.990]: rms  420 max  444 freq  -6633 +/- 111 delay   239 +/-   0
ptp4l[353.993]: rms 1698 max 3159 freq  -4695 +/- 2148 delay   239 +/-   0
ptp4l[354.996]: rms 2690 max 3897 freq  -1471 +/- 2492 delay   239 +/-   0
ptp4l[355.998]: rms 3114 max 3640 freq  -9355 +/- 962 delay   239 +/-   0
ptp4l[357.001]: rms 1597 max 2182 freq  -9749 +/- 118 delay   239 +/-   0
ptp4l[358.003]: rms  836 max 1124 freq  -8878 +/- 1084 delay   237 +/-   0
ptp4l[359.006]: rms 1657 max 1764 freq  -5404 +/- 660 delay   237 +/-   0
ptp4l[360.008]: rms 1064 max 1629 freq  -5066 +/- 753 delay   236 +/-   0
ptp4l[361.011]: rms  553 max  878 freq  -6780 +/- 456 delay   236 +/-   0
ptp4l[362.014]: rms  434 max  657 freq  -5875 +/- 484 delay   236 +/-   0
ptp4l[363.016]: rms  753 max  816 freq  -7552 +/- 298 delay   236 +/-   0
ptp4l[364.019]: rms  535 max  709 freq  -7286 +/- 692 delay   236 +/-   0
ptp4l[365.023]: rms  998 max 1078 freq  -5160 +/- 341 delay   236 +/-   0
ptp4l[366.025]: rms  561 max  879 freq  -5078 +/- 370 delay   236 +/-   0
ptp4l[367.028]: rms  814 max 1531 freq  -5000 +/- 1090 delay   235 +/-   0
ptp4l[368.031]: rms 1326 max 1955 freq  -2921 +/- 845 delay   236 +/-   0
ptp4l[369.034]: rms 1082 max 1257 freq  -5982 +/- 622 delay   235 +/-   0
ptp4l[370.037]: rms 1165 max 1218 freq  -7354 +/- 339 delay   235 +/-   0
ptp4l[371.040]: rms  737 max 1108 freq  -7641 +/- 395 delay   235 +/-   0
ptp4l[372.043]: rms  490 max  660 freq  -6178 +/- 368 delay   237 +/-   0
ptp4l[373.046]: rms  621 max  674 freq  -5334 +/- 125 delay   237 +/-   0
ptp4l[374.048]: rms  357 max  487 freq  -5195 +/-  36 delay   237 +/-   0
ptp4l[375.052]: rms  291 max  521 freq  -5721 +/- 388 delay   237 +/-   0
ptp4l[376.055]: rms  654 max 1082 freq  -5729 +/- 916 delay   237 +/-   0
ptp4l[377.058]: rms 1072 max 1474 freq  -4696 +/- 1321 delay   237 +/-   0
ptp4l[378.061]: rms 1439 max 1999 freq  -7943 +/- 429 delay   237 +/-   0
ptp4l[379.064]: rms  694 max 1248 freq  -7509 +/- 781 delay   237 +/-   0
ptp4l[380.066]: rms 1154 max 1509 freq  -9375 +/- 252 delay   237 +/-   0
ptp4l[381.069]: rms 1542 max 2717 freq  -6689 +/- 1864 delay   236 +/-   0
ptp4l[382.072]: rms 2601 max 3170 freq  -2280 +/- 352 delay   235 +/-   0
ptp4l[383.074]: rms  739 max 1327 freq  -3322 +/- 409 delay   235 +/-   0

Next message I will show you, the results disabling ptp4l.service and adding your comment to my command.

diegotxegp, 2020-07-20:
Here is the result with your comment added to my command.

ptp4l[234.075]: selected /dev/ptp0 as PTP clock
ptp4l[234.112]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[234.112]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[237.907]: port 1: new foreign master 00049f.fffe.ef0808-2
ptp4l[237.991]: port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[237.991]: selected local clock aabbcc.fffe.00094a as best master
ptp4l[237.991]: assuming the grand master role
ptp4l[239.907]: selected best master clock 00049f.fffe.ef0808
ptp4l[239.907]: port 1: MASTER to UNCALIBRATED on RS_SLAVE
ptp4l[240.418]: port 1: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[241.043]: rms 521030845502 max 1042061691457 freq -10405 +/- 4418 delay   231 +/-   0
ptp4l[242.045]: rms  917 max 1470 freq  -8541 +/- 1115 delay   231 +/-   0
ptp4l[243.047]: rms 1620 max 1720 freq  -5788 +/- 416 delay   231 +/-   0
ptp4l[244.048]: rms  670 max 1122 freq  -5957 +/- 231 delay   231 +/-   0
phc2sys
ptp4l[245.049]: rms  294 max  329 freq  -5993 +/-  61 delay   231 +/-   0
phc2sys[245.114]: reconfiguring after port state change
phc2sys[245.115]: selecting CLOCK_REALTIME for synchronization
phc2sys[245.115]: selecting enp3s0 as the master clock
phc2sys[245.115]: CLOCK_REALTIME phc offset 14168812892079905 s0 freq      +0 delay   4956
ptp4l[246.051]: rms  256 max  660 freq  -6407 +/- 345 delay   232 +/-   0
phc2sys[246.115]: CLOCK_REALTIME phc offset 14168812892068528 s1 freq  -11375 delay   4932
ptp4l[247.052]: rms 1452 max 1784 freq  -9038 +/- 675 delay   232 +/-   0
phc2sys[247.115]: CLOCK_REALTIME phc offset     -2659 s2 freq  -14034 delay   4908
ptp4l[248.053]: rms 1081 max 1671 freq  -6267 +/- 1037 delay   232 +/-   0
phc2sys[248.116]: CLOCK_REALTIME phc offset      -146 s2 freq  -12319 delay   4908
ptp4l[249.055]: rms 2428 max 3198 freq -11039 +/- 1769 delay   231 +/-   0
---> Ve al otro NODO y ejecuta: 'sudo tuned -d' y 'sudo ./ptp4l-phc2sys.sh'
nodo2@nodo2-N-A:~/Experimentos-Junio-2020$ phc2sys[249.116]: CLOCK_REALTIME phc offset     -3518 s2 freq  -15734 delay   4944
ptp4l[250.056]: rms  861 max 1657 freq  -9753 +/- 1146 delay   231 +/-   0
phc2sys[250.116]: CLOCK_REALTIME phc offset     -2834 s2 freq  -16106 delay   4956
ptp4l[251.058]: rms  886 max  967 freq  -7737 +/- 269 delay   233 +/-   0
phc2sys[251.116]: CLOCK_REALTIME phc offset       368 s2 freq  -13754 delay   4908
ptp4l[252.059]: rms 1171 max 1436 freq  -6303 +/- 463 delay   235 +/-   0
phc2sys[252.117]: CLOCK_REALTIME phc offset      2650 s2 freq  -11362 delay   4884
ptp4l[253.061]: rms 1750 max 1923 freq  -3957 +/- 516 delay   236 +/-   0
phc2sys[253.117]: CLOCK_REALTIME phc offset      4975 s2 freq   -8242 delay   4896
ptp4l[254.062]: rms  649 max  925 freq  -5948 +/- 860 delay   236 +/-   0
phc2sys[254.117]: CLOCK_REALTIME phc offset      2417 s2 freq   -9307 delay   4920
ptp4l[255.063]: rms  773 max  909 freq  -7295 +/-  93 delay   235 +/-   0
phc2sys[255.118]: CLOCK_REALTIME phc offset      -606 s2 freq  -11605 delay   4920
ptp4l[256.065]: rms  177 max  317 freq  -6639 +/- 247 delay   235 +/-   0
phc2sys[256.118]: CLOCK_REALTIME phc offset      -765 s2 freq  -11946 delay   4920
ptp4l[257.066]: rms  159 max  196 freq  -6294 +/-  23 delay   235 +/-   0
phc2sys[257.118]: CLOCK_REALTIME phc offset      -170 s2 freq  -11580 delay   4920
ptp4l[258.068]: rms  121 max  306 freq  -6443 +/- 167 delay   235 +/-   0
phc2sys[258.118]: CLOCK_REALTIME phc offset       -56 s2 freq  -11517 delay   4908
ptp4l[259.069]: rms  958 max 1208 freq  -4695 +/- 501 delay   235 +/-   0
phc2sys[259.118]: CLOCK_REALTIME phc offset      1586 s2 freq   -9892 delay   4908
ptp4l[260.070]: rms  663 max 1008 freq  -6147 +/- 784 delay   235 +/-   0
phc2sys[260.119]: CLOCK_REALTIME phc offset       302 s2 freq  -10700 delay   4908
ptp4l[261.072]: rms 3873 max 4884 freq  +1082 +/- 2255 delay   235 +/-   0
phc2sys[261.119]: CLOCK_REALTIME phc offset      6474 s2 freq   -4438 delay   4944
ptp4l[262.073]: rms 1582 max 2380 freq  -2635 +/- 2195 delay   234 +/-   0
phc2sys[262.119]: CLOCK_REALTIME phc offset      3893 s2 freq   -5077 delay   4944
ptp4l[263.075]: rms 2637 max 2776 freq  -7565 +/- 703 delay   234 +/-   0
phc2sys[263.119]: CLOCK_REALTIME phc offset     -3429 s2 freq  -11231 delay   4920
ptp4l[264.076]: rms 1436 max 2068 freq  -8026 +/- 185 delay   234 +/-   0
phc2sys[264.119]: CLOCK_REALTIME phc offset     -5348 s2 freq  -14178 delay   4944

diegotxegp, 2020-07-20:
And finally, here you are the results I get with ptp4l.service disabled at the LS1021ATSN platform and without tour comment you said to me to add. As you can see, the offsets are really low.

ptp4l[256.098]: rms    5 max   10 freq +14362 +/-   7 delay   241 +/-   0
phc2sys[257.093]: CLOCK_REALTIME phc offset         0 s2 freq   +9566 delay   4908
ptp4l[257.099]: rms    3 max    6 freq +14358 +/-   5 delay   241 +/-   0
phc2sys[258.093]: CLOCK_REALTIME phc offset        18 s2 freq   +9584 delay   4956
ptp4l[258.101]: rms    3 max    6 freq +14357 +/-   4 delay   241 +/-   0
phc2sys[259.093]: CLOCK_REALTIME phc offset       -22 s2 freq   +9549 delay   4920
ptp4l[259.102]: rms    4 max    6 freq +14360 +/-   5 delay   241 +/-   0
phc2sys[260.093]: CLOCK_REALTIME phc offset        32 s2 freq   +9597 delay   4944
ptp4l[260.104]: rms    3 max    6 freq +14358 +/-   5 delay   240 +/-   0
phc2sys[261.094]: CLOCK_REALTIME phc offset       -13 s2 freq   +9561 delay   4944
ptp4l[261.105]: rms    3 max    5 freq +14359 +/-   4 delay   240 +/-   0
phc2sys[262.094]: CLOCK_REALTIME phc offset       -11 s2 freq   +9559 delay   4896
ptp4l[262.106]: rms    4 max    8 freq +14358 +/-   6 delay   240 +/-   0
phc2sys[263.094]: CLOCK_REALTIME phc offset        -3 s2 freq   +9564 delay   4932
ptp4l[263.107]: rms    3 max    5 freq +14360 +/-   4 delay   241 +/-   0
phc2sys[264.094]: CLOCK_REALTIME phc offset        13 s2 freq   +9579 delay   4908
ptp4l[264.109]: rms    5 max   11 freq +14354 +/-   6 delay   240 +/-   0
phc2sys[265.094]: CLOCK_REALTIME phc offset       -57 s2 freq   +9513 delay   4908
ptp4l[265.110]: rms    3 max    6 freq +14358 +/-   4 delay   241 +/-   0
phc2sys[266.095]: CLOCK_REALTIME phc offset        15 s2 freq   +9568 delay   4932
ptp4l[266.111]: rms    5 max    8 freq +14358 +/-   6 delay   240 +/-   0
phc2sys[267.095]: CLOCK_REALTIME phc offset       -18 s2 freq   +9539 delay   4908
ptp4l[267.113]: rms    3 max    6 freq +14357 +/-   4 delay   241 +/-   0
phc2sys[268.095]: CLOCK_REALTIME phc offset        12 s2 freq   +9564 delay   4944
ptp4l[268.115]: rms    4 max    7 freq +14358 +/-   5 delay   240 +/-   0
phc2sys[269.095]: CLOCK_REALTIME phc offset       -21 s2 freq   +9535 delay   4908
ptp4l[269.116]: rms    3 max    6 freq +14358 +/-   4 delay   240 +/-   0
phc2sys[270.096]: CLOCK_REALTIME phc offset        23 s2 freq   +9572 delay   4944
ptp4l[270.117]: rms    4 max    7 freq +14357 +/-   5 delay   240 +/-   0
phc2sys[271.096]: CLOCK_REALTIME phc offset       -17 s2 freq   +9539 delay   4932
ptp4l[271.119]: rms    3 max    4 freq +14357 +/-   4 delay   240 +/-   0
phc2sys[272.096]: CLOCK_REALTIME phc offset       -17 s2 freq   +9534 delay   4944
ptp4l[272.120]: rms    3 max    6 freq +14358 +/-   4 delay   240 +/-   0
phc2sys[273.096]: CLOCK_REALTIME phc offset        17 s2 freq   +9563 delay   4920
ptp4l[273.121]: rms    3 max    5 freq +14358 +/-   4 delay   240 +/-   0
phc2sys[274.096]: CLOCK_REALTIME phc offset        -5 s2 freq   +9546 delay   4932
ptp4l[274.123]: rms    5 max   10 freq +14355 +/-   6 delay   240 +/-   0
phc2sys[275.097]: CLOCK_REALTIME phc offset        -7 s2 freq   +9543 delay   4944
ptp4l[275.124]: rms    3 max    5 freq +14357 +/-   4 delay   240 +/-   0
phc2sys[276.097]: CLOCK_REALTIME phc offset        29 s2 freq   +9576 delay   4896
ptp4l[276.125]: rms    3 max    6 freq +14356 +/-   5 delay   240 +/-   0
phc2sys[277.097]: CLOCK_REALTIME phc offset       -63 s2 freq   +9493 delay   4908
ptp4l[277.127]: rms    4 max    7 freq +14355 +/-   5 delay   240 +/-   0
phc2sys[278.097]: CLOCK_REALTIME phc offset         2 s2 freq   +9539 delay   4944
ptp4l[278.128]: rms    3 max    7 freq +14356 +/-   5 delay   240 +/-   0
phc2sys[279.097]: CLOCK_REALTIME phc offset        12 s2 freq   +9550 delay   4932
ptp4l[279.130]: rms    4 max    6 freq +14358 +/-   5 delay   240 +/-   0
phc2sys[280.098]: CLOCK_REALTIME phc offset       -15 s2 freq   +9526 delay   4956
ptp4l[280.131]: rms    3 max    6 freq +14356 +/-   5 delay   240 +/-   0
phc2sys[281.098]: CLOCK_REALTIME phc offset        -1 s2 freq   +9536 delay   4944
ptp4l[281.133]: rms    2 max    4 freq +14356 +/-   3 delay   241 +/-   0
phc2sys[282.098]: CLOCK_REALTIME phc offset       -22 s2 freq   +9515 delay   4920
ptp4l[282.134]: rms    4 max    7 freq +14355 +/-   5 delay   240 +/-   0
phc2sys[283.098]: CLOCK_REALTIME phc offset         6 s2 freq   +9536 delay   4955

vladimiroltean, 2020-07-20:
I don't know. The things I would check for are:
- is the kernel the same?
- is the ptp4l binary the same?
- are the config options in each respective case the same?
- are there ever 2 ptp4l instances running at the same time? (hint: `pidof ptp4l`)
- is phc2sys running on the LS1021A-TSN in either of those cases?
- don't these large offsets, in fact, occur under certain circumstances with the command-line invocation of ptp4l too?
- is ptp4l.service enabled at boot time or not? Does it make a difference if ptp4l is started early during boot, or late through command line? Does the order (first the systemd service, then the command line) matter?
- what time (`phc_ctl /dev/ptp1 get`) is the LS1021A-TSN advertising into the network, as GM, in both cases? Is it comparable?
I tested on my LS1021A-TSN and it makes no difference whether ptp4l is invoked through systemd or not. The difference is somewhere else.
If you don't figure it out you can also share the

diegotxegp, 2021-02-19:
Thank you so much for your help, Vladimir.

closed 2021-02-19 by diegotxegp