nxp-archive/openil #64: LS1028ARDB: how can use enetc-2 in u-boot for the tftpload

razr, 2020-07-22:
What do I need to be able to tftload from the enetc-2 on LS1028ARDB?
I do:

=> setenv ethact enetc-2
=> printenv
baudrate=115200
bootcmd=setenv bootargs root=/dev/mmcblk0p2 rootwait rw earlycon=uart8250,0x21c0500 console=ttyS0,115200 cma=256M video=1920x1080-32@60;mmcinfo;fatload mmc 0:1 ${dp_load} ${dp_file}; hd}
bootdelay=3
bootfile=Image
dp_file=ls1028a-dp-fw.bin
dp_load=0x90000000
dp_offset=0x2000
eth4addr=fa:97:d4:84:a2:99
eth5addr=fa:97:d4:84:a2:99
eth6addr=fa:97:d4:84:a2:99
eth7addr=fa:97:d4:84:a2:99
eth8addr=00:80:c8:3b:61:68
ethact=enetc-2
fdtaddr=0xb0000000
fdtcontroladdr=fbb31f08
fdtfile=fsl-ls1028a-rdb.dtb
loadaddr=0xa0000000
stderr=serial
stdin=serial
stdout=serial

Environment size: 713/8188 bytes
=> dhcp
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7

Abort
=>

vladimiroltean, 2020-07-22:
```
setenv ethact swp0
setenv ethprime swp0
```

razr, 2020-07-22:
yeap, it works. Thanks much

closed 2020-07-22 by razr