nxp-archive/openil #65: does OpenIL provide a Dockerfile which can be used to build OpenIL image?

razr, 2020-07-25:
I have tried to use https://github.com/openil/openil/blob/master/support/docker/Dockerfile to build an LS1028ARDB image as
```bash
docker build -t buildroot/base support/docker
docker run  -v $HOME/github/openil/openil:/home/br-user/openil -t -i  buildroot/base bash
```
and then inside a container
```bash
br-user@19414238c25f:~$ make nxp_ls1028ardb-64b_ubuntu_full_defconfig
br-user@19414238c25f:~$ make 2>&1 | tee build.log
```

A build fails since some packages are missing e.g. automaker, autotools-dev etc, 
```bash
tail -n 100 build.log

>>> qoriq-libnfc-nci R2.4 Patching libtool
>>> qoriq-libnfc-nci R2.4 Configuring
cd /home/br-user/openil/output/build/qoriq-libnfc-nci-R2.4 && ./bootstrap && ac_cv_lbl_unaligned_fail=yes ac_cv_func_mmap_fixed_mapped=yes ac_cv_func_memcmp_working=yes ac_cv_have_decl_malloc=yes gl_cv_func_malloc_0_nonnull=yes ac_cv_func_malloc_0_nonnull=yes ac_cv_func_calloc_0_nonnull=yes ac_cv_func_realloc_0_nonnull=yes lt_cv_sys_lib_search_path_spec="" ac_cv_c_bigendian=no  PATH="/home/br-user/openil/output/host/bin:/home/br-user/openil/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" AR="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-ar" AS="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-as" LD="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-ld" NM="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-nm" CC="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-gcc" GCC="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-gcc" CPP="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-cpp" CXX="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-g++" FC="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-gfortran" F77="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-gfortran" RANLIB="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-ranlib" READELF="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-readelf" STRIP="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-strip" OBJCOPY="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-objcopy" OBJDUMP="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-objdump" AR_FOR_BUILD="/usr/bin/ar" AS_FOR_BUILD="/usr/bin/as" CC_FOR_BUILD="/usr/bin/gcc" GCC_FOR_BUILD="/usr/bin/gcc" CXX_FOR_BUILD="/usr/bin/g++" LD_FOR_BUILD="/usr/bin/ld" CPPFLAGS_FOR_BUILD="-I/home/br-user/openil/output/host/include" CFLAGS_FOR_BUILD="-O2 -I/home/br-user/openil/output/host/include" CXXFLAGS_FOR_BUILD="-O2 -I/home/br-user/openil/output/host/include" LDFLAGS_FOR_BUILD="-L/home/br-user/openil/output/host/lib -Wl,-rpath,/home/br-user/openil/output/host/lib" FCFLAGS_FOR_BUILD="" DEFAULT_ASSEMBLER="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-as" DEFAULT_LINKER="/home/br-user/openil/output/host/bin/aarch64-linux-gnu-ld" CPPFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  " CXXFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  " LDFLAGS="" FCFLAGS=" -Os " FFLAGS=" -Os " PKG_CONFIG="/home/br-user/openil/output/host/bin/pkg-config" STAGING_DIR="/home/br-user/openil/output/host/aarch64-buildroot-linux-gnu/sysroot" INTLTOOL_PERL=/usr/bin/perl PATH="/home/br-user/openil/output/host/bin:/home/br-user/openil/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" ./configure LDFLAGS=-static --host=aarch64-linux-gnu
./bootstrap: 4: ./bootstrap: aclocal: not found
./bootstrap: 5: ./bootstrap: autoheader: not found
./bootstrap: 6: ./bootstrap: libtoolize: not found
./bootstrap: 7: ./bootstrap: automake: not found
./bootstrap: 8: ./bootstrap: autoreconf: not found
/bin/bash: ./configure: No such file or directory
package/pkg-generic.mk:254: recipe for target '/home/br-user/openil/output/build/qoriq-libnfc-nci-R2.4/.stamp_configured' failed
make: *** [/home/br-user/openil/output/build/qoriq-libnfc-nci-R2.4/.stamp_configured] Error 127
```

I refer to the OpenIL 1.8 user manual and wondering whether OpenIL provides Ubuntu docker containers for the build?
 
1.3.2 Host system requirements
OpenIL is designed to build in Linux systems. The following host environments have been verified to build the OpenIL.
• Ubuntu 16.10
• Ubuntu 16.04
• Ubuntu 14.04
• Ubuntu 18.04
While OpenIL itself builds most host packages it needs for the compilation, certain standard Linux utilities are expected to be
already installed on the host system. The following tables provide an overview of the mandatory and optional packages.

vladimiroltean, 2020-07-28:
No, OpenIL does not provide an official Dockerfile. If you have one that works for you maybe we can pick that up if you would like to.

razr, 2020-07-29:
yes, it would be great if OpenIL can provide a Dockerfile. I have one, still WIP, see attached, [Dockerfile.zip](https://github.com/openil/openil/files/4994985/Dockerfile.zip)
```
docker build -t buildroot/base support/docker
docker run  -v $HOME/github/openil/openil:/home/br-user/openil -v /etc/localtime:/etc/localtime:ro -v /etc/timezone:/etc/timezone:ro -h openil --privileged -t -i buildroot/base bash

br-user@openil:~/openil$ sudo update-binfmts --enable qemu-aarch64
br-user@openil:~$ make nxp_ls1028ardb-64b_ubuntu_full_defconfig
br-user@openil:~$ make 2>&1 | tee build.log
```

it builds packages for my LS1028A just fine, however, I have a problem with a rootfs. it fails at

```bash
>>>   Generating root filesystems common tables
rm -rf /home/br-user/openil/output/build/buildroot-fs
mkdir -p /home/br-user/openil/output/build/buildroot-fs
printf '        sshd -1 sshd -1 * /var/empty - - SSH drop priv user\n\n' >> /home/br-user/openil/output/build/buildroot-fs/full_users_table.txt
printf '        /var/empty d 755 root root - - - - -\n\n' > /home/br-user/openil/output/build/buildroot-fs/full_devices_table.txt
cat system/device_table.txt >> /home/br-user/openil/output/build/buildroot-fs/full_devices_table.txt
>>>   Generating filesystem image rootfs.ext2
mkdir -p /home/br-user/openil/output/images
rm -rf /home/br-user/openil/output/build/buildroot-fs/ext2
mkdir -p /home/br-user/openil/output/build/buildroot-fs/ext2
rsync -auH --exclude=/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM /home/br-user/openil/output/target/ /home/br-user/openil/output/build/buildroot-fs/ext2/target
echo '#!/bin/sh' > /home/br-user/openil/output/build/buildroot-fs/ext2/fakeroot
echo "set -e" >> /home/br-user/openil/output/build/buildroot-fs/ext2/fakeroot
echo "chown -h -R 0:0 /home/br-user/openil/output/build/buildroot-fs/ext2/target" >> /home/br-user/openil/output/build/buildroot-fs/ext2/fakeroot
PATH="/home/br-user/openil/output/host/bin:/home/br-user/openil/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /home/br-user/openil/support/scripts/mkusers /home/br-user/openil/output/build/buildroot-fs/full_users_table.txt /home/br-user/openil/output/build/buildroot-fs/ext2/target >> /home/br-user/openil/output/build/buildroot-fs/ext2/fakeroot
mkusers: user 'sshd' already exists with group 'nogroup' (wants 'sshd')
fs/ext2/ext2.mk:46: recipe for target '/home/br-user/openil/output/images/rootfs.ext2' failed
make: *** [/home/br-user/openil/output/images/rootfs.ext2] Error 1
```
I do not understand why it tries to create an sshd entry here. This file is empty if I build it natively.
```
printf '        sshd -1 sshd -1 * /var/empty - - SSH drop priv user\n\n' >> /home/br-user/openil/output/build/buildroot-fs/full_users_table.txt
```
This is because of I have no sshd user in the docker container?

vladimiroltean, 2020-07-29:
No, I think you are just running into a build system bug with the ubuntu skeleton.
Do you have this patch?
https://github.com/openil/openil/commit/5cb02af4ee68c522f2f2d7c579a451fd46a9f5e3
For the Ubuntu rootfs, I would suggest you to build the master branch rather than the release tag.

razr, 2020-07-29:
If I comment out this line, the build finishes without errors.
```
fs/common.mk:70:        $(call PRINTF,$(PACKAGES_USERS)) >> $(ROOTFS_FULL_USERS_TABLE)
```
I'm on the master branch, but I didn't pull the latest changes. Thanks for pointing me out to it. I'll test it later today.

razr, 2020-07-29:
it does not fail at ```sshd``` anymore but still does not work.
```
mke2fs 1.45.5 (07-Jan-2020)
Creating regular file /home/br-user/openil/output/images/rootfs.ext2
64-bit filesystem support is not enabled.  The larger fields afforded by this feature enable full-strength checksumming.  Pass -O 64bit to rectify.
Creating filesystem with 368170 4k blocks and 92160 inodes
Filesystem UUID: 660ebca8-116e-4bab-bdf8-836acee6fdab
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912

Allocating group tables: done
Writing inode tables: done
Creating journal (8192 blocks): done
Copying files into the device: __populate_fs: Could not allocate block in ext2 filesystem while writing file "libuv1:arm64.shlibs"
mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system
*** Maybe you need to increase the filesystem size (BR2_TARGET_ROOTFS_EXT2_SIZE)
fs/ext2/ext2.mk:48: recipe for target '/home/br-user/openil/output/images/rootfs.ext2' failed
make: *** [/home/br-user/openil/output/images/rootfs.ext2] Error 1

br-user@openil:~/openil$ grep BR2_TARGET_ROOTFS_EXT2_SIZE .config
BR2_TARGET_ROOTFS_EXT2_SIZE=""

br-user@openil:~/openil$ ls output/images/
bl2_sd.pbl  fip.bin  fsl-ls1028a-rdb.dtb  Image  ls1028a-dp-fw.bin  rcw_1300_sdboot.bin  rootfs.ext2  u-boot-dtb.bin  uboot-env.bin
```

vladimiroltean, 2020-07-29:
What config are you using? BR2_TARGET_ROOTFS_EXT2_SIZE is set to 2G in configs/nxp_ls1028ardb-64b_ubuntu_full_defconfig.

razr, 2020-07-29:
```
br-user@openil:~/openil$ rm .config
br-user@openil:~/openil$ make nxp_ls1028ardb-64b_ubuntu_full_defconfig
#
# configuration written to /home/br-user/openil/.config
#
br-user@openil:~/openil$ grep BR2_TARGET_ROOTFS_EXT2_SIZE .config
BR2_TARGET_ROOTFS_EXT2_SIZE=""
```

vladimiroltean, 2020-07-29:
```
make nxp_ls1028ardb-64b_ubuntu_full_defconfig O=output-ls1028ardb-ubuntu-full/                                                                                        
mkdir -p /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/lxdialog                                                                                                                   
PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \                                                                                                                                    
    obj=/opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config -C support/kconfig -f Makefile.br conf                                                                                      
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config -DCONFIG_=\"\"  -MM *.
c > /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/.depend 2>/dev/null || :                                                                                                        
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config -DCONFIG_=\"\"   -c co
nf.c -o /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/conf.o                                                                                                                      
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config -DCONFIG_=\"\"  -I. -c
 /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/zconf.tab.c -o /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/zconf.tab.o                                        
In file included from /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/zconf.tab.c:2484:                                                                                             
./util.c: In function ‘file_write_dep’:                                                                                                                                                              
./util.c:54:18: warning: ‘..config.tmp’ directive writing 12 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]                                                                     
   54 |  sprintf(buf, "%s..config.tmp", dir);                                                                                                                                                        
      |                  ^~~~~~~~~~~~                                                                                                                                                                
./util.c:54:2: note: ‘sprintf’ output between 13 and 4109 bytes into a destination of size 4097                                                                                                      
   54 |  sprintf(buf, "%s..config.tmp", dir);                                                                                                                                                        
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                         
In file included from /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/zconf.tab.c:2485:                                                                                             
./confdata.c: In function ‘conf_write’:                                                                                                                                                              
./confdata.c:778:23: warning: ‘.tmpconfig.’ directive writing 11 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]                                                                 
  778 |   sprintf(tmpname, "%s.tmpconfig.%d", dirname, (int)getpid());                                                                                                                               
      |                       ^~~~~~~~~~~                                                                                                                                                            
./confdata.c:778:3: note: ‘sprintf’ output between 13 and 4119 bytes into a destination of size 4097                                                                                                 
  778 |   sprintf(tmpname, "%s.tmpconfig.%d", dirname, (int)getpid());                                                                                                                               
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                
./confdata.c: In function ‘conf_write_autoconf’:                                                                                                                                                     
[... cut ...] oot-config/conf.o /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/zconf.tab.o  -o /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/conf
rm /opt/openil/output-ls1028ardb-ubuntu-full/build/buildroot-config/zconf.tab.c
  GEN     /opt/openil/output-ls1028ardb-ubuntu-full/Makefile
#
# configuration written to /opt/openil/output-ls1028ardb-ubuntu-full/.config
#
$ grep BR2_TARGET_ROOTFS_EXT2_SIZE output-ls1028ardb-ubuntu-full/.config
BR2_TARGET_ROOTFS_EXT2_SIZE="2G"
```

vladimiroltean, 2020-07-29:
I just moved to the latest master branch and I get the same behavior. I was on 619556cb2196 ("Support barematal on LX2160rdb REV2.") before. Let me see what changed.

razr, 2020-07-29:
thanks for looking into it, after I change it manually to "2G" it works.
```
br-user@openil:~/openil$ ls -la output/images/
total 3774132
drwxr-xr-x 2 br-user br-user       4096 Jul 29 22:55 .
drwxr-xr-x 6 br-user br-user       4096 Jul 29 22:55 ..
-rw-r--r-- 1 br-user br-user      82953 Jul 29 20:45 bl2_sd.pbl
-rw-r--r-- 1 br-user br-user  268435456 Jul 29 22:55 boot.vfat
-rw-r--r-- 1 br-user br-user     749212 Jul 29 20:45 fip.bin
-rw-r--r-- 1 br-user br-user      21061 Jul 29 20:44 fsl-ls1028a-rdb.dtb
-rw-r--r-- 1 br-user br-user   37595648 Jul 29 20:44 Image
-rw-r--r-- 1 br-user br-user     102464 Jul 29 20:29 ls1028a-dp-fw.bin
-rw-r--r-- 1 br-user br-user        236 Jul 29 20:44 rcw_1300_sdboot.bin
-rw-r--r-- 1 br-user br-user 2147483648 Jul 29 22:55 rootfs.ext2
lrwxrwxrwx 1 br-user br-user         11 Jul 29 22:55 rootfs.ext4 -> rootfs.ext2
-rw-r--r-- 1 br-user br-user 1204408320 Jul 29 22:55 rootfs.tar
-rw-r--r-- 1 br-user br-user 2483027968 Jul 29 22:55 sdcard.img
-rw-r--r-- 1 br-user br-user     707755 Jul 29 20:45 u-boot-dtb.bin
-rw-r----- 1 br-user br-user       8192 Jul 29 20:45 uboot-env.bin
```

vladimiroltean, 2020-07-29:
So the commit that changed this behavior is https://github.com/openil/openil/commit/df7776729e71f911ebbe2a9f5be40ca5ac9535af.

razr, 2020-07-29:
it is here
https://github.com/openil/openil/commit/df7776729e71f911ebbe2a9f5be40ca5ac9535af#diff-cffb8a37c9eae5f595095976094da161R28
it prints me this size, and apparently it is too small in my case.
```
dddd 1472680K
dddd 1472680K
mke2fs 1.45.5 (07-Jan-2020)
Creating regular file /home/br-user/openil/output/images/rootfs.ext2
64-bit filesystem support is not enabled.  The larger fields afforded by this feature enable full-strength checksumming.  Pass -O 64bit to rectify.
Creating filesystem with 368170 4k blocks and 92160 inodes
```
if I put "2G" it prints
```
dddd 1210304K
dddd 2G
mke2fs 1.45.5 (07-Jan-2020)
Creating regular file /home/br-user/openil/output/images/rootfs.ext2
64-bit filesystem support is not enabled.  The larger fields afforded by this feature enable full-strength checksumming.  Pass -O 64bit to rectify.
Creating filesystem with 524288 4k blocks and 131072 inodes
```

razr, 2020-07-30:
LS1028ARDB image boots fine, Is it a better way of getting an IP address over DHCP as
```
root@LS1028ARDB-Ubuntu:~# ifconfig eno2 up
root@LS1028ARDB-Ubuntu:~# ifconfig swp0 up
root@LS1028ARDB-Ubuntu:~# dhclient swp0
cmp: EOF on /tmp/tmp.666St3iTtM which is empty
root@LS1028ARDB-Ubuntu:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=115 time=8.35 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=115 time=8.11 ms
```
I'm not sure I understand why do I need to get eno2 first? If I do
```
root@LS1028ARDB-Ubuntu:~# ifconfig swp0 up
SIOCSIFFLAGS: Network is down
```
Is it any example of how to configure swp0 using netplan?

vladimiroltean, 2020-07-30:
I don't have a definitive answer on why you need to bring up eno2 first. It is the "DSA master" (aka management interface) of the Felix switch. Specifically, I don't know if there is any good reason for which the kernel cannot bring this interface up automatically.

I don't know about netplan but it can be done using systemd-networkd:
https://github.com/systemd/systemd/issues/7478#issuecomment-514562190
The feature is called "bind carrier". Maybe you need to raise an issue in whatever netplan's ticketing system is, and ask about DSA.

razr, 2020-07-31:
Thanks, let me look into it. 
NXP provides an example of how to configure interface using netplan here https://www.nxp.com/docs/en/user-guide/LSDKUG_Rev20.04_290520.pdf but only for the static IPs
```
4.1.11.6.5 Assign IP addresses to network interfaces
Static IP addresses can be assigned to network interfaces using the standard ifconfig or ip commands.
ifconfig <interface_name_in_Linux> <ip_address>
OR
ip address add <ip_address> dev <interface_name_in_linux>
Alternatively, Static IP addresses can also be assigned using netplan. Create a file called “config.yaml” in /etc/netplan. Using a
text editor, add the following lines to this config file and save it.
```

I still have a problem with a BR2_TARGET_ROOTFS_EXT2_SIZE.
I have built the image for the LS1046ARDB and it is the same behavior. should I open a separate issue for it?

```bash
Copying files into the device: __populate_fs: Could not allocate block in ext2 filesystem while writing file "libthai-data.list"
mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system
*** Maybe you need to increase the filesystem size (BR2_TARGET_ROOTFS_EXT2_SIZE)
fs/ext2/ext2.mk:48: recipe for target '/home/br-user/openil/output-ls1046ardb/images/rootfs.ext2' failed
make[1]: *** [/home/br-user/openil/output-ls1046ardb/images/rootfs.ext2] Error 1
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
```

As for the Dockerfile, do you want me to provide a pull request for it? There is a drawback with it since no one has fixed a ```sudo``` problem while creating a Ubuntu image and I still need to type the password twice. Other than that it works.

razr, 2020-08-10:
I have set _sudo_ for the user to _NOPASSWD_ in the Dockerfile, it works now without manual intervention.
```
RUN echo "${USER} ALL=(ALL) NOPASSWD:ALL" | sudo tee /etc/sudoers.d/${USER}
```

vladimiroltean, 2020-08-10:
We're working on an alternative way to grant the build user sudo privileges for the ubuntu rootfs builds, if you wait for a few more days it should appear in the master branch.

razr, 2020-08-10:
thanks for looking into it. 👍

vladimiroltean, 2020-08-17:
> As for the Dockerfile, do you want me to provide a pull request for it?

Since development also happens on an internal NXP git repository, the OpenIL github repo is not the central place for development but merely a mirror.
Could you please format your patches using `git format-patch` and `git send-email --to=dev@openil.org`?
https://openil.org/mailing.html
The process should be similar to the Linux kernel and any other project that uses mailing lists for development. You can read this for some rough guidelines:
https://www.kernel.org/doc/html/v4.17/process/submitting-patches.html
If you send your patch on the mailing list, it will be reviewed there and you'll get a notification when it is accepted. Then it will appear in Github after a resync.