nxp-archive/openil #66: LS1028ARDB: Problem on sleep command

JohnRama, 2020-07-29:
- Environment                              
  LS1028ARDB                               
                                           
- SW  
  - OpenIL                                 
  - master branch                          
                                           
- Problem                                  
  - How to duplicate                       
    1. run following script                
                                           
```
        while true;                        
        do                                 
            sleep 1;                       
            echo "hoge"                    
        done                  
```             
                                           
  - Problem                                
    - most of time, sleep does not return. 
      Why ??                               
                                           
    Would you please help on this ?        
    Thanks !!                              

John

vladimiroltean, 2020-07-29:
Question 1: could you please show the output of "cat /proc/timer_list"?
Question 2: could you please enable CONFIG_DEBUG_PREEMPT=y in the kernel and test again?

JohnRama, 2020-07-29:
Thanks for the quick reply.

Sorry, after rebooting LS1028ARDB,this problem never happened again...

>Question 1: could you please show the output of "cat /proc/timer_list"?
>Question 2: could you please enable CONFIG_DEBUG_PREEMPT=y in the kernel and test again?

>Question 3: what did you do to reach this condition? It does not occur immediately after a fresh boot, at least.
After playing with linuxptp and testptp, the system falled into this condition.
After rebooting the system, this problem never happened.

I'll answer remaining question once I could face the same problem again.

John

JohnRama, 2020-08-04:
Since i can not duplicate this problem again, I will close this thread for this moment.
Will open again once I faced the same problem.

John

closed 2020-08-04 by JohnRama

vladimiroltean, 2020-08-04:
The problem was solved with this commit anyway: https://github.com/openil/linux/commit/fc6b1274d1f58a049a128d3a00f2b4eed57ded05