nxp-archive/openil #67: LS1046ARDB: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000

razr, 2020-08-07:
I boot from SDCard as
```
make nxp_ls1046ardb-64b_ubuntu_full_defconfig O=output-ls1046ardb
make O=output-ls1046ardb 2>&1 | tee output-ls1046ardb/build.log

sudo dd if=.sdcard.img of=/dev/sdb status=progress
```
And get this kernel error. It was working before the latest commits.
```
OpenIL-Ubuntu ubuntu-18.04.4
LS1046ARDB-Ubuntu login: [   17.987668] 003: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[   17.996900] 003: Mem abort info:
[   18.000124] 003:   ESR = 0x86000005
[   18.003609] 003:   EC = 0x21: IABT (current EL), IL = 32 bits
[   18.009355] 003:   SET = 0, FnV = 0
[   18.012840] 003:   EA = 0, S1PTW = 0
[   18.016411] 003: user pgtable: 4k pages, 48-bit VAs, pgdp=00000009e6cef000
[   18.023289] 003: [0000000000000000] pgd=00000009e4c23003
[   18.028599] 003: , pud=0000000000000000
[   18.032430] 003: 
[   18.034349] 003: Internal error: Oops: 86000005 [#1] PREEMPT_RT SMP
[   18.040618] 003: Modules linked in:
[   18.044103] 003:  ip6table_filter
[   18.047413] 003:  ip6_tables
[   18.050288] 003:  cpufreq_powersave
[   18.053771] 003:  cpufreq_conservative
[   18.057515] 003:  ina2xx
[   18.060042] 003:  lm90
[   18.062395] 003: 
[   18.064314] 003: CPU: 3 PID: 34 Comm: ksoftirqd/3 Tainted: G        W         5.4.3-rt1 #1
[   18.072584] 003: Hardware name: LS1046A RDB Board (DT)
[   18.077720] 003: pstate: 40000085 (nZcv daIf -PAN -UAO)
[   18.082946] 003: pc : 0x0
[   18.085563] 003: lr : call_timer_fn.isra.30+0x20/0x80
[   18.090618] 003: sp : ffff800010183c90
[   18.094363] 003: x29: ffff800010183c90 
[   18.098195] 003: x28: 0000000000000000 
[   18.102027] 003: 
[   18.103946] 003: x27: ffff0009739ec450 
[   18.107778] 003: x26: 0000000000000000 
[   18.111609] 003: 
[   18.113527] 003: x25: 0000000000000000 
[   18.117358] 003: x24: ffffc820255f9000 
[   18.121189] 003: 
[   18.123107] 003: x23: ffffc820255e9000 
[   18.126938] 003: x22: dead000000000122 
[   18.130769] 003: 
[   18.132686] 003: x21: 0000000000000000 
[   18.136518] 003: x20: 0000000000000000 
[   18.140349] 003: 
[   18.142267] 003: x19: ffff000971d93a00 
[   18.146099] 003: x18: 0000000000000000 
[   18.149930] 003: 
[   18.151848] 003: x17: 0000000000000000 
[   18.155679] 003: x16: 0000000000000000 
[   18.159510] 003: 
[   18.161428] 003: x15: 0000000000000000 
[   18.165260] 003: x14: 0000000000000000 
[   18.169091] 003: 
[   18.171008] 003: x13: 0000000000000001 
[   18.174840] 003: x12: 0000000000004000 
[   18.178671] 003: 
[   18.180589] 003: x11: 0000000000000000 
[   18.184420] 003: x10: ffff0009739ec4e8 
[   18.188251] 003: 
[   18.190169] 003: x9 : 0000000000000001 
[   18.194001] 003: x8 : 0008000000000000 
[   18.197832] 003: 
[   18.199750] 003: x7 : ffff800010183d28 
[   18.203582] 003: x6 : ffff0009739ec4a0 
[   18.207413] 003: 
[   18.209330] 003: x5 : 0004000000000000 
[   18.213162] 003: x4 : ffff800010183d28 
[   18.216993] 003: 
[   18.218910] 003: x3 : 0000000000000080 
[   18.222742] 003: x2 : 0000000100000000 
[   18.226574] 003: 
[   18.228492] 003: x1 : 0000000000000000 
[   18.232323] 003: x0 : ffff00095f4231f0 
[   18.236154] 003: 
[   18.238072] 003: Call trace:
[   18.240949] 003:  0x0
[   18.243216] 003:  run_timer_softirq+0x418/0x4c0
[   18.247746] 003:  __do_softirq+0x11c/0x218
[   18.251842] 003:  run_ksoftirqd+0x50/0x70
[   18.255851] 003:  smpboot_thread_fn+0x214/0x310
[   18.260381] 003:  kthread+0x118/0x120
[   18.264041] 003:  ret_from_fork+0x10/0x18
[   18.268053] 003: Code: bad PC value
[   18.271538] 003: ---[ end trace 0000000000000004 ]---

OpenIL-Ubuntu ubuntu-18.04.4
[   39.403624] 003: rcu: INFO: rcu_preempt self-detected stall on CPU
[   39.409808] 003: rcu: 	3-...!: (5250 ticks this GP) idle=6ba/0/0x1 softirq=0/0 fqs=0 
[   39.417640] 003: 	(t=5250 jiffies g=2529 q=588)
[   39.422166] 003: rcu: rcu_preempt kthread starved for 5250 jiffies! g2529 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=3
[   39.432953] 003: rcu: RCU grace-period kthread stack dump:
[   39.438434] 003: rcu_preempt     I
[   39.441829] 003:     0    11      2 0x00000028
[   39.446268] 003: Call trace:
[   39.449142] 003:  __switch_to+0x198/0x220
[   39.453155] 003:  __schedule+0x288/0x690
[   39.457074] 003:  schedule+0x40/0xf0
[   39.460643] 003:  schedule_timeout+0x1c8/0x2d0
[   39.465083] 003:  rcu_gp_kthread+0x554/0x9a0
[   39.469349] 003:  kthread+0x118/0x120
[   39.473007] 003:  ret_from_fork+0x10/0x18
[   39.477014] 003: Task dump for CPU 3:
[   39.480669] 003: swapper/3       R
[   39.484064] 003:   running task    
[   39.487545] 003:     0     0      1 0x0000002a
[   39.491984] 003: Call trace:
[   39.494857] 003:  dump_backtrace+0x0/0x150
[   39.498949] 003:  show_stack+0x14/0x20
[   39.502692] 003:  sched_show_task+0x108/0x140
[   39.507046] 003:  dump_cpu_task+0x40/0x50
[   39.511051] 003:  rcu_dump_cpu_stacks+0xa4/0xe4
[   39.515577] 003:  rcu_sched_clock_irq+0x764/0xae0
[   39.520277] 003:  update_process_times+0x2c/0x50
[   39.524890] 003:  tick_sched_handle.isra.16+0x30/0x50
[   39.529938] 003:  tick_sched_timer+0x48/0xa0
[   39.534203] 003:  __hrtimer_run_queues+0x11c/0x1f0
[   39.538989] 003:  hrtimer_interrupt+0xd8/0x250
[   39.543428] 003:  arch_timer_handler_phys+0x28/0x40
[   39.548304] 003:  handle_percpu_devid_irq+0x80/0x140
[   39.553265] 003:  generic_handle_irq+0x24/0x40
[   39.557704] 003:  __handle_domain_irq+0x5c/0xb0
[   39.562229] 003:  gic_handle_irq+0x54/0xa0
[   39.566319] 003:  el1_irq+0xb8/0x180
[   39.569888] 003:  cpuidle_enter_state+0x13c/0x340
[   39.574589] 003:  cpuidle_enter+0x38/0x50
[   39.578594] 003:  call_cpuidle+0x2c/0x50
[   39.582512] 003:  do_idle+0x260/0x2c0
[   39.586167] 003:  cpu_startup_entry+0x24/0x40
[   39.590518] 003:  secondary_start_kernel+0x1b4/0x210
```
Here is the full boot log: [ls1046-boot.txt](https://github.com/openil/openil/files/5041216/ls1046-boot.txt)

vladimiroltean, 2020-08-07:
Hi, sorry for the breakage.
Could you please set your git remote for the kernel to this branch?
https://github.com/vladimiroltean/ls1028ardb-linux/tree/ethtool-consolidate-irq-coalescing
It is based on the linux-5.4.y branch of OpenIL, plus a few more backported patches that should fix this breakage in the igb driver.
If all is well, the patches should get merged very soon in OpenIL.

razr, 2020-08-07:
I didn't realize it was my i210 card :). yes, it boots fine now, **thank you**. I haven't done any further tests.
The assembly of Ubuntu rootfs is so slow even on my Xeon 32 cores machine, I'm wondering whether is not better to do here
[post-custom-skeleton-ubuntu-base-18.04.sh#L84](https://github.com/openil/openil/blob/master/board/nxp/common/ls/post-custom-skeleton-ubuntu-base-18.04.sh#L84)
```
debootstrap --components=main,restricted,universe,multiverse --include=<all extra packages from ubuntu-additional_packages_list>...
```
instead of adding them one by one later.

There is another glitch
```
>>>   Generating filesystem image rootfs.ext2
mkdir -p /home/br-user/openil/output-ls1046/images
rm -rf /home/br-user/openil/output-ls1046/build/buildroot-fs/ext2
mkdir -p /home/br-user/openil/output-ls1046/build/buildroot-fs/ext2
rsync -auH --exclude=/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM /home/br-user/openil/output-ls1046/target/ /home/br-user/openil/output-ls1046/build/buildroot-fs/ext2/target
rsync: opendir "/home/br-user/openil/output-ls1046/target/var/lib/snapd/void" failed: Permission denied (13)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
fs/ext2/ext2.mk:54: recipe for target '/home/br-user/openil/output-ls1046/images/rootfs.ext2' failed
make[1]: *** [/home/br-user/openil/output-ls1046/images/rootfs.ext2] Error 23
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
```

/home/br-user/openil/output-ls1046/target/var/lib/snapd/void has read-only permissions, I need to do
chmod 755 on it

vladimiroltean, 2020-08-07:
> `--include=<all extra packages from ubuntu-additional_packages_list>...`

I don't know, have you tried that and is it faster? If it is, I suppose we could.

> Is it the expectation that someone would compile natively on LS1046ARDB?

With the ubuntu-full image, yes, that is possible. If you don't want that maybe you should go with the plain ubuntu rootfs, or even with the plain buildroot rootfs.

razr, 2020-08-10:
yes, configs/nxp_ls1046ardb-64b_ubuntu_defconfig builds much faster, there are still some dev packages included. The full build takes around 49 mins on my machine.
I didn't measure time for the nxp_ls1046ardb-64b_ubuntu_full_defconfig, I think it takes a couple of hours, where most of the time it installs package by packages in the _for_ loops.