nxp-archive/openil #69: LS1028ARDB: Q on TFTP error: 'Undefined error code' (0)

JohnRama, 2020-08-07:
- Environment                                                                    
	- LS1028ARDB                                                                   
		- SD boot                                                                    
- SW                                                                             
  - OpenIL                                                                       
  	- master branch                                                              
                                                                                 
- Question                                                                       
	- I would like to update the kernel in the SD card from u-boot.
          I just followed the instruction of "3.8.3 Updating target images" OpenIL User Manual, 
          However got following error. 
          Would someone please tell me how you guys update the kernel from u-boot ?                                                             
                                                                                 
- What I tried                                                                   
		- Can not download Image                                                     

> 			=> tftpboot 83000000 openil/Image                                          
> 				Using swp0 device                                                        
> 				TFTP from server 192.168.99.1; our IP address is 192.168.99.46           
> 				Filename 'openil/Image'.                                                 
> 				Load address: 0x83000000                                                 
> 				Loading: *                                                               
> 				TFTP error: 'Undefined error code' (0)                                   
> 				Starting again

	- can download other binaries.    

> 			=> tftpboot 82000000 openil/fip.bin                                        
> 			hoge                                                                       
> 			Using swp0 device                                                          
> 			TFTP from server 192.168.99.1; our IP address is 192.168.99.46             
> 			Filename 'openil/fip.bin'.                                                 
> 			Load address: 0x82000000                                                   
> 			Loading: ################################################################# 
> 							 ################################################################# 
> 							 #################                                                 
> 							 563.5 KiB/s                                                       
> 			done                                                                       
> 			Bytes transferred = 749236 (b6eb4 hex)         

                            
                                                                                 
Thanks !!                                                                      
John

vladimiroltean, 2020-08-07:
Use another address for storing the image in DDR, for example 0xa0000000 or higher.
```
setenv autoload off
setenv serverip 192.168.1.1
setenv sd_tftp_boot 'setenv bootargs root=/dev/mmcblk0p2 rootwait rw earlycon=uart8250,0x21c0500 console=ttyS0,115200 cma=256M video=1920x1080-32@60; mmcinfo; fatload mmc 0:1 0x98000000 ls1028a-dp-fw.bin; hdp load 0x98000000 0x2000; dhcp && tftp a0000000 ls1028/fsl-ls1028a-rdb.dtb;tftp a1000000 ls1028/uImage;run qspi_hdploadcmd; bootm a1000000 - a0000000'
```

JohnRama, 2020-08-10:
Thanks, however I still am having the difficulty. 

I tried to download Image to 0xa0000000 and have the same error.
So, i guess I need to compress Image with gzip. 

Now Question is how can I generate Image.gz  with linux build system ?

When trying to change the config of linux kernel, I can not set KERNEL_GZIP. 
I guess this is  because it has the dependency to HAVE_KERNEL_GZIP, and HAVE_KERNEL_GZIP is not select via menuconfig.
So, I manually add following line to output/build/linux-OpenIL-v1.8-linux-202005/.config. 
CONFIG_HAVE_KERNEL_GZIP=y
Then, make linux-menuconfig from openil top directory. 
However HAVE_KERNEL_GZIP is not enabled...

Probably something wrong with my environment... but really appreciated any help here..

vladimiroltean, 2020-08-10:
> So, i guess I need to compress Image with gzip.

No, that's definitely not 'required'.
I could transfer an Image file using your command precisely.

```
NOTICE:  Fixed DDR on board

NOTICE:  4 GB DDR4, 32-bit, CL=11, ECC on
NOTICE:  BL2: v1.5(release):OpenIL-v1.8-202005-33-gfec4b355b72e
NOTICE:  BL2: Built : 00:55:32, May 28 2020
NOTICE:  BL31: v1.5(release):OpenIL-v1.8-202005-33-gfec4b355b72e
NOTICE:  BL31: Built : 00:55:32, May 28 2020
NOTICE:  Welcome to LS1028 BL31 Phase

U-Boot 2019.10 (May 28 2020 - 00:55:19 +0300)

SoC:  LS1028AE Rev1.0 (0x870b0010)
Clock Configuration:
       CPU0(A72):1300 MHz  CPU1(A72):1300 MHz
       Bus:      400  MHz  DDR:      1600 MT/s
Reset Configuration Word (RCW):
       00000000: 34004010 00000030 00000000 00000000
       00000010: 00000000 00b90000 0030c000 00000000
       00000020: 01e03150 00002580 00000000 00003296
       00000030: 00000000 00000008 00000000 00000000
       00000040: 00000000 00000000 00000000 00000000
       00000050: 00000000 00000000 00000000 00000000
       00000060: 00000000 00000000 200e705a 00000000
       00000070: bb580000 00000000
Model: NXP Layerscape 1028a RDB Board
Board: LS1028AE Rev1.0-RDB, Version: A, boot from SD
FPGA: v8 (RDB)
SERDES1 Reference : Clock1 = 100.00MHz Clock2 = 100.00MHz
DRAM:  3.9 GiB
DDR    3.9 GiB (DDR4, 32-bit, CL=11, ECC on)
Using SERDES1 Protocol: 47960 (0xbb58)
PCIe0: pcie@3400000 Root Complex: no link
PCIe1: pcie@3500000 Root Complex: no link
WDT:   Started with servicing (60s timeout)
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
Loading Environment from MMC... OK
EEPROM: NXID v1
In:    serial
Out:   serial
Err:   serial
Net:   eth0: enetc-0, eth2: enetc-2, eth4: swp0
Warning: swp1 (eth5) using random MAC address - f6:ee:6d:3a:0f:91
, eth5: swp1
Warning: swp2 (eth6) using random MAC address - 06:da:22:1c:3f:eb
, eth6: swp2
Warning: swp3 (eth7) using random MAC address - 02:37:1d:05:10:ab
, eth7: swp3
Hit any key to stop autoboot:  0
=> dhcp
BOOTP broadcast 1
DHCP client bound to address 10.0.0.111 (3 ms)
=> tftpboot 83000000 ls1028/Image
Using enetc-0 device
TFTP from server 10.0.0.1; our IP address is 10.0.0.111
Filename 'ls1028/Image'.
Load address: 0x83000000
Loading: #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         ###################
         6 MiB/s
done
Bytes transferred = 36977152 (2343a00 hex)
=> 
```

> Now Question is how can I generate Image.gz with linux build system ?

Uhm, `make`? Isn't it already available under `arch/arm64/boot/Image.gz` for you?

vladimiroltean, 2020-08-10:
I see the error code, as absurd as it is (0) is given by the server?
https://github.com/openil/u-boot/blob/master/net/tftp.c#L565
What TFTP server are you using?

vladimiroltean, 2020-08-10:
Ok, so some googling reveals this U-Boot mailing list discussion:
http://u-boot.10912.n7.nabble.com/tftp-errors-times-out-on-large-files-td10204.html

> the TFTP protocol has a file size limit of 32M unless one of the later
> RFC versions is followed.

Since I was able to transfer a 35.2 MB file, it means that my tftp server supports these RFC extensions it seems. I am using xinetd as tftp server on ubuntu 20.04.

JohnRama, 2020-08-13:
Thanks Vladimiroltean and sorry for the late response.

I dug a little bit into detail on this.
I just want to share my findings.

> I see the error code, as absurd as it is (0) is given by the server?
> https://github.com/openil/u-boot/blob/master/net/tftp.c#L565
> What TFTP server are you using?
I'm using atftpd at debian "jessie".
As you said, this tftp server returns back to the client with the error message "Undefined error code"
when the file exceed 32 MB (512 bytes/block x 65535 blocks), expecting the client to increase block size, more than 512 bytes/block.
https://github.com/seveas/atftp/blob/e60ad90917c727b607975c4077b5b4e1bc85b316/tftpd_file.c#L627

I tried with tftpd-hpa, which resulted into the same behavior with atftpd.

I tried with tftpd, which work for the file more than 32MB.
However, checking tftp messages with wireshark revealed that this tftp server does not use larger block size.
It just keeps using 512 bytes block and send more than 65535 blocks (which is against tftp protocol).

John

JohnRama, 2020-08-14:
> I would like to update the kernel in the SD card from u-boot.

Here is what I have eventually arrived after a struggle. I just want to share my way.

```
setenv temp_addr 0xa1000000                     
fatrm mmc 0:1 $bootfile                         
tftp $loadaddr openil/Image.gz;                 
unzip $loadaddr $temp_addr                      
fatwrite mmc 0:1 $temp_addr $bootfile $filesize 
```

Note: 
1. Image.gz can be created by chaining "Kernel image target name" to "Image.gz" from "Image" in openil config.
2. fatwrite/fatrm is available by enable CONFIG_ENV_IS_IN_FAT in uboot.

Thanks.
I'll close this thread.

closed 2020-08-14 by JohnRama