nxp-archive/openil #70: rsync: opendir "/home/br-user/openil/output-ls1046/target/var/lib/snapd/void" failed: Permission denied (13)

razr, 2020-08-17:
I do 
```
make nxp_ls1046ardb-64b_ubuntu_full_defconfig O=output-ls1046ardb
time make O=output-ls1046ardb 2>&1 | tee output-ls1046ardb/build.log
```
I get
```
>>>   Generating filesystem image rootfs.ext2
mkdir -p /home/br-user/openil/output-ls1046/images
rm -rf /home/br-user/openil/output-ls1046/build/buildroot-fs/ext2
mkdir -p /home/br-user/openil/output-ls1046/build/buildroot-fs/ext2
rsync -auH --exclude=/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM /home/br-user/openil/output-ls1046/target/ /home/br-user/openil/output-ls1046/build/buildroot-fs/ext2/target
rsync: opendir "/home/br-user/openil/output-ls1046/target/var/lib/snapd/void" failed: Permission denied (13)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
fs/ext2/ext2.mk:73: recipe for target '/home/br-user/openil/output-ls1046/images/rootfs.ext2' failed
make[1]: *** [/home/br-user/openil/output-ls1046/images/rootfs.ext2] Error 23
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2

real    123m45.873s
user    244m54.591s
sys     28m43.277s
```

```
br-user@openil:~/openil$ ls -la /home/br-user/openil/output-ls1046/target/var/lib/snapd/
total 44
drwxr-xr-x 11 br-user br-user 4096 Aug 17 13:04 .
drwxr-xr-x 32 br-user br-user 4096 Aug 17 13:24 ..
drwxr-xr-x  3 br-user br-user 4096 Aug 17 13:04 apparmor
drwxr-xr-x  2 br-user br-user 4096 Jul 10 15:59 auto-import
drwxr-xr-x  3 br-user br-user 4096 Aug 17 13:04 desktop
drwxr-xr-x  2 br-user br-user 4096 Jul 10 15:59 environment
drwxr-xr-x  2 br-user br-user 4096 Jul 10 15:59 firstboot
drwxr-xr-x  6 br-user br-user 4096 Aug 17 13:04 lib
drwxr-xr-x  3 br-user br-user 4096 Aug 17 13:04 snaps
drwxr-xr-x  3 br-user br-user 4096 Aug 17 13:04 ssl
d--x--x--x  2 br-user br-user 4096 Jul 10 15:59 void
```
to fix
```
br-user@openil:~/openil$ sudo chmod 775 /home/br-user/openil/output-ls1046/target/var/lib/snapd/void
```

vladimiroltean, 2020-08-19:
Could you please check if the problem is solved by the latest commit on the master branch
https://github.com/openil/openil/commit/58332418ce380755e686b4fddb99c7f7e0e7e115

razr, 2020-08-24:
Thank you, I'll check it later on this week

razr, 2020-09-09:
it works now, but there is another problem here:
```bash
installing python-pip ...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python-pip

************
ubuntu-package-installer failed
  Please running the below commands before re-compiling:
    rm -rf /home/br-user/openil/output-ls1046/target
    make skeleton-custom-dirclean
  Or
    make skeleton-custom-dirclean O=<output dir>
Makefile:741: recipe for target 'target-finalize' failed
make[1]: *** [target-finalize] Error 1
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
```
There are several packages which do not present in ```focal```. I need to comment ```additional_devel_packages_list```
I'm wondering how is a ubuntu based build tested before changes are committed to the github?

vladimiroltean, 2020-09-11:
It builds again after this commit:
https://github.com/openil/openil/commit/804ab769696336b7d459e754e23d562cc104748e

razr, 2020-09-11:
thanks, let me check it, however I see that ```hugepages``` is still present but this package does not exist in ```focal``` https://packages.ubuntu.com/search?suite=all&searchon=names&keywords=hugepages

vladimiroltean, 2020-09-11:
You can send patches using git-send-email --to=dev@openil.org.