nxp-archive/openil #72: Q on how to save defconfig in OpenIL environment

JohnRama, 2020-08-19:
- Environment                                                                     
	LS1028ARDB                                                                      
                                                                                  
- SW                                                                              
	- OpenIL                                                                        
	- master branch                                                                 
                                                                                  
This would be a silly question, but I'm still struggling to get familiar with     
openil environment.                                                               
                                                                                  
Question is that once you update the config for linux for example, how do you guys usually     
save the config ?                                                                 
                                                                                  
When checking the manual of build root, it describes make linux-update-defconfig. 
So, I've tried followings, but no success because                                
nxp_ls1028ardb-64b_defconfig uses fragment config files.                          
                                                                                  
$make linux-menuconfig                                                            
$make linux-update-defconfig                                                      
	Unable to perform linux-update-defconfig when fragment files are set            
	linux/linux.mk:594: recipe for target 'linux-update-defconfig' failed           
	make: *** [ ]linux-update-defconfig] Error 1                                    
	makoto-pc-linux :: src/openil/openil 2 » make linux-update-defconfig            
                                                                                  
I'm also interested in the method for another packages like uboot,  and busybox, etc...
Thanks for the help.                                                              
                                                                                  
John

vladimiroltean, 2020-08-20:
> no success because nxp_ls1028ardb-64b_defconfig uses fragment config files.

I find this phrase confusing.
Buildroot uses Kconfig, and Linux uses Kconfig. The `nxp_ls1028ardb-64b_defconfig` file is a defconfig for Buildroot, not for Linux. Also, it is a defconfig file, not a fragment. Buildroot does not support Kconfig fragments.
So you must be talking about the `BR2_LINUX_KERNEL_CONFIG_FRAGMENT_FILES` variable in the Buildroot defconfig.
With Buildroot it is possible to pass a list of Kconfig fragments in this variable, which are all applied after the architecture's defconfig (`arch/arm64/configs/defconfig`). These Kconfig fragments can be provided by the kernel repo itself (such as `lsdk.config`, or by the build system).
These days we try to use Kconfig fragments provided by the build system less and less.
This is why we have that variable set as this:

```
BR2_LINUX_KERNEL_CONFIG_FRAGMENT_FILES="$(LINUX_DIR)/arch/arm64/configs/lsdk.config"
```

This allows us to have the same configuration if we build the kernel from Buildroot or if we build it standalone.

> Question is that once you update the config for linux for example, how do you guys usually
> save the config ?

We modify the `arch/arm64/configs/lsdk.config` file in the kernel repository (or `arch/arm/configs/lsdk.config` for LS1021A).

As to how?
Either by hand (check the settings you want to enable with make menuconfig, and copy them manually to lsdk.config), or, or rare occasion, by regenerating the lsdk.config file manually.

There is an automated command for saving a defconfig with Linux (`make savedefconfig`), but there isn't one for saving a config fragment. So it is done (semi-manually) like this:

```
make defconfig
cp .config .config.defconfig
make lsdk.config
make menuconfig # apply your changes here
cp .config .config.lsdk
./scripts/diffconfig -m .config.defconfig .config.lsdk > arch/arm64/configs/lsdk.config
```

JohnRama, 2020-08-20:
Thanks Vladimiroltean for the detailed explanation and making the things clear. 
Really appreciated. 
I'm fine to close this issue. 

John

closed 2020-08-20 by JohnRama