nxp-archive/openil #73: vmlinux: ELF 64-bit LSB shared object

razr, 2020-08-20:
I do
```bash
br-user@openil:~/openil$ find output-ls1046/ -name vmlinux
output-ls1046/build/linux-linux-5.4.y/vmlinux

br-user@openil:~/openil$ file output-ls1046/build/linux-linux-5.4.y/vmlinux
output-ls1046/build/linux-linux-5.4.y/vmlinux: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), statically linked, BuildID[sha1]=574eb5241588d7d1a667a0513f1ba3bffdc8f021, with debug_info, not stripped
```

compare to the native kernel build (ELF 64-bit LSB executable)
```
$:~/linux_build-5.4.54$ find . -name vmlinux
./linux-5.4.54/vmlinux

$:~/linux_build-5.4.54$ file ./linux-5.4.54/vmlinux
./linux-5.4.54/vmlinux: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), statically linked, BuildID[sha1]=d7403abcde7872318c0a3eac6426d6d5dd45ed3a, with debug_info, not stripped
```

vladimiroltean, 2020-08-20:
Yes, and?
In one case you're cross-compiling the kernel, and in the other case you're compiling it natively.
Did you set the CC, ARCH and CROSS_COMPILE variables?

razr, 2020-08-20:
what is the difference between native and cross-compilation? The end result shall be an executable not a shared object or I'm missing here something?
```bash
br-user@openil:~/openil$ readelf -h output-ls1046/build/linux-linux-5.4.y/vmlinux
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           AArch64
  Version:                           0x1
  Entry point address:               0xffff800010080000
  Start of program headers:          64 (bytes into file)
  Start of section headers:          478472024 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         4
  Size of section headers:           64 (bytes)
  Number of section headers:         41
  Section header string table index: 40
```

vladimiroltean, 2020-08-20:
What is the problem you're debugging?

razr, 2020-08-24:
Sorry for replying late. it is about to run a vmlinux on the hypervisor.