nxp-archive/openil #74: Static - IP address issue | LS1021ATSN

Norman03, 2020-09-12:
1. I'm trying to assign a static IP address in /etc/network/interfaces file in Ls1021ATSN board (reference: section 8.2.5 in openil v1.7.1 guide). But I'm not able to see the static address assigned after bootup. So, ping is not working between the HOST PC and LS1021ATSN board.

2. Later I compiled the OPCUA examples as specified in v.1.7 documentation by enabling the "BR2_PACKAGE_OPEN62541_EXAMPLES" flag. But I'm not able to see the /usr/bin/sja-1105 binary in the openil after boot up. Rather I'm able see open62541_* binaries which I was able to run but I'm not able to browse the OPCUA server(LS1021ATSN) using client in my HOST.

vladimiroltean, 2020-09-12:
From your message I understand that you are really using a 1.7 build of openil, or are you trying to follow the steps from 1.7 on the latest build?

Norman03, 2020-09-12:
> From your message I understand that you are really using a 1.7 build of openil, or are you trying to follow the steps from 1.7 on the latest build?

Using latest openil build v1.9 and document 1.7

vladimiroltean, 2020-09-12:
That's what I thought.
The opc-sja1105 demo program is no longer supported in openil 1.8 and later, since the LS1021A-TSN has transitioned to a kernel implementation of the driver for sja1105. So the sja1105-tool user space driver and all programs derived from it are deprecated.
You are probably just interested in the demo itself, from OPC UA perspective. If you give me some time, I'll try to bring that demo program back to life by making it use "ethtool -S" instead of sja1105-tool.

Norman03, 2020-09-12:
Ok. Thanks.
Meantime, I'm trying to execute the open62541_* binaries in /usr/bin this should work right without any dependencies?