nxp-archive/openil #75: YANG models for the SJA1105 ports

Norman03, 2020-09-15:
Hi, Does openil 1.9 version supports yang model for sja1105 ports?

vladimiroltean, 2020-09-15:
Yes, it should support the IEEE 802.1Qbv YANG model when sysrepo-tsn is compiled with #define SYSREPO_TSN_TC:
https://github.com/openil/sysrepo-tsn/blob/master/src/qbv.c
I am led to understand that it will also support the IEEE 802.1Qci YANG model in the future, when sysrepo-tsn is reworked to use tc-gate instead of tsntool. For now, only 802.1Qbv, using tc-taprio.

mingkaihu-dorkforces, 2020-09-18:
The Qci configuration of sja1105 is supported.
https://github.com/openil/sysrepo-tsn/blob/master/src/qci_sf.c
https://github.com/openil/sysrepo-tsn/blob/master/src/qci_fm.c

mingkaihu, 2020-09-18:
Qci configuration is supported for sja1105.
https://github.com/openil/sysrepo-tsn/blob/master/src/qci_fm.c
https://github.com/openil/sysrepo-tsn/blob/master/src/qci_sf.c