nxp-archive/openil #76: Q on Seamless redundancy test on LS1028A

JohnRama, 2020-09-22:
I'm trying to evaluate seamless redundancy on LS1028A.                      

By following the "8.1.4.7.1 Sequence Generator test" in the OpenIL User Guide (Rev. 1.8, 05/2020),                                                                                               
I got an same result as described in 4. 

> 4. Capture frame on swp2 on TestCenter                                                                          
> We can get frames from swp2 on TestCenter, each frame adds the sequence number: 23450801, 23450802, 23450803… 
                                                                                              
Here is questions                                                                                                  
1. It looks this is not following the IEEE802.1CB-2017 standard.                                                   
 	 Would someone please teach me what standard this behavior is ?
2. Would someone please teach me how to enable IEEE802.1CB ? 
    I would expect the generated frame to have the R-TAG (ethertype = 0xf1c1) .                                                              
3.  I'm little bit confused for the device argument.

> 8.1.4.7.1 Sequence Generator test
> 2. On board A, run the commands:                                                                             
> tsntool> cbstreamidset --device swp0 --nullstreamid --nulldmac 0x7EA88C9B41DD --nullvid 1 --                 
> streamhandle 1                                                                                               
> tsntool> cbgen --device swp0 --index 1 --iport_mask 0x08 --split_mask 0x07 --seq_len 16 --                   
> seq_num 2048             
                                                                                         
Would someone please teach me why "-device swp0" ?? on the commands above ?                                                             
What happen if we set it to swp1, swp2, swp3  ??                                                                           
                                                                                                                   
> 8.1.4.7.2 Sequence Recover test
> 1. On board B, run the following commands:                                                                   
> tsntool> cbstreamidset --device swp2  --nullstreamid --nulldmac 0x7EA88C9B41DD --nullvid 1  --               
> streamhandle 1                                                                                               
> tsntool> cbrec --device swp0 --index 1 --seq_len 16 --his_len 31 --rtag_pop_en   

Would someone please teach me why "-device swp2" and "-device swp0", respectively on the command above  ??                                                             
What happen if we set it to something else ?

Really appreciated the help !!

John

vladimiroltean, 2020-09-22:
> It looks this is not following the IEEE802.1CB-2017 standard.
> Would someone please teach me what standard this behavior is ?

The IEEE 802.1CB draft 2.2 from 2016.

> Would someone please teach me how to enable IEEE802.1CB ?
> I would expect the generated frame to have the R-TAG (ethertype = 0xf1c1) .

This is not possible with the LS1028A, the hardware follows the draft implementation.
Although to be completely fair, there isn't any functional difference, it's just that the ratified standard has a 6-byte tag, 2 bytes of which are reserved, and the draft had 4 bytes. The EtherType can be customized in the LS1028A, but I fail to see the point in making it 0xf1c1 as long as the tag length is different.

> Would someone please teach me why "-device swp0" ?? on the commands above ?
> What happen if we set it to swp1, swp2, swp3 ??

Nothing, it works just as well. Since the tsntool netlink kernel API uses a net_device, then for IEEE 802.1CB, which is a global switch-level configuration, any switch net_device can be used for these commands. Odd, I know, but it is what it is.

> Would someone please teach me why "-device swp2" and "-device swp0", respectively on the command above ??
> What happen if we set it to something else ?

"tsntool> cbstreamidset --device swp2" because cbstreamidset is constructed on top of the MAC table (aka FDB) and the MAC table uses a destination port mask, so you need to specify the destination port here.
As for "tsntool> cbrec --device swp0", it doesn't matter here if it's swp0 or something else.

Have you seen this? It gives a practical example of 802.1CB and it explains the usage of the tsntool commands:
https://github.com/vladimiroltean/tsn-scripts/tree/8021cb-devel

JohnRama, 2020-09-23:
Great thanks, Vladimir

> The IEEE 802.1CB draft 2.2 from 2016.
Thank you so much for this !!

> The EtherType can be customized in the LS1028A, 
I see Ethertype is set to "0x2345" in my environment.
I want to customize it to use 0xf1c1, if it's easy.

> but I fail to see the point in making it 0xf1c1 as long as the tag length is different.
Sorry, I could not get the meaning of this sentence.
Would you please tell me more on this ?

If Ethertype modification is possible, my guess is to modify some code in 
net/ethernet/mscc/ocelot_tsn.c in the kernel. But could not figure it out that part ...

> Would someone please teach me why "-device swp0" ?? on the commands above ?
>     What happen if we set it to swp1, swp2, swp3 ??
>
> Nothing, it works just as well. Since the tsntool netlink kernel API [...]
Got it. Thanks !! 

> Would someone please teach me why "-device swp2" and "-device swp0", respectively on the command above ??
>     What happen if we set it to something else ?
>
> "tsntool> cbstreamidset --device swp2" [...]
Got it. Thanks !! 

> Have you seen this? It gives a practical example of 802.1CB and it explains the usage of the tsntool commands:
> https://github.com/vladimiroltean/tsn-scripts/tree/8021cb-devel

Thanks a lot !!  I'll take a look at them.

John

vladimiroltean, 2020-09-23:
> > but I fail to see the point in making it 0xf1c1 as long as the tag length is different.
> Sorry, I could not get the meaning of this sentence.
> Would you please tell me more on this ?

See these 2 pictures? They are screenshots from the 2016 draft:

[image]
[image]

Now see these 2 pictures? They are from the 2017 standard:

[image]
[image]

Notice how the draft from 2016
(a) did not decide on an EtherType yet
(b) has a 4 byte tag, as opposed to the 2017 standard which has 6 bytes?

So, as long as the sequence number of a FRER frame sent by the LS1028A will always be 2 bytes off compared to what another 802.1CB-2017 receiver will parse, what is the point of making the EtherType of the LS1028A be precisely 0xf1c1?

JohnRama, 2020-09-24:
Thanks for the clarification, Vladimir.

I understand that IEEE802.1CB-2017 has 6 bytes RTAG with Ethertype of 0xf1c1, 
while IEEE802.1CB-draft2.2 has 4 bytes tag with Ethertype undefined which LS1028A supports.
So these two are not compatible.

So, let's assuming that we have 2 LS1028A RDB boards, and run the test described the User Guide.
IEEE802.1CB-draft2.2 4byte tag consists of Ethertype and sequence number.
When running the test, Ethertype of the tag is set to 0x2345. 
I confirmed this with the ethernet bus monitoring tool (like wireshark).
I just want to modify this Ethertype to something another value.

I'm wondering from where this 0x2345 number is came from. 
Does this 4 byte tag is appended by hardware ?
I could not find out the code where this number is set in the device driver.
I could not find out the register from the LS1028A Reference Manual.

So do you know how to customize the Ethertype ?

> The EtherType can be customized in the LS1028A
I'm asking this question because you of this comment.
Thanks a lot. 

John

vladimiroltean, 2020-09-24:
See the ETH_P_8021CB definition here, change it to 0xf1c1:
https://github.com/openil/linux/blob/linux-5.4.y/drivers/net/dsa/ocelot/felix_tsn.c#L23

JohnRama, 2020-09-24:
Thanks a lot, Vladimir !!

Somehow, I used the different kernel tree, that's why I could not catch it.

John

JohnRama, 2020-09-29:
Confirmed it works.
Close the issue. 

Thanks a lot !! 

Jonn

```
diff --git a/drivers/net/dsa/ocelot/felix_tsn.c b/drivers/net/dsa/ocelot/felix_tsn.c
index b909d930dfa5..052b3a6cc312 100644
--- a/drivers/net/dsa/ocelot/felix_tsn.c
+++ b/drivers/net/dsa/ocelot/felix_tsn.c
@@ -20,7 +20,7 @@
 #include <net/tsn.h>
 #include "felix.h"

-#define ETH_P_8021CB           0x2345
+#define ETH_P_8021CB           0xf1c1
 #define FELIX_QSYS_HSCH_NUM    72
 /* MSCC TSN parameters limited */
 #define FELIX_PSFP_SFID_NUM    176
@@ -1534,6 +1534,8 @@ static int felix_seq_gen_set(struct net_device *ndev, u32 index,
                return -EINVAL;
        }

+       ocelot_write(ocelot, ETH_P_8021CB, SYS_SR_ETYPE_CFG);
+
        list_for_each_entry(tmp, &streamtable, list)
                if (tmp->handle == index)
                        streamid_multi_forward_set(ocelot, tmp->mac, tmp->vid,

```

closed 2020-09-29 by JohnRama

vladimiroltean, 2020-09-29:
Why do you need this anyway?

JohnRama, 2020-09-29:
No special meaning.