nxp-archive/openil #77: Configuring Qbv via NETCONF for LS1021A: "Set Qbv error: Resource temporarily unavailable"

emunicio, 2020-10-06:
Hi,

We are trying to configure Qbv in the the LS1021A-TSN (with the SJA-1105T) remotely through NETCONF following the examples in the OpenIL guide (1.9) (https://www.nxp.com/docs/en/user-guide/OPEN-LINUX-IND-UM-1.9.pdf), with netopeer2 client running  in another device. When we try to configure the LS1021A-TSN using the command 5 in page 210 "edit-config --target running --config=qbv-swp2-enable.xml", we get the following error:

"ERROR
	type:     application
	tag:      operation-failed
	severity: error
	path:     /ietf-interfaces:interfaces/interface[name='swp2']/ieee802-dot1q-sched:*//*
	message:  Set Qbv error: Resource temporarily unavailable"

The error shown in the journalctl is:

12:49:05 LS1021ATSN netopeer2-server[1798]: Error by processing of the commit request (session id=572384238): Invalid argument.
12:49:05 LS1021ATSN netopeer2-server[1798]: Commit operation failed with 1 error(s).

Similarly, when we add manually the scheduling configuration with e.g.,:
"tc qdisc add dev swp2 parent root taprio num_tc 8 map 0 1 2 3 4 5 6 7 queues 1@0 1@1 1@2 1@3 1@4 1@5 1@6 1@7 base-time 0 sched-entry S 80 50000 sched-entry S 40 50000 sched-entry S 3f 300000 flags 2"

the scheduling gets installed, but when we send the command to get the configuration:

"get-config --source running --filter-xpath /ietf-interfaces:interfaces/interface[name='swp2']/ieee802-dot1q-sched:gate-parameters"

does not return anything.

Is there maybe something we are missing? 
Looking to this, https://github.com/openil/openil/issues/75, may it be that we need to recompile sysrepo-tsn? Any hints on how to do this?

vladimiroltean, 2020-10-07:
Sorry for the delay, it was a bit more tricky for me to set up netopeer2-cli since I upgraded my PC to Ubuntu 20.04.
So I was able to run that command successfully.

```
[user@pc .../netconf/sysrepo-tsn/Instances] $ netopeer2-cli
> connect --host 10.0.0.142 --login root
Interactive SSH Authentication
Type your password:
Password:
> edit-config --target running --config=qbv-swp2-enable.xml
OK
>
```

And on the board:
```
[root@LS1021ATSN ~] # [ 4332.448752] 001: sja1105 spi0.1: Reset switch and programmed static config. Reason: Time-aware scheduling                                                                          

[root@LS1021ATSN ~] # tc qdisc show dev swp2     
qdisc taprio 100: root refcnt 9 tc 8 map 0 1 2 3 4 5 6 7 0 0 0 0 0 0 0 0
queues offset 0 count 1 offset 1 count 1 offset 2 count 1 offset 3 count 1 offset 4 count 1 offset 5 count 1 offset 6 count 1 offset 7 count 1
clockid invalid flags 0x7d8f4   base-time 0 cycle-time 17000 cycle-time-extension 0
        index 0 cmd S gatemask 0x5 interval 5000
        index 1 cmd S gatemask 0x7 interval 6000
        index 2 cmd S gatemask 0x7 interval 6000

qdisc pfifo 0: parent 100:8 limit 1000p
qdisc pfifo 0: parent 100:7 limit 1000p
qdisc pfifo 0: parent 100:6 limit 1000p
qdisc pfifo 0: parent 100:5 limit 1000p
qdisc pfifo 0: parent 100:4 limit 1000p
qdisc pfifo 0: parent 100:3 limit 1000p
qdisc pfifo 0: parent 100:2 limit 1000p
qdisc pfifo 0: parent 100:1 limit 1000p
```

The only thing that I needed to change was:

```diff
From cf9d64b2a0b707ec2f1bc98148a05f8b8fd75419 Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Wed, 7 Oct 2020 18:34:38 +0300
Subject: [PATCH] configs/ls1021atsn: use sysrepo-tsn with tc backend

The sja1105 does not support tsntool, we must configure its TSN offload
features (tc-taprio for 802.1Qbv, tc-gate for 802.1Qci) using the
iproute2 suite.

---
 configs/nxp_ls1021atsn_defconfig | 1 +
 1 file changed, 1 insertion(+)

diff --git a/configs/nxp_ls1021atsn_defconfig b/configs/nxp_ls1021atsn_defconfig
index df9e77a1abb4..ce6383df0136 100644
--- a/configs/nxp_ls1021atsn_defconfig
+++ b/configs/nxp_ls1021atsn_defconfig
@@ -82,6 +82,7 @@ BR2_PACKAGE_QORIQ_TSN_SCRIPTS=y
 
 # sysrepo-tsn
 BR2_PACKAGE_QORIQ_SYSREPO_TSN=y
+BR2_PACKAGE_QORIQ_SYSREPO_TSN_TC=y
 
 # basic packages and tools
 BR2_PACKAGE_MTD=y
-- 
2.25.1

```

Then rebuild with:

```
make qoriq-sysrepo-tsn-dirclean world
```

emunicio, 2020-10-09:
Thanks for the quick answer @vladimiroltean 
It worked perfectly!

closed 2020-10-09 by emunicio

crow1814, 2020-12-08:
Hi, vladimiroltean
I have followed your instructions to configure ls1021atsn, and the response is 'OK'
However, I find out that the feedback is "error" when I modify a value, such as the second "<sched:time-interval-value>6000</sched:time-interval-value>" to "<sched:time-interval-value>7000</sched:time-interval-value>".
The error message is:
```
edit-config --target running --config=qbv.xml
ERROR
type: application
tag: operation-failed
severity: error
message: Invalid argument
```
Is there maybe something I did wrong here?