nxp-archive/openil #78: Build is broken, toolchain URLs deprecated

J4z3, 2020-10-26:
Hello,

I'm trying to build OpenIL-1.9 for LS1021A-TSN, but it fails as some URLs are deprecated:
- http://sources.buildroot.net/toolchain-external-arm-arm/gcc-arm-9.2-2019.12-x86_64-arm-none-linux-gnueabihf.tar.xz
- http://sources.buildroot.net/gcc-arm-9.2-2019.12-x86_64-arm-none-linux-gnueabihf.tar.xz

These URLs are used when compiling and don't exist anymore.

Thanks in advance for your help

J4z3, 2020-10-26:
This issue only appears on the master branch.

vladimiroltean, 2020-10-26:
For me it is returning '302 Moved Temporarily' and then redirects. What is the message in your case?
[build.log](https://github.com/openil/openil/files/5439291/build.log)

J4z3, 2020-10-26:
> For me, developer.arm.com is checked first, where it is returning '302 Moved Temporarily' and then redirects. What is the message in your case?
> [build.log](https://github.com/openil/openil/files/5439291/build.log)

404 Not Found.

vladimiroltean, 2020-10-26:
Can I see a build log? This is not helpful.

J4z3, 2020-10-26:
Actually you are right that developer.arm is checked first.

```
>>> toolchain-external-arm-arm 2019.12 Downloading
--2020-10-26 17:10:41--  https://developer.arm.com/-/media/Files/downloads/gnu-a/9.2-2019.12/binrel/gcc-arm-9.2-2019.12-x86_64-arm-none-linux-gnueabihf.tar.xz
Resolving PROXY
Connecting to PROXY... connected.
Proxy tunneling failed: Proxy Authentication RequiredUnable to establish SSL connection.
--2020-10-26 17:10:41--  http://sources.buildroot.net/toolchain-external-arm-arm/gcc-arm-9.2-2019.12-x86_64-arm-none-linux-gnueabihf.tar.xz
Resolving PROXY
Connecting to PROXY... connected.
Proxy request sent, awaiting response... 404 Not Found
2020-10-26 17:10:42 ERROR 404: Not Found.

--2020-10-26 17:10:42--  http://sources.buildroot.net/gcc-arm-9.2-2019.12-x86_64-arm-none-linux-gnueabihf.tar.xz
Resolving PROXY
Connecting to PROXY... connected.
Proxy request sent, awaiting response... 404 Not Found
2020-10-26 17:10:42 ERROR 404: Not Found.
```
Where PROXY is the proxy address.

vladimiroltean, 2020-10-26:
> Proxy tunneling failed: Proxy Authentication RequiredUnable to establish SSL connection.

Fix your proxy.

closed 2020-10-26 by vladimiroltean

J4z3, 2020-10-26:
What you quoted is indeed a proxy issue, however I don't think the 404 errors are, since the URL I posted above do not exist.

vladimiroltean, 2020-10-26:
And you really think you're going to make it through the entire build with a proxy that mangles all HTTPS traffic? Or will you open another ticket for the next package that fails to download over HTTPS, and then the fallback happens to send 404 or simply not exist at all?