nxp-archive/openil #81: TQMLS1028A error when I am trying to delete any rule or create any rule "tc: command line is not complete"

rass92, 2021-01-11:
Dear Community,

I am using a TQMLS1028A with the following OpenIl version

NAME=OpenIL
VERSION=2018.12
ID=OpenIL
VERSION_ID=2018.12
PRETTY_NAME="OpenIL 2018.12"

When I try to delete the rules in any interface with the command
```
[root@OpenIL:~]# tc qdisc delete dev enp0s0f0 root
```
the output is this one
```
tc: command line is not complete, try "help"
```

additionally if I try to put a new rule like this one
```
tc qdisc add dev enp0s0f0 parent root handle 100 taprio num_tc 2 map 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 queues 0@0 1@1 base-time 1554445635681310809 sched-entry S 01 500000 sched-entry S 02 500000 clockid CLOCK_TAI
```
I get the following output even if I use `replace` instead of add, I get the same output

```
tc: invalid argument 'root' to 'command'
```

Any idea on what I am doing wrong? or why I cannot add or delete any rule?

The following output is when the command `tc qdisc show`

```
[root@OpenIL:~]# tc qdisc show
qdisc noqueue 0: dev lo root refcnt 2 
qdisc mq 0: dev enp0s0f0 root 
qdisc pfifo_fast 0: dev enp0s0f0 parent :8 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f0 parent :7 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f0 parent :6 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f0 parent :5 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f0 parent :4 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f0 parent :3 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f0 parent :2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f0 parent :1 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc mq 0: dev enp0s0f2 root 
qdisc pfifo_fast 0: dev enp0s0f2 parent :8 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f2 parent :7 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f2 parent :6 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f2 parent :5 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f2 parent :4 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f2 parent :3 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f2 parent :2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f2 parent :1 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc mq 0: dev enp0s0f6 root 
qdisc pfifo_fast 0: dev enp0s0f6 parent :8 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f6 parent :7 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f6 parent :6 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f6 parent :5 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f6 parent :4 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f6 parent :3 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f6 parent :2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev enp0s0f6 parent :1 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev swp0 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev swp1 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev swp2 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev swp3 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev swp4 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev swp5 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
```

Maybe this could be helpfull
```
[root@OpenIL:~]# tc --help
BusyBox v1.29.3 (2019-07-19 09:07:17 CEST) multi-call binary.

Usage: tc OBJECT CMD [dev STRING]
```

vladimiroltean, 2021-01-12:
Ask your board vendor to update to a newer OpenIL version (like the current one, 1.10) where the tc binary is provided by iproute2 and not by busybox, or update yourself. It shouldn't be too complicated.

> ```
> tc qdisc add dev enp0s0f0 parent root handle 100 taprio num_tc 2 map 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 queues 0@0 1@1 base-time 1554445635681310809 sched-entry S 01 500000 sched-entry S 02 500000 clockid CLOCK_TAI
> ```

This command almost works for me. The "queues 0@0 1@1" portion is invalid, you cannot have zero queues at traffic class 0, I assume you meant "queues 1@0 1@0". Other than that, the iproute2-based tc accepts the command.

> ```
> [root@OpenIL:~]# tc qdisc delete dev enp0s0f0 root
> ```

This command works for me too, with BR2_PACKAGE_IPROUTE2=y.

rass92, 2021-03-09:
I am sorry to say that at the end my board vendor told me that they are using yocto to make the OS for the baord

Here I was able to make the tc command worked.

Nevertheless thank you very much for taking your time to see my question and give me an answer.

closed 2021-03-09 by rass92