nxp-archive/openil #83: Any plan for regular sync with the original https://github.com/buildroot/buildroot?

scdrjcom, 2021-01-13:
Is there any any plan for regular sync with the original https://github.com/buildroot/buildroot?

mingkaihu-dorkforces, 2021-01-20:
Hi scdrjcom,
There is no regular cadence for the sync up, it's based on the features update, if there're major feature updates, we will sync with it.

BTW, it's also under discussion to use bitbake as the built framework for OpenIL. Which one do you prefer?

Thanks,
Mingkai

mingkaihu, 2021-01-20:
Hi scdrjcom,

There is no regular cadence for the sync up. It's based on feature updates, if there's major feature update, we will sync up with it.

BTW, it's also discussion to use the bitbake for the build framework. Which one do you prefer?

Thanks,
Mingkai

katracker, 2021-01-20:
> Hi scdrjcom,
> 
> There is no regular cadence for the sync up. It's based on feature updates, if there's major feature update, we will sync up with it.
> 
> BTW, it's also discussion to use the bitbake for the build framework. Which one do you prefer?
> 
> Thanks,
> Mingkai

Thanks for your response.  We use buildroot for our existing product. It is simple, easy to use and actively developed. It fulfills our requirement, so trying to continue using it for our next-gen product.

Will there be further updates in this repository?  Is there a plan to integrate these change in the original buildroot git?

scdrjcom, 2021-01-20:
> 
> 
> Hi scdrjcom,
> 
> There is no regular cadence for the sync up. It's based on feature updates, if there's major feature update, we will sync up with it.
> 
> BTW, it's also discussion to use the bitbake for the build framework. Which one do you prefer?
> 
> Thanks,
> Mingkai

Thanks for your response. We use buildroot for our existing product. It is simple, easy to use and actively developed. It fulfills our requirement, so trying to continue using it for our next-gen product.

Will there be further updates in this repository?

Is there a plan to integrate these change in the original buildroot git?