nxp-archive/openil #84: LS1021ATSN: make port on ETH1 mirror port

kru-alex, 2021-02-05:
I currently use the LS1021ATSN and have four devices connected to the interfaces swp2-5. I would like to use the port on ETH1 as the mirror port to monitor the traffic. 
With the commands given in the OpenIL user guide,

tc qdisc add dev swp2 clsact
tc filter add dev swp2 ingress matchall skip_sw \
        action mirred egress mirror dev swp3

it was just possible to mirror the packets among the switch ports. If I replace swp3 by eth1, I receive the error message:

RTNETLINK answers: Operation not supported
We have an error talking to the kernel

How is it possible to mirror packets from the swp ports to ETH1?

Best regards,
Alex

vladimiroltean, 2021-02-05:
Does this patch do what you want?
```diff
From 9002d017091e53bd0acdd2186173c5cd9a04c176 Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Sat, 6 Feb 2021 00:14:53 +0200
Subject: [PATCH] net: dsa: allow port mirroring towards foreign interfaces

To a DSA switch, port mirroring towards a foreign interface is the same
as mirroring towards the CPU port, since all non-DSA interfaces are
reachable through that. Tell the hardware to send the packets to the CPU
port and let the mirred action deal with them in software.

---
 net/dsa/slave.c | 10 +++++-----
 1 file changed, 5 insertions(+), 5 deletions(-)

diff --git a/net/dsa/slave.c b/net/dsa/slave.c
index b0571ab4e5a7..913a4a5e32a9 100644
--- a/net/dsa/slave.c
+++ b/net/dsa/slave.c
@@ -936,19 +936,19 @@ dsa_slave_add_cls_matchall_mirred(struct net_device *dev,
 	if (!act->dev)
 		return -EINVAL;
 
-	if (!dsa_slave_dev_check(act->dev))
-		return -EOPNOTSUPP;
-
 	mall_tc_entry = kzalloc(sizeof(*mall_tc_entry), GFP_KERNEL);
 	if (!mall_tc_entry)
 		return -ENOMEM;
 
+	if (dsa_slave_dev_check(act->dev))
+		to_dp = dsa_slave_to_port(act->dev);
+	else
+		to_dp = dp->cpu_dp;
+
 	mall_tc_entry->cookie = cls->cookie;
 	mall_tc_entry->type = DSA_PORT_MALL_MIRROR;
 	mirror = &mall_tc_entry->mirror;
 
-	to_dp = dsa_slave_to_port(act->dev);
-
 	mirror->to_local_port = to_dp->index;
 	mirror->ingress = ingress;
 
-- 
2.25.1

```

kru-alex, 2021-02-08:
Hello Vladimir,

thanks for the patch. I applied it, but unfortunately, I could not get the desired result.

In the man page of the command tc-matchall I found the following:

       skip_sw
              Do not process filter by software. If hardware has no
              offload support for this filter, or TC offload is not
              enabled for the interface, operation will fail.

If I issue the command
    ethtool -k swp2

among others, I get the line:
    hw-tc-offload: on

If I issue the command
    ethtool -k eth1

among others, I get the line:
    hw-tc-offload: off [fixed]

I think that is obstructing the operation, right?

vladimiroltean, 2021-02-10:
> I think that is obstructing the operation, right?

So skip_sw means to add the tc filter in the hardware datapath only. But since this ingress mirror needs to be executed partially by hardware (to extract the packets from the switch port and send them to the CPU) and partially by software (to redirect the packets from the switch CPU port towards the egress port), then of course you'd need to simply remove the "skip_sw" flag, such that the filter will be added both in the software and the hardware datapath.