nxp-archive/openil #85: LS1021ATSN/gPTP: Sync loss when accessing to clock to read

diegotxegp, 2021-02-05:
Hello,

My issue is that a sync loss happens when I do some tests in one of the boards of my network.

My network are a LS1021ATSN and two boards conected to it using gPTP. One board acts as a master, the switch as a bridge and the other board as a slave. I am using the daemon in each device, not the ptp4l.service (if ptp4l.service is recomendable, let me know).

With these instructions I synchronize the devices:
LS1021ATSN
sudo ptp4l -i swp2 -i swp2 -f /gPTP.cfg --tx_timestamp_timeout 20 -m

NODES
sudo ptp4l -i swp2 -i swp2 -f /gPTP.cfg --tx_timestamp_timeout 20 -m

So, it works perfectly with max offsets like 7-18ns until I run some tests where acceses to the time of the synchronized clock and suddenly this message raises.

ptp4l[3523.714]: rms    7 max   10 freq +15982 +/-   9 delay   232 +/-   0
ptp4l[3524.715]: rms    6 max   11 freq +15987 +/-   8 delay   233 +/-   0
ptp4l[3525.717]: rms    8 max   17 freq +15981 +/-  10 delay   234 +/-   0
ptp4l[3526.719]: rms    6 max   11 freq +15982 +/-   9 delay   233 +/-   0
ptp4l[3526.872]: timed out while polling for tx timestamp
ptp4l[3526.872]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[3526.872]: port 1 (enp3s0): send peer delay response failed
ptp4l[3526.872]: port 1 (enp3s0): SLAVE to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[3543.080]: port 1 (enp3s0): FAULTY to LISTENING on INIT_COMPLETE
ptp4l[3546.846]: port 1 (enp3s0): new foreign master 00049f.fffe.ef0808-1
ptp4l[3546.906]: port 1 (enp3s0): LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[3546.906]: selected local clock aabbcc.fffe.00094e as best master
ptp4l[3546.906]: port 1 (enp3s0): assuming the grand master role
ptp4l[3548.846]: selected best master clock aabbcc.fffe.00094a
ptp4l[3548.846]: port 1 (enp3s0): MASTER to UNCALIBRATED on RS_SLAVE
ptp4l[3549.128]: port 1 (enp3s0): UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[3549.879]: rms  176 max  297 freq +15694 +/- 128 delay   234 +/-   0
ptp4l[3550.880]: rms   43 max   70 freq +15868 +/-  55 delay   233 +/-   0
ptp4l[3551.882]: rms   63 max   73 freq +15976 +/-  16 delay   232 +/-   0
ptp4l[3552.883]: rms   39 max   58 freq +15994 +/-  10 delay   232 +/-   0
ptp4l[3553.885]: rms   13 max   28 freq +15977 +/-  13 delay   232 +/-   0
ptp4l[3554.886]: rms    7 max    9 freq +15966 +/-  10 delay   232 +/-   0
ptp4l[3555.888]: rms    9 max   15 freq +15959 +/-  10 delay   232 +/-   0

Could you help me?
I use the OpenIL version 1.8 and the Linuxptp 3.1 on the boards.

vladimiroltean, 2021-02-05:
Does the situation improve if you increase the tx_timestamp_timeout to a higher value like 50 ms?

diegotxegp, 2021-02-06:
Hello Vladimir,

> Does the situation improve if you increase the tx_timestamp_timeout to a higher value like 50 ms?

No at all. The same text appears. In fact, I tried a huge number for --tx_timestamp_timeout 10000 and this text appears:

ptp4l[1034.934]: rms    6 max   12 freq   -285 +/-   8 delay   235 +/-   0
ptp4l[1035.936]: rms    6 max   11 freq   -277 +/-   5 delay   235 +/-   0
ptp4l[1036.939]: rms    4 max    7 freq   -280 +/-   6
ptp4l[1038.091]: clockcheck: clock jumped backward or running slower than expected!
ptp4l[1038.092]: port 1 (enp3s0): SLAVE to LISTENING on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1038.092]: selected local clock aabbcc.fffe.00094e as best master
ptp4l[1038.191]: clockcheck: clock jumped forward or running faster than expected!
ptp4l[1039.017]: selected best master clock aabbcc.fffe.00094a
ptp4l[1039.017]: port 1 (enp3s0): LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[1039.932]: port 1 (enp3s0): UNCALIBRATED to LISTENING on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1039.932]: selected local clock aabbcc.fffe.00094e as best master
ptp4l[1039.932]: clockcheck: clock jumped backward or running slower than expected!
ptp4l[1040.441]: clockcheck: clock jumped forward or running faster than expected!
ptp4l[1040.575]: clockcheck: clock jumped forward or running faster than expected!
ptp4l[1041.017]: selected best master clock aabbcc.fffe.00094a
ptp4l[1041.017]: port 1 (enp3s0): LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[1043.103]: clockcheck: clock jumped backward or running slower than expected!
ptp4l[1043.104]: port 1 (enp3s0): UNCALIBRATED to LISTENING on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1043.104]: selected local clock aabbcc.fffe.00094e as best master
ptp4l[1043.106]: selected best master clock aabbcc.fffe.00094a
ptp4l[1043.106]: port 1 (enp3s0): LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[1043.106]: port 1 (enp3s0): rogue peer delay response
ptp4l[1043.106]: port 1 (enp3s0): UNCALIBRATED to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[1059.180]: port 1 (enp3s0): FAULTY to LISTENING on INIT_COMPLETE
ptp4l[1062.685]: selected local clock aabbcc.fffe.00094e as best master
ptp4l[1063.021]: port 1 (enp3s0): new foreign master 00049f.fffe.ef0808-1
ptp4l[1065.021]: selected best master clock aabbcc.fffe.00094a
ptp4l[1065.021]: port 1 (enp3s0): LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[1065.392]: port 1 (enp3s0): UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[1066.145]: rms  212 max  305 freq   -585 +/- 167 delay   236 +/-   1
ptp4l[1067.147]: rms   70 max  102 freq   -398 +/-  66 delay   239 +/-   0
ptp4l[1068.150]: rms   92 max  109 freq   -262 +/-  14 delay   239 +/-   0
ptp4l[1069.152]: rms   55 max   76 freq   -240 +/-   6 delay   238 +/-   0

diegotxegp, 2021-02-06:
Apart from the comment just above, I want to ask you some questions.

I have a test network with a LS1021ATSN platform and two boards connected to it on ETH2 and ETH3, according to the label in the front of the device. It would be swp2 and swp3 as I use in ptp4l.

What I want is create two different scenarios: one where the LS1021ATSN switch is the GM, and the boards the slaves which synchronized their clock to the GM time, and another where one board is the GM, and the LS1021ATSN acts as a bridge and the other board is a slave receiving the time through the switch. I have to remember that I want to use gPTP (802.1AS), not PTP (IEEE 1588). This detail is important.

My questions are:

1) Is it recommedable using the ptp4l.service on the LS1021ATSN switch and how would it be?
2) Is it better using the ptp4l daemon in each device (LS1021ATSN, board 1 and board 2) instead of the ptp4l.service?
3) What steps would you do if you did that?
4) I want that slaves clocks never go back in time. Never go back. How would I do that? --step_threshold? It is by default at 0.0.
5) I am using the OpenIL version 1.8 with an inner linuxptp version 2.0 in the LS1021ATSN platform, and a linuxptp version 3.1 in the boards. Does it influence in some bad behaviour?
6) In latest version of OpenIL, do you have solve errors for the LS1021ATSN platform? Do you recommend me to update my OpenIL OS?

These are the questions I have right now in my mind.

I just want to create that network I told you and synchronize the time among them to do some tests with global clock.

Thank you so much for your prompt reply.

vladimiroltean, 2021-02-06:
> No at all. The same text appears.

No it doesn't, in the first log you had:
```
ptp4l[3526.872]: timed out while polling for tx timestamp
ptp4l[3526.872]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
```
and in the second log you don't.
But now I notice that the logs you're sharing are on some other device, not the LS1021A-TSN board?

> it works perfectly with max offsets like 7-18ns until I run some tests where acceses to the time of the synchronized clock and suddenly this message raises

What are you doing exactly to break it? Can I see some logs on the actual board?

> Is it recommedable using the ptp4l.service on the LS1021ATSN switch and how would it be?

The /usr/lib/systemd/system/ptp4l.service systemd service that is preinstalled uses /etc/linuxptp.cfg which should provide a good starting point for an 802.1AS bridge on swp2-swp5. I don't know of any issues with it. It may not be what you need for your testing though.

> Is it better using the ptp4l daemon in each device (LS1021ATSN, board 1 and board 2) instead of the ptp4l.service?

Again, it depends what you need it to do. If by ptp4l daemon you mean starting/stopping ptp4l directly from a script, you could do that too.

> What steps would you do if you did that?

I might just happen to have some examples which might give you some ideas. Keep in mind that you'll probably need to adapt them heavily though - just use them as a starting point.
```
Board A ETH0 <-> Board B SWP2
Board A ETH1 <-> Board B SWP3

 +---------------------------------------------------+
 | LS1021A-TSN board B (bridge)                      |
 |                                                   |
 |   +-----------+   +-----------+   +-----------+   |
 |   |           |   |           |   |           |   |
 |   |           |   |           |   |           |   |
 |   |           |   |     +--------+|           |   |
 |   |    SWP5   |   |    SWP3   |  ||    ETH1   |   |
 |   +-----------+   +-----------+  |+-----------+   |
 |   |           |   |           |  ||           |   |
 |   |           |   |           |  ||           |   |
 |   |           |   |     +------+ ||           |   |
 |   |    SWP4   |   |    SWP2   || ||    ETH0   |   |
 +---------------+---------------+|-|------------+---+
                                  +-----------------------+
                                    |                     |
                                    +------------------+  |
                                                       |  |
 +---------------------------------------------------+ |  |
 | LS1021A-TSN board A (sender & receiver)           | |  |
 |                                                   | |  |
 |   +-----------+   +-----------+   +-----------+   | |  |
 |   |           |   |           |   |           |   | |  |
 |   |           |   |           |   |           |   | |  |
 |   |           |   |           |   |     +-----------+  |
 |   |    SWP5   |   |    SWP3   |   |    ETH1   |   |    |
 |   +-----------+   +-----------+   +-----------+   |    |
 |   |           |   |           |   |           |   |    |
 |   |           |   |           |   |           |   |    |
 |   |           |   |           |   |     +--------------+
 |   |    SWP4   |   |    SWP2   |   |    ETH0   |   |
 +---------------+---------------+---------------+---+
```

The point of this example is the measure the forwarding latency of the switch when the time-aware scheduler is enabled, and some packets are sent in-band with the schedule. OpenIL has a program available called "isochron" which can measure the packet latency by taking MAC-level hardware timestamps of those packets at the sender and at the receiver, and compare them (it assumes that the sender and receiver are synchronized over PTP). There is also the possibility
that the isochron sender and receiver run on the same board, and this is precisely what the test does: it sets up an isochron sender on ETH0 of board A, a receiver on ETH1 of the same board, and make board B a time-aware bridge whose egress port is swp3.

What I did in the attached scripts is simply start/stop PTP on demand, and configure isochron to
send 10 packets and measure their latency (I'll spare you the more subtle aspects of the test as they are probably not relevant to the topic which is just PTP).

Create these script files in a folder and transfer that folder to both boards:
[... 327 lines cut ...]
adc93ad9ad0e net: dsa: sja1105: fix PTP timestamping with large tc-taprio cycles
9b51bf97c1af net: dsa: sja1105: offload the Credit-Based Shaper qdisc
da1075c01423 net: dsa: sja1105: request promiscuous mode for master
4976214678b3 net: dsa: allow drivers to request promiscuous mode on master
359e3150df30 net: dsa: sja1105: avoid invalid state in sja1105_vlan_filtering
2eb54d78875a docs: net: dsa: sja1105: document the best_effort_vlan_filtering option
ab3838b00166 net: dsa: sja1105: implement VLAN retagging for dsa_8021q sub-VLANs
841daf26e3f7 net: dsa: sja1105: implement a common frame memory partitioning function
a6058bad75ff net: dsa: sja1105: add packing ops for the Retagging Table
7441ca14aec9 net: dsa: sja1105: add a new best_effort_vlan_filtering devlink parameter
b7be3a8cd9a0 net: dsa: tag_sja1105: implement sub-VLAN decoding
4e14879736f3 net: dsa: tag_8021q: support up to 8 VLANs per port using sub-VLANs
9feda4edb6bb net: dsa: sja1105: prepare tagger for handling DSA tags and VLAN simultaneously
[... cut ...]
3bd72c75802c net: dsa: sja1105: deny alterations of dsa_8021q VLANs from the bridge
3382628c4d2f net: dsa: sja1105: keep the VLAN awareness state in a driver variable
a0713e3365d0 net: dsa: tag_8021q: introduce a vid_is_dsa_8021q helper
aac931ec928b net: dsa: sja1105: Revert pre-mainline implementation of best_effort_vlan_filtering
```

So yes, there are some changes. In terms of bug fixes this is probably the most relevant, it first appeared in OpenIL 1.9:
```
adc93ad9ad0e net: dsa: sja1105: fix PTP timestamping with large tc-taprio cycles
```

diegotxegp, 2021-02-07:
> No it doesn't, in the first log you had:
>
> ```
> ptp4l[3526.872]: timed out while polling for tx timestamp
> ptp4l[3526.872]: increasing tx_timestamp_timeout may correct this issue, but it is likely [...]

The log is from one of the boards connected to the LS1021ATSN platform. In particular the board which acts as a master on the network I told you: master (node 1) -> bridge (LS1021ATSN) -> slave (node 2). In fact, the log of the LS1021ATSN is not alterated when the node 1 is.

> > it works perfectly with max offsets like 7-18ns until I run some tests where acceses to the time of the synchronized clock and suddenly this message raises
> 
> What are you doing exactly to break it? Can I see some logs on the actual board?
> 

What I do it is a test which accesses 1 million times to REALTIME_CLOCK, where I have the PHC time mapped with phc2sys. Then, the node 1 loses sync until it recover about 10 seconds after.

> The /usr/lib/systemd/system/ptp4l.service systemd service that is preinstalled uses /etc/linuxptp.cfg which should provide a good starting point for an 802.1AS bridge on swp2-swp5. I don't know of [...]
For master (LS1021ATSN) --> slave (any node) I used the ptp4l daemon for each device like I told you in the first message.
'''
With these instructions I synchronize the devices:
LS1021ATSN
sudo ptp4l -i swp2 -i swp2 -f /gPTP.cfg --tx_timestamp_timeout 20 -m (gPTP.cfg with gmCapable 0)

NODES
sudo ptp4l -i enp3s0 -f /gPTP.cfg --tx_timestamp_timeout 20 -m
'''
I did this because using ptp4l.service for this example, the LS1021ATSN delivered a different time to the PHC of the node, unlike the time I force to it using phc_ctl /dev/ptp1 set 1234432534.32432523 (for example). Some months ago, you told me in another converstation, that I must change the time of the LS1021ATSN because there was a error if I kept the epoch time (1970). In addition, using the ptp4l.service the offsets are higher. Conversely, For the other case, master (node 1) --> bridge (LS1021ATSN) --> slave (node 2), I used the ptp4l.service on the LS1021ATSN because it took the time from the master and deliver it to the salve with good offsets. If I use the ptp4l daemon as well.

Using /etc/linuxptp.cfg, does the 1021ATSN platform know that I am using gPTP.cfg in the other nodes or should I use /etc/gPTP.cfg in the LS1021ATSN platform? In the begining I used linuxptp.cfg for enable and disable the gmCapable option to use the LS1021ATSN platform, but after these problems of sync of loss I start using the /etc/gPTP.cfg to change the gmCapable option in case the previous way it was correct. Cast light on this, please. Remember that I want to use gPTP (802.1AS).

> > Is it better using the ptp4l daemon in each device (LS1021ATSN, board 1 and board 2) instead of the ptp4l.service?
> 
> Again, it depends what you need it to do. If by ptp4l daemon you mean starting/stopping ptp4l directly from a script, you could do that too.
> 

Ok. I prefer doing the same mechanism for both situations: [1] slave (node 1) <-- master (LS1021) --> slave (node 2) and [2] master (node 1) --> bridge (LS1021ATSN) --> slave(node 2).

> > What steps would you do if you did that?
> 
> I might just happen to have some examples which might give you some ideas. Keep in mind that you'll probably need to adapt them heavily though - just use them as a starting point.
> 
> ```
> Board A ETH0 <-> Board B SWP2
> Board A ETH1 <-> Board B SWP3
> 
>  +---------------------------------------------------+
>  | LS1021A-TSN board B (bridge)                      |
>  |                                                   |
>  |   +-----------+   +-----------+   +-----------+   |
>  |   |           |   |           |   |           |   |
>  |   |           |   |           |   |           |   |
>  |   |           |   |     +--------+|           |   |
>  |   |    SWP5   |   |    SWP3   |  ||    ETH1   |   |
>  |   +-----------+   +-----------+  |+-----------+   |
>  |   |           |   |           |  ||           |   |
>  |   |           |   |           |  ||           |   |
>  |   |           |   |     +------+ ||           |   |
>  |   |    SWP4   |   |    SWP2   || ||    ETH0   |   |
>  +---------------+---------------+|-|------------+---+
>                                   +-----------------------+
>                                     |                     |
>                                     +------------------+  |
>                                                        |  |
>  +---------------------------------------------------+ |  |
>  | LS1021A-TSN board A (sender & receiver)           | |  |
>  |                                                   | |  |
>  |   +-----------+   +-----------+   +-----------+   | |  |
>  |   |           |   |           |   |           |   | |  |
>  |   |           |   |           |   |           |   | |  |
>  |   |           |   |           |   |     +-----------+  |
>  |   |    SWP5   |   |    SWP3   |   |    ETH1   |   |    |
[... 342 lines cut ...]

And why do you set the same time to CLOCK_REALTIME and /dev/ptp1 if they will be the same when you use phc2sys a little below?

What does mean "freq 0"?

BTW. When we watch the log when the network is syncronized, it appears RMS, max offset, freq... What does freq mean? I can observe that each device has different values. Is not it the frequency that check/sync/poll the offset? Should not I be the same for everyone?

[... cut ...] al these configs are new and the description confusing.

Finally, I can tell you that if I synchronize the LS1021ATSN platform as a master and one board as a salve, in this case, if I run my tests, the sync is not lost. It seems it happens only when the LS1021ATSN is a bridge. I have to check with the parameters you said (the code below). Previously, I only used "--tx_timestamp_timeout 20".

'''
--gmCapable 0 --tx_timestamp_timeout 50 \
--step_threshold 0.00002 --first_step_threshold 0.00002
'''

diegotxegp, 2021-02-08:
Using the commands I could see in your code:

[NODE 2]
schedule ptp4l -i enp3s0 -f /etc/ptp4l_cfg/gPTP.cfg \
--gmCapable 1 --tx_timestamp_timeout 50 \
--step_threshold 0.00002 --first_step_threshold 0.00002

[LS1021ATSN]
phc_ctl CLOCK_REALTIME set 1612270628.000000000 && phc_ctl CLOCK_REALTIME freq 0
phc_ctl /dev/ptp1 set 1612270628.000000000 && phc_ctl /dev/ptp1 freq 0

schedule ptp4l -i swp2 -i swp3 -f /etc/ptp4l_cfg/gPTP.cfg \
--gmCapable 0 --tx_timestamp_timeout 50 \
--step_threshold 0.00002 --first_step_threshold 0.00002

[NODE 1]

schedule ptp4l -i enp3s0 -f /etc/ptp4l_cfg/gPTP.cfg \
--gmCapable 0 --tx_timestamp_timeout 50 \
--step_threshold 0.00002 --first_step_threshold 0.00002

And trying to change a little the values...

--tx_timestamp_timeout 1000
--step_threshold 1.0
--first_step_threshold 1.0

Still the problem happens. Everything works fine until I run my test where I do 1 million accesses to the CLOCK_REALTIME. When I change --tx_timestamp_timeout to 10000 instead of 50, one message appears with something like "Clockcheck: clock jumped backward or running slower than expected!"

ptp4l[1409.536]: rms    6 max   13 freq +18913 +/-   8 delay   234 +/-   0
ptp4l[1410.538]: rms    6 max   10 freq +18901 +/-   5 delay   234 +/-   0
ptp4l[1410.636]: timed out while polling for tx timestamp
ptp4l[1410.637]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[1410.637]: port 1 (enp3s0): send peer delay request failed
ptp4l[1410.637]: port 1 (enp3s0): SLAVE to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[1426.860]: port 1 (enp3s0): FAULTY to LISTENING on INIT_COMPLETE
ptp4l[1429.963]: port 1 (enp3s0): new foreign master 00049f.fffe.ef0808-1
ptp4l[1430.226]: port 1 (enp3s0): LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[1430.226]: selected local clock aabbcc.fffe.00094e as best master
ptp4l[1430.226]: port 1 (enp3s0): assuming the grand master role
ptp4l[1431.964]: selected best master clock aabbcc.fffe.00094a
ptp4l[1431.964]: port 1 (enp3s0): MASTER to UNCALIBRATED on RS_SLAVE
ptp4l[1432.953]: port 1 (enp3s0): UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
ptp4l[1433.830]: rms   38 max   65 freq +18837 +/-  19 delay   235 +/-   0
ptp4l[1434.833]: rms    8 max   15 freq +18869 +/-  11 delay   235 +/-   0

Thank you for your help, Vladimir.

diegotxegp, 2021-02-09:
Excuse me, Vladimir. Do you have any published paper evaluating the LS1021ATSN?

vladimiroltean, 2021-02-10:
> Still the problem happens. Everything works fine until I run my test where I do 1 million accesses to the CLOCK_REALTIME. When I change --tx_timestamp_timeout to 10000 instead of 50, one message appears with something like "Clockcheck: clock jumped backward or running slower than expected!"

Could you show me how to reproduce the 1 million accesses to CLOCK_REALTIME?

> Do you have any published paper evaluating the LS1021ATSN?

Nope, the documentation in the OpenIL user manual at https://openil.org/guide_list.html is all there is. If there's any analysis in particular that you would like to see, I guess you could ask that it gets included in next versions.

diegotxegp, 2021-02-10:
> Could you show me how to reproduce the 1 million accesses to CLOCK_REALTIME?

I discovered that most of the fault is by the print-to-file execution. After creating an array with one million of accesses to the CLOCK_MONOTONIC and CLOCK_REALTIME, the ptp4l communication fails when I want to print all that information in a file. I am trying to rewrite the code to open and close the file more properly.

> > Do you have any published paper evaluating the LS1021ATSN?
> 
> Nope, the documentation in the OpenIL user manual at https://openil.org/guide_list.html is all there is. If there's any analysis in particular that you would like to see, I guess you could ask that it gets included in next versions.

Can I ask for it? To whom?

diegotxegp, 2021-02-10:
Ah. And what is the difference among the versions of OpenIL; ls1021atsn_deconfig, ls1021atsn_ubuntu_deconfig and ls1021atsn_ubuntu_full_deconfig?

vladimiroltean, 2021-02-10:
> Can I ask for it? To whom?

Just ask here.

> what is the difference among the versions of OpenIL; ls1021atsn_deconfig, ls1021atsn_ubuntu_deconfig and ls1021atsn_ubuntu_full_deconfig?

`nxp_ls1021atsn_defconfig` uses packages compiled from source using the native Buildroot makefiles
`nxp_ls1021atsn_ubuntu_defconfig` and `ls1021atsn_ubuntu_full_defconfig` use a minimal rootfs skeleton with just the essential packages compiled by Buildroot, assembled together with an Ubuntu 18.04 rootfs for armv7 which includes the apt package manager. The full defconfig contains more apt packages preinstalled than the normal one.

In general I would recommend using the default `nxp_ls1021atsn_defconfig` build and using menuconfig there to build all the packages that you need. The building for the Ubuntu filesystems has been a bit flaky in the past.

diegotxegp, 2021-02-19:
Thank you so much for your help, Vladimir.

closed 2021-02-19 by diegotxegp