nxp-archive/openil #90: LS1021ATSN: Could not enable to tag VLAN when using Mode 3(best_effort_vlan_filtering).

Leo1726, 2021-04-05:
# Enironment
```
LS1021ATSN
- Ubuntu 18.04
- OpenIL v1.9
```
# Step to reproduce
Connect as **Node 1(192.168.200.222)-->{swp4}Switch(192.168.200.23){swp2}-->Node 2(192.168.200.220)**

## Expected behavior:
1. Node 1 send normal Ethernet frame, then switch could tag VID=1 and output it to Node 2.
2. Switch send frame with tag VID=1, and output to Node 2.

## Test Method:
Use ping IP, then observe on Wireshark v3.3.4

## VLAN Config
```
port vlan ids
swp2	 1 PVID Egress Untagged
swp4	 1 PVID
br0	 1 PVID Egress Untagged

```
## Log

# Actual Behavior

[image]

```
Node 1 ping Node 2: 
$ ping 192.168.200.220
```
[image]
- This realized expected behavior 1.
- Then, let best_effort_vlan_filtering true, Mode 3.
[image]

```
Switch ping Node 2: 
$ ping 192.168.200.220
```
[image]
- This realized expected behavior 2.
- However, if I try to use Node 1 ping Node 2, it works with no VLAN Tag.
```
Node 1 ping Node 2: 
$ ping 192.168.200.220
```
[image]
**- Dose it mean that expected behavior 1 & 2 could not coexist, which means Mode 3 is not the expansion of Mode 2???**

vladimiroltean, 2021-04-07:
Can you please let me know if these 3 kernel patches solve your issue?
1:
```
From 2349286d74897403311c6d2cfb8e2ead6d85b364 Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Wed, 7 Apr 2021 19:01:55 +0300
Subject: [PATCH 1/3] net: dsa: sja1105: use the bridge pvid in
 best_effort_vlan_filtering mode

The best-effort VLAN filtering mode is the sja1105 driver's attempt to
allow frame tagging towards the CPU with a unique VLAN ID corresponding
to the source port at the same time as allowing the bridge to freely
alter the VLAN table. It works by making the switch classify all untagged
ingress traffic to a secret pvid managed by net/dsa/tag_8021q.c.
Also, VLAN-tagged frames are retagged to another secret VLAN managed by
tag_8021q. Both these VLANs managed by tag_8021q are called "rx_vid".
The retagged rx_vid has some bits which encode a "sub-VLAN", and the
pvid-based rx_vid has those sub-VLAN bits set to zero. Software looks at
the rx_vid and knows what port and original VLAN the packet came from.

There is a huge oversight in the setup created by the sja1105 driver for
the best-effort VLAN filtering mode. That is:

ip link add br0 type bridge vlan_filtering 1
ip link set swp2 master br0
ip link set swp4 master br0
bridge vlan del dev swp4 vid 1
bridge vlan add dev swp4 vid 1 pvid

Then we send an untagged packet from swp2 and expect the switch to
forward it to swp4 and deliver it with VLAN 1 added. It is forwarded,
except the packet is egress-untagged.

This happens because of the way in which tag_8021q works (there is a
detailed picture in net/dsa/tag_8021q.c, above dsa_8021q_setup_port).
In the example above, the tag_8021q pvid of swp2 is 1026. This VLAN is
added to all other switch ports, to allow untagged traffic forwarding.
On all ports, the tag_8021q pvid of 1026 is installed as egress-untagged,
in order to hide the existence of DSA tag_8021q from the user.

This is fine, except when the real (bridge) pvid is egress-tagged, it
isn't. The user _wants_ to see this VLAN in the outside world, and we
can't really do that, because the sja1105 driver doesn't use that VLAN
but another one which the user knows nothing about.

As a side note, this only happens for untagged traffic on the ingress
port. If the packet arrives as pvid-tagged (i.e. tagged with VID 1)
on a port with tag_8021q, then the packet is classified to VLAN ID 1
(the bridge pvid) as opposed to the tag_8021q pvid. So we don't have
the same problem.

Consider the following more generic example:

Port             | sw0p0 sw0p1 sw0p2 sw0p3  |   sw1p0 sw1p1 sw1p2 sw1p3
=================+==========================+=============================
tag_8021q rx_vid | 1024  1025  1026  1027   |   1088  1089  1090  1091
Bridge VLAN      |  1     1     2     1     |    3     2     2     1
Bridge flags     | pvid        pvid         |   pvid  pvid
                 | untag untag              |

VLAN 1024 is added to sw0p1, sw0p2, sw0p3, sw1p0, sw1p1, sw1p2, sw1p3
as untagged.

The following pattern emerges:
A VLAN which is pvid on any port in the bridging domain (therefore has a
tag_8021q rx_vid) and is egress-tagged on another (potentially the same)
port will leak the tag_8021q VLAN. Every egress-tagged bridge VLAN that
is a pvid on another port must have a retagging rule from the tag_8021q
rx_vid to the bridge VLAN.

[... 290 lines cut ...]
+	list_for_each_entry(v, vlan_list, list) {
+		if (v->port == port && v->vid == vid) {
 			/* Already added */
-			return 0;
+			if (v->untagged == untagged && v->pvid == pvid)
+				/* Nothing changed */
+				return 0;
+
+			/* It's the same VLAN, but some of the flags changed
+			 * and the user did not bother to delete it first.
+			 * Update it and trigger sja1105_build_vlan_table.
+			 */
+			v->untagged = untagged;
+			v->pvid = pvid;
+			return 1;
+		}
+	}
 
 	v = kzalloc(sizeof(*v), GFP_KERNEL);
 	if (!v) {
-- 
2.25.1

```
Thanks.

Leo1726, 2021-04-08:
I will try, thanks so much. By the way, I met quite similar situation in LS1028ATSN. How could I solve that since LS1028ATSN has combined mode 2 and 3 together ?

vladimiroltean, 2021-04-08:
What do you mean exactly by "combined mode 2 and 3 together"? You mean that one switch operates in mode 2 and another in mode 3?

Leo1726, 2021-04-08:
Sorry for misunderstanding. I meant that I wanted to use LS1028ATSN to realize my expected behavior:
1. Node 1 send normal Ethernet frame, then switch could tag VID=1 and output it to Node 2.
2. Switch send frame with tag VID=1, and output to Node 2.

I found quite similar probelm as LS1028ATSN. I just could realize expected behavior 1 or 2, not both of them.

Thanks!

vladimiroltean, 2021-04-08:
Understood. There is nothing specific to the board with this issue. I can reproduce it on the LS1028A-TSN too.
Please note that while my patches fix the issue you have reported, they introduce another issue with PTP RX timestamping in best-effort VLAN filtering mode. I am currently investigating it. Sorry for the trouble.

Leo1726, 2021-04-08:
So, how could I reproduce these three patches on LS1028A-TSN since there is no sja1105 driver on it? I couldn't find these files to change them.

vladimiroltean, 2021-04-08:
The kernel is located here: https://github.com/openil/linux
And the driver for the switch is at its usual location: drivers/net/dsa/sja1105/
To add the patches to the kernel you can either:
(a) add the patches to the OpenIL/Buildroot build system: https://buildroot.org/downloads/manual/manual.html
(b) compile the kernel using a standalone toolchain:
```
export ARCH=arm64
make defconfig lsdk.config
make -j 8 Image.gz dtbs 2>&1 | tee build.log && mkimage -A arm64 -O linux -T kernel -C gzip -a 0x80080000 -e 0x80080000 -d arch/arm64/boot/Image.gz uImage && cp uImage arch/arm64/boot/dts/freescale/*ls1028a*.dtb /srv/tftpboot/ls1028/
```

vladimiroltean, 2021-04-19:
Hi,

I don't think there is a way to solve the problem without limitations. Let me
try to explain what the options are.
DSA models switches as port multiplexers with hardware-offloaded forwarding, so
packets sent to/from the network stack towards the switch need to be presented
separately for each front-facing switch port. This is done because DSA switch
network interfaces are first and foremost regular network devices, and must be
able to support any higher-level protocol (for example, IP forwarding in
software).
The SJA1105 switch is somewhat special in that it does not offer any indication
about the switch port on which a packet came when it forwards that packet to
the host port. Link-local traffic (STP, PTP) is an exception because the
respective protocols explicitly require explicitly targeting a physical port as
opposed to a bridging domain, so the SJA1105 switch offers a native way to
send/receive this kind of packets to/from a specific physical port.
To work around the hardware limitation and offer per-port general purpose
traffic termination, the DSA driver for SJA1105 makes use of
net/dsa/tag_8021q.c and reserves two VLAN ranges:
- 1024-2047 for "RX VLANs". The packets received on a switch port will be
  tagged with a VLAN ID derived from the number of the switch port, and
  software will decode this VLAN ID and replace it with the real value.
  For autonomous forwarding, the RX VLAN of a port is added to the membership
  list of all the other ports too, and this VLAN is made egress-untagged so
  that it is popped and clients attached to the switch do not see it.
- 2048-3071 for "TX VLANs". Every TX VLAN contains just the CPU port and one of
  the front-facing ports, and is used for precise steering of packets from the
  network stack towards the corresponding physical ports.
When the switch ports are VLAN-unaware (either bridged or standalone), all
packets received on a port are classified to that port's RX VLAN from
tag_8021q (for example 1024), regardless of the VLAN ID from the packet - that
is to say, 1024 is the pvid. So 2 stations having MAC addresses
00:01:02:03:04:05 (connected to swp0) and 00:01:02:03:04:06 (connected to swp1)
will be learnt by the switch in VLAN 1024 and 1025. To ensure that packets do
not get forwarded between ports based on flooding (the destinations are not
unknown to the FDB), we enable Shared VLAN Learning which makes the switch look
only at the destination MAC address for the forwarding decision lookup, and
ignore the VLAN ID. So with this setting, it works because VLAN 1024 and 1025
span the same set of ports.
For ports under a VLAN-aware bridge, there is no simple way to deduce which
port a packet came from, because the bridge layer controls the VLANs and their
flags (pvid, untagged). And imposing restrictions such as "you cannot have the
same VLAN installed on two ports" would be unproductive, because it would not
allow forwarding of any packet except to the CPU and back. So nothing is done
by default, DSA does not attempt to deduce the source port on RX and this is
why general purpose RX/TX through the swpN net devices is unavailable when they
operate in this mode. PTP and STP are still available because they rely on
packet traps and not on VLANs.
The third mode (best-effort VLAN filtering) is an attempt to combine modes one
and two. We allow the bridge to add VLANs to the SJA1105 switch as long as they
do not overlap with the 1024-3071 range used by tag_8021q. And for they most
part, they indeed do not overlap, with one exception: the pvid (more on this
later).
For every VLAN added by the user (either on the bridge or through 8021q
upper interfaces), the code for best-effort VLAN filtering creates a mapping in
the VLAN Retagging table, such that when a packet with VLAN 100 is flooded, the
copy of it which reaches the CPU port will be received with a VLAN ID managed
by tag_8021q. Software can again look up this mapping table based on the
tag_8021q VLAN and identify the source port, switch ID and original VLAN ID,
and then send the packet up the stack for further processing. This part works
as expected. What doesn't is the fact that the bridge is led into having an
incorrect idea of what is the pvid of each physical port. When you set the pvid
on the first switch port to 1, in hardware, the pvid is still that of the
tag_8021q module (1024) and not that of the bridge (1). Then, when untagged
packets are forwarded to a second bridge port where VLAN 1 is egress-tagged,
the expectation is that untagged packets will have the VLAN tagged pushed on
egress. But the reality is that they don't, because, even though VLAN 1 is
committed to hardware with the proper port membership and flags, ingress
untagged packets are not classified to it, so the egress tagging settings have
no effect. If you would configure tag_8021q to set VLAN ID 1024 as
[... 101 lines cut ...]
100, we would have to keep some sort of lookup table (or maybe extend the one
we use for imprecise RX towards the bridge) such that received traffic with
VLAN 100 will set skb->dev = swp2, but this will mean that swp3.100 is not
possible. Also, having VLAN 100 as a bridge VLAN is not possible either.
To allow having both swp2.100 and swp3.100, the only option is to use VLAN
retagging again, such that VLAN 100 coming from swp2 is seen by DSA as having
VLAN ID 1042, and VLAN 100 coming from swp3 is seen as having 1043. But again
there are the pitfalls of VLAN retagging, which are that:
[... cut ...]
mode. It would basically be more similar to mode 2 (fully VLAN aware) than it
is to mode 3. So we could enable Independent VLAN Learning (so you could have
the same MAC address learnt in multiple VLANs, with potentially different
destination ports), and you would have access to way more bridge VLANs than in
mode 3 (2K vs 32).
So in order to know how to proceed further, I need to know your exact
requirements regarding what must work and what mustn't. Thanks for the
understanding, and sorry for the huge afterthought.

Leo1726, 2021-04-28:
Hi, vladimiroltean, thanks a lot!
I think maybe first probable solution could give a quicker way to solve it.
Btw, if the third one could realize, that will be better.
```
Keep swp0's pvid as 1 (the bridge pvid), and perform retagging of VLAN 1 on
swp0 towards port 4 (the host port) and VLAN 1024, but do not drop the
original packets, instead keep both, and drop what you don't need in
software. This limits the maximum useful bandwidth of the CPU port to 500
Mbps.
```

vladimiroltean, 2021-07-09:
Hello,
Sorry for the long delay.
The git tree linked below is an implementation of the proposed method 3 (delete the VLAN retagging code and add support for imprecise RX and imprecise TX through a VLAN-aware bridge):
https://github.com/vladimiroltean/linux/tree/sja1105-bridge-fwd-offload-v1
The development was done on top of the mainline kernel, which is at v5.13.
Please let me know if it solves all issues you are seeing, then we can discuss backporting to v5.10 or other ways of integrating these changes into OpenIL.