nxp-archive/openil #96: PPS signal on LS1021A TSN?

aslaamshaafi, 2021-06-23:
Hi, 
Currently, I am using LS1021A TSN for clock synchronization. I wonder if the switch has exposed PPS port to measure clock synchronization? If yes, may I learn how to configure it?
I could not find much information about it in user manual.
https://www.nxp.com/docs/en/user-guide/OPEN-LINUX-IND-UM-1-10.pdf

vladimiroltean, 2021-06-24:
Try this
```
cat pps.sh
#!/bin/bash

# enum ptp_pin_function
PTP_PF_NONE=0
PTP_PF_EXTTS=1
PTP_PF_PEROUT=2
PTP_PF_PHYSYNC=2

PTP1_CHANNEL0=0
PTP1_PEROUT_PERIOD="1 1"

PTP1_PEROUT_START="0 0"

echo "$PTP_PF_PEROUT $PTP1_CHANNEL0" > /sys/class/ptp/ptp1/pins/ptp_clk
echo "$PTP1_CHANNEL0 $PTP1_PEROUT_START $PTP1_PEROUT_PERIOD" > /sys/class/ptp/ptp1/period
```
There is a LED on the board (D10) which should blink when the signal from PTP_CLK is high. Using the PTP periodic output (perout) functionality you can modify the period of the waveform but not the duty cycle (that is fixed at 50%).
Hope this helps.

aslaamshaafi, 2021-06-24:
Thanks for the reply, Actually I want to compare the PPS signal of the LS1021A TSN (acting as master) with the end-station (acting as slave) on oscilloscope. I attached the setup I am using. Please let me know if it is possible with the LS1021A TSN. If so, let me know which port I can use.
[image]

vladimiroltean, 2021-06-24:
The PTP_CLK pin (routed to the D10 LED and the TP1 test point) is emitted by the sja1105 switch.
The LS1021A SoC does not have the option to emit PPS options because the IEEE 1588 pins are pinmuxed with the RGMII port eth2, the same port that is used as a host port for the switch.
So the answer is: any of swp2, swp3, swp4, swp5.