nxp-archive/openil #104: sja1105-tool no spide under /dev on ls102a TSN board

riavix, 2021-11-17:
Hi,
I am interested to configure sja1105 switch on ls1021a tsn board. I have configured spi user support (in linux-menuconfig) and I can see on my target 
/sys/bus/spi//devices/spi0.1
but I can not find /dev/spidev0.1 ( that is the  user interface requested by sja1105-tool to configure the switch)
Looking in dmesg command I have found following error:

[    0.980283] sja1105 spi0.1: Probed switch chip: SJA1105T
[    1.132327] sja1105 spi0.1: Unexpected {device ID, part number}: 0x1e00030e 0x9a83
[    1.132343] sja1105 spi0.1: Device ID check failed: -19

What I am missing?

kernel version is 5.4.3-rt1
Thank you in advance

vladimiroltean, 2021-11-17:
You are on the wrong path.
The sja1105-tool is a user space driver whereas the errors you are seeing in dmesg are coming from a kernel driver. With the kernel driver you don't need the user space driver, you should concentrate on why the kernel driver does not work.
As the error message is suggesting, the driver queries the switch's device ID twice. First it recognizes the switch as SJA1105T, the second time it fails to recognize the device ID since it is corrupted. So it is a SPI communication issue.
First I would try updating to a newer kernel and see if the behavior persists. If it does, I would try to reduce the spi-max-frequency property from arch/arm/boot/dts/ls1021a-tsn.dts to something lower and see if the behavior changes. This is relatively unlikely, but there may be an issue with the board.

riavix, 2021-11-18:
Thank you for your answer.
So I'll try what you are suggesting.
Do you mean that I don't need sja1105-tool to "play" with sja1105 switch?
In which way I can change behaviour of the switch form user space?

vladimiroltean, 2021-11-18:
Please follow this documentation with the kernel driver:
https://www.kernel.org/doc/html/latest/networking/dsa/sja1105.html
In short, since one network interface is created for each switch port, the regular user space interaction with network interfaces is expected.

riavix, 2021-11-18:
Thank you again, Vladimir!