nxp-archive/openil #111: Ls1021 atsn qbv demo performance  doesn’t well

liing0228, 2022-06-10:
Hi sir,
I follow the step ls1021atsn demo in opening v1.10 user guide 
My qbv schedule on 1021 is:
                        Basetime 0
                               0x80 50000
                               0x81 250000
                               0x40 100000
                               0x88 100000
                       Flag 2
My cycle time is 500000ns.
I use three host that one is sender that send OpCUA tsn punsub multicast layer2 packet with priority 3 every 0.5ms,and one is receiver and last one is iperf3 launcher.
However when I start use iperf3 the pubsub receiver packetloss rate can high to 3-10 % .
When there is no iperf3 traffic the packet loss will be only 0.x%.
I am sure my ptp4l and phc2sys work well.
And my pubsub sender Tx schedule is every 0.3ms in the 0.5ms cycle time .
And the iperf3 result shows that the bandwidth been cut through half.
And I fount that the QoS protect the high priority packet better than TAS .
So that I don’t know what’s wrong?

vladimiroltean, 2022-06-11:
I don't understand this: "When there is no iperf3 traffic the packet loss will be only 0.x%."
Why would you have any packet loss at all even with no background traffic? I think that's the problem you need to start looking at.

liing0228, 2022-06-12:
> I don't understand this: "When there is no iperf3 traffic the packet loss will be only 0.x%."
> Why would you have any packet loss at all even with no background traffic? I think that's the problem you need to start looking at.

@vladimiroltean 
Sorry that I didn’t say clear .
Actually  that is no  packet really loss.
What i mean  the loss% is that the packet didn’t arrive in  the right time slot open.
For example I send packet every 0.5ms but when there’s iperf3 traffic will 2-10% packet didn’t arrive in between 0.5 ms interval .
When there is no iperf will only 0.x% packet didn’t arrive in 0.5ms