nxp-archive/openil #115: 802.1CB test on LS1028ARDB board

pzm919, 2022-10-23:
Hello，I'm doing a 802.1CB test on LS1028ARDB board.
There is only one LS1028ARDB board in the current experimental environment. All 4 hosts are used to connect to the four ports of the switch. The connection topology is as follows, pc1-swp3, pc2-swp0, pc3-swp1, pc4-swp2, First of all, after I establish a bridge through the command, I then perform packet capture comparison analysis on pc1, pc2, pc3, and pc4 respectively, so as to verify the frame copy function.
By running the command
tsntool> cbstreamidset --device swp0 --index 1 --nullstreamid --nulldmac 0x7EA88C9B41DD --
nullvid 1 --streamhandle 1
tsntool> cbgen --device swp0 --index 1 --iport_mask 0x08 --split_mask 0x07 --seq_len 16 --
seq_num 2048
I changed 0x7EA88C9B41DD to the physical address 2C:53:4A:07:DF:06 of the host connected to the swp0 port, and set the destination address of the sending stream to 2C:53:4A:07:DF:06 ,
The result is that I can receive the same changed packets on all hosts connected to the outgoing port, is this normal? And how can only the specified port be able to receive the copied packets? It is mentioned in the manual that the serial number will be added to the package. Where is the serial number added? What experimental results show that 802.1CB achieves frame replication? Is swp0 fixed in the cbsteramidset instruction? and is 0x7EA88C9B41DD the destination address representing the stream to be copied?
[image]
[image]
@jacmet @felipec @ceyusa @lwalkera @charles-dyfis-net @vladimiroltean

vladimiroltean, 2022-10-23:
> The result is that I can receive the same changed packets on all hosts connected to the outgoing port, is this normal?

Since you use `--split_mask 0x07` (bitwise 0b111), it behaves as instructed. It splits the CB stream to ports 0, 1, 2.

> Where is the serial number added?

More details here:
https://github.com/vladimiroltean/isochron/issues/10

> What experimental results show that 802.1CB achieves frame replication?

This one, for example:
https://github.com/vladimiroltean/isochron/tree/master/frer

> Is swp0 fixed in the cbsteramidset instruction?

Yes, more or less.
More details: https://github.com/vladimiroltean/isochron/blob/master/frer/8021cb-load-config.sh#L185

> and is 0x7EA88C9B41DD the destination address representing the stream to be copied?

Yes.

pzm919, 2022-10-24:
Thank you very much for your reply! I use the packet sending tool to send data from pc1 to pc2 by setting the source MAC address to 2c:53:4a:07:df:04 and the destination MAC address to 2c:53:4a:07:df:06 , I want to copy this packet so that it can be received on pc3 at the same time, but pc4 does not need to receive this packet, how should I do it? Change --split_mask? What should it be set to? Is the --device in the directives cbstreamidset and cbgen a port for stream identification and stream replication? Do the two have to be the same? Must be set to swp0? What does each parameter of cbstreamidset and cbgen mean? Could you please explain a little bit? Thank you for your help.
@vladimiroltean

pzm919, 2022-10-24:
> > The result is that I can receive the same changed packets on all hosts connected to the outgoing port, is this normal?
> 
> Since you use `--split_mask 0x07` (bitwise 0b111), it behaves as instructed. It splits the CB stream to ports 0, 1, 2.
> Thank you very much for your reply.
What should --split_mask be set to, so that only the specified port can copy the CB flow? For example, it only splits the cb flow to ports 0 and 1, but not port 2. The port that sets the flow identification must be the destination mac address where the packet is sent Is that the port it's connected to? Is stream replication related to ports?