nxp-archive/openil #117: about IEEE 1588/802.1AS

ysdo89, 2023-07-11:
hello.
I am referring to the Open Industrial User Guide, Rev. 1.10, 12/2020, and I am using the LS1021ATSN board.
I want to get the same data as the graph included in the guide (Figure 7. Offset from master in start up state).

### **How can I measure the data behind that graph?**

vladimiroltean, 2023-07-11:
Something like this:

Modify gPTP.cfg, such that rms statistics are not printed, but full offset and frequency log for each servo adjustment:

```
logSyncInterval -3
summary_interval -3
```

Then save program output to log files:
```
ptp4l -f gPTP.cfg -m | tee ptp4l.log &
ts2phc -f ts2phc.cfg -a --transportSpecific 0x1 -m | tee ts2phc.log
```

Then plot it with gnuplot:

```
#!/bin/bash

set -e -u -o pipefail

cat ptp4l.log | awk \
	'/ master offset / && / s2 / { \
	time=gensub(/^ptp4l\[(.*)\]:/, "\\1", "g", $1); \
	print time, $4; \
	}' > ptp4l.plot

gnuplot --persist -e 'set xlabel "Seconds since boot"; set ylabel "Offset from PTP master (nanoseconds)"; set term svg; set terminal svg enhanced background rgb "white"; set output "ptp4l_offsets.svg"; plot "ptp4l.plot" with dots;'

cat ts2phc.log | awk \
	'/ offset / && / s2 / { \
	time=gensub(/^ts2phc\[(.*)\]:/, "\\1", "g", $1); \
	print time, $4; \
	}' > ts2phc.plot

gnuplot --persist -e 'set xlabel "Seconds since boot"; set ylabel "Offset from main PHC (nanoseconds)"; set term svg; set terminal svg enhanced background rgb "white"; set output "ts2phc_offsets.svg"; plot "ts2phc.plot" with dots;'
```

The splitting of the plots in a start-up phase (convergence to offset 0) and a stable state (s2) phase was done manually, so that the variations in the stable state are more visible on a separate plot (larger scale).

P.S. I'm not sure if Open Industrial Linux is maintained any longer (it's superseded by NXP Real-Time Edge). Please ask the question in full, in a way that doesn't require digging up old documentation, if you want to get an answer from the general public, or open an NXP support ticket otherwise.

ysdo89, 2023-07-12:
Hello!, Thank you so much for your quick response.
I'm having trouble following the process you suggested and would like to ask a few more questions.
We connected to the experimental environment as shown in the figure(Figure 6. Setup for time-aware bridge verification
) in the manual.
However, I'm getting the following error, do you know why?

[root@LS1021ATSN /etc/ptp4l_cfg] # `ptp4l -f gPTP.cfg -m | tee ptp4l.log &`

[1] 1710
[root@LS1021ATSN /etc/ptp4l_cfg] # ptp4l[67621.482]: ioctl SIOCETHTOOL failed: No such device
ptp4l[67621.508]: ioctl SIOCETHTOOL failed: No such device
ptp4l[67621.532]: ioctl SIOCETHTOOL failed: No such device
ptp4l[67621.556]: ioctl SIOCETHTOOL failed: No such device
ptp4l[67621.580]: ioctl SIOCETHTOOL failed: No such device
ptp4l[67621.605]: ioctl SIOCETHTOOL failed: No such device
ptp4l[67621.629]: ioctl SIOCETHTOOL failed: No such device
ptp4l[67621.629]: PTP device not specified and automatic determination is not supported. Please specify PTP device.
failed to create a clock

Additionally, the `gPTP.cfg` file is set up as follows.

> 
> [global]
> gmCapable 1
> priority1 248
> priority2 248
> logAnnounceInterval 0
> logSyncInterval -3
> summary_interval -3  // added
> syncReceiptTimeout 3
> neighborPropDelayThresh 800
> min_neighbor_prop_delay -20000000
> assume_two_step 1
> path_trace_enabled 1
> follow_up_info 1
> transportSpecific 0x1
> ptp_dst_mac 01:80:C2:00:00:0E
> network_transport L2
> delay_mechanism P2P
> step_threshold 0.00002
> tx_timestamp_timeout 20
> boundary_clock_jbod 1
> 
> [sw0p0]
> [sw0p1]
> [sw0p2]
> [sw1p0]
> [sw1p1]
> [sw1p2]
> [sw1p3]

vladimiroltean, 2023-07-12:
Do you have network interfaces called sw0p0, sw0p1, sw0p2, sw1p0, sw1p1, sw1p2, sw1p3? It's like making an account with the name Your Name.

ysdo89, 2023-07-12:
I typed exactly what the manual says. Now that you mention it, I think you're right about removing it.

ysdo89, 2023-07-12:
We entered the contents of the manual. We removed this content.
We still get the following error

[root@LS1021ATSN /etc/ptp4l_cfg] # `no interface specified`

ysdo89, 2023-07-12:
[sw0p0] -> [eth0] 
                  [eth1]